zigpy/zha-device-handlers #1580: [BUG] TS0002 switched on and off as a group from HA

IIIEII, 2022-05-23:
**Describe the bug**
A have many 2-gang and 4-gang tuya zigbee switches. I'm connecting them to HomeAssistant using ZHA integration with Sonoff ZigBee 3.0 USB Dongle. They are identified as zhaquirks.tuya.ts000x.Switch_(2/4)G_GPP.
When I press physical buttons on switch it works normal, but if I switch any single entity in HA it switches all 2 or 4 buttons together.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect 2-gang switch to power
2. Go to Settings > Devices & Services > "Configure" button for ZHA integration > "+ Add Device" button
3. New device (TS0002 by _TZ3000_fvixbt4b, Quirk: zhaquirks.tuya.ts000x.Switch_2G_GPP) with two entities (both turned off) is discovered
4. Turn on any entity
5. Both buttons on switch clicked (relay turn on sound) and showed they turned on with light-emitting diodes
6. In HA we see that both devices are turned on too
7. Turn off any entity and they both turned off same way.

**Expected behavior**
If I turn on/off single entity in HA I'm waiting to turn on/off only one button at switch

<details>
<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_fvixbt4b",
  "model": "TS0002",
  "class": "zhaquirks.tuya.ts000x.Switch_2G_GPP"
}
```

</details>

<details>
[... 416 lines cut ...]
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Received command 0x0A (TSN 207): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-05-23 23:39:14 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.tz3000_fvixbt4b_ts0002_12e3d3f8_on_off, old_state=<state light.tz3000_fvixbt4b_ts0002_12e3d3f8_on_off=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_brightness=None, friendly_name=_TZ3000_fvixbt4b TS0002 12e3d3f8 on_off, supported_features=8 @ 2022-05-23T23:38:57.335885+03:00>, new_state=<state light.tz3000_fvixbt4b_ts0002_12e3d3f8_on_off=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_brightness=None, friendly_name=_TZ3000_fvixbt4b TS0002 12e3d3f8 on_off, supported_features=8 @ 2022-05-23T23:39:14.260182+03:00>>
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x257A), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=206, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\xCE\x0B\x0A\x00')
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=206)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x257A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=33, SecurityUse=<Bool.false: 0>, TimeStamp=13785899, TSN=0, Data=b'\x08\xCE\x0A\x00\x00\x10\x00', MacSrcAddr=0x257A, MsgResultRadius=29)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Received ZCL frame: b'\x08\xCE\x0A\x00\x00\x10\x00'
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=206, command_id=10, *is_reply=True)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Received command 0x0A (TSN 206): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=206, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy.zcl] [0x257A:1:0x0006] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x257A), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=206, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\xCE\x0B\x0A\x00')
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-05-23 23:39:14 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=206)
```

</details>

**Additional context**
Here is switch that I'm testing: https://aliexpress.ru/item/1005003479909650.html
I'm using this sonoff zigbee dongle: https://aliexpress.ru/item/1005003763454756.html
I have home Assistant Core 2022.5.3 installed with docker homeassistant/raspberrypi2-homeassistant

MattWestb, 2022-05-24:
I think its one updated version that is needing the famous tuya magic spell being casted on them for working OK. Its the same with the LIDL 3 power strip with updated firmware that  have getting the casting implanted and working OK after hat.

IIIEII, 2022-05-24:
Hello @MattWestb!
Could you please tell me little more about "tuya magic spell"? Maybe there are some instructions or other issues with it described?

MattWestb, 2022-05-24:
I think @javicalle is knowing little more and have doing one quirk with it and i have doing one with some help but it completely different device.
Most is documented in the issues for tuya TS004F and also LIDL 3 AC 4 USB power stripe issues.

javicalle, 2022-05-24:
@IIIEII could you test with a local quirk?
There is a modified version of `tx000x.py`:

<details>
<summary>tx000x.py</summary>

```python
"""tuya TS000X Switches."""

import asyncio

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)

class Switch_1G_GPP(CustomDevice):
    """Tuya 1 gang switch module with restore power state support."""

    signature = {
        MODEL: "TS0001",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=266 device_type=256
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
[... 360 lines cut ...]
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

If you are not familiar with, there are some guides here to install local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693

IIIEII, 2022-05-24:
@javicalle after applying this changes 2-gang switch works as expected! I tried same for Switch_4G_GPP with no luck, do you have any ideas about it?

javicalle, 2022-05-24:
Ummm, it would be just to add the class inheritance like:
```python
class Switch_4G_GPP(TuyaMagicSpell, CustomDevice):
```

It's the device signature resolving the `Switch_4G_GPP` quirk? If not, add the device signature to see if there is something diferent.

IIIEII, 2022-05-25:
It's resolving to `Switch_4G_GPP` and I added `class Switch_4G_GPP(TuyaMagicSpell, CustomDevice)` right after testing 2-gang. I've been surprised when it didn't help me with 4-gang 🤷‍♂️

<details>
<summary>4-gang device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_gbdxbmwz",
  "model": "TS0004",
  "class": "zhaquirks.tuya.ts000x.Switch_4G_GPP"
}
```

</details>

javicalle, 2022-05-25:
Have you tried to remove the device from HA and add again?

IIIEII, 2022-05-25:
Yes, I tried it. Maybe this evening I'll do some more test with a few switches some of which haven't been added to HA before.

MattWestb, 2022-05-25:
@javicalle the tuya magic spell can fixing most devises by being "casted" one time or more and can being done any time and the device is being nice (the LIDL power stripe version) and can needing it after power retested.
Some device need it being casted only one time (until battery is out **and** being retested at the same time) and it must being made then its paring and in the first second it have getting the network key and before its requesting updating the TC-Link key (the TS004F implementation). 
The last version is normally for devices that is changing its hardware setting like adding more endpoints or changing cluster config and the second version for device that is not battery powered.

IIIEII, 2022-05-25:
At least one newly added 4-gang switch works well. I'm gonna continue tests to identify when it works and when not.

IIIEII, 2022-05-26:
I finally did a completely fresh configuration (removed all switches from HA, restarted and added them again) and now everything works well. I was not able to repeat those conditions under which rejoin didn't help me, so let's think it was an accident.

MattWestb, 2022-05-27:
@IIIEII Some device is needing being deleted and new joined for the magic spell working OK then its needed being in init mode then getting it so very likely your device is this type.

IIIEII, 2022-06-01:
BTW, could it be added to project to be available without local code rewrite? Or it's something unique that works only with my devices and can't be used widely?

javicalle, 2022-06-02:
Sure!
This is not a trivial one, but not difficult to do yourself. You can create a PR just editing the file from the web:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts000x.py

The new class is already in the mainstream, so don't add to the file, just import it like:
```python
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice
```
(keep the 2 blank lines after the import statement)

Then add the class inheritance (with the new name) as you have done in your local quirk:
```python
class Switch_2G_GPP(EnchantedDevice, CustomDevice):
```

IMO that would be added to all the classes in file (`Switch_1G_GPP` to `Switch_4G_GPP`).

I can give you support in any doubt or problem you have in the process.

Once the PR is merged, the new version of the zha-device-handlers library is generated and it is included in HA (probably for v2022.07), the change will be available to everyone and you will be able to remove the local quirk from your instance.

IIIEII, 2022-06-07:
@javicalle Hi! Should I do something else with this PR?

javicalle, 2022-06-07:
Something is not fine with your PR.
It is created in your repo and not in the 'oficial' one (https://github.com/zigpy/zha-device-handlers).

Can you try to merge your changes in your repo?
Once merged it's possible that would ask you to create a PR to the zigpy repo. That would be the good one.

IIIEII, 2022-06-07:
🤦‍♂️ it was my mistake, thanks a lot!

closed 2022-06-08 by dmulcahey

incognitto-indago, 2023-06-26:
Hello everybody!

I hope you will still take a look at this.

I have been using Zha in HA for a while and have more than 10 smart switches. Recently, I have bought new Zemismart/Tuya Zigbee Light Switches (TS0004 and TS0003). Unlike my previous ones, each one of these new ones are switching on and off as group when I use the entities in Home Assistant. If I press the physical buttons on the switch, they do work well, and they turn on and off just the one I pressed. If I try to turn on or off just one button via Home Assistant, all the buttons on that switch will also turn on or off.

I have same-model switches that work fine here, but they were bought about year ago. I suspect it has something to do with their firmware, but I am not familiar nor confortable with tweaking the internal working of the switches.

I tried to cast the Tuya magic spell via the custom quirk but I suspect something else is used to identify the models to match the quirk.

My new models that are acting as a group are: Device info TS0003 by _TZ3000_empogkya and TS0004 by _TZ3000_a37eix1s

Could you help me? Perhaps I have to edit the tx000x.py for it to correct recognize my switches? Is there anything else to change other than the "MODEL:"?

Again, thank you!

incognitto-indago, 2023-06-26:
Or should I open another issue for this similar problem?

Thank you!

MattWestb, 2023-06-26:
It device is normally working OK but the newer need "tuya magic spell" being casted then they is being added to the system for decoupling the hardware then getting commands from the host system.
Post diagnostic for both devices so its possible see if they system have the quirk loaded for them.

incognitto-indago, 2023-06-26:
Thanks. Will create a new issue then.