zigpy/zha-device-handlers #1582: [Device Support Request] TS130F - _TZ3000_kmsbwdol

Oulewappeur, 2022-05-24:
**Is your feature request related to a problem? Please describe.**
This device gets recognized and connected by ZHA out of the box. 
It's a switch for two shutters/curtains/covers with physical buttons (up, pause, down) for both outputs of them
However, the percentages are the wrong way around (so when it's closed it says its 0% and when it's open it says it's 100%), and I'm not able to calibrate properly.

I've been trying to fix it myself, but I have never done this before and I'm probably doing it completely wrong!

There are also two toggles which seem to do nothing
And the rssi is not shown either.

Product that I purchased:
[https://nl.aliexpress.com/item/1005003308586552.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60161453943&aff_fcid=b68f242928344655ae91979f35502ba9-1653409970693-02737-_u76szN&tt=MG&aff_fsk=_u76szN&aff_platform=default&sk=_u76szN&aff_trace_key=b68f242928344655ae91979f35502ba9-1653409970693-02737-_u76szN&shareId=60161453943&businessType=ProductDetail&platform=AE&terminal_id=a4e2cde960db4c23a78d82206988e934&afSmartRedirect=y](url)

**Describe the solution you'd like**
A roller shutter that works properly would be nice!

**Device signature**
> {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_kmsbwdol",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}

**Diagnostic information** 

> {
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
[... 288 lines cut ...]
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "WINDOW_COVERING_DEVICE"
      },
      {

        "name": "WINDOW_COVERING_DEVICE"
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": "Rolluiken Master Bedroom",
    "device_reg_id": "e9c111e1ea39aa847286fb5476f54e7e",
    "area_id": "c6c9153cd0da4c2996d3b14d43157310"
  }
}

**Additional context**
[image]
[image]
[image]

javicalle, 2022-05-24:
Your device doesn't match any existing quirk because of double switch. Existing quirks already fix the position issue.

There is a custom quirk that I think that can match your device:
<details>
<summary>ts130f.py</summary>

```
"""Device handler for loratap TS130F smart curtain switch."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaZBExternalSwitchTypeCluster

ATTR_CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008
CMD_GO_TO_LIFT_PERCENTAGE = 0x0005

class TuyaWithBacklightOnOffCluster(CustomCluster):
    """TuyaSmartCurtainOnOffCluster: fire events corresponding to press type."""

    cluster_id = OnOff.cluster_id

    LIGHT_MODE_1 = {0x8001: 0}
    LIGHT_MODE_2 = {0x8001: 1}
    LIGHT_MODE_3 = {0x8001: 2}

    attributes = {0x8001: ("backlight_mode", t.enum8)}

class TuyaCoveringCluster(CustomCluster, WindowCovering):
    """TuyaSmartCurtainWindowCoveringCluster: Allow to setup Window covering tuya devices."""

    attributes = WindowCovering.attributes.copy()
    attributes.update({0xF000: ("tuya_moving_state", t.enum8)})
    attributes.update({0xF001: ("calibration", t.enum8)})
    attributes.update({0xF002: ("motor_reversal", t.enum8)})
    attributes.update({0xF003: ("calibration_time", t.uint16_t)})

    def _update_attribute(self, attrid, value):
        if attrid == ATTR_CURRENT_POSITION_LIFT_PERCENTAGE:
            # Invert the percentage value (cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757)
            value = 100 - value
        super()._update_attribute(attrid, value)

    async def command(
        self, command_id, *args, manufacturer=None, expect_reply=True, tsn=None
    ):
        """Override default command to invert percent lift value."""
        if command_id == CMD_GO_TO_LIFT_PERCENTAGE:
            percent = args[0]
            # Invert the percentage value (cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757)
            percent = 100 - percent
            v = (percent,)
[... 373 lines cut ...]
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

If you are not familiar with, there are some guides here to install local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693

javicalle, 2022-05-24:
And please, edit your first comment and fix the "device signature" and "Diagnostic information" putting your the info inside the ``` marks and respecting all the breaklines:
````
<details>
<summary>Device signature / Diagnostic Information</summary>

```
Paste the comments here.
```

</details>
````

Oulewappeur, 2022-05-24:
> Your device doesn't match any existing quirk because of double switch. Existing quirks already fix the position issue.
>
> There is a custom quirk that I think that can match your device:
> <details> [...]

Thanks for your reply. I already noticed none of them were duals, but I was assuming adding an extra endpoint would solve this.

I'll give the file you added a try tomorrow, bedtime for now!

Also updated the opening post :)

javicalle, 2022-05-24:
Thanks for fixing the format.
I've updated the code because I had forgotten about the new endpoint in the replacement part.

Goodnight 😴

Oulewappeur, 2022-05-25:
Well, this is looking good!

[image]

Going to run some tests and will comeback with the results

Oulewappeur, 2022-05-25:
Alright I did some testing and what an improvement! I have now been able to properly calibrate them and they give the correct percentages! Thanks @javicalle !!!

There is only one remark, but I'm not sure if it's related to this, or within HomeAssistant itself.
If I change to an X percentage via the slider, it stops at that actual point, but the status in HA still shows "Opening" or "Closing" while it should actually be "Open".
[image]
This only changes after I actually press the "stop" button. Only when it reaches the 100% it automatically jumps from opening to open.

It sometimes gives the wrong movement status, so if I go from percentage 100 to 80, it should show status "Closing" but instead, it shows status "Opening"

Next to that, there is also this cluster called TuyaWithBacklightOnOffCluster, I'm not sure what values I can give this, and my assumption is that I can the backlight of the buttons, but I want to see what the possibilities are.

<details>
<summary>I found something in a different topic </summary>

You have all found functions on the _TuyaZBOnOffAttributeCluster_:
https://github.com/zigpy/zha-device-handlers/blob/039a31f1ca15821e800bdb347eb47fdb02455f31/zhaquirks/tuya/__init__.py#L789-L819
And also switch type on the _TuyaZBExternalSwitchTypeCluster_:
https://github.com/zigpy/zha-device-handlers/blob/039a31f1ca15821e800bdb347eb47fdb02455f31/zhaquirks/tuya/__init__.py#L913-L928
Child lock is not "found" and is not implanted in ZHA or Z2M.

_Originally posted by @MattWestb in https://github.com/zigpy/zha-device-handlers/issues/1442#issuecomment-1076641024_
</details>

Does that mean I can put in 0, 1 and 2 as value, and what would be the result of each value?

PR #1807 (2022-10-04): Update TS130f covers quirk
A couple of changes here:
1. remove the duplicated quirk definition `TuyaMoesDiyRf433TS130F` (exactly the same as `TuyaTS130FTI2`)
2. add a new `TuyaTS130Double_GP` quirk

Fix: #1582
closed 2022-10-04 by dmulcahey