zigpy/zha-device-handlers #1584: [Device Support Request] _TZE200_n9clpsht TS0601 Immax Neo Keypad

Celer21, 2022-05-25:
Hi,
i bought Immax Neo Keypad for alarm system. It is zigbee 3.0 device (TUYA).
I would like to get support this keypad in ZHA.
It is possible?
Thank you.

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_n9clpsht",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.41",
    "supervisor": "2022.05.3",
    "host_os": "Home Assistant OS 8.1",
    "docker_version": "20.10.14",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "browser_mod": {
      "version": "1.5.3",
      "requirements": []
    },
    "hacs": {
[... 416 lines cut ...]
[0xBAB0:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
[0xBAB0:1:0x0201]: Attribute report 'MoesBHTThermostat'[occupied_heating_setpoint] = 2000
climate.tze200_aoclfnxz_ts0601_thermostat: Attribute 'occupied_heating_setpoint' = 2000 update
[0x9E4E:1:0xef00] Received ZCL frame: b'\x09\x2F\x01\x00\x36\x65\x01\x00\x01\x01'
[0x9E4E:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=47, command_id=1, *is_reply=True)
[0x9E4E:1:0xef00] Unknown cluster command 1 b'\x006e\x01\x00\x01\x01'
[0x9E4E:1:0xef00] Received command 0x01 (TSN 47): b'\x006e\x01\x00\x01\x01'
[0x9E4E:1:0xef00] No explicit handler for cluster command 0x01: b'\x006e\x01\x00\x01\x01'
[0x9DC5](TS0121): Device seen - marking the device available and resetting counter
[0x9DC5](TS0121): Update device availability -  device available: True - new availability: True - changed: False
[0x55FC](TS011F): Device seen - marking the device available and resetting counter
[0x55FC](TS011F): Update device availability -  device available: True - new availability: True - changed: False
[0xBAB0:1:0xef00] Received ZCL frame: b'\x09\x01\x01\x56\xD8\x10\x02\x00\x04\x00\x00\x00\x14'
[0xBAB0:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=1, command_id=1, *is_reply=True)
[0xBAB0:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:get_data(param=Command(status=86, tsn=216, command_id=528, function=0, data=[4, 0, 0, 0, 20]))
[0xBAB0:1:0xef00] Received command 0x01 (TSN 1): get_data(param=Command(status=86, tsn=216, command_id=528, function=0, data=[4, 0, 0, 0, 20]))
[0xBAB0:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
[0xBAB0:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
```

</details>

**Additional context**

[image]

Celer21, 2022-05-29:
I sold this device and bought supported one.

closed 2022-05-29 by Celer21

hesspoint, 2022-11-29:
Yes also got it. I can connect it with Home Assistant via ZHA but there are no entities and I cannot get any inputs via zha_event 

Anyone got this working? Or is that keypad not supported and I better return it?

besiktas97, 2024-01-02:
I am having the same issue. Which one did you bought Celer21 ?

Celer21, 2024-01-02:
> I am having the same issue. Which one did you bought Celer21 ?

I bought this one from Linkind https://www.zigbee2mqtt.io/devices/ZS130000078.html#linkind-zs130000078 and switch to Zigbee2MQTT. So i did not tested this one in ZHA.