zigpy/zha-device-handlers #1585: [BUG] Elko Super TR thermostat not working.

bendikrb, 2022-05-25:
**Describe the bug**
See attached screenshots.
Related to;
#1520
#999 

**Screenshots**
[image]
[image]

<details>
<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4919, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0x0b04"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "ELKO",
  "model": "Super TR",
  "class": "zhaquirks.elko.smart_super_thermostat.ElkoSuperTRThermostat"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
    "os_version": "5.13.0-41-generic",
    "supervisor": "2022.05.3",
    "host_os": "Ubuntu 21.10",
    "docker_version": "20.10.8",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "fontawesome": {
      "version": "2.1.5",
[... 519 lines cut ...]
    "endpoint_names": [
      {
        "name": "THERMOSTAT"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "22dc4bbe60414a0ebda147336f27daea",
    "area_id": "4504c49c9cab4ee9a2baaae74ae35325"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
2022-05-25 16:19:33 WARNING (MainThread) [homeassistant.components.zha.entity] climate.elko_super_tr_344b22fe_thermostat: can't set 'heat' mode. Supported modes are: [<HVACMode.OFF: 'off'>]
```

</details>

**Additional context**
Add any other context about the problem here.

thgjerde, 2022-06-12:
Broken for me too.

<details><summary>Debug log</summary>

```
2022-06-12 22:50:46 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=climate, service=set_hvac_mode, service_data=entity_id=climate.elko_super_tr_entre_thermostat, hvac_mode=heat>
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy.zcl] [0xE3AC:1:0x0201] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), manufacturer=0, tsn=19, command_id=<GeneralCommand.Write_Attributes: 2>, *is_reply=False)
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy.zcl] [0xE3AC:1:0x0201] Sending request: Write_Attributes(attributes=[Attribute(attrid=0x001C, value=TypeValue(type=SystemMode, value=<SystemMode.Heat: 4>)), Attribute(attrid=0x0406, value=TypeValue(type=Bool, value=<Bool.true: 1>)), Attribute(attrid=0x0411, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xE3AC), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=513, TSN=19, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x04\x00\x00\x13\x02\x1C\x00\x30\x04\x06\x04\x10\x01\x11\x04\x10\x00')
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=19)
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=513, SrcAddr=0xE3AC, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=138, SecurityUse=<Bool.false: 0>, TimeStamp=5550473, TSN=0, Data=b'\x0C\x00\x00\x13\x0B\x02\x84', MacSrcAddr=0xE3AC, MsgResultRadius=29)
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy.zcl] [0xE3AC:1:0x0201] Received ZCL frame: b'\x0C\x00\x00\x13\x0B\x02\x84'
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy.zcl] [0xE3AC:1:0x0201] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), manufacturer=0, tsn=19, command_id=11, *is_reply=True)
2022-06-12 22:50:46 DEBUG (MainThread) [zigpy.zcl] [0xE3AC:1:0x0201] Decoded ZCL frame: ElkoSuperTRThermostatCluster:Default_Response(command_id=2, status=<Status.UNSUP_MANUF_GENERAL_COMMAND: 132>)
2022-06-12 22:50:46 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE3AC:1:0x0201]: wrote {'system_mode': <SystemMode.Heat: 4>, 'device_on': 1, 'night_lowering': 0} attrs, Status: Default_Response(command_id=2, status=<Status.UNSUP_MANUF_GENERAL_COMMAND: 132>)
2022-06-12 22:50:46 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140601249930496] 'uint8_t' object is not iterable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1704, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1741, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 717, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/climate.py", line 413, in async_set_hvac_mode
    if await self._thrm.async_set_operation_mode(HVAC_MODE_2_SYSTEM[hvac_mode]):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 243, in async_set_operation_mode
    if not await self.write_attributes({"system_mode": mode}):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 298, in write_attributes
    return self.check_result(res)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 306, in check_result
    return all(record.status == Status.SUCCESS for record in res[0])
TypeError: 'uint8_t' object is not iterable
2022-06-12 22:50:47 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD789](Super TR): Device seen - marking the device available and resetting counter
2022-06-12 22:50:47 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD789](Super TR): Update device availability -  device available: True - new availability: True - changed: False
```

</details>

The `Status.UNSUP_MANUF_GENERAL_COMMAND` seems suspect to me.

torge70, 2022-07-05:
I have just added this thermostat to HA. Only 2 sensors showed up, all unavailable.
I have downloaded the quirk from #1638 above and installed it in custom_zha_quirks folder. HA now shows 7 sensors but they are all unknown/unavailable.

[image]

closed 2022-07-08 by dmulcahey

torge70, 2022-07-08:
I have installed the latest "smart_super_thermostat.py" from github in custom_zha_quirks. I can control the device from HA but reading data does not work correctly. Most entities returns no correct data and if i toggle og change setting on the thermostat, it's not updated in HA. I have tried to read the code or find documentation on how quirks work but I'm stuck.

How can I help getting the thermostat to work in HA?

bendikrb, 2022-07-11:
Still not working after #1638 (I have tried removing the device, restarting and then re-added it).
The climate entity now shows hvac state/mode and temperature, but I'm not able to adjust temperature or set hvac mode.

[image]

[image]

PR #1638 (2022-07-08): Fixes Elko Super TR
Fixes #1585