zigpy/zha-device-handlers #1588: [BUG] OSRAM Lightify Dimming Switches are not sending events.

IMarvinTPA, 2022-05-26:
**Describe the bug**
Sylvania Osram Lightify+ Dimming Switch are not working.
I have two of these switches, previously used on a Lightify network.
One of them only reports Dim Up and Dim Down events.  The other one reports no events.

**To Reproduce**
I paired the switches by holding both buttons for several seconds.
I went to the events developer tools and filtered for zha_event
Watched as mostly nothing happened except for one of them sends the move_with_on_off, move, and stop commands.

**Expected behavior**
I expected all five events for quick presses as well as long presses (and the release).

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3130",
  "class": "zhaquirks.centralite.cl_3130.CentraLite3130"
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Not sure where to look for these.
```

</details>

**Additional context**
Using Sonoff 3.0 USB Dongle Plus and running HA from docker on a linux host on an intel system.

dmulcahey, 2022-05-26:
I have many of these and they work fine. Try reconfiguring them.

IMarvinTPA, 2022-05-26:
I got "The device reconfiguration failed. Additional information may be available in the logs." and the yellow dot is on the OnOff binding.

closed 2022-11-30 by github-actions[bot]