zigpy/zha-device-handlers #1597: [Device Support Request] Add device TZE200_2hf7x9n3

caslubas, 2022-06-02:
Hi, Sorry for my bad english.

I recently added a device with the help of some members, following the steps described in #1535. Then I created a PR for the device to be integrated into the ZHA, I was successful with the integration, now I have a new device (TZE200_2hf7x9n3), but I am not having success this time, even following the mentioned procedure to the letter.

After editing the MODELS_INFO session, adding the device, I deleted it, restarted HA and added the device again (resetting device).

The session, from the ts0601_switch.py file, looks like this:

MODELS_INFO: [
             # ("_TZE200_kyfqmmyl", "TS0601"), ## candidate reported in #716
             ("_TZE200_tz32mtza", "TS0601"),
             ("_TZE200_2hf7x9n3", "TS0601"),

But HA didn't show the controls as expected (3 switches).

That's why I ask for help.

What should I do to try to solve the problem?

<details>

<summary>{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_2hf7x9n3",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}</summary>

</details>

<details>

<summary>{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.6.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.12",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Campo_Grande",
    "os_name": "Linux",
    "os_version": "5.10.0-13-amd64",
    "supervisor": "2022.05.3",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.14",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "alexa_media": {
      "version": "4.0.1",
      "requirements": [
        "alexapy==1.25.6",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "hacs": {
      "version": "1.25.5",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "nodered": {
      "version": "1.0.8",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.30.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.75",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 43720,
    "manufacturer": "_TZE200_2hf7x9n3",
    "model": "TS0601",
    "name": "_TZE200_2hf7x9n3 TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 164,
    "rssi": -59,
    "last_seen": "2022-06-02T12:52:23",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    },
    "entities": [],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "e17a5c10ef96517aebca8b1541398ce5",
    "area_id": null
  }
}</summary>

</details>

<details>
<summary>[Additional logs](https://caslubas.duckdns.org:8123/frontend_latest/34ab12a7.js:110:808 Uncaught TypeError: Cannot read properties of undefined (reading 'entity_id')
13:08:59 – (ERRO) components/system_log/__init__.py
[0xA177:1:0xef00] Unknown cluster command 17 b'\x00e\xd1'
13:07:59 – (ALERTA) /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py - a mensagem ocorreu pela primeira às 13:07:56 e apareceu 18 vezes
Setup of platform growatt_server is taking longer than 60 seconds. Startup will proceed without waiting any longer.
13:01:26 – (ERRO) Sensor
Waiting on integrations to complete setup: sensor.growatt_server
13:01:20 – (ALERTA) bootstrap.py
Setup of sensor platform growatt_server is taking over 10 seconds.
13:00:36 – (ALERTA) Sensor)</summary>

</details>
[image]

javicalle, 2022-06-03:
And here comes the new GPP versions...

Add the new `GreenPowerProxy` import like:
```python
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
```

Then add the new class, for example between `TuyaSingleSwitchTO` and `TuyaDoubleSwitchTO` (keep the 2 blank lines separation between classes):
```python
class TuyaSingleSwitch_GP(TuyaSingleSwitchTO):
    """Tuya single channel switch with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_2hf7x9n3", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```

You know how it follows 😉

javicalle, 2022-06-03:
BTW, python-masters help will be appreciate here.

The file has up to 6 switch device. I refuse to replicate the code from all devices.
Any elegant way to extend classes so as not to replicate code?

I have found 2 variants:
* update at `__init__()`:
https://github.com/zigpy/zha-device-handlers/blob/a55235f253a2d1ea98f04b6518792e32e6c91ebe/zhaquirks/xbee/xbee_io.py#L18-L29
* the copy & update
https://github.com/zigpy/zha-device-handlers/blob/a55235f253a2d1ea98f04b6518792e32e6c91ebe/zhaquirks/tuya/__init__.py#L873-L877

Any recommendation at this topic?

caslubas, 2022-06-03:
> And here comes the new GPP versions...
> 
> Add the new `GreenPowerProxy` import like:
> 
> ```python
> from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
> ```
> 
> Then add the new class, for example between `TuyaSingleSwitchTO` and `TuyaDoubleSwitchTO` (keep the 2 blank lines separation between classes):
> 
> ```python
> class TuyaSingleSwitch_GP(TuyaSingleSwitchTO):
>     """Tuya single channel switch with GreenPowerProxy cluster device."""
> 
>     signature = {
>         MODELS_INFO: [
>             ("_TZE200_2hf7x9n3", "TS0601"),
>         ],
>         ENDPOINTS: {
>             # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1
>             # input_clusters=[0, 4, 5, 61184]
>             # output_clusters=[10, 25]>
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaOnOffManufCluster.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
>             # input_clusters=[]
>             # output_clusters=[33]
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> 
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     MoesSwitchManufCluster,
>                     TuyaOnOff,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         }
>     }
> ```
> 
> You know how it follows wink

hello javicalle,

I made the addition of GreenPowerProxy in the ts0601_switch.py file,

I added the device to the TuyaTripleSwitchTO class, in model_info, looking like this:

<details>
MODELS_INFO: [
             ("_TZE200_tz32mtza", "TS0601"),
             ("_TZE200_2hf7x9n3", "TS0601"),
         ]
</details>

So I saved the file, deleted the old device from the ZHA, restarted the HA, added the device again, but it still has no controls (3 switches).

What procedure should I try?

javicalle, 2022-06-03:
No, no, your device need a new device signature like the `TuyaSingleSwitch_GP` that I proposed.
If your device is a 'triple' switch, you need a class like:
```python
class TuyaTripleSwitch_GP(TuyaSwitch):
    """Tuya single channel switch with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_2hf7x9n3", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

Remove the previous `TuyaSingleSwitch_GP` and any reference to model `_TZE200_2hf7x9n3` from other classes.

caslubas, 2022-06-03:
I made the necessary corrections, adding the TuyaTripleSwitch_GP class, now the 3 controls appear, but they do not turn off or on the lamps

Here's the new data:

<details>

<summary>Device signature</summary>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_2hf7x9n3",
  "model": "TS0601",
  "class": "ts0601_switch.TuyaTripleSwitch_GP"
}
</details>

<details>

<summary>Diagnostic information</summary>
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.6.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.12",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Campo_Grande",
[... 250 lines cut ...]
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "d8879505136fdc81f1eef9cf5c994821",
    "area_id": "cozinha"
  }
}
</details>

<details>
<summary>[Additional logs]</summary>
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:1411
First occurred: 16:49:56 (4 occurrences)
Last logged: 16:49:56

[0xF5FC:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=230, dp=7, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0xF5FC:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=231, dp=8, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0xF5FC:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=232, dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0xF5FC:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=238, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))

</details>

javicalle, 2022-06-03:
Try this changes:

1. add a new import `EnchantedDevice` like:
```python:
from zhaquirks.tuya.mcu import (
    EnchantedDevice,
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)
```
2. modify the new class inheritance like:
```python
class TuyaTripleSwitch_GP(EnchantedDevice, TuyaSwitch):
```

But there are also 4 new DPs: 7, 8, 9 and 16 probably device will extra functions.

Can you post any link to product or vendor?

caslubas, 2022-06-03:
The changes were made, looking like this:

<details>
<summary>ts0601_switch.py file:</summary>
"""Tuya DP based switches."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import MoesSwitchManufCluster, TuyaOnOff, TuyaOnOffManufCluster
from zhaquirks.tuya.mcu import (
    EnchantedDevice,
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaSingleSwitchTI(TuyaSwitch):
    """Tuya single channel switch time on in cluster device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000,0x0004, 0x0005,0x000a, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 10, 61184] output_clusters=[25]>
        MODELS_INFO: [("_TZE200_7tdtqgwv", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }
[... 281 lines cut ...]
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
</details>

Link de venda do produto, selecione:
Black-ZY-US, need neutral wire, 3-gang
[Link de venda do produto, selecione:
Black-ZY-US, need neutral wire, 3-gang
[https://pt.aliexpress.com/item/4001181765345.html?spm=a2g0s.8937460.0.0.15fb2e0eNfgcxp&gatewayAdapt=glo2bra](https://pt.aliexpress.com/item/4001181765345.html?spm=a2g0s.8937460.0.0.15fb2e0eNfgcxp&gatewayAdapt=glo2bra)

Continues without turning on or off

javicalle, 2022-06-03:
Ouch! Wrong approach.
Remove the `EnchantedDevice` inheritance, is no needed
```python
class TuyaTripleSwitch_GP(TuyaSwitch):
```
Replace the `TuyaOnOff` in the replacement part like:
```python
replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
```

caslubas, 2022-06-03:
The device is now not discovered!

javicalle, 2022-06-03:
Any errors in the logs? 

Double check the file format, there must be a typo somewhere.

javicalle, 2022-06-03:
I have updated the previous comment with the new `replacement`:
https://github.com/zigpy/zha-device-handlers/issues/1597#issuecomment-1146313309

caslubas, 2022-06-03:
Now yes! we were successful!
Now we have the device with all controls working properly!
Thank you for your patience @javicalle

javicalle, 2022-06-03:
I'm curious about what can be these new DPs. Maybe are nothing, but you will have these errors in your logs. If it is ok to you, go ahead with the PR 👍🏻

caslubas, 2022-06-03:
This warning message appeared in the log:

<details>
<summary>Logger: zigpy.zcl</summary>
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:1411
First occurred: 19:48:51 (6 occurrences)
Last logged: 19:48:57

[0x3BD9:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=252, dp=8, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0) , raw=b'\x00\x00\x00\x00', *payload=0)))
[0x3BD9:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=254, dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0) , raw=b'\x00\x00\x00\x00', *payload=0)))
[0x3BD9:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=0, dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0) , raw=b'\x00\x00\x00\x00', *payload=0)))
[0x3BD9:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=2, dp=8, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0) , raw=b'\x00\x00\x00\x00', *payload=0)))
[0x3BD9:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=4, dp=7, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0) , raw=b'\x00\x00\x00\x00', *payload=0)))
</details>

Can I disregard the log message, which seems to be just an "alert" and open a Pull Request?

javicalle, 2022-06-04:
> Can I disregard the log message, which seems to be just an "alert" and open a Pull Request?

It's fine to me 🤷🏻‍♂️ 

'router' will be the correct word.

PR #1604 (2022-06-08): New Tuya GPP switches
Here comes the new Tuya switches with the GreePowerProxy cluster.
Note that they seem to make use of the `NO_MANUFACTURER_ID` for messaging with the MCU.

Definitions for 1,2,3 and 4 switch devices have been added, although we only have 2 and 3 switch models_info. 
To facilitate the inclusion of new devices, those with 1 and 4 switches have also been defined with a fake model_info.

Fixes: #1597
Fixes: #1603
closed 2022-06-04 by caslubas