zigpy/zha-device-handlers #1602: [Device Support Request] TS0601 _TZE200_r0jdjrvi

ForresT1984, 2022-06-04:
The smart curtain motor only integrates as a router, so no entities show up. I can’t control the motor at all 

I hope there will be a quirk for the motor in an update. 

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_r0jdjrvi",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```
Hopefully some one can help. 
[image]

javicalle, 2022-06-05:
I believe that you can add your model info at the `TuyaMoesCover0601` (`ts0601_cover.py` file) and `TUYA_COVER_COMMAND` (`__init__.py` file).

There is a good guide to create your local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

You need to copy the `ts0601_cover.py` and `__init__.py` files to your local zha_quirk folder and edit the content:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_cover.py

ForresT1984, 2022-06-06:
> You need to copy the `ts0601_cover.py` and `__init__.py` files to your local zha_quirk folder and edit the content:

Thanks for the reply, 

Just to be certain,

I copy the files via File editor, make the change in `Configuration.yaml` , adding the custom quirks folder. 
Then i copy the files and add them to the folder. 

In  `__init__.py` i add the cover type '_TZE200_r0jdjrvi' to the list of devices?
In `ts0601_cover.py` i copy the text you provided in your command, but as i see it, it is searching for a specific device, so i have to copy the content of the device as in my post to your posted code?

Again thanks for your help so far.

javicalle, 2022-06-06:
You need to import the `GreenPowerProxy` at the `ts0601_cover.py` file like:
```python
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, OnOff, Ota, Scenes, Time
```

I'm not sure, but maybe you also need to make the tuya imports relative:
```python
# from zhaquirks.tuya import (
from . import (
```
Only if there's problems with the `zhaquirks.tuya` classes 

> so i have to copy the content of the device as in my post to your posted code?

I'm not sure what you mean here. My code already have the reference to the `_TZE200_r0jdjrvi`model.

ForresT1984, 2022-06-06:
I just got a lot further, 

I have the open, close and stop button: 
[image]

I can also click them.
But when i do, i get this error message; Kan service cover/close_cover niet aanroepen. '_TZE200_r0jdjrvi' It is partially in dutch. I think in English it will be something like this: can't call service cover/close_cover. '_TZE200_r0jdjrvi'

javicalle, 2022-06-06:
Any other info in your logs?

ForresT1984, 2022-06-06:
`` Last logged: 11:48:50

    [140020887927824] '_TZE200_r0jdjrvi'
    [140020792716400] '_TZE200_r0jdjrvi'

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1704, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1741, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 717, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 141, in async_close_cover
    res = await self._cover_channel.down_close()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 65, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 1138, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_r0jdjrvi'
``

This is the log.

Yes i tried to repair. did it without issues.
Even deleted the device and re-add it.

Still same problem

javicalle, 2022-06-06:
You need to add your device in the `TUYA_COVER_COMMAND` (in the `__init__.py` file)
Something like:
```python
TUYA_COVER_COMMAND = {
    "_TZE200_r0jdjrvi": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
...
```

ForresT1984, 2022-06-06:
Did it, but unfortunately still the same error message:

``[140227074703808] '_TZE200_r0jdjrvi'

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1704, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1741, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 717, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 135, in async_open_cover
    res = await self._cover_channel.up_open()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 65, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 1138, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_r0jdjrvi'
``

javicalle, 2022-06-06:
Try to change the relative import in your local `ts0601_cover` file:
```python
# from zhaquirks.tuya import (
from . import (
```

Once you get it working in HA we can try to address the inverted issue.

ForresT1984, 2022-06-06:
I don't get what you mean:

From this:

` from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)`
To this:

`from . import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)`

javicalle, 2022-06-06:
Yes, that is the change.

ForresT1984, 2022-06-06:
`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 339, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_cover.py", line 13, in <module>
    from . import (
ImportError: attempted relative import with no known parent package
`

Sorry, got an error again on restarting.

javicalle, 2022-06-06:
Ummmm, could you try with:
```python
from custom_zha_quirks import ( TuyaManufacturerWindowCover, TuyaManufCluster, TuyaWindowCover, TuyaWindowCoverControl, )
```

Delete any `__pycache__` folder inside your local `custom_zha_quirks` folder and restart HA.

ForresT1984, 2022-06-06:
Oke,

now i won't get an error message when i press the button, but nothing is happening wen i press the open, stop or close button. 
And open and close are still inverted. 

Again, thanks for all the help so far, on my own i never would gotten this far.

javicalle, 2022-06-06:
The Tuya cover aren't easy devices to quirk...

We would need the logs of the device to understand what could be happening. Enable in HA the debug logging for ZHA and attach the logs from operating the device from HA and/or from the remote:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Please, try to keep the logs as simple as possible, attaching only the relevant info for every command.
And put your logs inside some code markdown:
````
```
example of logs from HA opening command inside code marks
```
````
or even:
````
<details>
<!-- extra blank lines are important. Don't remove -->
<summary>Open command</summary>

```
example of logs from HA opening command inside code marks
```

</details>
````

ForresT1984, 2022-06-06:
<details>

<summary>Open command</summary>

```
2022-06-06 14:52:59 DEBUG (MainThread) [zigpy.zcl] [0xF381:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=58, command_id=0, *is_reply=False)
2022-06-06 14:52:59 DEBUG (MainThread) [zigpy.zcl] [0xF381:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 0]))
2022-06-06 14:52:59 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 58 under 59 request id, data: b'0541113a0000000104000100'
2022-06-06 14:52:59 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (27, 59, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0xF381 endpoint=1>, 260, 61184, 1, b'\x05A\x11:\x00\x00\x00\x01\x04\x00\x01\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)

2022-06-06 14:56:49 DEBUG (MainThread) [zigpy.zcl] [0xF381:1:0xef00] Received ZCL frame: b'\t\x97\x02\x00P\x06\x01\x00\x01\x00'
2022-06-06 14:56:49 DEBUG (MainThread) [zigpy.zcl] [0xF381:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=151, command_id=2, *is_reply=True)
2022-06-06 14:56:49 DEBUG (MainThread) [zigpy.zcl] [0xF381:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:set_data_response(param=Command(status=0, tsn=80, command_id=262, function=0, data=[1, 0]))
2022-06-06 14:56:49 DEBUG (MainThread) [zigpy.zcl] [0xF381:1:0xef00] Received command 0x02 (TSN 151): set_data_response(param=Command(status=0, tsn=80, command_id=262, function=0, data=[1, 0]))
2022-06-06 14:56:49 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf381>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09970200500601000100'
```

</details>

Is this what you need?

javicalle, 2022-06-06:
Yes, that's perfect.
Can you attach the same command from the remote?

javicalle, 2022-06-09:
(Just a little, not too much...)

**Disclaimer**: _There will be dragons here_. It is very likely that the code does not work or works very erratically (mixed up/down/stop. The version is completely experimental and has not been minimally tested before.

There is my proposal:

<details>
<summary>ts0601_mcu_cover.py</summary>

```python
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOffNM,
)

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: (
                "accurate_calibration",
                t.enum8,
                True,
            ),  # 0: calibration started, 1: calibration finished
            0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse
            0xF003: ("travel", t.uint16_t, True),  # 30 to 9000 (units of 0.1 seconds)
        }
    )
[... 205 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaOnOffNM,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

Remove any previous quirk and create the new file in your local quirk folder.
Delete cache and restart.

javicalle, 2022-06-10:
@ForresT1984 have you removed the previous quirk?
I can't see where is the code using the `TuyaManufCluster`. Attach the full exception to see where the error comes from.

ForresT1984, 2022-06-14:
> 

Yes, i removed the old quirk files. 

This is the full logbook error
`
[140225519239520] '_TZE200_r0jdjrvi'

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1704, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1741, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 717, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 135, in async_open_cover
    res = await self._cover_channel.up_open()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 65, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 1138, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_r0jdjrvi'
`

javicalle, 2022-06-14:
Caught!

There was an error in the `replacement` part (the `TuyaWindowCovering` reference):
```python
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaOnOffNM,
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

Could you fix it and try again?

ForresT1984, 2022-06-14:
This file ``__init__.py`` also need to be deleted?

When i do i get this error message:
``
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/mcu/__init__.py:216
First occurred: 12:26:05 (2 occurrences)
Last logged: 12:26:08

[0xF381:1:0xef00] No cluster_dp found for 1, on_off
``

And this one
`` 
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/mcu/__init__.py:243
First occurred: 12:26:05 (2 occurrences)
Last logged: 12:26:08

[0xF381:1:0xef00] no MCU command for data TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True, manufacturer=-1)
``
All i got in the folder is this file. The others are removed.
[image]

javicalle, 2022-06-14:
I don't know if you are using the `custom_zha_quirks` mechanism or you are editing the `site-packages` files.
My proposal is a new file that can (must?) work with the latest version from zhaquirks.

If you have edited the `/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/mcu/init.py` (or any else) you need to restore the package version.
If you have added a `__init__.py` file to your `custom_zha_quirks` folder, remove it.

ForresT1984, 2022-06-14:
I'm using the custom_zha_quirks.

I never edited the files you say, i don't even know how to access them.

``
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/mcu/__init__.py:243
First occurred: 15:04:59 (2 occurrences)
Last logged: 15:05:10

[0xF381:1:0xef00] no MCU command for data TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True, manufacturer=-1)
``

javicalle, 2022-06-14:
I can't see the image with your folder contents, but you only need one file inside: `ts0601_mcu_cover.py`

The error is related to an `on_off` command. Usually this would be a switch. I don't know how can be related to the cover. There is a `TuyaOnOffNM` left in the code (that would be a switch for the `motor_steering`, but not implemented yet).

Are the cover commands reporting any log?

javicalle, 2022-06-14:
Can you get and attach the logs when you operate the cover form HA? Just like you did in https://github.com/zigpy/zha-device-handlers/issues/1602#issuecomment-1147418638

kolmakova, 2022-06-20:
@javicalle @ForresT1984 Hey, look into https://github.com/zigpy/zha-device-handlers/issues/1294. This motor have the same device signature and guys found a solution there. My device is **_TZE200_nogaemzt**, has same signature, too, and this solution works, but buttons still aren't working. I'm only able to use silder and control covering percentage, but this is much better than nothing.

javicalle, 2022-06-20:
Without the debug log from the buttons press I am not able to see where this `on_off` attribute comes from.

You can try to change the `dp_to_attribute` 1 and 4 in `TuyaWindowCoverManufCluster` class like this:

```python
        1: DPToAttributeMapping(
            TuyaWindowCovering.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.ENUM,
        ),
        .../...
        4: DPToAttributeMapping(
            TuyaWindowCovering.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.ENUM,
            endpoint_id=2,
        ),
```

ForresT1984, 2022-06-24:
Sorry, for the delay, it was a bit hectic at work. I will try to upload the log this weekend.

ForresT1984, 2022-06-24:
<details>

<summary>Open command via RC</summary>

```
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Received ZCL frame: b'\t\xc6\x02\x00\xc7\x02\x02\x00\x04\x00\x00\x00\x00'
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=198, command_id=2, *is_reply=True)
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=199, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Received command 0x02 (TSN 198): set_data_response(data=TuyaCommand(status=0, tsn=199, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-06-24 08:47:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC958:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 0
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=198, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-06-24 08:47:26 DEBUG (MainThread) [homeassistant.components.zha.cover] setting position: 0
2022-06-24 08:47:26 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.tze200_r0jdjrvi_ts0601_fb63f387_window_covering, old_state=<state cover.tze200_r0jdjrvi_ts0601_fb63f387_window_covering=closed; current_position=0, friendly_name=_TZE200_r0jdjrvi TS0601 fb63f387 window_covering, supported_features=15 @ 2022-06-24T08:47:02.421998+02:00>, new_state=<state cover.tze200_r0jdjrvi_ts0601_fb63f387_window_covering=open; current_position=100, friendly_name=_TZE200_r0jdjrvi TS0601 fb63f387 window_covering, supported_features=15 @ 2022-06-24T08:47:26.447910+02:00>>
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 198 under 10 request id, data: b'18c60b0200'
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc958>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09c60200c70202000400000000'
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x535c>, 1, 260, 61184, b'\tV\x01\x00\x00\x13\x02\x00\x04\x00\x00\x02\t', 0, 175, 255, 247, 39, 62, 2, -45]
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Received ZCL frame: b'\tV\x01\x00\x00\x13\x02\x00\x04\x00\x00\x02\t'
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=86, command_id=1, *is_reply=True)
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=0, dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\t\x02\x00\x00', *payload=521)))
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Received command 0x01 (TSN 86): get_data(data=TuyaCommand(status=0, tsn=0, dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\t\x02\x00\x00', *payload=521)))
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=86, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-24 08:47:26 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
```

</details>

<details>

<summary>Close command via RC</summary>

```
2022-06-24 08:47:01 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x535c>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'092d010002020200040000016c'
2022-06-24 08:47:01 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 208, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x535C endpoint=1>, 260, 61184, 1, b'\x18-\x0b\x01\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
[... cut ...]
2022-06-24 08:47:02 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 209: 00
2022-06-24 08:47:02 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xd1 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xc958 endpoint=1>, status: 0x00
2022-06-24 08:47:02 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-06-24 08:47:02 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
```

</details>

javicalle, 2022-06-24:
Your device is sending reports for DP_2 (payload `0` for close and `100` for open):
```
TuyaWindowCoverManufCluster:set_data_response(
  data=TuyaCommand(
    status=0, 
    tsn=196, 
    dp=2, 
    data=TuyaData(
      dp_type=<TuyaDPType.VALUE: 2>, 
      function=0, 
      raw=b'd\x00\x00\x00', 
      *payload=100
    )
  )
)
```

So it seems that the open/close commands must be translated to the `current_position_lift_percentage` attribute. I can try with another implementation, but which command triggers the stop command?
Please attache the logs from the RC stop button.

javicalle, 2022-06-24:
This would work for the open and close commands from HA. Replace it in the `TuyaWindowCovering` class:
```python
        # (upopen, downclose, stop)
        if command_id in (0x0002):  # ¿0x0003: continue?
            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_attr="curtain_switch",
                attr_value=TUYA2ZB_COMMANDS[command_id],  # convert tuya2zigbee command
                expect_reply=expect_reply,
                manufacturer=manufacturer,
            )
            self.endpoint.device.command_bus.listener_event(
                TUYA_MCU_COMMAND,
                cluster_data,
            )
            return foundation.GENERAL_COMMANDS[
                foundation.GeneralCommand.Default_Response
            ].schema(command_id=command_id, status=foundation.Status.SUCCESS)

        # (go_to_lift_percentage)
        elif command_id in (0x0000, 0x0001, 0x0005):
            if command_id == 0x0000:
                lift_value = 100
            elif command_id == 0x0001:
                lift_value = 0
            elif command_id == 0x0005:
                lift_value = args[0]

            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_attr="current_position_lift_percentage",
                attr_value=lift_value,
                expect_reply=expect_reply,
                manufacturer=manufacturer,
            )
            self.endpoint.device.command_bus.listener_event(
                TUYA_MCU_COMMAND,
                cluster_data,
            )
            return foundation.GENERAL_COMMANDS[
                foundation.GeneralCommand.Default_Response
            ].schema(command_id=command_id, status=foundation.Status.SUCCESS)

```

ForresT1984, 2022-06-25:
I get an error when i move the slider with the new code. The stop button on the remote is not registered in home assistant.

 <details>

<summary>Error code</summary>

```
2022-06-25 11:06:04 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF9A4](TRADFRI bulb GU10 WW 400lm): Device seen - marking the device available and resetting counter
2022-06-25 11:06:04 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF9A4](TRADFRI bulb GU10 WW 400lm): Update device availability -  device available: True - new availability: True - changed: False
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x535c>, 1, 260, 61184, b'\t,\x01\x00\x00\x13\x02\x00\x04\x00\x00\x02\x0b', 0, 175, 255, 244, 149, 65, 2, -46]
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Received ZCL frame: b'\t,\x01\x00\x00\x13\x02\x00\x04\x00\x00\x02\x0b'
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=44, command_id=1, *is_reply=True)
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=0, dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x0b\x02\x00\x00', *payload=523)))
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Received command 0x01 (TSN 44): get_data(data=TuyaCommand(status=0, tsn=0, dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x0b\x02\x00\x00', *payload=523)))
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=44, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x535C:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-06-25 11:06:04 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.tze200_yvx5lh6k_ts0601_humidity, old_state=<state sensor.tze200_yvx5lh6k_ts0601_humidity=52.2; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=luchtkwaliteitsensor.zolder humidity @ 2022-06-25T11:06:02.117920+02:00>, new_state=<state sensor.tze200_yvx5lh6k_ts0601_humidity=52.3; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=luchtkwaliteitsensor.zolder humidity @ 2022-06-25T11:06:04.609050+02:00>>
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 44 under 110 request id, data: b'182c0b0100'
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x535c>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'092c010000130200040000020b'
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 110, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x535C endpoint=1>, 260, 61184, 1, b'\x18,\x0b\x01\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 110]
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 110: 00
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x6e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x535c endpoint=1>, status: 0x00
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x50f8>, 1, 260, 2820, b'\x18\xd4\n\x08\x05!\x90\x02', 0, 175, 239, 244, 149, 65, 2, -64]
2022-06-25 11:06:04 DEBUG (MainThread) [zigpy.zcl] [0x50F8:1:0x0b04] Received ZCL frame: b'\x18\xd4\n\x08\x05!\x90\x02'
[... cut ...]
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaOnOffNM,
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

javicalle, 2022-06-25:
My fault. Replace in the first if:
```python
        # (upopen, downclose, stop)
        if command_id == 0x0002:  # ¿0x0003: continue?
```

> The stop button on the remote is not registered in home assistant.

Just to be sure.
* Does the remote have a stop button?
* Does the remote stop button stop the cover?

The remote control command should leave some trace in the HA logs even if it is not reflected in the UI. We need to know what command the remote is sending in order to emulate it in HA.

ForresT1984, 2022-06-25:
Does the remote have a stop button? Yes
    Does the remote stop button stop the cover? Yes

Did the changes, nothing, sorry. 

Here is the log file:

<details>

<summary>Open via HA</summary>

```
2022-06-25 15:14:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x553A:1:0x0b04]: async_update
2022-06-25 15:14:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x553A:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_current', 'rms_voltage']
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy.zcl] [0x553A:1:0x0b04] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=108, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy.zcl] [0x553A:1:0x0b04] Sending request: Read_Attributes(attribute_ids=[1291, 1288, 1285])
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 108 under 109 request id, data: b'006c000b0508050505'
2022-06-25 15:14:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1052:1:0x0b04]: async_update
2022-06-25 15:14:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1052:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_current', 'rms_voltage']
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy.zcl] [0x1052:1:0x0b04] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=110, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy.zcl] [0x1052:1:0x0b04] Sending request: Read_Attributes(attribute_ids=[1291, 1288, 1285])
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 110 under 111 request id, data: b'006e000b0508050505'
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 103]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (24, 105, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x5DCD endpoint=1>, 260, 2820, 1, b'\x00h\x00\x0b\x05\x08\x05\x05\x05', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 38>, 105]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (24, 107, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x50F8 endpoint=1>, 260, 2820, 1, b'\x00j\x00\x0b\x05\x08\x05\x05\x05', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 38>, 107]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (24, 109, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x553A endpoint=1>, 260, 2820, 1, b'\x00l\x00\x0b\x05\x08\x05\x05\x05', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 38>, 109]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (24, 111, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x1052 endpoint=1>, 260, 2820, 1, b'\x00n\x00\x0b\x05\x08\x05\x05\x05', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 38>, 111]
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 103: 00
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x67 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x24ec endpoint=1>, status: 0x00
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 105: 00
2022-06-25 15:14:28 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x69 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x5dcd endpoint=1>, status: 0x00
[... cut ...]
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 16: 00
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x10 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xc958 endpoint=1>, status: 0x00
```

</details>

tony-el-hawi, 2022-08-23:
Hello hard working fellas. Are there any new updates regarding this issue?
I face the exact same problem with the Tuya zigbee curtain motor (manufacturer=_TZE200_r0jdjrvi, model=TS0601). I manage to add the device in ZHA but no entities are associated with it. Did the zha quirk approach work? is there another way to resolve the conundrum?

codypet, 2022-09-16:
Hi everyone.  Is it possible to tag along this and get support for this model roller shade as well?  _TZE200_7eue9vhc I'm having the same issue of a cover not coming up.   Perhaps the two are related?

javicalle, 2022-09-22:
@tony-el-hawi if your motor don't follow the `WindowCovering` standar you will need a quirk to integrate it in HA.
@codypet without all the info it's hard to say. It will be better if you open a new issue with all the info possible.

javicalle, 2022-09-22:
I haven't been able to detect the `stop` command in the logs.
It would be necessary for someone to test enabling the logs and repeatedly press the stop button on the remote to be able to identify the command in the logs.
If the device does not report the stop command, the implementation approach will have to be changed

codypet, 2022-09-22:
@javicalle No worries.  I ended up learning my way through doing it myself and its solution is #1763.  Its been working flawlessly on my instance since I updated the quirk.

caradas, 2022-11-21:
Hello, is something missing to complete this?
I have the motor and managed to get the above mentionend quirk to add the entities. But they seem to have no function at all.

javicalle, 2022-11-21:
If you have need to configure the local quirk, or you are in a old version of HA o your device is not the same because that quirk has been merged and included in the latests versions of HA.
Please open another issue with all the required info to get help.

javicalle, 2022-11-21:
Fixed by: #1763

closed 2022-11-21 by javicalle