zigpy/zha-device-handlers #1603: [Device Support Request]

caslubas, 2022-06-05:
I have a new device to integrate, like other times, I started by adding the device to the MODELS_INFO of the TuyaDoubleSwitchTO class, but it didn't show the 2 controls (2-gang).

Here is the link to purchase the device:
Select: Black-ZY-US, need neutral wire, 2-gang, US.
[_TZE200_7deq70b8](https://pt.aliexpress.com/item/4001181765345.html?spm=a2g0s.8937460.0.0.3e642e0enT2jWC&gatewayAdapt=glo2bra)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_7deq70b8",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.6.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.12",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Campo_Grande",
    "os_name": "Linux",
    "os_version": "5.10.0-13-amd64",
    "supervisor": "2022.05.3",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.14",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
[... 439 lines cut ...]
[0x79B7:1:0x0000] Attribute report received: 0xFFDF='�v.*\x13�}.*e�}.*\x12'
[0x79B7:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=21, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
[0x79B7:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
Duplicate 21 TSN
[0x4737](TS0601): Device seen - marking the device available and resetting counter
[0x4737](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x3BD9](TS0601): Device seen - marking the device available and resetting counter
[0x3BD9](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x3BD9:1:0x0000] Received ZCL frame: b'\x08\xa7\n\x01\x00 D\xe2\xff 0\xe4\xff \x00'
[0x3BD9:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=167, command_id=10, *is_reply=True)
[0x3BD9:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=68)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=48)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])
[0x3BD9:1:0x0000] Received command 0x0A (TSN 167): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=68)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=48)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])
[0x3BD9:1:0x0000] Attribute report received: app_version=68, 0xFFE2=48, 0xFFE4=0
[0x3BD9:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=167, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
[0x3BD9:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
[0x4A6E](MS01): Device seen - marking the device available and resetting counter
[0x4A6E](MS01): Update device availability -  device available: True - new availability: True - changed: False
[0x9750](TH01): Device seen - marking the device available and resetting counter
[0x9750](TH01): Update device availability -  device available: True - new availability: True - changed: False
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-06-05:
The quirk for this device would be:
```python
class TuyaDoubleSwitch_GP(TuyaSwitch):
    """Tuya double channel switch with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_7deq70b8", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

caslubas, 2022-06-05:
Yes, I think so, it is like this in my file, with the changes in the initial lines, in which GreenPowerProxy was added in line 3, looking like this:
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

and

from zhaquirks.const import (
     DEVICE_TYPE,
     ENDPOINTS,
     INPUT_CLUSTERS,
     MODELS_INFO,
     OUTPUT_CLUSTERS,
     PROFILE_ID,
)

Note, that I removed the EnchantedDevice option, because it gave an error at the time of "merge", but in my file it has, the copy of the file that is working correctly is like this:

<details>
<summary>My File ts0601_swicth.py</summary>
```
"""Tuya DP based switches."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    EnchantedDevice,
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaSingleSwitchTI(TuyaSwitch):
    """Tuya single channel switch time on in cluster device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000,0x0004, 0x0005,0x000a, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 10, 61184] output_clusters=[25]>
        MODELS_INFO: [("_TZE200_7tdtqgwv", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
[... 288 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```
</details>

caslubas, 2022-06-05:
How do I put entire code so it doesn't appear "split"?

javicalle, 2022-06-05:
Keep the blank lines inside the markdown (before and after the ```), like this:

````
<details>
<summary>ts0601_swicth.py</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>
````

javicalle, 2022-06-05:
> Note, that I removed the EnchantedDevice option

It was added during a test, but not necessary. Just the `GreenPowerProxy` and the new classes.

caslubas, 2022-06-05:
Success!, now the device has displayed the controls, are working fine!
Thanks again for your help and patience with me!

Should I open a new PR or will you take care of it?
I don't want to steal credit from those who deserve it.

<details>
<summary>My ts0601_switch.py file now looks like this:</summary>

```yaml
"""Tuya DP based switches."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaSingleSwitchTI(TuyaSwitch):
    """Tuya single channel switch time on in cluster device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000,0x0004, 0x0005,0x000a, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 10, 61184] output_clusters=[25]>
        MODELS_INFO: [("_TZE200_7tdtqgwv", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOff,
                ],
[... 345 lines cut ...]
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
Don't remove the extra line breaks outside the ``` marks.
```

</details>

javicalle, 2022-06-05:
Check this part:
````
```yaml """Tuya DP based switches.""" from zigpy.profiles import zha from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zhaquirks.const import (
DEVICE_TYPE,
ENDPOINTS,
INPUT_CLUSTERS,
MODELS_INFO,
OUTPUT_CLUSTERS,
PROFILE_ID,
)
...
````

and replace with (new break line and `python` formatting):
````
```python
"""Tuya DP based switches.""" from zigpy.profiles import zha from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zhaquirks.const import (
DEVICE_TYPE,
ENDPOINTS,
INPUT_CLUSTERS,
MODELS_INFO,
OUTPUT_CLUSTERS,
PROFILE_ID,
)
...
````

javicalle, 2022-06-05:
> Should I open a new PR or will you take care of it?
> I don't want to steal credit from those who deserve it.

I have create already the PR #1604. I hope that you don't mind.
This PR include all the quirks for 1 to 4 gang devices.

caslubas, 2022-06-05:
> I have create already the PR #1604. I hope that you don't mind

care? No, I do not care.
After all, the credit is all yours!
For me, the important thing is to have the device working and learning along the way!

caslubas, 2022-06-05:
> Check this part:
> 
> ...
> ```
> 
> and replace with (new break line and `python` formatting):
> 

Too many errors, ZHA integration is failing after modification!

<details>
<summary>Logs:</summary>
Logger: homeassistant.config_entries
Source: components/zha/__init__.py:99
First occurred: 14:25:57 (1 occurrences)
Last logged: 14:25:57

Error setting up entry socket://192.168.100.114:8888 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 339, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 846, in exec_module
  File "<frozen importlib._bootstrap_external>", line 983, in get_code
  File "<frozen importlib._bootstrap_external>", line 913, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_switch.py", line 1
    """Tuya DP based switches.""" from zigpy.profiles import zha from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
                                  ^
SyntaxError: invalid syntax
</details

javicalle, 2022-06-05:
> Too many errors, ZHA integration is failing after modification!

Upssss, no, no. This comment was for editing your Github comment, no the quirk code.

caslubas, 2022-06-05:
Sorry, I'm learning!
Can I close the issue?

javicalle, 2022-06-05:
There's no need. It will be closed when PR #1604 merged. 👍🏻

javicalle, 2022-06-05:
But... I think it will be better to close the PR #1601

PR #1604 (2022-06-08): New Tuya GPP switches
Here comes the new Tuya switches with the GreePowerProxy cluster.
Note that they seem to make use of the `NO_MANUFACTURER_ID` for messaging with the MCU.

Definitions for 1,2,3 and 4 switch devices have been added, although we only have 2 and 3 switch models_info. 
To facilitate the inclusion of new devices, those with 1 and 4 switches have also been defined with a fake model_info.

Fixes: #1597
Fixes: #1603
closed 2022-06-08 by dmulcahey