zigpy/zha-device-handlers #1608: [Device Support Request] _TZE200_fcooykb4 TS0601 LED controller

nchee1421, 2022-06-06:
**Is your feature request related to a problem? Please describe.**

The device is a  Skydance WZ1 LED dimmable controller and connects to ZHA but there are no entities associated with it.

The product website is as follows:
http://www.iskydance.com/index.php?c=product_show&a=index&id=1583

**Describe the solution you'd like**

I'm a newbie and am not sure how to get it working. I think installing a quirk will do the trick but am not sure what quirk to use and how/whether I need to modify it for this device.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_fcooykb4",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
2022-06-02 15:57:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=33), 0, 34, b'!<\x00')
2022-06-02 15:57:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'5667219ea8d515b26f944a25aa559249bd4e05a8ccf267d9147e'
2022-06-02 15:57:19 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6667a19e54f8f1777e'
2022-06-02 15:57:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2022-06-02 15:57:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 54 (sendBroadcast) received: b'00d2'
2022-06-02 15:57:19 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7667b1ed512a158459944a2daa55929b634e27ab123164aac1c67a117e'
2022-06-02 15:57:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2022-06-02 15:57:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0500003600000008000000d2ff000000ffff03213c00'
2022-06-02 15:57:19 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_UNKNOWN: 8>, groupId=0, sequence=210), 255, 0, 0x0000, 255, 255, b'!<\x00']
2022-06-02 15:57:19 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2022-06-02 15:57:19 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](EZSP): started initialization
2022-06-02 15:57:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[... 182 lines cut ...]
2022-06-02 15:57:33 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2022-06-02 15:57:33 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'ec4db9419bfeff277184ffcd00'
2022-06-02 15:57:33 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x4dec, 84:71:27:ff:fe:9b:41:b9, 255, -51, []]
2022-06-02 15:57:33 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x4dec, 84:71:27:ff:fe:9b:41:b9, 255, -51, [])
2022-06-02 15:57:33 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4469b1ed542e14b259954b25ab5592e46383cbe612312483f8ccbc76be423fa7ebcdda6f8fffc7c9d5d2698c5423a9ec763ca5ea7582469ed50d39a3e7e913361c07bbe5ca628a459a4d994f9ff7c3de45f31e884302d10f381bbe5cda7e'
2022-06-02 15:57:33 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2022-06-02 15:57:33 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010000010100010000adffcdec4dffff4308050adfff423c0000000004000000001200000000120000000007000000000706992b2a1207992b2a1200000000070000000007000000000791992b2a1292992b2a1202'
2022-06-02 15:57:33 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=173), 255, -51, 0x4dec, 255, 255, b'\x08\x05\n\xdf\xffB<\x00\x00\x00\x00\x04\x00\x00\x00\x00\x12\x00\x00\x00\x00\x12\x00\x00\x00\x00\x07\x00\x00\x00\x00\x07\x06\x99+*\x12\x07\x99+*\x12\x00\x00\x00\x00\x07\x00\x00\x00\x00\x07\x00\x00\x00\x00\x07\x91\x99+*\x12\x92\x99+*\x12']
2022-06-02 15:57:33 DEBUG (MainThread) [zigpy.zcl] [0x4dec:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
2022-06-02 15:57:33 DEBUG (MainThread) [zigpy.zcl] [0x4dec:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65503, value=<TypeValue type=CharacterString, value=>)]]
2022-06-02 15:57:33 DEBUG (MainThread) [zigpy.zcl] [0x4dec:1:0x0000] Attribute report received: 65503=
2022-06-02 15:57:33 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x4dec, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=5), 39, b'\x18\x05\x0b\n\x00')
2022-06-02 15:57:33 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'456e219c54c658b658944a24ab1593499c4b00aef5cb6c81fd6eb57e'
2022-06-02 15:57:50 WARNING (MainThread) [zigpy.zcl] [0x4dec:1:0xef00] Unknown cluster-specific command 17
2022-06-02 15:57:51 WARNING (MainThread) [zigpy.zcl] [0x4dec:1:0xef00] Unknown cluster-specific command 17
2022-06-02 15:57:51 WARNING (MainThread) [zigpy.zcl] [0x4dec:1:0xef00] Unknown cluster-specific command 17
2022-06-02 15:57:51 WARNING (MainThread) [zigpy.zcl] [0x4dec:1:0xef00] Unknown cluster-specific command 17
2022-06-02 15:57:51 WARNING (MainThread) [zigpy.zcl] [0x4dec:1:0xef00] Unknown cluster-specific command 17
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

whitelynx, 2022-11-28:
I also have one of these devices, and would love to see it supported. It looks like there was also an earlier issue for this device that has been closed due to lack of activity: https://github.com/zigpy/zha-device-handlers/issues/1095

In that older issue, someone also linked to this PR for Zigbee2Mqtt that added support for this device: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3620/files

If I have time, I can try taking a look at adding support for this; however, I'm pretty short on time and I've never worked in this repo before, so I'd be starting from scratch. If anyone with more experience in this repo sees this and has time to implement it (or is willing to give me some coaching on adding a new device to the repo), I'd love to offer support any way I can.

closed 2023-07-10 by github-actions[bot]

szogun1987, 2024-02-04:
Could we get back to this issue?
I tried to add _TZE200_fcooykb4 to TuyaDoubleSwitchDimmer in ts0601_dimmer.py  it kinda works (I'm able to turn on or off 1 LED strip. it declares support for dimming but doesn't work. The second light entity is detected but doesn't work