zigpy/zha-device-handlers #1610: [Device Support Request] TS011F _TZ3000_3zofvcaa Zemismart Zigbee Smart Wall Socket ZTB242 (2 AC plugs + 1 USB type A + 1 USB type C)

w35l3y, 2022-06-06:
**Is your feature request related to a problem? Please describe.**
Device is detected and reports status OK when the command is made from the device itself.
When the command is made from the HA, it triggers all endpoints at once.

**Describe the solution you'd like**
Add support for sending commands from HA for each endpoint separately.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
[... 345 lines cut ...]
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:1:0x0006] Received command 0x0A (TSN 212): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:1:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:2:0x0006] Received ZCL frame: b'\x18\xd5\n\x00\x00\x10\x00'
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:2:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=213, command_id=10, *is_reply=True)
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:2:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:2:0x0006] Received command 0x0A (TSN 213): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:2:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-06-06 17:13:31 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x578B:4:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:3:0x0006] Received ZCL frame: b'\x18\xd6\n\x00\x00\x10\x00'
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:3:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=214, command_id=10, *is_reply=True)
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:3:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:3:0x0006] Received command 0x0A (TSN 214): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:3:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:4:0x0006] Received ZCL frame: b'\x18\xd7\n\x00\x00\x10\x00'
2022-06-06 17:13:31 DEBUG (MainThread) [zigpy.zcl] [0x578B:4:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=215, command_id=10, *is_reply=True)
2022-06-06 17:13:32 DEBUG (MainThread) [zigpy.zcl] [0x578B:4:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:32 DEBUG (MainThread) [zigpy.zcl] [0x578B:4:0x0006] Received command 0x0A (TSN 215): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-06-06 17:13:32 DEBUG (MainThread) [zigpy.zcl] [0x578B:4:0x0006] Attribute report received: on_off=<Bool.false: 0>
```

</details>

**Additional context**
Does anyone have any suggestion from I could start of ?
I have already copied and modified some examples from https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py and https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_switch.py but no success.

MattWestb, 2022-06-06:
LIDL is having one 3 AC with 4 USB that was working OK but new batch having one new firmware that is making the same as you device.
Try patching this quirk for your device https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/ts011f_plug.py and very likely you can getting all endpoints working OK from ZHA.

Its called "tuya magic spell" ;-))

w35l3y, 2022-06-06:
I added `cast_tuya_magic_spell_task` and `cast_tuya_magic_spell` to the code and now it is green for `TuyaZBOnOffAttributeCluster`
It used to return an yellow mark.

[image]

I didn't add `TuyaBasicCluster` because it doesn't seem to be referenced anywhere.

I had to remove and repair the device to make it work... I am still making some tests.

Should I PR at tuya or lidl folder (or elsewhere) ?

MattWestb, 2022-06-07:
Looks great !!!
I was thinking deleting and paring was not needed but i was wrong and good to knowing.
If you is taking the power way and putting it back after one minute is it working OK ?
If not you need the basic cluster implementation 2
.
One PR is more the welcome and i like using tuya then in the end all is made with tuya code and its hundred of brands out there.

javicalle, 2022-06-07:
Hi there,
IMHO `zhaquirks/tuya/ts011f_plug.py` would be a good place to put the quirk.

There is a `EnchantedDevice` class that I believe that can be extended to get the TMS working:
* https://github.com/zigpy/zha-device-handlers/blob/b6556169bf641e8f2eb882e8ace4a5a1ab4552d5/zhaquirks/tuya/mcu/__init__.py#L575-L587

Just extend your device implementation from it.

MattWestb, 2022-06-07:
@javicalle Shall we rewriting our exiting quirks with spells to the new class for the TS004F and   the LIDL TS011F or leaving them as they is ?

I think its bad having the spell in all quirks and not in one imported function.

w35l3y, 2022-06-07:
Just for future reference...

TS004F : https://github.com/zigpy/zha-device-handlers/blob/b6556169bf641e8f2eb882e8ace4a5a1ab4552d5/zhaquirks/tuya/ts004f.py#L187-L199
TS011F : https://github.com/zigpy/zha-device-handlers/blob/b6556169bf641e8f2eb882e8ace4a5a1ab4552d5/zhaquirks/lidl/ts011f_plug.py#L35-L95
https://github.com/zigpy/zha-device-handlers/blob/b6556169bf641e8f2eb882e8ace4a5a1ab4552d5/zhaquirks/lidl/ts011f_plug.py#L127-L133

javicalle, 2022-06-07:
I'm not sure if the code of the LIDL device is the same since it seems much more complex to me (it has `tries`, `rejoin`, ...) and I don't dare to modify it.
I think that apart from the LIDL we only have the `TuyaSmartRemote004FDMS`, can it be? Pretty sure we can refactor it to use the new class.

MattWestb, 2022-06-07:
The TS004F is in real world only only being sent to the device then its being new joined then the quirk is being loaded. Then HA is starting is being triggered but ZHA have not initiated the coordinator so its not being sent to the device.
Its exactly what the TS004F is needing and is working well.

Some devices that is main powered can needing the spell casted after power off and on then the TS004F method is not working and need being triggered in one other way.
I have not testing the LIDL method then my power strips is on old firmware that is working OK without the spell and its have other cluster settings on the end points and its not being updated by tuya ZBGW. I think its kicking the device so its rejoining (with the network OK) and being in init mode then sending the spell = very good is its working OK.

Some TS0602 / MCU device looks needing it and also most (O)LED temperature / humidity / light sensors combis.

Only critical is timing and the attribute read and write is needed being sent after getting the network key and before the device us getting the updated TC link key (its being in init mode).

Or letting the spell in the quirks that needing it for not distorting working devices ?

PR #1612 (2022-06-08): Add support TS011F _TZ3000_3zofvcaa ZTB242
Zemismart Zigbee Smart Wall Socket (2 AC plugs + 1 USB type A + 1 USB type C)

Fixes #1610
closed 2022-06-08 by dmulcahey