zigpy/zha-device-handlers #1611: [BUG] Danalock V3 disconnection

circa1665, 2022-06-06:
**Describe the bug**
As noted in these two threads:
https://community.home-assistant.io/t/integration-for-danalock-v3-zigbee/179574/85
https://forum.phoscon.de/t/danalock-v3-new-fw-0-20-0-losing-connection/1838/43

And in this PR:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6057

The [Danalock V3](https://zigbee.blakadder.com/Danalock_V3-BTZB.html) remains connected to the Zigbee network for 12-36hrs and then disconnects, the batteries must then be removed to reboot the device.

As noted in the PR linked above: after some time, the lock seemingly prefers to connect directly to the coordinator. When the lock changes its parent from a router to the coordinator, the zigbee connection then drops.

Could a similar change as described in the linked PR "[adding DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6057/files)" be applied to the Z-stack-firmware to help with the Danalock V3 connection issues?

There is also a separate issue that the Phoscon/Deconz devs are attempting to resolve with Danalock (and will require Danalock to issue a firmware update), as noted in the linked PR: "The lock detects than a parent is not reachable (aka powered off) and correctly rejoins via another parent. But it doesn't detect if the parent router plays nice, when commands are send to the gateway via a router the lock doesn't rejoin the network if it doesn't receive APS ACKs or ZCL Default Reponses from the gateway for a few times. The fix should be fairly simple and bring robustness like the Philips Hue motion sensor has."

**To Reproduce**

1. Add Danalock V3 to ZHA in Home Assistant.
2. After 12-36hrs lock disconnects from the zigbee network and can no longer be locked or unlocked via ZHA. The lock state and battery level are no longer reported.
3. Remove batteries from the device to reboot the lock.
4. The lock will reconnect to the Zigbee network and functions as expected for the next 12-36hrs.

**Expected behavior**
Stable connection to the Zigbee network.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4444, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x0101",
        "0x0b05",
        "0xfcff"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Danalock",
  "model": "V3-BTZBE",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```
{
  "home_assistant": {
[... 235 lines cut ...]
      {
        "name": "DOOR_LOCK"
      }
    ],
    "user_given_name": "Front door lock",
    "device_reg_id": "8d5506ad14fe41cae4efa3ce8625b070",
    "area_id": "4f23624c4a4d489fa5fd0e62097bdf74"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

RamiRihawi, 2022-08-04:
I have the same lock with the exact same issue.
I am also following the Phoscon work on this issue, hopefully Danalock company will issue an update soon.

closed 2023-02-07 by github-actions[bot]

calamarain, 2023-02-07:
👍