zigpy/zha-device-handlers #1620: _TZE200 _3towulqd TS0601 Tuya Motion Sensor still not working

itsolon, 2022-06-14:
hi, javicalle, i have installed new intance with few modules.. for test and enabled logging.
for your interest here you can see everything

<details>

<summary>Device log extracted</summary>

```yaml

2022-06-13 18:48:53 WARNING (MainThread) [zigpy.zcl] [0x4269:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=87, dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xaa\x02\x00\x00', *payload=682)))
2022-06-13 18:48:53 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=37, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-13 18:48:53 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
2022-06-13 18:48:53 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4269), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=37, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x25\x0B\x02\x81')
2022-06-13 18:48:53 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-13 18:48:53 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=37)
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x4269, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=153, SecurityUse=<Bool.false: 0>, TimeStamp=2117896, TSN=0, Data=b'\x09\x26\x02\x00\x58\x01\x04\x00\x01\x01', MacSrcAddr=0x4269, MsgResultRadius=29)
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Received ZCL frame: b'\x09\x26\x02\x00\x58\x01\x04\x00\x01\x01'
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=38, command_id=2, *is_reply=True)
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Decoded ZCL frame: PirMotionManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=88, dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>)))
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Received command 0x02 (TSN 38): set_data_response(data=TuyaCommand(status=0, tsn=88, dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>)))
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=88, dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>))
2022-06-13 18:48:54 WARNING (MainThread) [zigpy.zcl] [0x4269:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=88, dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>)))
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=38, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4269), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=38, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x26\x0B\x02\x81')
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-13 18:48:54 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=38)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x4269, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=153, SecurityUse=<Bool.false: 0>, TimeStamp=2211875, TSN=0, Data=b'\x09\x27\x02\x00\x59\x0C\x02\x00\x04\x00\x00\x02\xAA', MacSrcAddr=0x4269, MsgResultRadius=29)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Received ZCL frame: b'\x09\x27\x02\x00\x59\x0C\x02\x00\x04\x00\x00\x02\xAA'
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=39, command_id=2, *is_reply=True)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Decoded ZCL frame: PirMotionManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=89, dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xaa\x02\x00\x00', *payload=682)))
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Received command 0x02 (TSN 39): set_data_response(data=TuyaCommand(status=0, tsn=89, dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xaa\x02\x00\x00', *payload=682)))
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=89, dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xaa\x02\x00\x00', *payload=682))
2022-06-13 18:48:55 WARNING (MainThread) [zigpy.zcl] [0x4269:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=89, dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xaa\x02\x00\x00', *payload=682)))
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=39, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4269), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=39, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x27\x0B\x02\x81')
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=39)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x4269, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=149, SecurityUse=<Bool.false: 0>, TimeStamp=2225736, TSN=0, Data=b'\x09\x28\x02\x00\x5A\x01\x04\x00\x01\x00', MacSrcAddr=0x4269, MsgResultRadius=29)
2022-06-13 18:48:55 DEBUG (MainThread) [zigpy.zcl] [0x4269:1:0xef00] Received ZCL frame: b'\x09\x28\x02\x00\x5A\x01\x04\x00\x01\x00'
[... cut ...]
    },
    "entities": [
      {
        "entity_id": "sensor.tze200_3towulqd_ts0601_51c662cb_power",
        "name": "_TZE200_3towulqd TS0601"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "IAS_ZONE"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "792a6da6e8efa645d188a045299e8e15",
    "area_id": null
  }
}

```

</details>

_Originally posted by @itsolon in https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1154156204_

javicalle, 2022-11-06:
Old thread, but have preference over the #1896 

Have you tried the quirk from https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1219634462 ?
The signature don't seem to match the proposed quirk.

javicalle, 2022-11-21:
For anyone that will to try the proposed quirk and perform some tests, configure your local quirk folder and create a new file with the proposed quirk. 
If you don't know how to set the custom quirk this is a good guide to follow:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once saved, remove any `__pycache__` folder, restart HA and repair the device.

Check if the device's signature has changed.
If something is not working, detail the behavior and attach logs from the device.

ctml91, 2022-12-29:
I have two ZG-204ZL, one of them stops updating for a period of time (1-6 hours) and then starts updating again. The sensor continues to flash red when motion is detected during this period except the state remains stuck in occupied or clear (whichever it was when it froze).

I'm wondering if this is an issue with these devices or it's an issue with zigbee2mqtt. Unfortunately I don't have tuya hub so I don't think I can verify which it is.

javicalle, 2022-12-29:
Here is ZHA device handlers, nothing to do with Z2M

Does your device have the same signature?
Have you tried any quirk?

svetlisashkov, 2023-01-09:
I have the exact same _TZE200 _3towulqd TS0601 Tuya Motion Sensor device, and I tried the  quirk from[ #1599 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1219634462) but it didn't work for me...

hphde, 2023-03-15:
Tried the same - quirk did not work

mahagr, 2023-05-11:
My 2 cents: it's hard to get the light sensor to show up, it looks like the lux information is not being sent most of the time...?

@hphde Try to reconnect the device and point it with a flashlight or something while it is connecting to the hub. That said, even then it seems to be hard to get the lux value to show up.

hphde, 2023-05-11:
Ok. This is funny. Somewhere between my comment and today the sensor appeared. First occurence is 2nd of May checking the timeline but this seems to be the oldest entry what HA stores for all sensors?

closed 2023-11-14 by github-actions[bot]