zigpy/zha-device-handlers #1622: [Device Support Request] TS0601 TZE200_sh1btabb Tuya Garden Water Valve

MQTT-ESP8266, 2022-06-14:
I cannot get this Tuya water valve to run. There are no entities shown in the device manager.

It would be great if this device could be added to tuva quirks so that it is usable with ZHA. There already exists a solution for zigbee2mqtt (link below). Maybe that makes it easier.

<details>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_sh1btabb",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.45",
    "supervisor": "2022.05.3",
    "host_os": "Home Assistant OS 8.2",
    "docker_version": "20.10.14",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.73",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
[... 320 lines cut ...]
[0xBFE0](TS0201): Device seen - marking the device available and resetting counter
[0xBFE0](TS0201): Update device availability -  device available: True - new availability: True - changed: False
[0x7CA8:1:0xef00] Received ZCL frame: b'\t\x9b\x02\x00\x94g\x02\x00\x04\x00\x00\x00\x00'
[0x7CA8:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=155, command_id=2, *is_reply=True)
[0x7CA8:1:0xef00] Unknown cluster command 2 b'\x00\x94g\x02\x00\x04\x00\x00\x00\x00'
[0x7CA8:1:0xef00] Received command 0x02 (TSN 155): b'\x00\x94g\x02\x00\x04\x00\x00\x00\x00'
[0x7CA8:1:0xef00] No explicit handler for cluster command 0x02: b'\x00\x94g\x02\x00\x04\x00\x00\x00\x00'
[0x7CA8:1:0xef00] Received ZCL frame: b'\t\x9c\x02\x00\x95\x02\x01\x00\x01\x00'
[0x7CA8:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=156, command_id=2, *is_reply=True)
[0x7CA8:1:0xef00] Unknown cluster command 2 b'\x00\x95\x02\x01\x00\x01\x00'
[0x7CA8:1:0xef00] Received command 0x02 (TSN 156): b'\x00\x95\x02\x01\x00\x01\x00'
[0x7CA8:1:0xef00] No explicit handler for cluster command 0x02: b'\x00\x95\x02\x01\x00\x01\x00'
[0x7CA8:1:0xef00] Received ZCL frame: b'\t\x9d\x02\x00\x96f\x03\x00\x0810:21:46'
[0x7CA8:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=157, command_id=2, *is_reply=True)
[0x7CA8:1:0xef00] Unknown cluster command 2 b'\x00\x96f\x03\x00\x0810:21:46'
[0x7CA8:1:0xef00] Received command 0x02 (TSN 157): b'\x00\x96f\x03\x00\x0810:21:46'
[0x7CA8:1:0xef00] No explicit handler for cluster command 0x02: b'\x00\x96f\x03\x00\x0810:21:46'

```

</details>

**Additional context**
- This is the Garden water valve in question: (https://www.aliexpress.com/item/1005004222098040.html)
- There already exists a solution for zigbee2mqtt: https://github.com/MoonshineSG/zigbee2mqtt-custom/blob/13b0b39084ff6fa37a88c667a4303c566d49b234/QT-05M.js

jerrm, 2022-06-14:
Wild A** Guess, but try editing the quirk found here: https://github.com/zigpy/zha-device-handlers/issues/1571

Changing the "MODELS_INFO:" line to match your device.

[How to install a custom quirk.](https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226/2)

MQTT-ESP8266, 2022-06-14:
I tried it like that and it also created a device with an On/Off-Entity but it does nothing. The valve stays silent.

javicalle, 2022-06-15:
It looks like the same quirk should work as in this case:
* https://github.com/zigpy/zha-device-handlers/issues/1551

The proposed version is the following, but there are several DPs whose meaning and functionality are unknown, or what is the best way to map them in HA.

```python
"""Tuya valve devices."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaLevelControl,
    TuyaLevelControlNM,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaValveManufCluster(TuyaMCUCluster):
    """Tuya valve manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF03: ("dp_3", t.uint32_t, True),
            0xEF65: ("dp_101", t.uint32_t, True),
            0xEF66: ("dp_102", t.uint32_t, True),  # <-- new cluster attribute with fake IDs (102=0x66).Format= ID: ("name", type, True)
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            TuyaLevelControlNM.ep_attribute,
            "current_level",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (x * 255) // 100,
            dp_converter=lambda x: (x * 100) // 255,
        ),
        3: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_3",
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_101",
            dp_type=TuyaDPType.VALUE,
        ),
        102: DPToAttributeMapping(        # <-- device DP102
            TuyaMCUCluster.ep_attribute,  # <-- reference to the cluster which has the attribute
            "dp_102",                     # <-- attribute "name"
            dp_type=TuyaDPType.VALUE,     # <-- DP Type it is related to the attribute type
        ),
        110: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
            # output_clusters=[0x000a, 0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster,
                    TuyaOnOffNM,
                    TuyaLevelControlNM,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

Some DPs are mapped to the `TuyaValveManufCluster` attributes. The same can be done with the rest of DPs just following the indications of DP_102

What is expected to work:
* on/off switch function
* battery entity

What is expected NOT to work:
* everything else?

Regards.

MQTT-ESP8266, 2022-06-15:
I tried to integrate this quirk but I get an error:
`Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_sh1btabb.py:27
First occurred: 14:14:09 (1 occurrences)
Last logged: 14:14:09

Error setting up entry socket://192.168.0.73:8888 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 339, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_sh1btabb.py", line 27, in <module>
    from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaLevelControlNM' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/mcu/__init__.py)
`

MQTT-ESP8266, 2022-06-15:
I tested it but it does not work. The valve is not opperated by the generated entity nor is the battery-level shown
[image]
.

javicalle, 2022-06-15:
Please, attach the logs from operating the device. both from the physical device and from HA, identifying each one with the associated event.

javicalle, 2022-06-15:
There is the info about how enable the debug log level for ZHA:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Once enabled (you will need to restart HA), look at the `home-assistant.log` file while you interacts with device. This will allow you to identify logs with events.

The events logs will give us the info needed to map the DPs to functionallity.

MQTT-ESP8266, 2022-06-15:
I switched on debuging and this are the entries during manually switching the valve 3x on and off:

```
2022-06-16 00:18:11 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2022-06-16 00:18:11 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'ef001801180004000000000000000000060003000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2022-06-16 00:18:11 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-06-16 00:18:12 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 170 (getValue) received: b'0001f4'
2022-06-16 00:18:12 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 244
2022-06-16 00:18:12 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 924, MAC_TX_BROADCAST = 1107, MAC_RX_UNICAST = 114, MAC_TX_UNICAST_SUCCESS = 22, MAC_TX_UNICAST_RETRY = 0, MAC_TX_UNICAST_FAILED = 0, APS_DATA_RX_BROADCAST = 3, APS_DATA_TX_BROADCAST = 3, APS_DATA_RX_UNICAST = 27, APS_DATA_TX_UNICAST_SUCCESS = 13, APS_DATA_TX_UNICAST_RETRY = 8, APS_DATA_TX_UNICAST_FAILED = 4, ROUTE_DISCOVERY_INITIATED = 4, NEIGHBOR_ADDED = 1, NEIGHBOR_REMOVED = 1, NEIGHBOR_STALE = 0, JOIN_INDICATION = 1, CHILD_REMOVED = 0, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 244]
2022-06-16 00:18:12 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA546](TS0201): Device seen - marking the device available and resetting counter
2022-06-16 00:18:12 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA546](TS0201): Update device availability -  device available: True - new availability: True - changed: False
2022-06-16 00:18:22 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2022-06-16 00:18:22 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBFE0](TS0201): Device seen - marking the device available and resetting counter
2022-06-16 00:18:22 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBFE0](TS0201): Update device availability -  device available: True - new availability: True - changed: False
2022-06-16 00:18:22 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'f3001c01180004000000000000000000060003000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2022-06-16 00:18:22 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-06-16 00:18:22 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 170 (getValue) received: b'0001f4'
2022-06-16 00:18:22 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 244
2022-06-16 00:18:22 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 928, MAC_TX_BROADCAST = 1111, MAC_RX_UNICAST = 114, MAC_TX_UNICAST_SUCCESS = 22, MAC_TX_UNICAST_RETRY = 0, MAC_TX_UNICAST_FAILED = 0, APS_DATA_RX_BROADCAST = 3, APS_DATA_TX_BROADCAST = 3, APS_DATA_RX_UNICAST = 27, APS_DATA_TX_UNICAST_SUCCESS = 13, APS_DATA_TX_UNICAST_RETRY = 8, APS_DATA_TX_UNICAST_FAILED = 4, ROUTE_DISCOVERY_INITIATED = 4, NEIGHBOR_ADDED = 1, NEIGHBOR_REMOVED = 1, NEIGHBOR_STALE = 0, JOIN_INDICATION = 1, CHILD_REMOVED = 0, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 244]
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating Devices ...
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating already existing Device 10971 0189968
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating already existing Device 11795 0551968
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating already existing Device 11795 1038488
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating already existing Device 11959 0075424
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating already existing Device 11959 0076432
2022-06-16 00:18:23 INFO (SyncWorker_4) [pyfritzhome.fritzhome] Updating already existing Device 11795 0551240
[... cut ...] DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 244]
2022-06-16 00:18:36 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7C3F](TH01): Device seen - marking the device available and resetting counter
2022-06-16 00:18:36 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7C3F](TH01): Update device availability -  device available: True - new availability: True - changed: False
```

javicalle, 2022-06-15:
These logs don't have the format I would expect.
Can you attache the logs from manipulating the switch in HA?

MQTT-ESP8266, 2022-06-16:
This log was written during pairing of the water valve and then manipulating the switch in HA several times.

<details>
<summary>home-assitant.log</summary>

```
2022-06-16 11:39:09 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 244
2022-06-16 11:39:09 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 176, MAC_TX_BROADCAST = 220, MAC_RX_UNICAST = 7, MAC_TX_UNICAST_SUCCESS = 1, MAC_TX_UNICAST_RETRY = 0, MAC_TX_UNICAST_FAILED = 0, APS_DATA_RX_BROADCAST = 1, APS_DATA_TX_BROADCAST = 1, APS_DATA_RX_UNICAST = 2, APS_DATA_TX_UNICAST_SUCCESS = 1, APS_DATA_TX_UNICAST_RETRY = 4, APS_DATA_TX_UNICAST_FAILED = 2, ROUTE_DISCOVERY_INITIATED = 2, NEIGHBOR_ADDED = 0, NEIGHBOR_REMOVED = 0, NEIGHBOR_STALE = 0, JOIN_INDICATION = 0, CHILD_REMOVED = 0, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 244]
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: a4:c1:38:4a:ea:11:11:2f
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7E33](TS0601): started configuration
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:ZDO](TS0601): 'async_configure' stage succeeded
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0006]: Performing cluster binding
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0006]: Configuring cluster attribute reporting
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=0)]
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0006]: finished channel configuration
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0008]: Performing cluster binding
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0008]: bound 'level' cluster: Status.SUCCESS
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0008]: Configuring cluster attribute reporting
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0008]: Successfully configured reporting for '{'current_level': (1, 900, 1)}' on 'level' cluster: [ConfigureReportingResponseRecord(status=0)]
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0008]: finished channel configuration
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0000]: Configuring cluster attribute reporting
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0000]: finished channel configuration
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0001]: Performing cluster binding
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0001]: bound 'power' cluster: Status.SUCCESS
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0001]: Configuring cluster attribute reporting
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0001]: Successfully configured reporting for '{'battery_voltage': (3600, 10800, 1), 'battery_percentage_remaining': (3600, 10800, 1)}' on 'power' cluster: [ConfigureReportingResponseRecord(status=0)]
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0001]: finished channel configuration
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0019]: finished channel configuration
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0006]: 'async_configure' stage succeeded
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0008]: 'async_configure' stage succeeded
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0000]: 'async_configure' stage succeeded
2022-06-16 11:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7E33:1:0x0001]: 'async_configure' stage succeeded
[... cut ...] >)
2022-06-16 11:40:33 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tze200_sh1btabb_ts0601_2f1111ea_level_on_off: turned on: {'on_off': Default_Response(command_id=1, status=<Status.SUCCESS: 0>)}
2022-06-16 11:40:33 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.tze200_sh1btabb_ts0601_2f1111ea_level_on_off, old_state=<state light.tze200_sh1btabb_ts0601_2f1111ea_level_on_off=off; supported_color_modes=[<ColorMode.BRIGHTNESS: 'bright

MQTT-ESP8266, 2022-06-17:
Is this the log you were looking for?

javicalle, 2022-06-20:
The device is receiving the commands and accepting it, but not seems to be the good ones.
After checking the attached link, it seems to me a pretty diferent quirk:
* https://github.com/MoonshineSG/zigbee2mqtt-custom/blob/master/QT-05M.js

There is a new version:

```python
"""Tuya valve devices."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)

from zigpy.zcl.clusters.security import IasZone

from zhaquirks import LocalDataCluster

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_TYPE,
)
from zhaquirks.tuya import NoManufacturerCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaLevelControl,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaLevelControlNM(NoManufacturerCluster, TuyaLevelControl):
    """Tuya LevelControl cluster with NoManufacturerID."""

class WaterSensorCluster(LocalDataCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Water_Sensor}

class TuyaValveManufCluster(TuyaMCUCluster):
    """Tuya valve manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF65: ("dp_101", t.uint32_t, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaLevelControlNM.ep_attribute,
            "current_level",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (x * 255) // 100,
            dp_converter=lambda x: (x * 100) // 255,
        ),
        3: DPToAttributeMapping(
            WaterSensorCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_101",
            dp_type=TuyaDPType.VALUE,
        ),
        102: DPToAttributeMapping(
            TuyaLevelControlNM.ep_attribute,
            "current_level",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (x * 255) // 100,
            dp_converter=lambda x: (x * 100) // 255,
            endpoint_id=2,
        ),
        110: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
            # output_clusters=[0x000a, 0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster,
                    TuyaOnOffNM,
                    TuyaLevelControlNM,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }

```

MQTT-ESP8266, 2022-06-20:
It still does not work. It is the same as before. A HA-device is created with a switchable entity but this entity does not switch the valve. I attached the log from pairing the valve and then manipulating the valve manually by pressing the physical button on the valve and then switching the newly created entity

<details>
<summary>home-assitant.log</summary>

```
14:36:17 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2022-06-20 14:36:17 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'380043003b000e000000000001000100070007000200010002000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2022-06-20 14:36:17 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-06-20 14:36:17 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 170 (getValue) received: b'0001f4'
2022-06-20 14:36:17 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 244
2022-06-20 14:36:17 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 56, MAC_TX_BROADCAST = 67, MAC_RX_UNICAST = 59, MAC_TX_UNICAST_SUCCESS = 14, MAC_TX_UNICAST_RETRY = 0, MAC_TX_UNICAST_FAILED = 0, APS_DATA_RX_BROADCAST = 1, APS_DATA_TX_BROADCAST = 1, APS_DATA_RX_UNICAST = 7, APS_DATA_TX_UNICAST_SUCCESS = 7, APS_DATA_TX_UNICAST_RETRY = 2, APS_DATA_TX_UNICAST_FAILED = 1, ROUTE_DISCOVERY_INITIATED = 2, NEIGHBOR_ADDED = 1, NEIGHBOR_REMOVED = 0, NEIGHBOR_STALE = 0, JOIN_INDICATION = 0, CHILD_REMOVED = 0, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 244]
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=17), 0, 18, b'\x11<\x00')
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'7d331221a9622ae94d59947c25aa5592499c4e36abffcd76b7fd7b437e'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3212a1a9622a15a0359d7e'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'247d3321a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc477d38757e'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 54 (sendBroadcast) received: b'0012'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4212b1a9112a10b259a24a25aa5592499c5cd8abedce9874fed75f89860b7e'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'050000360000000000000012ff000000ffff03113c00'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=18), 255, 0, 0x0000, 255, 255, b'\x11<\x00']
2022-06-20 14:36:21 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5313a1a9fb2a1536917e'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 175 (addTransientLinkKey) received: b'00'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TRUST_CENTER_POLICY: 0>, <EzspDecisionBitmask.ALLOW_UNSECURED_REJOINS|ALLOW_JOINS: 3>)
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'361021a9012a15b1594fd67e'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6410a1a9012a159c2b7e'
2022-06-20 14:36:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2022-06-20 14:36:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 85 (setPolicy) received: b'00'
2022-06-20 14:36:22 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7410b1a96b2a134ea6944a13aa5592499c4e27b9ffce674a927e'
2022-06-20 14:36:22 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2022-06-20 14:36:22 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'06fcff0000360000000000000012120000'
[... cut ...]
2022-06-20 14:37:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7FED:2:0x0008]: Reading attributes in chunks: ['start_up_current_level']
2022-06-20 14:37:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7FED:2:0x0008]: initializing uncached channel attributes: ['current_level'] - from cache[False]
2022-06-20 14:37:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7FED:2:0x0008]: Reading attributes in chunks: ['current_level']
2022-06-20 14:37:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.ba

javicalle, 2022-06-20:
Sorry but I can't see the logs from HA interaction.
Enable the debug logs and attach only relñevant info from device. If you are not able to get the info i couldn't help you.

rmharris62, 2022-07-09:
This was modeled off the wrong valve.  Here's the _correct_ Zigbee2MQTT solution:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4374/files

MQTT-ESP8266, 2022-07-11:
> Sorry but I can't see the logs from HA interaction. Enable the debug logs and attach only relevant info from device. If you are not able to get the info i couldn't help you.

Sorry for the late reply. I was on holidays. I tried again several times but the logfiles did not bring up anything new. It is always the same result.

> Apparently TZE200_arge1ptm isn't really the same valve. So this quirk won't really work.
> 
> Here's Zigbee2MQTT's working solution for the TZE200_sh1btabb valve:
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/pull/4374/files

That is interesting. Is there anyway to translate that into a quirk that I could test?

rmharris62, 2022-07-12:
Unfortunately, I'm not familiar enough with python to know how to make a quirk, I just happened to come across this when I was trying to get my own valve working.

@javicalle are you able to translate the Zigbee2MQTT solution into a quirk? I'm also willing / able to test it.

galbers, 2022-08-04:
Anyone get any further with this? I too have this water valve and would love to have it working correctly

cheerc, 2022-08-07:
> 

1. Try to use Zigbee2MQTT Edge rather than Zigbee2MQTT.
  a. Turn off Zigbee2MQTT first.
  b. Go to Supervisor and Install Zigbee2MQTT Edge
  c. Your settings will not change either using Edge version or not.
2. Download this file: https://github.com/MoonshineSG/zigbee2mqtt-custom/blob/master/QT-05M.js
3. Put the file above into config/zigbee2mqtt
4. Turn on Zigbee2MQTT Edge
5. Pair the device

This works for me. Worth to try it and share your result here
[image]
.

infinitejest, 2022-08-14:
I looked at the Zigbee2MQTT library and I was able to make the battery and the on/off work. The problem is that the ZHA quirk was set up on the wrong channel/Datapoint. By changing the datapoint to 2 for the switch and 108 for the Battery level; it now works as intended. I will try to see if I can make the other entities work so it has parity with Z2m

`"""Tuya valve devices."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaLevelControl,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaLevelControlNM(NoManufacturerCluster, TuyaLevelControl):
    """Tuya LevelControl cluster with NoManufacturerID."""

class TuyaValveManufCluster(TuyaMCUCluster):
    """Tuya valve manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF03: ("dp_3", t.uint32_t, True),
            0xEF65: ("dp_101", t.uint32_t, True),
            0xEF66: ("dp_102", t.uint32_t, True),  # <-- new cluster attribute with fake IDs (102=0x66).Format= ID: ("name", type, True)
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        1: DPToAttributeMapping(
            TuyaLevelControlNM.ep_attribute,
            "current_level",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (x * 255) // 100,
            dp_converter=lambda x: (x * 100) // 255,
        ),
        3: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_3",
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_101",
            dp_type=TuyaDPType.VALUE,
        ),
        102: DPToAttributeMapping(        # <-- device DP102
            TuyaMCUCluster.ep_attribute,  # <-- reference to the cluster which has the attribute
            "dp_102",                     # <-- attribute "name"
            dp_type=TuyaDPType.VALUE,     # <-- DP Type it is related to the attribute type
        ),
        108: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        2: "_dp_2_attr_update",
        1: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        108: "_dp_2_attr_update",
    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
            # output_clusters=[0x000a, 0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster,
                    TuyaOnOffNM,
                    TuyaLevelControlNM,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
`

hkoci, 2022-08-15:
Thanks, @infinitejest - this is perfect! :) I can confirm this works on my QT06 valve. I just wanted to add that there is a newer iteration model branded as the same Giex device on Aliexpress but with the manufacturer code _TZE200_a7sghmms; I added this on the MODELS_INFO list and this quirk pairs successfully with this type of water valve as well.

The light entity toggle does take a couple of seconds after triggering to actually open or close the valve - though this was the case with Z2M. Also, my battery level seems to be at 200%, I don't think this was an issue on Z2M. I will wait for the battery to decrease and see if the result changes.

javicalle, 2022-08-15:
The `TuyaPowerConfigurationCluster` is doubling the `battery_percentage_remaining`.
To fix it you can create a new class:
```python
class TuyaMCUPowerConfiguration(PowerConfiguration, TuyaLocalCluster):
    """Tuya PowerConfiguration."""

```

You will need to import the `TuyaLocalCluster`:
```python
from zhaquirks.tuya import NoManufacturerCluster, TuyaLocalCluster
```

And use it in the replacement part.

MQTT-ESP8266, 2022-08-15:
Thanks you, @infinitejest !
I can confirm that this works for the TZE200_sh1btabb Model 
[image]

neilsimp1, 2022-08-17:
Hmm it doesn't quite work for me.
It pairs, and if I press the button on the device I see the switch toggle in HASS.
However, nothing seems to stop the flow of water. If I hook it to my hose, water flows through whether the switch is on or off.

Any ideas?

infinitejest, 2022-08-17:
Did you check the model of valve you have ?

neilsimp1, 2022-08-17:
`TS0601
by _TZE200_sh1btabb `

The button triggers the switch on/off in HASS but doesn't have any effect on the water flowing through.

Before this quirk the button worked to start/stop the water flow but obviously didn't work in HASS.

infinitejest, 2022-08-17:
I just wanted to make sure is not a hardware issue. That the valve works when manually operated. Otherwise only idea is go to the logs for zha to try to figure out what channel is your valve sending info too

simoautio, 2022-08-17:
> `TS0601 by _TZE200_sh1btabb `
> 
> The button triggers the switch on/off in HASS but doesn't have any effect on the water flowing through.
> 
> Before this quirk the button worked to start/stop the water flow but obviously didn't work in HASS.

I had the same issue initially, but removing the device, restarting home assistant and re-pairing did the trick. Battery percentage shows correctly, and the on/off toggle works.

@infinitejest, thanks for your efforts! if you need any logs or anything with the remaining entities I can try to help. Would be happy to reward you for your work if you have one of those "buy me coffee" or patreon things!

infinitejest, 2022-08-17:
I currently i don't have those. But it would be great to work together to make it work.  So far i was able to fix water flow. Not as useful but shows 0 if off and 1 lf on.

neilsimp1, 2022-08-17:
The best I can conclude in my case is that it's a device issue.
Even unpaired from my network, the button on the device seems to do nothing. I do see a light on it light up but that is it.

Odd, considering this button did work when I got it and it's really just been sitting for a few weeks. Oh well. Unless there is some factory reset option I might be out of luck with this one.

Thank you @infinitejest for the effort on the quirk!

MQTT-ESP8266, 2022-08-17:
I had the same issue. Turns out that I left the device with valve on and the batteries were drained. New batteries - no problem.

infinitejest, 2022-08-18:
I think I can add most entities; but what I dont know is how to make home assistant recognize them. The way the quirk was originally set up is as a light swtich. But these type of devices do not have water compsumption or start time, End time etc. Anyone knows how to be able to make zha recognize these even though they are no officially entities in a light or swtich ?

oyagev, 2022-08-20:
I confirm this is working for TZE200_sh1btabb device. 
However as @infinitejest mentioned, it is recognized as a Light device, lacking some essential features like start & finish, consumption, etc. Would be wonderful to see those too.

Thanks so much for adding it, much waited for 🙏👏

oyagev, 2022-08-20:
@infinitejest I just noticed that when triggering the device on (light.turn_on), it automatically turns off after 1 min. 
Can it be that this trigger take a time length param that defaults to 1 min?

neilsimp1, 2022-08-22:
I was able to get mine working after taking it apart completely and putting it back together. So not a battery issue but I guess something in the valve solenoid was loose? Not sure but it works.

I notice the same issue as @oyagev though, it shuts off after about a minute. I can provide logs if it will help.

esmoyer, 2022-08-26:
I just got the TZE200_sh1btabb and have been playing around with it in Z2M and have found that it would initially turn off after about 1 minute by default. I changed the "runtime" to 15 minutes and then when I would turn it on it would run for 15 minutes and turn off. I just discovered this quirk (thank you @infinitejest) and am going to see if the device will retain the 15 minute run time from Z2M with ZHA (kind of doubt it). Otherwise, will need to figure out how to send that runtime to the device in ZHA before turning it on to get it to run for more than a minute at a time.

infinitejest, 2022-08-26:
I agree; exposing the other entities with the ability to change it is the way to fix all of that. I can get the device to show me all the data of the other entities. However, what I have not been able to figure out yet is how to re-route the data to Zigbee-certified entities. I been trying to send the data to the Measurements and Sensing entities in the zigbee protocol but so far that crashed my ZHA network. So I had to create another instance of HA to see if I can make that work. I will keep you updated in my progress. If someone has tips on how to route data from one entity to another I will appreciate the tip.

esmoyer, 2022-08-26:
Awesome, I'm not overly familiar with the quirk stuff with ZHA but might start reading up on it. I did test moving the valve from Z2M where I had the run time set to 15 minutes. Added it to ZHA and turned the switch on in HA and it ran for 15 minutes before turning off automatically. So it seems as a temporary work around, one could add it to Z2M (need to have a spare dongle) set the run time (up to 24 hours) and then use HA to turn it on and off having what was set on the valve itself as a "fail safe".

robertvorthman, 2022-09-27:
I'm not sure how, but some setting I changed allows my water valve to stay on longer than 1 minute.

In the HA device settings page, Manage Clusters, I went through all the cluster attributes like "time", "standard_time", "time_status", etc and entered the value  "2".  I also entered "2" in the "Manufacturer Code Override" field.  My water valve has been on for 20 minutes now.  If somebody understands the cluster attributes better than me maybe they can figure out what happened, but it is at least possible to run the valve longer than 1 minute.

I have the _TZE200_sh1btabb version, delivered by Amazon today.

cloudbr34k84, 2022-11-05:
just reading through this as i just got the exact same device as @MQTT-ESP8266 
[image]

mine has a different manufacture name, but thats it
"manufacturer": "_TZE200_a7sghmms",
  "model": "TS0601",
  "class": "zigpy.device.Device"

hkoci, 2022-11-05:
> just reading through this as i just got the exact same device as @MQTT-ESP8266 [image]
> 
> mine has a different manufacture name, but thats it "manufacturer": "_TZE200_a7sghmms", "model": "TS0601", "class": "zigpy.device.Device"

See above, just change the manufacturer code in the code provided by @infinitejest https://github.com/zigpy/zha-device-handlers/issues/1622#issuecomment-1214475892 to:

`        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"),("_TZE200_a7sghmms", "TS0601")],`

My current file is essentially the same, have used it since last summer when on vacation to water the flowers. Enable ZHA custom quirks and place the configuration file in the folder you located for the quirks.

```
"""Tuya valve devices."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaLevelControl,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaLevelControlNM(NoManufacturerCluster, TuyaLevelControl):
    """Tuya LevelControl cluster with NoManufacturerID."""

class TuyaValveManufCluster(TuyaMCUCluster):
    """Tuya valve manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF03: ("dp_3", t.uint32_t, True),
            0xEF65: ("dp_101", t.uint32_t, True),
            0xEF66: ("dp_102", t.uint32_t, True),  # <-- new cluster attribute with fake IDs (102=0x66).Format= ID: ("name", type, True)
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        1: DPToAttributeMapping(
            TuyaLevelControlNM.ep_attribute,
            "current_level",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (x * 255) // 100,
            dp_converter=lambda x: (x * 100) // 255,
        ),
        3: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_3",
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_101",
            dp_type=TuyaDPType.VALUE,
        ),
        102: DPToAttributeMapping(        # <-- device DP102
            TuyaMCUCluster.ep_attribute,  # <-- reference to the cluster which has the attribute
            "dp_102",                     # <-- attribute "name"
            dp_type=TuyaDPType.VALUE,     # <-- DP Type it is related to the attribute type
        ),
        108: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        2: "_dp_2_attr_update",
        1: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        108: "_dp_2_attr_update",
    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"),("_TZE200_a7sghmms", "TS0601")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
            # output_clusters=[0x000a, 0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster,
                    TuyaOnOffNM,
                    TuyaLevelControlNM,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
 
```

cloudbr34k84, 2022-11-05:
oh man awesome!
"Though there were some connection issues with when the toggle does not actually toggle but this has not happened frequently" To be honest when i was testing this in Tuya, i had the same issue. It came good after 1 hr or so?

hkoci, 2022-11-05:
> oh man awesome! "Though there were some connection issues with when the toggle does not actually toggle but this has not happened frequently" To be honest when i was testing this in Tuya, i had the same issue. It came good after 1 hr or so?

Pretty much, I used it as a sunrise automation with HA when it was the summer.

Yeah, I believe its the h/w if it occurs on Tuya as well - you'll enjoy it for sure. Made my life a lot easier on vacation.

cloudbr34k84, 2022-11-05:
> Pretty much, I used it as a sunrise automation with HA when it was the summer.
>
> Yeah, I believe its the h/w if it occurs on Tuya as well - you'll enjoy it for sure. Made my life a lot easier on [...]

i just like to buy these things and play around. but my goal is to use it with this. Basically the liquid fertilizer injects into into hose.
https://www.ezflo.com.au/product/ez-flo-3-2l/

hkoci, 2022-11-05:
> i just like to buy these things and play around. but my goal is to use it with this. Basically the liquid fertilizer injects into into hose. https://www.ezflo.com.au/product/ez-flo-3-2l/

Looks impressive 😲  - hope this goes well! We used the cheap irrigation system with the sprinklers around the garden, and might look into adding fertilizer now.

cloudbr34k84, 2022-11-05:
> Looks impressive 😲 - hope this goes well! We used the cheap irrigation system with the sprinklers around the garden, and might look into adding fertilizer now.

i only have real grass out the front, no in ground irrigation, just an osculating fan water sprayer. 

The quirks works, but there are not alot of entities in the quirk??

hkoci, 2022-11-05:
> i only have real grass out the front, no in ground irrigation, just an osculating fan water sprayer.
>
> The quirks works, but there are not alot of entities in the quirk??

With ZHA there is only battery level and a light entity to toggle the water valve

[image]

If you use Zigbee2MQTT, you'll see [more entities exposed](https://www.zigbee2mqtt.io/devices/QT06.html) like:

- irrigation_target
- cycle_irrigation_num_times
- cycle_irrigation_interval
- irrigation_start_time
- irrigation_end_time
- last_irrigation_duration
- water_consumed

which are missing on the zha quirk atm.

cloudbr34k84, 2022-11-05:
> With ZHA there is only battery level and a light entity to toggle the water valve
>
> [image]
>
> If you use Zigbee2MQTT, you'll see [more entities exposed](https://www.zigbee2mqtt.io/devices/QT06.html) [...]

Yeah thats all i have, light and battery, how come we cant get the the others??

cloudbr34k84, 2022-11-05:
FYI if you want to change the light switch to a switch
MODIFYING THE DEVICE TYPE - https://www.home-assistant.io/integrations/zha/#modifying-the-device-type
As not all device manufacturers follow the Zigbee standard, at times a device can be incorrectly classified. For example, a switch could be classified as a light.

To correct the device type, also called domain, add the following to your configuration.yaml and restart Home Assistant:

zha:
  device_config:
    84:71:27:ff:fe:93:17:24-1:    # format: {ieee}-{endpoint_id}
      type: "switch"              # corrected device type
YAML
{ieee} is the device hardware address which can be read from the Home Assistant UI when looking at Device info. From device info, you can find the {endpoint_id} by viewing the Zigbee device signature.
under zha in config file add
device_config:

hkoci, 2022-11-05:
Thank you @cloudbr34k84 , it has caused a lot of issues with homekit manipulating the device as a brightness slider!

hkoci, 2022-11-05:
> Yeah thats all i have, light and battery, how come we cant get the the others??

I think the issue is that originally, we were having issues porting over the quirk from Z2M and then @infinitejest realised that the datapoints were incorrect for the battery and valve switches.

If we were to improve the script, it would be to add the missing entities - most likely, comparing their script data points and ours.

The [referenced data points](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/giex.js) from Z2M are the following:

- giexWaterValveState: 2,
- giexWaterValveMode: 1,
- giexWaterValveIrrigationTarget: 104,
- giexWaterValveWaterConsumed: 111,
- giexWaterValveIrrigationStartTime: 101,
- giexWaterValveIrrigationEndTime: 102,
- giexWaterValveLastIrrigationDuration: 114,
- giexWaterValveBattery: 108,
- giexWaterValveCycleIrrigationNumTimes: 103,
- giexWaterValveCycleIrrigationInterval: 105,
- giexWaterValveCurrentTempurature: 106,

cloudbr34k84, 2022-11-05:
> > > > > > > > oh man awesome! "Though there were some connection issues with when the toggle does not actually toggle but this has not happened frequently" To be honest when i was testing this in Tuya, i had the same issue. It came good after 1 hr or so?
> > > > > > > 
> > > > > > > 
> > > > > > > Pretty much, I used it as a sunrise automation with HA when it was the summer.
> > > > > > > Yeah, I believe its the h/w if it occurs on Tuya as well - you'll enjoy it for sure. Made my life a lot easier on vacation.
> > > > > > 
> > > > > > 
> > > > > > i just like to buy these things and play around. but my goal is to use it with this. Basically the liquid fertilizer injects into into hose. https://www.ezflo.com.au/product/ez-flo-3-2l/
> > > > > 
> > > > > 
> > > > > Looks impressive 😲 - hope this goes well! We used the cheap irrigation system with the sprinklers around the garden, and might look into adding fertilizer now.
> > > > 
> > > > 
> > > > i only have real grass out the front, no in ground irrigation, just an osculating fan water sprayer.
> > > > The quirks works, but there are not alot of entities in the quirk??
> > > 
> > > 
> > > With ZHA there is only battery level and a light entity to toggle the water valve
> > > [image]
> > > If you use Zigbee2MQTT, you'll see [more entities exposed](https://www.zigbee2mqtt.io/devices/QT06.html) like:
> > > 
> > > * irrigation_target
> > > * cycle_irrigation_num_times
> > > * cycle_irrigation_interval
> > > * irrigation_start_time
> > > * irrigation_end_time
> > > * last_irrigation_duration
> > > * water_consumed
> > > 
> > > which are missing on the zha quirk atm.
> > 
> > 
> > Yeah thats all i have, light and battery, how come we cant get the the others??
> 
> I think the issue is that originally, we were having issues porting over the quirk from Z2M and then @infinitejest realised that the datapoints were incorrect for the battery and valve switches.
> 
> If we were to improve the quirk, it would be to add the missing entities - most likely, comparing their script data points and ours.
> 
> The [referenced data points](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/giex.js) from Z2M are the following:
> 
> * giexWaterValveState: 2,
> * giexWaterValveMode: 1,
> * giexWaterValveIrrigationTarget: 104,
> * giexWaterValveWaterConsumed: 111,
> * giexWaterValveIrrigationStartTime: 101,
> * giexWaterValveIrrigationEndTime: 102,
> * giexWaterValveLastIrrigationDuration: 114,
> * giexWaterValveBattery: 108,
> * giexWaterValveCycleIrrigationNumTimes: 103,
> * giexWaterValveCycleIrrigationInterval: 105,
> * giexWaterValveCurrentTempurature: 106,

I have no idea how to do any of this??

hkoci, 2022-11-05:
> I have no idea how to do any of this??

Me neither just an experiment for me tbh 😆 

One day I will code a quirk!

cloudbr34k84, 2022-11-05:
> Me neither just an experiment for me tbh 😆
>
> One day I will code a quirk!

bugger! well thanks for helping me, im trying to see if i can find some other quirks for devices i have, that arent working

mbo18, 2022-11-15:
If you want the valve to be detected as a switch instead of a light, change this ligne in the `replacement` part of the quirk:
`DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,`
with:
`                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
`

cloudbr34k84, 2022-11-16:
Is anyone else having an issue with this quirk where it just shuts off by its self after 3-4 mins
[image]

mbo18, 2022-11-16:
> Is anyone else having an issue with this quirk where it just shuts off by its self after 3-4 mins
> [image]

It may be setup like this in the valve. To change this you need a tuya gateway or move to Z2M to change this setting or wait for a better quirk

cloudbr34k84, 2022-11-16:
I have a tuya gateway and connected it and it's fine on there . So frustrating

mbo18, 2022-11-16:
I've made some progress, can someone try this quirk:

<details>

<summary>Click me</summary>

```python
"""Giex Tuya valve devices."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaPowerConfigurationCluster,
)

# info from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/giex.js
GIEX_TUYA_VALVE_MODE_ATTR = 0xEF01 # Mode [0] duration [1] capacity
GIEX_TUYA_VALVE_START_TIME_ATTR = 0xEF65 # Last irrigation start time (GMT)
GIEX_TUYA_VALVE_END_TIME_ATTR = 0xEF66 # Last irrigation end time (GMT)
GIEX_TUYA_VALVE_NUM_TIMES_ATTR = 0xEF67 # Number of cycle irrigation times, set to 0 for single cycle, min=0 max=100
GIEX_TUYA_VALVE_TARGET_ATTR = 0xEF68 # Irrigation target, duration in seconds or capacity in litres (depending on mode), min=0, max=3600
GIEX_TUYA_VALVE_INTERVAL_ATTR = 0xEF69 # Cycle irrigation interval in seconds, min=0, max=3600
GIEX_TUYA_VALVE_DURATION_ATTR = 0xEF72 # Last irrigation duration

_LOGGER = logging.getLogger(__name__)

class GiexTuyaValveMode(t.enum8):
    """Giex valve mode enum."""

    duration = 0x00
    capacity = 0x01

class TuyaValveWaterConsumed(Metering, TuyaLocalCluster):
    """Tuya Valve Water consumed cluster."""

    VOLUME_LITERS = 0x0007

    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {0x0300: VOLUME_LITERS}

class GiexTuyaValveManufCluster(TuyaMCUCluster):
[... 105 lines cut ...]
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaPowerConfigurationCluster,
                    TuyaValveWaterConsumed,
                    GiexTuyaValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
 ```

</details>

I will try to make changes in order for us to be able to change the valve mode between `duration` or `capacity`, and also change the target, number of cycle or circle interval. But I think that I will need the help of some quirks experts!

infinitejest, 2022-11-16:
> # info from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/giex.jsGIEX_TUYA_VALVE_MODE_ATTR = 0xEF01 # Mode [0] duration [1] capacityGIEX_TUYA_VALVE_START_TIME_ATTR = 0xEF65 # Last irrigation start time (GMT)GIEX_TUYA_VALVE_END_TIME_ATTR = 0xEF66 # Last irrigation end time (GMT)GIEX_TUYA_VALVE_NUM_TIMES_ATTR = 0xEF67 # Number of cycle irrigation times, set to 0 for single cycle, min=0 max=100GIEX_TUYA_VALVE_TARGET_ATTR = 0xEF68 # Irrigation target, duration in seconds or capacity in litres (depending on mode), min=0, max=3600GIEX_TUYA_VALVE_INTERVAL_ATTR = 0xEF69 # Cycle irrigation interval in seconds, min=0, max=3600GIEX_TUYA_VALVE_DURATION_ATTR = 0xEF72 # Last irrigation duration

That is awesome.  I meant to figure this out,. But didn't have the time.
You can see most entities in the home assistant logs but could not figure
out how to translate to the quirk.

Great job

[35 more replies]

closed 2022-11-19 by dmulcahey