zigpy/zha-device-handlers #1623: [Device Support Request] Schneider Electric / LK Wiser 550D6001 four-button Battery device

mrwee, 2022-06-15:
**Is your feature request related to a problem? Please describe.**
The device is recognized by zha and I can make one (1) out of four devices work in hass. The rest does not generate any events.

**Describe the solution you'd like**
Full support for this device

<details>
```
<!-- Device signature can be acquired by clicking on the "Zigbee Device Signature" button in the device settings view -->
<summary>Device signature</summary>
```

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "21": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0xff17"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0102"
      ]
    },
    "22": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xff17"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0102"
      ]
    },
    "23": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xff17"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0102"
[... 340 lines cut ...]
      {
        "name": "DIMMER_SWITCH"
      }
    ],
    "user_given_name": "OliverWiz",
    "device_reg_id": "1c04faff9affbc4d780aaa88f01f62b3",
    "area_id": null
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Don't know which log this should be.
```

</details>

**Additional context**
All four devices have been updated to latest (as of 15-06-2022) SW version 2.9.1 via the LK Wiser Gateway.
I have tried both the two button and four button mode (press-and-hold button 1 + 4 for 10 seconds to change mode). Same result. For the one device, zha_events reflect the correct button setup.

axellebot, 2022-08-23:
Found something related to Zigbee2MQTT : [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/c88b2b8038ea6e19648986eeaef43c13592e5193/devices/schneider_electric.js#L532-L554)

axellebot, 2022-08-26:
Can you please provide output file when calling `zha_toolkit.scan_device` service by installing [ZHA Toolkit](https://github.com/mdeweerd/zha-toolkit).

NB : Because of https://github.com/mdeweerd/zha-toolkit/issues/83, you will need to create `scans/FLS/AIRLINK` directories first.

axellebot, 2022-08-31:
[zigbee2mqtt](https://www.zigbee2mqtt.io/devices/550D6001.htm)

> Depending on the firmware version the device may support both 1-channel and 2-channel mode. In 1-channel mode both the upper and lower buttons works as the upper buttons. On some firmware versions (incl. those sold as Elko EKO07117) the device starts out in 1-channel mode and must be switched into 2-channel mode either by using the Elko / Wiser gateway or by holding down button 1 and 4 (upper left and lower right) for approx. 10 seconds - the led will flash red and the become green once successfull.

[zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/d1f00202a79a5cd4b6548dd1d15057895a3e6666/devices/schneider_electric.js#L292-L311)

> When in 2-gang operation mode, unit operates out of endpoints 21 and 22, otherwise just 21