zigpy/zha-device-handlers #1624: [Device Support Request] Fantem 4-in 1 Sensor (_TZ3210_zmy9hjay  TS0202)

wryandginger, 2022-06-15:
**Is your feature request related to a problem? Please describe.**
This is a motion sensor that exposes motion, temperature, humidity, and illuminance. Device pairs and 4 entities show, but temperature and humidity do not update unless you manually read the cluster. 
(See: [Discussion Here](https://community.home-assistant.io/t/tuya-zigbee-multi-sensor-4-in-1-smart-pir-motion-humidity-light-temperature-sensor/409780) )

[Item link on AliExpress](https://de.aliexpress.com/item/3256801692659675.html?gatewayAdapt=glo2deu4itemAdapt)

**Describe the solution you'd like**
I'd like the temperature and humidity to update automatically.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0001",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0106",
      "in_clusters": [
        "0x0001",
        "0x0400"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3210_zmy9hjay",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.6.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 312 lines cut ...]

```
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4aff>, 1, 260, 1280, b'\t`\x00\x05\x00\x00\x00\x00\x00', 0, 175, 255, 112, 20, 19, 8, -45]
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy.zcl] [0x4AFF:1:0x0500] Received ZCL frame: b'\t`\x00\x05\x00\x00\x00\x00\x00'
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy.zcl] [0x4AFF:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=96, command_id=0, *is_reply=True)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy.zcl] [0x4AFF:1:0x0500] Decoded ZCL frame: IasZone:status_change_notification(zone_status=<ZoneStatus.Tamper|Alarm_1: 5>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy.zcl] [0x4AFF:1:0x0500] Received command 0x00 (TSN 96): status_change_notification(zone_status=<ZoneStatus.Tamper|Alarm_1: 5>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy.zcl] [0x4AFF:1:0x0500] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=96, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy.zcl] [0x4AFF:1:0x0500] Sending reply: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-06-15 14:00:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4AFF:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 96 under 211 request id, data: b'18600b0000'
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4aff>, ep: 1, profile: 0x0104, cluster_id: 0x0500, data: b'096000050000000000'
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 211, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x4AFF endpoint=1>, 260, 1280, 1, b'\x18`\x0b\x00\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 211]
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 211: 00
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xd3 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x4aff endpoint=1>, status: 0x00
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-06-15 14:00:18 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
```

</details>

wryandginger, 2022-06-20:
I was thinking this was going to be similar to the NEO NAS-PD07, so I looked there to see if I could just add this device to the model info, but that didn't work.

I'm not sure if this is any help, but looking at the clusters and some other quirks, I think this might be using 
[class ZemismartTempHumidity(CustomDevice)](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0201_zemismart.py) for the temp and humidity since the clusters seem to line up.

I tried to make a custom quirk using just the ts0201_zemismart.py, replacing the model info -- nothing happened.
I then tried my best to mash up the NAS-PD07 and Zemismart quirks, but that crashed ZHA because I clearly don't know what I'm doing.

harishvelli, 2022-09-26:
Following. I have the same issue and when I use the automation from https://community.home-assistant.io/t/tuya-zigbee-multi-sensor-4-in-1-smart-pir-motion-humidity-light-temperature-sensor/409780/7 it ends up randomly triggering the motion detection to be detected eventho nothing has triggered it.

Fierydwarf, 2023-01-01:
Following. Same issue using ZHA.

wryandginger, 2023-01-01:
I checked my sensors and I'm not seeing any false triggers. Does that happen when only you're using my automation? 

Are you on battery or USB?
In one forum post somewhere -- zigbee2mqtt, probably? -- I read that certain battery voltages can cause false motion events. If you're using batteries, make sure they're new regular lithium batteries and not a rechargeable. Test with mains power (no batteries) to see if that does anything different. 

Have you tried adjusting the sensitivity of the sensor?
Go to Device Information --> Manage Zigbee Device --> IasZone (Endpoint 1m 0x0500 in) --> current_zone_sensitivity_level. Then you can read/write the value --1, 2, or 3. Level 2 seems to be what all mine are set to. Keep in mind, some electronic devices, heating devices, and other stuff in your house can be causing the motion to trigger.  

It may be worthwhile testing the sensors by connecting them to a USB battery bank and putting in a closed cabinet. If they're still randomly tripping with no light/motion/automation then it may be a defective sensor.

MiguelLZPF, 2023-02-27:
Following. Same issue using ZHA.

wryandginger, 2023-02-27:
Just to add, this is a supported device in [Z2M](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/fantem.js)

I'm happy to test, but I don't have the skills to write a quirk from scratch. 

I think I was close with looking at NAS-PD07, but I know the endpoints don't match up.

19jm81, 2023-03-12:
Sensor work really well with Z2M, support for ZHA would be awesome, I'd like to simplify my setup with ZHA, but no support for this sensors keeps me from doing that atm.

wryandginger, 2023-05-16:
Additional info for developing a quirk -- I'll try to take a look at these more closely this weekend to tinker.

Here are some drivers that are written for other platforms:

- [Deconz](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4511)
- [Hubitat](https://community.hubitat.com/t/release-tuya-zigbee-multi-sensor-4-in-1-pir-motion-sensors-and-mmwave-presence-radars-w-healthstatus/92441)
- [Smartthings](https://community.smartthings.com/t/edge-driver-mc-zigbee-drivers-for-motion-open-close-moisture-smoke-co-sensors-and-others-devices/231689/1165)

closed 2023-11-19 by github-actions[bot]