zigpy/zha-device-handlers #1627: [BUG] ZHA sends events only by one key (qbkg03lm)

Shurik85, 2022-06-17:
**Describe the bug**
ZHA sends events only by one key, the second key is simply ignored and not fixed in zha_event. The problem is observed on the quirks: zhaquirks.xiaomi.aqara.ctrl_neutral.CtrlNeutral.
The device type is incorrectly determined - this device is not a router.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Developer Panel - Event'
2. Click on 'zha_event'

**Expected behavior**
Pressing the right key should display the event.

**Screenshots**
[image]

<details>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=1, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x000a",
        "0x0019"
      ],
      "out_clusters": [
        "0x0000",
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0010"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0010"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0006",
        "0x0012"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.ctrl_neutral2",
[... 335 lines cut ...]
    "user_given_name": "\u0414\u0435\u0442\u0441\u043a\u0430\u044f_\u0432\u044b\u043a\u043b\u044e\u0447\u0430\u0442\u0435\u043b\u044c",
    "device_reg_id": "640581fb8395214b65b67d51d8d8ff9c",
    "area_id": "detskaia"
  }
}

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-06-18:
The current quirk only handles events for 1 button.

A new quirk needs to be made for 2 gang devices. My proposal is the following one:

```python
"""Xiaomi aqara wall switch devices."""
import logging

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    BinaryOutput,
    DeviceTemperature,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks import EventableCluster
from zhaquirks.const import (
    ARGS,
    ATTRIBUTE_ID,
    ATTRIBUTE_NAME,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    CLUSTER_ID,
    COMMAND,
    COMMAND_ATTRIBUTE_UPDATED,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
    VALUE,
)
from zhaquirks.xiaomi import (
    LUMI,
    BasicCluster,
    OnOffCluster,
    XiaomiCustomDevice,
    XiaomiPowerConfiguration,
)

ATTRIBUTE_ON_OFF = "on_off"
DOUBLE = "double"
HOLD = "long press"
PRESS_TYPES = {0: "long press", 1: "single", 2: "double"}
SINGLE = "single"
STATUS_TYPE_ATTR = 0x0055  # decimal = 85
XIAOMI_CLUSTER_ID = 0xFFFF
XIAOMI_DEVICE_TYPE = 0x5F01
XIAOMI_DEVICE_TYPE2 = 0x5F02
XIAOMI_DEVICE_TYPE3 = 0x5F03

_LOGGER = logging.getLogger(__name__)

# click attr 0xF000
[... 355 lines cut ...]
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 0, ATTRIBUTE_NAME: ATTRIBUTE_ON_OFF, VALUE: 2},
        },
        (COMMAND_HOLD, BUTTON_2): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 6,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 0, ATTRIBUTE_NAME: ATTRIBUTE_ON_OFF, VALUE: 0},
        },
        (COMMAND_RELEASE, BUTTON_2): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 6,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 0, ATTRIBUTE_NAME: ATTRIBUTE_ON_OFF, VALUE: 1},
        },
        (COMMAND_DOUBLE, BUTTON_2): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 6,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 0, ATTRIBUTE_NAME: ATTRIBUTE_ON_OFF, VALUE: 2},
        },
    }
```

I have removed the 2 gang devices from the current quirk and added to the new one.

Shurik85, 2022-06-19:
Excellent. Everything is working.

PR #1628 (2022-06-28): Add switch events for Aqara 2 gang devices
This PR adds the switch events for the second gang in Aqara 2 gang devices.
The models `lumi.ctrl_neutral2` and `lumi.switch.b2lacn02` has been migrated to the new quirk.
Despite the models have changed the ir quirk no breaking change is expected here.

Fixes: #1627
closed 2022-06-28 by dmulcahey