zigpy/zha-device-handlers #1630: [Device Support Request] TS004F _TZ3000_rco1yzb1

TomBursch, 2022-06-23:
**Is your feature request related to a problem? Please describe.**
I connected a smart button, which I bought from Lidl in Germany. The button is not firing any events and shows up as a switch inside of home assistant.

**Describe the solution you'd like**
Although mentioned [here](https://github.com/zigpy/zha-device-handlers/pull/1437), that this device should work, I couldn't get it to recognize a tap.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000",
        "0xe001"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_rco1yzb1",
  "model": "TS004F",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.6.7",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.9.12",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.4.0-117-generic",
    "run_as_root": true
  },
  "custom_components": {
    "monitor_docker": {
      "version": "1.10",
      "requirements": [
        "aiodocker==0.21.0",
        "python-dateutil==2.8.2"
      ]
    },
    "hacs": {
      "version": "1.25.5",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "alexa_media": {
      "version": "4.0.2",
      "requirements": [
        "alexapy==1.26.0",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "garbage_collection": {
      "version": "4.7.8",
      "requirements": [
        "python-dateutil>=2.8.2"
      ]
    },
    "rki_covid": {
      "version": "1.5.8",
      "requirements": [
        "rki-covid-parser==1.3.3"
      ]
    },
    "localtuya": {
      "version": "4.0.0",
      "requirements": []
    },
    "dwd_weather": {
      "version": "1.2.22",
      "requirements": [
        "simple_dwd_weatherforecast==1.1.5",
        "markdownify==0.6.5"
      ]
    },
    "samsungtv_tizen": {
      "version": "1.5.9",
      "requirements": [
        "websocket-client==0.56.0",
        "wakeonlan==1.1.6",
        "numpy==1.21.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.30.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.75",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 52452,
    "manufacturer": "_TZ3000_rco1yzb1",
    "model": "TS004F",
    "name": "_TZ3000_rco1yzb1 TS004F",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4098,
    "power_source": "Battery or Unknown",
    "lqi": 57,
    "rssi": null,
    "last_seen": "2022-06-23T12:06:34",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0104",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0004",
            "0x0006",
            "0x1000",
            "0xe001"
          ],
          "out_clusters": [
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x000a",
            "0x0019",
            "0x1000"
          ]
        }
      }
    },
    "entities": [
      {
        "entity_id": "switch.deckenschalter_tom_on_off",
        "name": "_TZ3000_rco1yzb1 TS004F"
      },
      {
        "entity_id": "button.deckenschalter_tom_identify",
        "name": "_TZ3000_rco1yzb1 TS004F"
      },
      {
        "entity_id": "sensor.deckenschalter_tom_power",
        "name": "_TZ3000_rco1yzb1 TS004F"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "DIMMER_SWITCH"
      }
    ],
    "user_given_name": "Deckenschalter Tom",
    "device_reg_id": "69fa3f70183ede8370e913d52cfe5eb2",
    "area_id": "tom_s_zimmer"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
The device can be seen here: https://www.lidl.de/p/silvercrest-smart-button-mit-magnetischer-wandhalterung-zigbee-smart-home/p100339621

MattWestb, 2022-06-23:
I have looking  for that device but LIDL.AT have not getting it so i cant testing it.
Then making the quirk for ROK we was lifting one device class with only TS004F and hoping new devices shall using it but tuya/LIDL have adding one more cluster (0xE001) so its not being loaded.
You can trying adding the 0xE001 cluster as in cluster in the device signature after this line https://github.com/zigpy/zha-device-handlers/blob/71c95b76751ada487314eda2bc3bde5246f7dea0/zhaquirks/tuya/ts004f.py#L346 in one local quirk and see if the device is loading the quirk for the device.
The device shall sending Light commands if its working OK but i dont knowing if the system is congratulating the binding OK so its boradcasting them to the coordinator and need logs for fixing it.
You can trying adding  the device as one light in one ZHA light group with 2 or more lights and the device shall broadcasting the commands to the last added group if its working like the ROK. 

And on more question witch coordinator is you using ?

TomBursch, 2022-06-24:
I'm currently on the road will test it out in a week or so when I'm back. I'm using a cheap Sonoff CC2531 Dongle. Thanks for the quick response!

Pr0mises, 2022-06-24:
@MattWestb I just bought this smart button and willing to debug it.
How do I add the 0xE001 cluster? What should the line 347 look like?

MattWestb, 2022-06-24:
On line https://github.com/zigpy/zha-device-handlers/blob/71c95b76751ada487314eda2bc3bde5246f7dea0/zhaquirks/tuya/ts004f.py#L62
Adding  `, TuyaZBExternalSwitchTypeCluster` in the end so its being imported.
Adding the cluster like this:
https://github.com/zigpy/zha-device-handlers/blob/c379b942cfd303c47d163d4e07ad25b4e978e21e/zhaquirks/tuya/ts130f.py#L354
and if you like but i think its not needed in the replacement (then we dont have found some useful function for this device on it) like this:
https://github.com/zigpy/zha-device-handlers/blob/c379b942cfd303c47d163d4e07ad25b4e978e21e/zhaquirks/tuya/ts130f.py#L383
If all is going well its shall getting the quirk loaded.
I hope you is getting events from the device then more users have porblem getting it from TS004F remotes.

Pr0mises, 2022-06-24:
Well, there's only on_off as entity. 
Somehow, my entity names do not change with the device name now...

I do get events for the long press tho

MattWestb, 2022-06-24:
> Well, I've only on_off as entity and somehow my entity names do not change with the device name now...

You is meaning one switch in ZHA GUI ?
If yes is not one light you can sending on off to.

Zigbee controller is sending commands and you need listening to them and using then for triggering in automatons.

Pr0mises, 2022-06-24:
Yes I've only the switch on_off.
I've multiple zigbee bulbs and other switches though.
I control on/off of a light with an automation now.
How should it even be possible to sync the button to a light?

tim-geier, 2022-06-25:
I just bought the same button, and had the same problem. So I made a quirk for it and now it's working fine for me. All events described in the manual are coming through (single click = on, double-click = off, hold = dim step, release = dim stop) and the weird button disappears.

<details>
<summary>Quirk</summary>

```python
"""Lidl TS004F Smart Button SSBM A1"""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    CLUSTER_ID,
    COMMAND,
    COMMAND_MOVE,
    COMMAND_MOVE_SATURATION,
    COMMAND_OFF,
    COMMAND_ON,
    COMMAND_STEP,
    COMMAND_STOP,
    COMMAND_STOP_MOVE_STEP,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PARAMS,
    PROFILE_ID,
    RIGHT,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)
from zhaquirks.tuya import TuyaZBOnOffAttributeCluster, TuyaZBExternalSwitchTypeCluster 

class LidlSmartRemote004F(CustomDevice):
    """Lidl one button smart remote - model SSBM A1"""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2080, device_version=1, input_clusters=[0, 1, 3, 4, 4096, 57345], output_clusters=[3, 4, 5, 6, 8, 10, 25, 4096])
        MODEL: "TS004F",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LightLink.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    Time.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.NON_COLOR_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,  # Is needed for adding group then binding is not working.
                    LightLink.cluster_id,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    Time.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    LevelControl.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

    device_automation_triggers = {
        (COMMAND_ON, BUTTON_1): {COMMAND: COMMAND_ON, ENDPOINT_ID: 1},
        (COMMAND_OFF, BUTTON_1): {COMMAND: COMMAND_OFF, ENDPOINT_ID: 1},
        (COMMAND_STEP, BUTTON_1): {COMMAND: COMMAND_STEP, ENDPOINT_ID: 1},
        (COMMAND_STOP, BUTTON_1): {COMMAND: COMMAND_STOP, ENDPOINT_ID: 1},
    }
```
</details>

If this also works for anyone else I can open a pull request to add support for this button.

MattWestb, 2022-06-25:
### Great work !!

Look on the DA for ROK that is having more automatons then you is missing one and i think changing the names to the same as ROK is more logic (only `LONG_PRESS, BUTTON)`.
https://github.com/zigpy/zha-device-handlers/blob/71c95b76751ada487314eda2bc3bde5246f7dea0/zhaquirks/tuya/ts004f.py#L67

Also if you like testing if its have support for "tuya scenes" that the ROK is having.
Try very fast tipple press and see is the device is sending light command or tuya scene events.
Also then the quirk is loaded you can reading and writing the attribute 
`switch_mode (id: 0x8004)` on the 
`TuyaSmartRemoteOnOffCluster (Endpoint id: 1, Id: 0x0006, Type: out)`  (Command = 0x00
Event = 0x01) and waking up the device you shall getting `none` or the mode its using.

tim-geier, 2022-06-25:
I changed the names like you suggested, definitely makes more sense this way.
Which automation do you think is missing? I looked at the ROK file, but I don't get it, the lidl button does not support any rotation...

I tried the triple press and the button lights up for a second or so but it didn't send any events to zha. It seems like it did switch to another mode however because no matter how you click it, it does not send any event anymore. With another triple press it switches back to normal mode and works fine again.
There is something in the manual about an scene link mode (translated from german "Modus Szenenverknüpfung") but according to the manual the button should blink 3 times if the mode switch was successful, this does not happen.

Could you please explain how do I 
> reading and writing the attribute `switch_mode (id: 0x8004)` on the `TuyaSmartRemoteOnOffCluster (Endpoint id: 1, Id: 0x0006, Type: out)` (Command = 0x00 Event = 0x01) 

or where I can read how to do that? I'm completely new to zha and ZigBee, so I don't really understand what you mean with all of that, sorry.

MattWestb, 2022-06-25:
Great then its nor one rotating dimmer instead of one 4 button dimmer switch !!

You is using only step but not the the mode / direction.
ROK in command mode is the rotating made with 2 speeds (device is using 3 but the real fast is not working good so no DA is made for it) and can you taking the `step_size` parameter away and its only using the direction/ step mode.
https://github.com/zigpy/zha-device-handlers/blob/71c95b76751ada487314eda2bc3bde5246f7dea0/zhaquirks/tuya/ts004f.py#L144-L155.

Reading setting attribute in ZHA: From the device card the tree` . `left from "reconfigure device" and then "manage cluster". On the cluster panel select cluster (OnOff but tuya version) and then zigbee attribute. Clicking "get zigbee attribute" and waking up the device.
For writing put 0 or 1 in `value` and clicking "set zigbee attribute".

The tripple press look interesting but is you is not getting any events that it have changing the attribute for mode its not not the same as ROK.
Putt ltlte more logging for ZHA and looking in the log if you is getting some interesting sent from it then you is changing mode and also if its sending other not known commands that we can using. Z2M have not finding any extras but its very likely its having it.
Tuya is s´ding very strange no standard things and is using Zigbee words for other function so its not easy knowing what they is meaning / doing.

The most importing you have getting the events working and doing working device automatons for the "normal" functions.
Perhaps the device is needing "tuya magic" being casted on it (the original 4 button need it) but i can testing it and adding the function later then LIDL is selling in in Austria then i have waiting for it coming ober one half year now (from was seeing it being Zigbee certificated) but i can trying getting all working if you have time and knowledge for doing more testing.

Pr0mises, 2022-07-01:
@tim-geier 
I tested your quirk had to add this line: 
`TuyaZBExternalSwitchTypeCluster.cluster_id,`
to `OUTPUT_CLUSTERS: [`

what button entities do you have? I've still only on_off

green-Flamingo, 2022-08-13:
@tim-geier I just tried your quirk and for some reason it still doesn't work for me. I do see the triggers when I select the device in an automation (see screenshot), but no matter how long, often, or strong I press the button I do not receive a signal.

I am pretty new to the whole ZHA topic, so any input is highly appreciated.

evlo, 2022-10-27:
So for me, it kinda worked with my zigstar lan gateway and HA server in cloud, but when I switched to yeellow, it reacts with 3 minute delay or not at all even.

evlo, 2022-10-27:
Lol ok, so in correct mode what one needs to do to make it work, is to to create automation that sets device to off when it is on, it is same with tuya I think

erkr, 2022-11-17:
Hi, I connected the same device. Although it pairs fine, the functionality is very limited.

Also for me single switch is added as an entity. Single press results in on, double press in off. 

Selecting the device in an automation doesn't expose much more: on, off and toggle.
I would like to see no switch, but only events like: 
- single press
- double press
- rotate left, right...

Some device info. I hope these can contribute to create a quirk.
The bottom pairs without a quirk. Device Signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000",
        "0xe001"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_rco1yzb1",
  "model": "TS004F",
  "class": "zigpy.device.Device"
} 
```

theorlangur, 2023-02-21:
https://github.com/zigpy/zha-device-handlers/issues/1630#issuecomment-1166291884

this one worked for me with a modification:

`
device_automation_triggers = {
        (SHORT_PRESS, BUTTON): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (LONG_PRESS, BUTTON): {COMMAND: COMMAND_STEP, CLUSTER_ID: 8, ENDPOINT_ID: 1},
        (LONG_RELEASE, BUTTON): {COMMAND: COMMAND_STOP, CLUSTER_ID: 8, ENDPOINT_ID: 1},
    }
`

except for the long press - don't get those events for some reason. But I do get 'long release'

erkr, 2023-08-21:
Still not settled into official release

erkr, 2024-02-17:
🙏

arnoakavdb, 2024-03-27:
Any news about, this?I'm figuring the same problem...

Pr0mises, 2024-03-27:
> Any news about, this?I'm figuring the same problem...

I gave up using the button as the battery draw was insanely high and had to repair it like once per month

MattWestb, 2024-03-28:
> I gave up using the button as the battery draw was insanely high and had to repair it like once per month

I have trying getting some of this but LIDL in Austria have not getting them so i cant testing.
The DMS (DimmerSwitch) can start draining batteries is well known but its depends of its parent if the parents is doing the network management OK so have one good router near as parent (and not the coordinator) is important and my is running for over 2 years now also the ROK (rotating dimmer).

erkr, 2024-03-28:
Mines are also running fine without draining batteries. I made a dummy quirk to eliminate the default on off button and act on the device click events (single or double clicks). Gave up on rotation as those are also not working when paired with a Tuya bridge.

closed 2024-10-01 by github-actions[bot]