zigpy/zha-device-handlers #1631: [Device Support Request]

overas, 2022-06-24:
**Is your feature request related to a problem? Please describe.**
Device joins zigbee network and it's recognized by home-assistant. 

In Norway the smart plug is sold as Apex smart plug 16A EU version https://shop.evasmart.no/produkt/smarthus/smart-plug#product-tabs1. It is also known as Datek HLU2909K.
Documentation: https://dsh.datek.no/smartplug/index.html and https://dsh.datek.no/SmartPlug/Apex_SmartPlug_UserManual.pdf
Zigbee2mqtt already support the device: https://www.zigbee2mqtt.io/devices/HLU2909K.html

Entities are created for:
On/Off switch: It works
Electrical power measurement (W): Values updated correctly.
Electrical measurement ac frequency (Hz): Values updated correctly.
Electrical measurement rms current (A): Values updated correctly.
Electrical measurement rms voltage (V): Values updated correctly.
Temperature (C, Celsius): Values updated correctly.
Electrical measurement power factor: Entity is created but values are never updated.

The problem is Electrical measurement power factor sensor which ZHA create as "sensor.datek_pop_dd6ba0fe_electrical_measurement_power_factor" with state "Unknown". I believe the sensor gives a lot of `WARNING (MainThread) [zigpy.zcl] [0x3B22:1:0xfeed] Unknown cluster command 42 b'` in the HA log.

[image]

[image]

**Describe the solution you'd like**
Please consider adding support for fixing the electrical measurement power factor.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4919, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0402",
        "0x0b04",
        "0x1000",
        "0xfee1",
        "0xfeed"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Datek",
  "model": "PoP",
  "class": "zigpy.device.Device"
}
[... 465 lines cut ...]
      }
    ],
    "user_given_name": null,
    "device_reg_id": "15372fe168724e74df5a49d181287b35",
    "area_id": null
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
2022-06-24 11:13:05 WARNING (MainThread) [zigpy.zcl] [0x5E0E:1:0xfeed] Unknown cluster command 42 b''
2022-06-24 11:13:08 WARNING (MainThread) [zigpy.zcl] [0x3B22:1:0xfeed] Unknown cluster command 42 b''
```

</details>

**Additional context**
[image]
[image]
[image]

closed 2023-07-10 by github-actions[bot]

andersi, 2023-08-25:
Installed Home Assistant 2023.8.4 today. Still the same problem.

`Logger: zigpy.zcl
Source: runner.py:179
First occurred: 11:54:23 (406 occurrences)
Last logged: 15:38:16

[0x8320:1:0xfeed] Unknown cluster command 42 b''
[0xDE7F:1:0xfeed] Unknown cluster command 42 b''`

overas, 2023-08-26:
As a workaround i filtered the log to avoid these messages. I hope somebody make a zha quirk for Datek powerplug.
```
logger:
  default: info
  logs:
    custom_components.pyscript: info
  filters:
    zigpy.zcl:
      - "Unknown cluster command 42 b"
      - "Unknown cluster command 6 b"
```

Tom-ahawk, 2023-10-03:
I am getting this issue spamming my log in HA:
`2023-10-03 18:48:29.248 WARNING (MainThread) [zigpy.zcl] [0xDE54:1:0xfeed] Unknown cluster command 42 b''`

Talked to the producer (Eva Smarthus ) and they say:
"It looks like a cluster which saying who is the parent, how many times restart and coverage. Its not a error, but must likely ZHA not supporting this."
"Det ser ut til at det er cluster som sender hvem som er parent, hvor mange ganger den har restartet og dekning. Det skal ikke være noe feil, men det er mulig ZHA ikke har støtte for det. "