zigpy/zha-device-handlers #1632: [Device Support Request] TS011F _TZ3000_cfnprab5 (XENON 4AC+2USB power strip)

tim-geier, 2022-06-24:
**Is your feature request related to a problem? Please describe.**
I have bought [this ](https://www.amazon.de/dp/B09TW36J98/) power strip from amazon, it should be working with zha (according to [https://zigbee.blakadder.com/Xenon_SM-SO306.html](https://zigbee.blakadder.com/Xenon_SM-SO306.html) but the sockets cannot be controlled. The device can be added to Home Assistant and the setup wizard shows that everything went fine and that the device can be used. The five switches get recognized but the device does not respond at all. The "Last seen" timestamp in Home Assistant also never changes from the time I pair the device.
The only difference I could spot is my model has an additional "-EZ". The supported device is model SM-SO306, I received the model SM-SO306-EZ, but I dont know whats the difference (if there is one) and how to get it working.
I think it needs to use something like the quirk for "ts011f.Plug_4AC_2USB", since thats basically the device I have, but something does not match. My device looks exactly like the one pictured in #1106.

**Describe the solution you'd like**
Would be nice if my power strip could be controlled like intended or somebody could tell me what I'm doing wrong.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
[... 322 lines cut ...]
        "name": "ON_OFF_PLUG_IN_UNIT"
      },
      {
        "name": "ON_OFF_PLUG_IN_UNIT"
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "a5810d76391e9ba0b41d40a56ed441cd",
    "area_id": null
  }
}
```

</details>

<details>
<summary>Additional logs (pairing)</summary>

See attached file, body was too long if pasted in here...
[Pairing_log.txt](https://github.com/zigpy/zha-device-handlers/files/8979850/Pairing_log.txt)

</details>

MattWestb, 2022-06-24:
Your device is nearly the same its not having one extra cluster that is missing on some endpoints.
https://github.com/zigpy/zha-device-handlers/blob/71c95b76751ada487314eda2bc3bde5246f7dea0/zhaquirks/tuya/ts011f_plug.py#L439
Deleting the missing extra cluster in the signature and replace on the endpoints in one local quirk and testing if ZHA is loading the quirk.

javicalle, 2022-06-24:
Try with a quirk like this one:
```python
"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaNewManufCluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Plug_4AC_2USB_xenon(CustomDevice):
    """Tuya 4 outlet + 2 USB surge protector with restore power state support."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=266
            # device_version=1
            # input_clusters=[3, 4, 5, 6, 57344, 57345]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
[... 121 lines cut ...]
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

```

tim-geier, 2022-06-24:
Thank you both for the quick reply, I tried @javicalle quirk, but in the pairing log there is an error
```
Considering <class 'ts011f.Plug_4AC_2USB_xenon'>
Fail because input cluster mismatch on at least one endpoint
```

<details>
<summary>Full log</summary>

```
[0xE612](lumi.sensor_magnet.aq2): Device seen - marking the device available and resetting counter
[0xE612](lumi.sensor_magnet.aq2): Update device availability -  device available: True - new availability: True - changed: False
New device 0xc8b7 (a4:c1:38:84:36:44:41:d7) joined the network
[0xc8b7] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xb7\xc8\xd7AD6\x848\xc1\xa4\x8e'
[0xc8b7:zdo] ZDO request ZDOCmd.Device_annce: [0xC8B7, a4:c1:38:84:36:44:41:d7, 142]
Tries remaining: 3
[0xc8b7] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc8b7] Extending timeout for 0x1b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x1b\x00\xb7\xc8\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0xc8b7] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0xc8b7] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1d\x00\xb7\xc8\x06\x01\x02\x03\x04\x05\xf2'
[0xc8b7] Discovered endpoints: [1, 2, 3, 4, 5, 242]
[0xc8b7] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=2 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=3 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=4 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=5 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0xc8b7:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x1f\x00\xb7\xc8\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
[0xc8b7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 0xE000
Unknown cluster 0xE001
[0xc8b7:2] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'!\x00\xb7\xc8\x10\x02\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0xc8b7:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xc8b7:3] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'#\x00\xb7\xc8\x10\x03\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0xc8b7:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xc8b7:4] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'%\x00\xb7\xc8\x10\x04\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0xc8b7:4] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=4, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xc8b7:5] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC8B7 ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b"'\x00\xb7\xc8\x10\x05\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00"
[0xc8b7:5] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=5, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xc8b7:242] Discovering endpoint information
Tries remaining: 3
[... cut ...]
[0x2953:5:0x0006] Decoded ZCL frame: OnOff:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=Bool, value=<Bool.true: 1>))])
[10579:5:0x0006]: finished channel initialization
[10579:5:0x0006]: 'async_initialize' stage succeeded
[10579](TS011F): power source: Mains
[10579](TS011F): completed initialization
```
</details>

@MattWestb with what should I replace on the endpoints? Sorry if I ask stupid questions, this is the first week I am running Home Assistant and I dont really have an idea how all of this works, yet. I need to read a lot more how all of this ZigBee stuff works :)

MattWestb, 2022-06-24:
Endpoint 1, 2 , 3, 4 and 5 put one   `#` in the beginning if the line and the line is being ignored.

javicalle, 2022-06-24:
Ouuuuch, I believe that i have fixed it.
Can you try again?

javicalle, 2022-06-24:
Not in endpoint 1

MattWestb, 2022-06-24:
Very true @javicalle  !!!
Dont forgetting deleting the cluster it in the replacement then its working and making one PR so its not being no not working cluster that confusing user and the system.

tim-geier, 2022-06-24:
Unfortunately still an error with @javicalle edited version
```
Checking quirks for _TZ3000_cfnprab5 TS011F (a4:c1:38:84:36:44:41:d7)
Considering <class 'ts011f.Plug_4AC_2USB_xenon'>
Fail because input cluster mismatch on at least one endpoint
```

<details>
<summary>Full log</summary>

```
New device 0x35fb (a4:c1:38:84:36:44:41:d7) joined the network
[0x35fb] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x11\xfb5\xd7AD6\x848\xc1\xa4\x8e'
[0x35fb:zdo] ZDO request ZDOCmd.Device_annce: [0x35FB, a4:c1:38:84:36:44:41:d7, 142]
Tries remaining: 3
[0x35fb] Requesting 'Node Descriptor'
Tries remaining: 2
[0x35fb] Extending timeout for 0x45 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'E\x00\xfb5\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x35fb] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x35fb] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'G\x00\xfb5\x06\x01\x02\x03\x04\x05\xf2'
[0x35fb] Discovered endpoints: [1, 2, 3, 4, 5, 242]
[0x35fb] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=2 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=3 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=4 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=5 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x35fb:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'I\x00\xfb5\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
[0x35fb:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 0xE000
Unknown cluster 0xE001
[0x35fb:2] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'K\x00\xfb5\x10\x02\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0x35fb:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0x35fb:3] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'M\x00\xfb5\x10\x03\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0x35fb:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0x35fb:4] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'O\x00\xfb5\x10\x04\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0x35fb:4] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=4, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0x35fb:5] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'Q\x00\xfb5\x10\x05\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0x35fb:5] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=5, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0x35fb:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x35FB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'S\x00\xfb5\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[... cut ...]
        "0xe001"
      ],
```
but the in clusters in the quirk file are these five
```
   INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
```
shouldn't these match? Where can I find where Identify.cluster_id and so on are defined, so I can find out which is which and maybe delete the ?wrong? one?

MattWestb, 2022-06-24:
`Identify.cluster_id` is 0x03 / 3 so deleting it in the signature for the endpoints that is not having it.

All cluster on all endpoints must matching for the system to loading the quirk.

tim-geier, 2022-06-24:
The quirk now looks like this:

```
"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaNewManufCluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Plug_4AC_2USB_xenon(CustomDevice):
    """Tuya 4 outlet + 2 USB surge protector with restore power state support."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=266
            # device_version=1
            # input_clusters=[3, 4, 5, 6, 57344, 57345]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
[... 113 lines cut ...]
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```
and it is loaded for my power strip, but the switches still don't work.
[image]

I can set the switch in Home Assistant but after a few seconds it goes back to the old position and the power strip does not switch on/off.

javicalle, 2022-06-24:
Ouuuuuch^2

I'm editing from mobile phone, so no easy to review code.
I have already try to fix it in my initial comment.

javicalle, 2022-06-24:
Can you try with another approach?

1. Import the `TuyaOnOffNM` class (replace the import line):
```python
from zhaquirks.tuya.mcu import EnchantedDevice, TuyaOnOffNM
```
2. Use the new class in just one endpoint (in the `replacement` part) just to test, for example:
```python
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffNM,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
```

Restart and check all the switches again. Maybe one of them would work with the change.

MattWestb, 2022-06-24:
@javicalle its looks being one updated version then its having better cluster confing  (more zigbee standard) as the original and very likely tuya have doing some tuning like LIDL have doing with the last shipped version that is needing tuya magic being casted for working OK.

Its implanted in the LIDL quirk folder and can being good trying if no manufacture is not working.

@tim-geier looks better only need finding what tuya have cooking this time for getting it working OK !!

tim-geier, 2022-06-24:
@javicalle sure, I can try whatever brings us closer to a solution :)
If i change the second endpoint to TuyaOnOffNM, then I get the following error, if I try to change the second switch:
[image]

@MattWestb thanks for the hint, I will try to cobble something together with the tuya magic from the lidl plugs. But this will have to wait until tomorrow...

javicalle, 2022-06-24:
The 'NO_MANUFACTURER' will require more changes...

Restore the `TuyaOnOffNM` cluster and replace the class defition with:
```python
class Plug_4AC_2USB_xenon(EnchantedDevice):
```

tim-geier, 2022-06-25:
Still no luck with replacing the class definition and reverting the second endpoint back to normal.
<details>
<summary>Pairing log</summary>

```[0x0000](ConBee II): Device seen - marking the device available and resetting counter
[0x0000](ConBee II): Update device availability -  device available: True - new availability: True - changed: False
[0xE612](lumi.sensor_magnet.aq2): Device seen - marking the device available and resetting counter
[0xE612](lumi.sensor_magnet.aq2): Update device availability -  device available: True - new availability: True - changed: False
New device 0xbefb (a4:c1:38:84:36:44:41:d7) joined the network
[0xbefb] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xfb\xbe\xd7AD6\x848\xc1\xa4\x8e'
[0xbefb:zdo] ZDO request ZDOCmd.Device_annce: [0xBEFB, a4:c1:38:84:36:44:41:d7, 142]
Tries remaining: 3
[0xbefb] Requesting 'Node Descriptor'
Tries remaining: 2
[0xbefb] Extending timeout for 0x1b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x1b\x00\xfb\xbe\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0xbefb] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0xbefb] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1d\x00\xfb\xbe\x06\x01\x02\x03\x04\x05\xf2'
[0xbefb] Discovered endpoints: [1, 2, 3, 4, 5, 242]
[0xbefb] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=2 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=3 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=4 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=5 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0xbefb:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x1f\x00\xfb\xbe\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
[0xbefb:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 0xE000
Unknown cluster 0xE001
[0xbefb:2] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'!\x00\xfb\xbe\x10\x02\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0xbefb:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xbefb:3] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'#\x00\xfb\xbe\x10\x03\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0xbefb:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xbefb:4] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'%\x00\xfb\xbe\x10\x04\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00'
[0xbefb:4] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=4, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xbefb:5] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBEFB ieee=a4:c1:38:84:36:44:41:d7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b"'\x00\xfb\xbe\x10\x05\x04\x01\n\x01\x01\x04\x04\x00\x05\x00\x06\x00\x01\xe0\x00"
[0xbefb:5] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=5, profile=260, device_type=266, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 0xE001
[0xbefb:242] Discovering endpoint information
Tries remaining: 3
[... cut ...] But if I try to change one of the switches before it disappears and shows up again, the power strip responds and turns on/off all sockets, but this only works one time, then its completely dead until I remove it from the network and pair it again, as always.
- The light on the power strip does not stop flashing after Home Assistant shows device ready. I dont have any other gateway to test it with but I would assume the pairing light would stop flashing after the device is successfully connected.

javicalle, 2022-06-25:
The spell is casted at the first pairing:
```
[0xBEFB:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=45, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0xBEFB:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 0, 1, 5, 7, 65534])
.../...
device - 0xBEFB:a4:c1:38:84:36:44:41:d7 entering async_device_initialized - is_new_join: True
device - 0xBEFB:a4:c1:38:84:36:44:41:d7 has joined the ZHA zigbee network
[0xBEFB](TS011F): started configuration
.../...
[0xBEFB:1:0x0000] Received ZCL frame: b'\x18-\x01\x04\x00\x00B\x10_TZ3000_cfnprab5\x00\x00\x00 \x03\x01\x00\x00 C\x05\x00\x00B\x06TS011F\x07\x00\x000\x01\xfe\xff\x000\x00'
[0xBEFB:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=45, command_id=1, *is_reply=True)
[0xBEFB:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(
  status_records=[
    ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZ3000_cfnprab5')), 
    ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=3)), 
    ReadAttributeRecord(attrid=0x0001, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=67)), 
    ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS011F')), 
    ReadAttributeRecord(attrid=0x0007, status=<Status.SUCCESS: 0>, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>)), 
    ReadAttributeRecord(attrid=0xFFFE, status=<Status.SUCCESS: 0>, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>))])
```

But as you said, seems that device keeps pairing again and again, every time with a diferent nwk address, but I can see any reason for that.

This is something new with the last change (the `EnchantedDevice` part) or already happens before the change?

tim-geier, 2022-06-25:
This happens with every configuration, regardless of the active quirk or without any. I just didn't know thats not normal until now.

javicalle, 2022-08-11:
Looking at `zigbee-herdsman-converters` library for some 'inspiration' I have found something that probably not related to that case but...

There is a case in which I interpret that there is some kind of 'bind' between each of the sockets and the 'coordinator'. I'm not sure what you mean or how it could be implemented here, but does what I say make sense to anyone?
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L1347-L1355

javicalle, 2022-08-11:
Ummmm, there is a especific converter for this device:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/useelink.js#L25-L44

From my understand the quirk is casting the Tuya spell and then doing this reporting binding bettwen endpoints. I'm going to try to understand what that second part does.

yruss972, 2022-08-11:
@javicalle I noticed that converter but I'm not sure it's actually the same device.
They list the model as SM-SO306E-2W which I've had for maybe a year already and it works out of the box in ZHA using the ts011f_plug.Plug_4AC_2USB quirk. It shows for me as having the manufacturer: _TZ3000_o005nuxx.

I actually bought these devices thinking they were the SM-SO306E-2W which I had working but got the SM-SO306E-1Z model with the _TZ3000_cfnprab5 manufacturer instead and, as we saw above, they don't have all the same endpoints- 

Could that have something to do with the "spell"? I'm not really sure what the spell is supposed to do but I saw someone mention that it enables additional endpoints? maybe once spelled, they have the same signatures?

Does using EnchantedDevice cast the spell or is something more complicated needed?
FYI, I tried to cast the spell on the device manually using zha-toolkit but it only generated a mess of errors so I gave up.

javicalle, 2022-08-11:
Hi there Yonah,
 the `_TZ3000_cfnprab5` manufacturer is the one in the `useelink.js` file. Is one of the few references I have found so I will take as a starting point.

That Tuya 'spell' is something that has been discovered sniffing the pairing process with the Tuya gateway. Some (recent) Tuya devices need to get it to behave a little more 'standar' zigbee device.

And yes, the `EnchantedDevice` class is in command to 'cast' that spell during the pairing proccess.

But (if I'm not wrong) most users are reporting that the pairing proccess keep going in bucle, so maybe device is expecting something else appart that spell...

javicalle, 2022-08-11:
I have some kind of crazy quirk based on nothing but I'm away from my development PC. If someone want to try...

My proposal would be:
* create a `OnOffGroupCluster` class just like in: 
  * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lutron/lzl4bwhl01remote.py#L40-L42 
* use it in the `replacement` part

It would be something like (not tested):
```python
"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import GroupBoundCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaNewManufCluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class OnOffGroupCluster(GroupBoundCluster, OnOff):
    """On/Off Cluster which only binds to a group."""

class Plug_4AC_2USB_xenon(EnchantedDevice):
    """Tuya 4 outlet + 2 USB surge protector with restore power state support."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
[... 122 lines cut ...]
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffGroupCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

```

What would do that `GroupBoundCluster`? I'm not sure. Maybe it would not even start...

yruss972, 2022-08-11:
I don't know how this could possibly be helpful but after a bunch of random changes, both my devices are working with my quirk. Approximately I think this is what happened:

1. I enabled debugging as follows:
```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zhaquirks: debug
```
2. I removed the broken devices from zha and reboot my host and noticed a bunch of really weird uart errors - seemed to me like a race condition- that the raspbee I use wasn't ready yet when ZHA started trying to use it and HA got all borked.

3. I restarted (not rebooted) HA and those errors went away.
4. I went into the ZHA integration configure menu and clicked update without changing anything
5. I went into the ZHA integration system options menu and clicked update without changing anything
6. I reloaded the ZHA integration
7. I paired the devices again and it magically worked :?

Since it worked, I've done a couple cycles of restarting HA and reloading the ZHA integration.
I've also disabled the debugging just in case the debugging affected something and it all still works.

I'm attaching my quirk.
[ts011f_plug.py.txt](https://github.com/zigpy/zha-device-handlers/files/9311415/ts011f_plug.py.txt)

javicalle, 2022-08-11:
I'm so glad someone finally got it to work!!!

Could you test if in this device works some of the Tuya attributes?:
* `backlight_mode` (0,1,2)
* `power_on_state` (Off:0, On:1, LastState:2)

Thanks in advanced.

MattWestb, 2022-08-11:
@javicalle The sniffs i have seen with tuya magic spell its looks like tuya is not configuring reporting and the device is reporting to the coordinator as default and also if its have group its reporting the the group. It can being some device is needing more reporting configured but try without first and see if its working OK and adding if needed.

yruss972, 2022-08-12:
@javicalle Happy to test.
Quick description of the device:
There is one physical button which globally turns on or off the entire power strip.
There are 6 lights on the powerstrip, one for the global on/off button, one for each "switch" state.
The light for the global button can be blue, red, or a mix of blue and red.

backlight_mode:
0: the global on/off button is unlit whether on or off
1: the global on/off button is lit red when **on** and unlit when **off**
2: the global on/off button is lit red when **off** and unlit when **on** 
3: the global on/off button is lit red whether on or off (returns SwitchBackLight.undefined_0x03)
4: the global on/off button is unlit whether on or off (returns SwitchBackLight.undefined_0x04)
Getting/setting the value only works on the first TuyaZBOnOffAttributeCluster. On the others, the attribute shows in HA always has a value of None and setting it doesn't seem to do anything.

child_lock:
0: shows as false, the global on/off button works as expected
1: shows as true, the global on/off button has no effect, HA can still manage the switches

power_on_state:
0: Off - when the strip looses power and regains power, the entire strip remains off (as opposed to turning on but leaving all the individual sockets unpowered
1: On - when the strip looses power and regains power, the entire strip turns on including all sockets and despite their individual last power states
2: Last State - when the strip looses and regains power, the individual sockets return to the last power state, if the global button is used, all sockets are turned on despite the last power state. 

I don't see anything else that I can easily guess how to test but happy to try.

javicalle, 2022-08-12:
> @javicalle The sniffs i have seen with tuya magic spell its looks like tuya is not configuring reporting and the device is reporting to the coordinator as default and also if its have group its [...]

Thanks @MattWestb for the review. My impression was that the `EnchantedDevice` part was already tested by several users, that's why I was looking for something diferent. As you have pointed, it seems that there is no need to anything else than the Tuya spell.
But I'm curious about why in `zigbee-herdsman-converters` are casting the spell and also configuring the 'reporting.bind' (whatever it means).
Looking at the `GroupBoundCluster`, my assumption is that a new cluster is generated to link all the plugs so that they can all be managed at the same time (as the button usually does on power strips), but the truth is that I am assuming a lot of things without solid base.

javicalle, 2022-08-12:
@yruss972 Thanks for testing and for your detailed response.

The first `TuyaZBOnOffAttributeCluster` is the cluster for the plug1, right?
I think being able to define an aditional global switch would be a huge improvement for these types of devices (I think the Tuya app represents it that way). IMO that switch would be a candidate to handle those 'global' settings. In any case, it is not something simple to handle and I would not handle it in this same issue.

Do you feel able to create the PR for this device? I can try to give you support in whatever you need.

MattWestb, 2022-08-12:
If having one in group cluster on one endpoint ZHA is adding the device / endpoint of the device in group manager and can being added as one light in ZHA groups (Zigbee groups) that using broadcast for commands.
If adding one group out cluster to one endpoint is normally one controller that is sending commands to one bonded group and ZHA is adding bound functionality in the device cards cluster management.

tuya have doing strange device like the TS04F (dimmer switch / remote) that cant being bound to one group but have in cluster that is possible adding groups to as one light and its sending commands to the group (110T not Zigbee standard).

One light switch (one rely for receiving on/off commands / end point and one controller / endpoint the sending commands) shall having group cluster on its endpoints and in and out for working OK as one normal light and one normal light controller on the same endpoint and adding groups and binding is making it working autonomous then the host / NCP is off line.

c11umw, 2022-08-18:
> I don't know how this could possibly be helpful but after a bunch of random changes, both my devices are working with my quirk. 

This worked for me also using the file that was attached.
I followed all the steps except (6) and it didn't work, but doing step (6) (I reloaded the ZHA integration) made the difference and now I can control the switches independently

Ordered 2 more to replace WiFi strips that lose connection when the TV is on (right next to my ISP router) :)

edwardtfn, 2022-08-30:
Same for me... It worked fine when using @yruss972's quirk.
After uploading the quirk I've removed the device and then added again and looks like everything is working fine now.

Thanks!!

MattWestb, 2022-08-30:
Then 3 user have success with the posted quirk is some one interesting making one PR so its being integrated in ZHA in the future and dont need using local quirk for getting the device working OK ?

yruss972, 2022-09-07:
I've submitted a PR for the basic quirk but I don't know how to add support for the features mentioned here: https://github.com/zigpy/zha-device-handlers/issues/1632#issuecomment-1213118577

If anyone can give me guidance there, I'd love to get as much working as possible.

closed 2022-09-12 by dmulcahey

harveyorg, 2022-10-07:
pasting the code from @yruss972 into the ts011f_plug.py worked for me. 
 -setup the dev containers extension in vs code (trying to work in bash in the container console in Portainer is hard work)
- opened the file, pasted in the Plug_4AC_2USB_cfnprab5 class, saved
- restarted HA
- removed the device from ZHA and went through pairing again

done - bit concerned my hack will be overwritten during the next update though. 
 
[image]

dmulcahey, 2022-10-16:
@harveyorg can you open a PR with the changes?

KaiserSoft, 2022-10-17:
Very nice, I want to add my "works for me as well" to this fix.
I added the class found in the comment https://github.com/zigpy/zha-device-handlers/issues/1632#issuecomment-1212383687 to the bottom of /usr/local/lib/python3.10/site-packages/zhaquirks/tuya/ts011f_plug.py

Removed the device, restarted HA, rejoined the device and it works as expected!

Previously it triggered all outlets when I tried to turn a single one on. Now it works like the one I purchased a long time ago. I can control every outlet individually. Thanks a lot!

fasteddie-g, 2022-11-13:
Appreciate all the comments/suggestions above.  I've read through several times and tried repeating the steps with no luck.  

I have a HomeAssistant Blue running the "latest" version of HomeAssistant with all updates installed.   I use a HubZ for Zwave and Zigbee devices.   I have 36 Zwave devices and 15 Zigbee devices.    

I wanted a power strip where I could control individual outlets, so I bought [this US version Zigbee power strip](https://www.amazon.com/gp/product/B09YPR1W53), Model: SM-SO306-U on the box, and Model: S-SO306 on the actual power strip.  FCC ID:  2AJ5F-SM-SO306.   It's being recognized in Home Assistant as TS011F by _TZ3000_cfnprab5 (this is the same as others above).

The [manufacturer link is here](http://www.xenon.cn/surge-protector/sm-so306-u.html), and it's also mentioned on [this Zigbee blackadder site](https://zigbee.blakadder.com/Xenon_SM-SO306.html).   On the manufacturer's site, its noted as SM-SO306-U2Z.

In researching my issue, it appears there are 2 symptoms:  toggle one plug on/off affects all plugs, or, unable to toggle any plugs (my symptom).  I've tried adding multiple quirks including [the one mentioned above] (https://github.com/zigpy/zha-device-handlers/files/9311415/ts011f_plug.py.txt) by yruss972  (following same procedure used), but all result in the same inability to control any of the plugs.  

I'm lost as to what my next steps are.  Any help/guidance would be greatly appreciated.

<details>
<summary>Diagnostic Info</summary>

```Python
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v0.8.23",
      "requirements": [
        "packaging>=20.8"
      ]
    },
    "hacs": {
      "version": "1.28.3",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "keymaster": {
      "version": "v0.0.82",
      "requirements": []
    },
    "fordpass": {
      "version": "0.1.37",
      "requirements": []
    },
    "alexa_media": {
      "version": "4.3.2",
      "requirements": [
        "alexapy==1.26.4",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "dwains_dashboard": {
      "version": "3.3.0",
      "requirements": []
    }
  },
  "integration_manifest": {
[... 469 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xe001": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
```
</details>

MattWestb, 2022-11-13:
Your device is unknown and need being added in the quirk so the system is knowing it and can applying the fix.
```
    "manufacturer": "_TZ3000_cfnprab5",
    "model": "TS011F",
    "name": "_TZ3000_cfnprab5 TS011F",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
```

fasteddie-g, 2022-11-13:
Thanks for the quick reply.  The original poster had the same exact info in his diagnostic info

```yaml
   "manufacturer": "_TZ3000_cfnprab5",
    "model": "TS011F",
    "name": "_TZ3000_cfnprab5 TS011F",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    ```

I agree the device is unknown, I just do not understand how to make it known despite my searches.  Is it as simple as editing the ts011f_plug.py file to include info to identify/map the device?  Thanks in advance for any guidance you can provide.

MattWestb, 2022-11-13:
I have looking on the original posted device signature and its having the same "_TZ3000_cfnprab5" but your is having the same but the signature looks complete different so its not helping.

Can you delete the device from ZHA and waiting one minute and adding it agen and then all is stable and working "normal" downloading the diagnostic from the device card and posting it one more time ?

javicalle, 2022-11-13:
Hi @fasteddie-g ,
 yours has another flawor from Tuya devices and will need a new quirk to work. 

Please, restore the original `tx011f_plug.py` file version and add this quirk at the end of the file:

<details open>
<summary>Plug_4AC_2USB_Var04</summary>

```python
class Plug_4AC_2USB_Var04(EnchantedDevice):
    """Tuya 4 outlet + 2 USB (variation 04))."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=["0x0000","0x0003","0x0004","0x0005","0x0006","0xe001"]
            # output_clusters=[]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            2: {
                # input_clusters=["0x0000","0x0003","0x0004","0x0005","0x0006","0xe001"]
                # output_clusters=[]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                # input_clusters=["0x0000","0x0003","0x0004","0x0005","0x0006","0xe001"]
                # output_clusters=[]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                # input_clusters=["0x0000","0x0003","0x0004","0x0005","0x0006","0xe001"]
                # output_clusters=[]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
[... 74 lines cut ...]
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

```

</details>

Remove any `__pycache__` folder from your local quirk, restart HA and repair your device.

fasteddie-g, 2022-11-13:
Thank you both!   @javicalle  Thank you for taking the time to provide the updated class. I believe I made the changes as recommended, I didn't see any __pycache__ folder anywhere in file editor (/config or subdirs like zha_quirks/), but HA still doesn't appear to be picking up the quirk when the device is added, and the behavior is the same where I'm unable to control any of the individual plugs (or the entire strip).   I've modified templates to get some unknown Zwave devices to work in the past (like Yale locks), but I'm new to Zigbee.  So I sincerely appreciate the assistance & suggestions.  

Just to provide context in case I am missing something, or following an incorrect process.  Below are my configuration.yaml, Diagnostic info, and updated ts011f_plug.py file.   When I try to add the device, I follow these steps:
1. Ensure all targeted configs (configuration.yaml, quirks files, etc) are in their desired state.
2. Power strip device (_TZ3000_cfnprab5 TS011F) is removed from Zigbee Home Automation (ZHA)
3. Restart HomeAssistant
4. Once HomeAssistant is fully loaded
    a. ZHA - "Add device" 
    b. Power strip - Hold power button for 5 seconds to reset/put into pairing mode
    c. Wait ~1 minute and check status of device in HomeAssistant

Any idea on next steps?  Anything else I should be checking/confirming or doing?

<details>
<summary>configuration.yaml</summary>

```Yaml
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate
group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks/

zha_toolkit:

logger:
  # The next line sets the default logging level, for all python modules.
  # It seems "recommended" to set it to avoid too much logging.
  default: warning
  logs:
    custom_components.zha_toolkit: debug
    zigpy: debug
    zhaquirks: debug
    homeassistant.components.zha: debug
```
</details>

<details>
<summary>Updated /config/zha_quirks/ts011f_plug.py</summary>

```Python
"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
[... 867 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xe001": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```
</details>

javicalle, 2022-11-13:
Don't worry, all the steps seem fine to me.
There was an error on the proposed quirk, I have fixed all the `DEVICE_TYPE` definitions from the signature part (they are `ON_OFF_LIGHT`).
Fix or replace the quirk, check if there's a `__pycache__` folder, restart HA and repair the device.

fasteddie-g, 2022-11-13:
@javicalle Thank you again.  Progress - it picked up the quirk this time!!  But the general experience is the same - unable to control the device from HA. 

<details>
<summary>Diagnostic info</summary>

```Yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v0.8.23",
      "requirements": [
        "packaging>=20.8"
      ]
    },
    "hacs": {
      "version": "1.28.3",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "keymaster": {
      "version": "v0.0.82",
      "requirements": []
    },
    "fordpass": {
      "version": "0.1.37",
      "requirements": []
    },
    "alexa_media": {
      "version": "4.3.2",
      "requirements": [
        "alexapy==1.26.4",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "dwains_dashboard": {
      "version": "3.3.0",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.85",
      "zigpy-deconz==0.19.0",
[... 540 lines cut ...]
                "attribute_name": "on_off",
                "value": 1
              }
            },
            "unsupported_attributes": {
              "0x8002": {
                "attribute_name": "power_on_state"
              },
              "0x4003": {
                "attribute_name": "start_up_on_off"
              }
            }
          },
          "0xe001": {
            "endpoint_attribute": "tuya_external_switch_type",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
</details>

javicalle, 2022-11-13:
When you say "_unable to control the device from HA_", which is the behavior? Is device disabled? Is enabled but don't change its status? Status change in HA but not in the device? 

There is another try for your device. Over your current version do the following:
1. at the begining of file, add the `` import:
```python
from zhaquirks.tuya.mcu import EnchantedDevice, TuyaOnOffNM
```

2. Use the new class in any of the **replacement** clusters, i.e. in cluster 1:
```python
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffNM,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
```

Remove, restart, repair and test if the behavior for switch 1 changes.

fasteddie-g, 2022-11-13:
Thanks for the quick reply.  I'll try the suggestions shortly and report back. 

> When you say "unable to control the device from HA", which is the behavior? Is device disabled? Is enabled but don't change its status? Status change in HA but not in the device?

The device appears normal in HA, and I can toggle any of the plugs/switches in the UI, but they have no impact on the device.  Also, if I turn the entire power strip on or off, it does not update in HA.  Visually, at the bottom of this reply is a screen shot of what I see right now.  It shows the switches as all on, because I joined the device to HA when the plug was on.  But if toggle any of them to off, in ~5 seconds it will just revert to back on.

<details>
<summary>Logs around the time I just tried to toggle one of the switches to off</summary>
```yaml
2022-11-13 16:22:52.363 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x576B), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=230, profile_id=260, cluster_id=12, data=Serialized[b'\x18\xe5\nU\x009\x85\xeb\xd1>'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-54)
2022-11-13 16:22:52.364 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Received ZCL frame: b'\x18\xe5\nU\x009\x85\xeb\xd1>'
2022-11-13 16:22:52.364 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=229, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-13 16:22:52.365 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Decoded ZCL frame: AnalogInputCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.4099999964237213))])
2022-11-13 16:22:52.365 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Received command 0x0A (TSN 229): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.4099999964237213))])
2022-11-13 16:22:52.365 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Attribute report received: present_value=0.4099999964237213
2022-11-13 16:22:53.361 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x576B), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=231, profile_id=260, cluster_id=12, data=Serialized[b'\x18\xe7\nU\x009\x14\xae\xc7>'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-54)
2022-11-13 16:22:53.361 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Received ZCL frame: b'\x18\xe7\nU\x009\x14\xae\xc7>'
2022-11-13 16:22:53.361 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=231, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-13 16:22:53.362 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Decoded ZCL frame: AnalogInputCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.38999998569488525))])
2022-11-13 16:22:53.362 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Received command 0x0A (TSN 231): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.38999998569488525))])
2022-11-13 16:22:53.363 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Attribute report received: present_value=0.38999998569488525
2022-11-13 16:22:55.150 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x29E0](ZB-CL01): last_seen is 57022.99984884262 seconds ago and ping attempts have been exhausted, marking the device unavailable
2022-11-13 16:22:55.150 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x29E0](ZB-CL01): Update device availability - device available: False - new availability: False - changed: False
2022-11-13 16:22:55.171 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x12F9](ZB-CL01): Device seen - marking the device available and resetting counter
2022-11-13 16:22:55.171 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x12F9](ZB-CL01): Update device availability - device available: True - new availability: True - changed: False
2022-11-13 16:22:55.365 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x576B), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=232, profile_id=260, cluster_id=12, data=Serialized[b'\x18\xea\nU\x009\xb8\x1e\x85>'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-54)
2022-11-13 16:22:55.365 DEBUG (MainThread) [zigpy.zcl] [0x576B:2:0x000c] Received ZCL frame: b'\x18\xea\nU\x009\xb8\x1e\x85>'
[... cut ...] us.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=0))])
2022-11-13 16:24:23.193 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4F93](TS0505B): last_seen is 55026.01232743263 seconds ago and ping attempts have been exhausted, marking the device unavailable
2022-11-13 16:24:23.194 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4F93](TS0505B): Update device availability - device available: False - new availability: False - changed: False
```
</details>

[image]

fasteddie-g, 2022-11-13:
I updated the quirk file, added the additional import line at the top, and modified endpoints 1 and 2 with the suggested code.  Similar user experience where changes in the UI doesn't impact the device.

<details>
<summary>Logs when I tried to toggle all 5 switches to off</summary>

```Yaml

2022-11-13 16:43:54.474 DEBUG (MainThread) [zigpy.zcl] [0x5605:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=21, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-13 16:43:54.475 DEBUG (MainThread) [zigpy.zcl] [0x5605:1:0x0006] Sending request: off()
2022-11-13 16:43:55.282 DEBUG (MainThread) [zigpy.zcl] [0x5605:2:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=23, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-13 16:43:55.282 DEBUG (MainThread) [zigpy.zcl] [0x5605:2:0x0006] Sending request: off()
2022-11-13 16:43:55.746 DEBUG (MainThread) [zigpy.zcl] [0x5605:3:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=25, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-13 16:43:55.746 DEBUG (MainThread) [zigpy.zcl] [0x5605:3:0x0006] Sending request: off()
2022-11-13 16:43:55.754 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x18F5](ZB-CL01): last_seen is 151350.56500911713 seconds ago and ping attempts have been exhausted, marking the device unavailable
2022-11-13 16:43:55.755 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x18F5](ZB-CL01): Update device availability - device available: False - new availability: False - changed: False
2022-11-13 16:43:56.217 DEBUG (MainThread) [zigpy.zcl] [0x5605:4:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=27, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-13 16:43:56.218 DEBUG (MainThread) [zigpy.zcl] [0x5605:4:0x0006] Sending request: off()
2022-11-13 16:43:56.641 DEBUG (MainThread) [zigpy.zcl] [0x5605:5:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=29, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-13 16:43:56.642 DEBUG (MainThread) [zigpy.zcl] [0x5605:5:0x0006] Sending request: off()
2022-11-13 16:43:59.187 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5605:1:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: 'Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'
2022-11-13 16:43:59.723 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x12F9](ZB-CL01): Device seen - marking the device available and resetting counter
2022-11-13 16:43:59.723 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x12F9](ZB-CL01): Update device availability - device available: True - new availability: True - changed: False
2022-11-13 16:43:59.993 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5605:2:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: 'Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'
2022-11-13 16:44:00.456 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5605:3:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: 'Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'
2022-11-13 16:44:00.762 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA9C0](ZB-CL01): last_seen is 54811.98769211769 seconds ago and ping attempts have been exhausted, marking the device unavailable
2022-11-13 16:44:00.762 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA9C0](ZB-CL01): Update device availability - device available: False - new availability: False - changed: False
2022-11-13 16:44:00.926 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5605:4:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: 'Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'
2022-11-13 16:44:01.351 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5605:5:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: 'Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'
2022-11-13 16:44:02.738 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4F93](TS0505B): last_seen is 56205.55712080002 seconds ago and ping attempts have been exhausted, marking the device una [... cut ...]
                "attribute_name": "power_on_state"
              },
              "0x4003": {
                "attribute_name": "start_up_on_off"
              }
            }
          },
          "0xe001": {
            "endpoint_attribute": "tuya_external_switch_type",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```
</details>
|
[add_device_log.txt](https://github.com/zigpy/zha-device-handlers/files/9998417/add_device_log.txt)

MattWestb, 2022-11-14:
After adding `EnchantedDevice` to the device class you need deleting the device in ZHA and waiting one minute and adding it new.
If not doing it the the "tuya magic" is not casted and all switches is being as one from the GUI.

botswanabub, 2022-11-19:
> what quirk is associated with it? I copied the original fix (which basically removes identity.cluster_id and TuyaZBE000Cluster.cluster_id from endpoints 2,3,4,5) in a class called Plug_4AC_2USB_cfnprab5. Now, even though this quirk has been reinstalled by a monthly update it's still work (I assume it's cached and would break if I tried to pair again).

hi, can u show your config in the ts*** file ... I tried it too without success.

botswanabub, 2022-11-19:
finally worked, not sure why, I think because of deleting __pycache__ in the tuya folder. have to checkup and come back.

[image]

MaxMSchneider, 2023-03-10:
Hey all,

I recently purchased this power strip to replace my other one running on localtuya. Problem is, I can't get it to pair even.

While pairing in ZHA it is discovered, but can not finish the interview as it seems. Tried pairing via only the coordinator as well as the closest router(s). The strange behavior is that it seems to leave the network, constantly swapping to another one.

Anyone had this issue, too?

javicalle, 2023-03-10:
Please open a new issue with all the required info.
Attach also the logs from the pairing process.

ingoratsdorf, 2023-07-08:
While this is closed, it's the best place for another version of
TS011F
by _TZ3000_cfnprab5

My version has 4AC plug points and 4 USB ports and no power restore etc. Really only 5 switches and nothing else.
The quirk for that one works for me:

```
"""TS011F multiboard with 4 outlet + 4 USB without any metering or restore"""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Scenes,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)

class Plug_4AC_2USB_TZ3000_cfnprab5(CustomDevice):
    """Tuya 4 outlet + 4 USB without any metering or restore"""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            1: {
                # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0100
                # input_clusters=[0, 3, 4, 5, 6, 0xe001]
                # output_clusters=[]>
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            2: {
                # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0100
                # input_clusters=[0, 3, 4, 5, 6, 0xe001]
                # output_clusters=[]>
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
[... 100 lines cut ...]
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

```

commit 4a950f7f (2022-09-12): Fixes #1632 (#1735)