zigpy/zha-device-handlers #1633: [BUG] Wrong commands Hue Dimmer Switch 324131137411(EU) RWL021

Homergy, 2022-06-26:
**Describe the bug**
Same issue as was posted here [https://github.com/zigpy/zha-device-handlers/issues/878](url) but since that ticket is closed I believe it would be a good idea to make a new post. I have 2 old Hue dimmers they are up to date and both have the exact same behavior.
If I read it correctly the changes to fix this were later pulled? [https://github.com/zigpy/zha-device-handlers/pull/884](url)
Can I possibly manually integrate them? (I'm very new to this)

**To Reproduce**
Steps to reproduce the behavior:
1. Add Hue Dimmer RWL021 324131137411 to ZHA
2. Go to Developer Tools --> Events --> Listen to events --> "zha_event"
3. See that the commands are not what they are supposed to be, only 3 commands "on" , "off_with_effect", "step"
4. Automations and blueprints don't work due to wrong commands

**Expected behavior**
Recieve the new updated commands when pressing a button:
 ''on_short_release''--> 'button_1_single_press'
 ''on_hold'' --> 'button_1_long_press'
 ''on_double_press'' --> 'button_1_double_press'
 ''up_short_release''--> 'button_2_single_press'
 ''up_hold'' --> 'button_2_long_press'
 ''up_double_press'' --> 'button_2_double_press'
 ''down_short_release'' --> 'button_3_single_press'
 ''down_hold'' --> 'button_3_long_press'
 ''down_double_press'' --> 'button_3_double_press'
 ''off_short_release'' --> 'button_4_single_press'
 ''off_hold'' --> 'button_4_long_press'
 ''off_double_press'' --> 'button_4_double_press'

**Screenshots**
A possible solution would be to reconfigure the device, unfortunately this doesn't work:
[image]

<details>
<summary>Device signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4107, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=0, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0830",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0xfc00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "RWL021",
  "class": "zhaquirks.philips.rwlfirstgen.PhilipsRWLFirstGen"
}
```

</details>

MrCamby, 2022-06-26:
Same for me

Homergy, 2022-06-26:
@TheJulianJES as requested a new topic and tagged you.

TheJulianJES, 2022-06-26:
> If I read it correctly the changes to fix this were later pulled? [https://github.com/zigpy/zha-device-handlers/pull/884](url)
> Can I possibly manually integrate them? (I'm very new to this)

The changes were pulled into Home Assistant (meaning they're already included in Home Assistant).

I have the same exact dimmer (same signature) and it's working just fine for me here and the quirk is correctly applying for your dimmer. So it's a different issue than the older one.

You could try to re-pair your dimmer to Home Assistant again or pressing a button on the dimmer during reconfiguring.

Homergy, 2022-06-26:
@TheJulianJES You're a hero!!!
How could the solution be so simple... I adapted blueprint, paired and re-paired,... while all I had to do was press a button during reconfiguring... a bit ashamed I did not know that was a trick.

I have 2 dimmers tried removing it from HA, re-pairing to Hue check for upgrades (which there were none) and repairing it back to ZHA. Both still had the issue.
Pressing a button while reconfiguring did do the trick!
Still got an error but it works:
[image]

The strange (or normal?) thing is that it now sends 3 commands while pressing the on button:
<details>
<summary>"zha_event" log for just one on press button</summary>
```
Event 40 fired 8:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:2:0xfc00",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 2,
        "cluster_id": 64512,
        "command": "on_press",
        "args": {
            "button": "on",
            "press_type": "press",
            "command_id": 0,
            "args": [
                1,
                3145728,
                0,
                33,
                0,
                0
            ]
        },
        "params": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-26T18:33:39.012458+00:00",
    "context": {
        "id": "01G6GMMVY4TX301RGYCC786HWH",
        "parent_id": null,
        "user_id": null
    }
}
Event 39 fired 8:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:2:0xfc00",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 2,
        "cluster_id": 64512,
        "command": "on_short_release",
        "args": {
            "button": "on",
            "press_type": "short_release",
            "command_id": 0,
            "args": [
                1,
                3145728,
                2,
                33,
                0,
                0
            ]
        },
        "params": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-26T18:33:38.801927+00:00",
    "context": {
        "id": "01G6GMMVQH886Q152YTX2WDVKT",
        "parent_id": null,
        "user_id": null
    }
}
Event 38 fired 8:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:1:0x0006",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": [],
        "params": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-26T18:33:38.730748+00:00",
    "context": {
        "id": "01G6GMMVNAHB6173VP5MX6S2F4",
        "parent_id": null,
        "user_id": null
    }
}
```
</details>
However it works fine now thx for the quick support!

MrCamby, 2022-06-26:
For me it does still not work.
I tried reconfiguring it multiple times now, but it does not work for me.

[image]

dmulcahey, 2022-06-26:
> For me it does still not work. I tried reconfiguring it multiple times now, but it does not work for me.
> 
> [image]

Are you pressing a button on the remote to keep it awake while reconfigure is running?

MrCamby, 2022-06-26:
Yes, I pressed every button (one after another, aswell as only one button) with a delay of maybe a second between each click.

closed 2022-12-30 by github-actions[bot]