zigpy/zha-device-handlers #1634: [Device Support Request] Tuya wall light switches TS0601 _TZE200_ 1-gang, 2-gang, 3-gang, 4-gang

ProgrammerIamNot, 2022-06-26:
Hi, 
I bought a bunch of Tuya switches from one manufacturer. 
They are: 
1-gang: TS0601 _TZE200_gbagoilo
2-gang: TS0601 _TZE200_nh9m9emk
3-gang: TS0601 _TZE200_go3tvswy
4-gang: TS0601 _TZE200_mexisfik

The devices can be paired but no entities are available. 

Can these be added to supported devices, please?

Thank you.

<details>

<summary>Device signature</summary>

```yaml

####### 1-gang switch #######

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_gbagoilo",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

####### 2-gang switch #######

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
[... 300 lines cut ...]
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": "Entrance",
    "device_reg_id": "db9a98b73f23c1a7e3dbab1a0210c7e1",
    "area_id": "living_room"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
will add if needed
```

</details>

**Additional context**
I also purchased a dimmer from the same manufacturer and it works
Dimmer: TS0601 _TZE200_1agwnems

javicalle, 2022-06-28:
These seem to be some of the new GPP Tuya devices.
Could you attach some manufacturer or vendor link to help other users to identify the device?

These would work just adding the model info in the existing quirks:
* enable the `custom_quirks_path`
  * https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274
* copy the `ts0601_switch.py` file inside the folder
  * https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0601_switch.py
* add the model info of every device to its GPP class:
  * TuyaSingleSwitch_GP (replace the `YYYYYYYYYYYY` value):
```python
        MODELS_INFO: [
            ("_TZE200_gbagoilo", "TS0601"),
        ],
```
  * TuyaDoubleSwitch_GP:
```python
        MODELS_INFO: [
            ("_TZE200_7deq70b8", "TS0601"),
            ("_TZE200_nh9m9emk", "TS0601"),
        ],
```
  * TuyaTripleSwitch_GP:
```python
        MODELS_INFO: [
            ("_TZE200_2hf7x9n3", "TS0601"),  # reported #1597
            ("_TZE200_go3tvswy", "TS0601"),  # reported #1597
        ],
```
  * TuyaQuadrupleSwitch_GP (replace the `XXX` value):
```python
        MODELS_INFO: [
            ("_TZE200_mexisfik", "TS0601"),
        ],
```
* restart HA and check

ProgrammerIamNot, 2022-06-28:
Thanks, @javicalle for your response. I'll give it a try.

I  bought the switches via Alibaba [here](https://www.alibaba.com/product-detail/US-AU-Standard-Smart-Zigbee-Switch_1600481439953.html?spm=a2700.galleryofferlist.normal_offer.d_title.57fb3511HPMHCg)
1-gang Zigbee switch on their website is [here](http://www.szmakegood.com/show-35-25.html).
Looks like this:
[image]

ProgrammerIamNot, 2022-06-28:
So it did work for me.
The only thing I needed to fix for 1-gang is changing its replacement input cluster from TuyaOnOff to TuyaOnOfNM. Must have been just a copy-paste mistake there.
Thanks again, @javicalle . I appreciate your support.
[image]

ProgrammerIamNot, 2022-06-28:
Just in case for future reference in order to turn pairing mode any button needs to be toggled five times and then held down for ~5 sec until the lights start flashing.

javicalle, 2022-06-28:
> for 1-gang is changing its replacement input cluster from TuyaOnOff to TuyaOnOfNM. Must have been just a copy-paste mistake there.

Ouuuuch. Yes it is.

Once tested, you can create the PR to add your devices to the mainstream?

Just edit from the web the file content with the detected changes and follow the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py

This way other users will be able to use the devices in ZHA thanks to you!

shadmanhabib, 2022-12-19:
I've created a pull request which adds these device models. I just installed a 2-gang with the same device model as yours @ProgrammerIamNot and was looking for a solution.

Found this thread and thought I'd implement it for everyone. Hopefully all works well!

closed 2023-07-10 by github-actions[bot]

putimir, 2025-06-05:
I found a solution, for anyone that faces the same problem:

As seen from the logs, the device presents itself as 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.**ON_OFF_LIGHT**: 256>, 'input_clusters': [0, 4, 5, 61184], 'output_clusters': []}}}), empahasis on ON_OFF_LIGHT.

The quirk is looking (all devices in the quirk) for a SMART_PLUG device:

` DEVICE_TYPE: zha.DeviceType.SMART_PLUG,`

So in order to make this  switch to be recognised by the quirk, you have to:

1. edit the MODELS_INFO: to include the correct id (_TZE200_wvovwe9h in this case)
2. edit the DEVICE_TYPE to ON_OFF_LIGHT:

`DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,`