zigpy/zha-device-handlers #1635: [BUG] Woox R7060 Irrigation Controller

overas, 2022-06-26:
**Describe the bug**
Woox R7060 Irrigation Controller device joins zigbee network and it's recognized by home-assistant.
ZHA has support for the device already: https://github.com/zigpy/zha-device-handlers/issues/939. Zigbee2mqtt has also support for this device: https://www.zigbee2mqtt.io/devices/R7060.html. 

The device gives  a lot of `WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'` in the HA log.

Entities are created for:
On/Off switch (switch.vanning_bringebar_on_off): It works.
Sensor battery power in procent (sensor.vanning_bringebar_power): Values updates correctly.
Sensor lqi (sensor.tz3210_eymunffl_ts0101_3d24d4fe_basic_lqi): Values updates correctly. The sensor is disabled and must be enabled before working.
Sensor rssi (sensor.tz3210_eymunffl_ts0101_3d24d4fe_basic_rssi):  Values do not updated. Give state "Unknown". The sensor is disabled and must be enabled before working.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a scheduled irrigation automation which open the switch for watering and closing the switch after watering.
```
  - service: switch.turn_on
    data: {}
    target:
      entity_id: switch.vanning_pergola_on_off
  - delay:
      hours: 0
      minutes: 30
      seconds: 0
      milliseconds: 0
  - service: switch.turn_off
    data: {}
    target:
      entity_id: switch.vanning_pergola_on_off
```
2. In the HA log I get a lot of these warnings `WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b' `. I have six Woox R7060 Irrigation Controller placed in my garden. The watering starts 06:00 and ends 08:30.

**Expected behavior**
No warnings in the HA log.

**Screenshots**
[image]
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_eymunffl",
  "model": "TS0101",
  "class": "zigpy.device.Device"
}
```

[... 301 lines cut ...]
2022-06-25 08:20:46 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:46 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:47 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:47 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:48 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:48 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:49 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:20:49 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Nf\x04\x00\x01\x00'
2022-06-25 08:50:47 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:48 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:48 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:49 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:49 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:50 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:50 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:51 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:51 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:51 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
2022-06-25 08:50:52 WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-06-28:
Hi,  I will try to clarify some things.

> ZHA has support for the device already: #939. 

No, not really. Issue #939 dealt with this same issue, but no action was taken on it, so there is no quirk created for the device and your device is handled as an standard Zigbee device:
```
"class": "zigpy.device.Device"
```

> Zigbee2mqtt has also support for this device: https://www.zigbee2mqtt.io/devices/R7060.html.

Yes, but only the voltage (_Voltage of the battery in millivolts_) is diferent here.

> The device gives a lot of `WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02'` in the HA log.

This is due to the reports that the device is sending by a manufacturer cluster.
As the manufacturer does not document what these reports are, we can only try to guess their meaning based on the functionality of the device and the reported values.

For this reason, in #939 they were asked to use a quirk implementation that can provide information on the type of data and value reported by the device.

And that is what I am going to suggest to you.

Configure the custom_quirk in your installation and create the following quirk inside the folder:
```python
"""Woox irrigation controller."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    OnOff,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import (
    TUYA_MCU_COMMAND,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
)

class WooxR7060(CustomDevice):
    """Woox R7060 Irrigation Controller."""

    signature = {
        MODELS_INFO: [("_TZ3210_eymunffl", "TS0101")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaMCUCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaMCUCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

It's expected that these `Unknown cluster command 6` will become something like:
```
2022-06-25 15:19:25 WARNING (MainThread) [zigpy.zcl] [0xC958:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
```

And that could give to us some information about what is in these reports and lead to a quirk for the device.

overas, 2022-07-01:
Maybe not relevant, but here are Woox specifications for r7060: 
https://wooxhome.com/products-c10/other-c4/woox-r7060-smart-garden-irrigation-control-p61

> And that could give to us some information about what is in these reports and lead to a quirk for the device.

[image]

`➜  config grep -i -C5 "No 'handle_active_status_report'" home-assistant.log`
[extract_home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9027446/extract_home-assistant.log)

overas, 2022-07-12:
I did a scan_device with zha-toolkit. Here are the result:

{
    "ieee": "bc:33:ac:ff:fe:46:1a:a5",
    "nwk": "0xab7b",
    "model": "TS0101",
    "manufacturer": "_TZ3210_eymunffl",
    "manufacturer_id": "0x4098",
    "endpoints": [
        {
            "id": 1,
            "device_type": "0x0051",
            "profile": "0x0104",
            "in_clusters": {
                "0x0000": {
                    "cluster_id": "0x0000",
                    "title": "Basic",
                    "name": "basic",
                    "attributes": {
                        "0x0000": {
                            "attribute_id": "0x0000",
                            "attribute_name": "zcl_version",
                            "value_type": [
                                "0x20",
                                "uint8_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 3
                        },
                        "0x0001": {
                            "attribute_id": "0x0001",
                            "attribute_name": "app_version",
                            "value_type": [
                                "0x20",
                                "uint8_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 69
                        },
                        "0x0002": {
                            "attribute_id": "0x0002",
                            "attribute_name": "stack_version",
                            "value_type": [
                                "0x20",
                                "uint8_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 0
                        },
                        "0x0003": {
                            "attribute_id": "0x0003",
                            "attribute_name": "hw_version",
                            "value_type": [
                                "0x20",
                                "uint8_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 1
                        },
                        "0x0004": {
                            "attribute_id": "0x0004",
                            "attribute_name": "manufacturer",
[... 375 lines cut ...]
                    "commands_received": {},
                    "commands_generated": {}
                }
            },
            "out_clusters": {
                "0x000a": {
                    "cluster_id": "0x000a",
                    "title": "Time",
                    "name": "time",
                    "attributes": {},
                    "commands_received": {},
                    "commands_generated": {}
                },
                "0x0019": {
                    "cluster_id": "0x0019",
                    "title": "Ota",
                    "name": "ota",
                    "attributes": {},
                    "commands_received": {},
                    "commands_generated": {}
                }
            }
        }
    ]
}

giulio12, 2023-05-22:
Do you know the status of this issue? I am having problems with this device

overas, 2023-05-23:
The Woox R7060 Irrigation Controller works without any warning logging now.
`The device gives a lot of WARNING (MainThread) [zigpy.zcl] [0x48A5:1:0xef00] Unknown cluster command 6 b'\x00Of\x04\x00\x01\x02' in the HA log.`

I use ZHA and Home Assistant 2023.5.3. I think ZHA have device_quirks for this device now.
[image]

ZHA diagnostic:
[zha-7c3d20ba7a7f008249aa59370c96c47d-_TZ3210_eymunffl TS0101-88244f8c0b5769b7fc66a8404f1122f5.json.txt](https://github.com/zigpy/zha-device-handlers/files/11541867/zha-7c3d20ba7a7f008249aa59370c96c47d-_TZ3210_eymunffl.TS0101-88244f8c0b5769b7fc66a8404f1122f5.json.txt)

giulio12, 2023-05-23:
Thank you for the answer. May I ask you which coordinator are you using? I have the conbee2 and I think that might the problem

overas, 2023-05-24:
I use a ZigStar Stick v4 (https://zig-star.com/projects/zigbee-stick-v4/). Some people have reported with conbee unstable network with wrong firmware https://github.com/Koenkk/zigbee2mqtt/issues/9554

Lifth1309, 2023-05-30:
I'm also getting these kinds of errors:
`
[0x77EE:1:0xef00] Unknown cluster command 6 b'\x00\xabf\x04\x00\x01\x00'`

I'm also running into errors when i'm trying to do a zha_toolkit scan_device on it:

```
got Status.UNSUP_GENERAL_COMMAND status for discover_commands starting 0
Failed 'discover_attributes_extended' starting 0x0005/0x0000. Error:
Failed 'discover_attributes_extended' starting 0x0000/0x0000. Error:
Couldn't read 0x0005/0xfffe:
Couldn't read 0xef00/0x0001: 
```

My Coordinator is a Sonoff Zigbee 3.0 USB Dongle.
I've got two of those Woox R7060, both are behaving the same way.

I'm have zero experience with quirks so far - is it possible, that i am somehow missing a zha quirk?

overas, 2023-05-31:
From the ZHA diagnostic it seems the zigbee device do not use any quirks.
```
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 5062,
    "manufacturer": "_TZ3210_eymunffl",
    "model": "TS0101",
    "name": "_TZ3210_eymunffl TS0101",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4098,
    "power_source": "Battery or Unknown",
```
Maybe it is the coordinator which logs the message

closed 2023-12-04 by github-actions[bot]