zigpy/zha-device-handlers #1636: [BUG] lumi.plug.maus01 only reporting electrical_measurement (Watts)

stuartf, 2022-06-27:
**Describe the bug**
Paired a brand new Aqara Model ZNCZ12LM manufactured by Xiaomi to Home Assistant, it shows a number of entities in state `Unknown` and electrical_measurement which seems to accurately report the wattage being used by the plugged in device. 

**To Reproduce**
Steps to reproduce the behavior:
1. Pair device
2. View entities
3. See that they are not populated

**Expected behavior**
Configured entities should show actual data from the plug

**Screenshots**
[image]

<details>

<summary>lumi.plug.maus01</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    },
    "100": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x000f"
      ],
[... 362 lines cut ...]
      {
        "name": "OCCUPANCY_SENSOR"
      }
    ],
    "user_given_name": "Washer",
    "device_reg_id": "e24f857ca8691a225ceadd17ca25eb12",
    "area_id": "laundry"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

stuartf, 2022-07-07:
I'm finding that the electrical_measurement (Watts) and the switch disappear after restarting Home Assistant, making the plug useless until re-paired.

stuartf, 2022-07-07:
It appears that since upgrading the switch, electrical_measurement (Watts), and the analog_input (that seems to just be Watts with a single decimal precision) no longer disappear on restart, which is nice. But all the other entities still show as `Unknown`.
[Here's the updated diagnostics](https://github.com/zigpy/zha-device-handlers/files/9067516/zha-f97ea53801e9a0b3202e634d8284ff9f-LUMI.lumi.plug.maus01-e24f857ca8691a225ceadd17ca25eb12.json.1.txt)

dmulcahey, 2022-07-07:
Can you try removing the entities. I bet if you click one HA will say it’s orphaned and give you the option to delete it.

multicm, 2022-07-15:
Just wanted to jump in and say I am experiencing the same issue, all sensors except watts is showing as "unknown"

tias2001, 2022-08-03:
I have the same issue
Conbee II 
deCONZ = dresden elektronik deCONZ protocol: ConBee I/II, RaspBee I/II
by ZHA
Plug
Nwk: 0x0857
Device Type: Router
LQI: 255
RSSI: -40
Last Seen: 2022-08-03T09:44:33
Power Source: Mains
Quirk: zhaquirks.xiaomi.aqara.plug.Plug

11jwolfe2, 2022-12-09:
I am also having this issue as power consumption is not found! Seems like The EU version had a similar issue but was resolved earlier this year. Anyone know how to fix on the US version? Willing and able to dive in on diagnostics for this issue!

TheJulianJES, 2023-01-04:
This should partly be fixed in Home Assistant [2023.1.0b5](https://github.com/home-assistant/core/releases/tag/2023.1.0b5) and later by:
- https://github.com/zigpy/zha-device-handlers/pull/2057

You might need to fully remove the plug and re-pair it to get the functionality working.

Note that the issue also states: "There are several other functions supported by the device that I'll add in a subsequent PR."

thomasmb, 2023-01-12:
I added 2 of these EU aqara smart plugs (paired after upgrading to 2023.1) – and power measurements are currently not reported.
[image]

TheJulianJES, 2023-01-12:
@thomasmb This issue is about different plugs and has nothing to do with your issue.

Try the following:
If you're using a Conbee, update its firmware to a somewhat recent version and re-pair the device.
If you're using an EZSP based dongle (like SkyConnect), then update to the latest HA version and re-pair the device.

If you still have issues, create a new issue on this repo and fill-out the issue template.

closed 2023-07-18 by github-actions[bot]