zigpy/zha-device-handlers #1641: [Device Support Request] GLEDOPTO GL-C-002P

FireHelmet, 2022-06-30:
Hello !

I have a new GLEDOPTO controller, details are [here](https://www.gledopto.eu/5in1-zigbee-controller-mini-thin-gledopto). I was able to pair with Home-Assistant, entities are present but the controls (on/off, brightness, temperature, etc) are not working.

Thank you for your help!

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4687, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "GLEDOPTO",
  "model": "GL-C-006P",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.5.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "4.18.8-odroidc2",
    "run_as_root": true
  },
  "custom_components": {
    "momentary": {
      "version": "0.5.3",
      "requirements": []
    },
    "dyson_cloud": {
      "version": "0.15.0",
      "requirements": [
        "libdyson==0.8.7"
      ]
    },
    "dyson_local": {
      "version": "0.16.4-4",
      "requirements": [
        "libdyson==0.8.11"
      ]
    },
    "lifx": {
      "version": "2022.06.03",
      "requirements": [
        "aiolifx==0.8.1",
        "aiolifx_effects==0.2.2"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.73",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 35784,
    "manufacturer": "GLEDOPTO",
    "model": "GL-C-006P",
    "name": "GLEDOPTO GL-C-006P",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4687,
    "power_source": "Mains",
    "lqi": null,
    "rssi": null,
    "last_seen": "2022-06-30T12:09:57",
    "available": false,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4687, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "11": {
          "profile_id": 260,
          "device_type": "0x010c",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x0300",
            "0x1000"
          ],
          "out_clusters": [
            "0x0019"
          ]
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    },
    "entities": [
      {
        "entity_id": "button.led_strip_identify",
        "name": "GLEDOPTO GL-C-006P"
      },
      {
        "entity_id": "light.led_strip_level_light_color_on_off",
        "name": "GLEDOPTO GL-C-006P"
      },
      {
        "entity_id": "select.led_strip_on_off_startuponoff",
        "name": "GLEDOPTO GL-C-006P"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "COLOR_TEMPERATURE_LIGHT"
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": "Led strip",
    "device_reg_id": "3cc1f10bb2f9fb903d97bda720ba7ba6",
    "area_id": "salon"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

TheJulianJES, 2022-06-30:
The ``device signature`` and ``diagnostic information`` you submitted seems to be for a ``GL-C-006P`` (and not a ``GL-C-002P``).
Are you sure that you entered the correct information? (Does the controller really report itself as another model?)

Also, you could (backup and) upgrade Home Assistant to the latest release.

FireHelmet, 2022-06-30:
Yes, the model detected by ZHA is related to the mode of the device. I can switch mode (Dim, CCT, RGB, RGBW, RGBCCT) by pushing opt button. So, the device model is GL-C-002P but the software model depends of the mode.

I will upgrade and do a new pairing

FireHelmet, 2022-06-30:
Thanks for your support!

I will do it tomorrow and I keep you informed

TheJulianJES, 2022-07-01:
Apparently there's a firmware bug in the device. This might be related: https://github.com/Koenkk/zigbee2mqtt/issues/11202

FireHelmet, 2022-07-02:
Oh okay, I will take a look so. Do you know how can I verify the current firmware version before doing the upgrade?

FireHelmet, 2022-07-08:
Hello @TheJulianJES 

I followed the instruction given by Sunxperious, here : https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1173047823

I was able to start the upgrade but my device doesn't stay paired to the coordinator and so the upgrade stop.

2022-07-08 21:24:02 DEBUG (MainThread) [zigpy.zcl] [0xCC02:11:0x0019] OTA image_block handler for 'GLEDOPTO GL-C-008P': field_control=FieldControl.MinimumBlockPeriod, manufacturer_id=4687, image_type=5145, file_version=352399361, file_offset=5200, max_data_size=48, request_node_addr=Noneblock_request_delay=0
2022-07-08 21:24:02 DEBUG (MainThread) [zigpy.zcl] [0xCC02:11:0x0019] OTA upgrade progress: 2.4

Could you help me?

Thank you for your support!

acburnett, 2022-09-19:
I'm having similar issues with the  GL-C-002P. When set to RGB mode, the device reports as a GL-C-003P. However I dont get a color wheel, I only get a white temperature control slider that doesn't do anything. On/Off behaves properly but light color cant be changed at all. On the newest firmware for the device and home assistant is fully up to date.

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4687, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x0102",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "GLEDOPTO",
  "model": "GL-C-003P",
  "class": "zigpy.device.Device"
}

closed 2023-07-11 by github-actions[bot]

vukitoso, 2023-11-18:
Hello. Please tell me if it is possible to configure the controller via z2m so that after turning off and on the electricity, the light is always OFF.
Example:
- I turned on the light
- the electricity in the apartment was turned off
- I went to bed
- they provide electricity at night! This is where it needs to be always OFF.
On controllers GL-C-001P, GL-C-002P, GL-C-008P, GL-C-008P (mini), can you make the light always OFF?
The controller will always be in mode: RGB+CCT: GL-C-008P

MattWestb, 2023-11-18:
I think its better asking that at Z2M and not ZHA git issue