zigpy/zha-device-handlers #1645: [Device Support Request] MoesHouse Presence sensor

sparkydave1981, 2022-07-07:
**Is your feature request related to a problem? Please describe.**
Device support

**Describe the solution you'd like**
Please provide support for [MoesHouse (Tuya) Zigbee Presence Sensor ](https://www.moeshouse.com/products/smart-zigbee-human-presence-detector-pir-mmwave-radar-detection-sensor-ceiling-mount)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ikvncluo",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.6.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.12",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Perth",
    "os_name": "Linux",
    "os_version": "5.15.45",
    "supervisor": "2022.06.2",
    "host_os": "Home Assistant OS 8.2",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "braviatv_psk": {
      "version": "0.4.2",
      "requirements": [
        "pySonyBraviaPSK==0.2.4"
      ]
    },
    "browser_mod": {
      "version": "1.5.3",
[... 292 lines cut ...]
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "Ensuite Presence",
    "device_reg_id": "af96549acbdde6ed48a449434bf1a679",
    "area_id": "739384a020964a66982b7f118ece74c1"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
[image]

javicalle, 2022-07-22:
Can you try to add the device to the `MmwRadarMotion` quirk?

If you are not familiar, there are some guides about it. Maybe the most popular can be:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

You will need to copy the `ts0601_motion.py` file in your local `custom_zha_quirks` folder:
* https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0601_motion.py

Add your manufacturer in the `MmwRadarMotion` MODELS_INFO:
```python
        MODELS_INFO: [
            ("_TZE200_ar0slwnd", "TS0601"),
            ("_TZE200_ikvncluo", "TS0601"),
            ("_TZE200_sfiy5tfs", "TS0601"),
            ("_TZE200_mrf6vtua", "TS0601"),
            ("_TZE200_ztc6ggyl", "TS0601"),
        ],
```

Save & restart.

Then check if device signature changes, that would mean that the device loads the quirk.

javicalle, 2022-07-23:
If the quirk is working, do you mind to create the PR to add your device to the mainstream?

Just edit from the web the file content with the detected changes and follow the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_motion.py
This way other users will be able to use the devices in ZHA thanks to you!

sparkydave1981, 2022-07-24:
This has been added in another issue:
https://github.com/zigpy/zha-device-handlers/pull/1591

However it really needs to be improved such that the GUI has control of the device settings rather than only providing the two sensor entities.

sparkydave1981, 2022-07-25:
At the risk of having this in two locations, the other Issue (1591) has already been merged so any further investigation work probably needds to be done here instead.

I'm happy to do the testing if someone can guide me a little.

This is what I see for that cluster:
[image]
[image]

So I take it we need to line those up with these somehow?

// TuYa Smart Human Presense Sensor
    tshpsPresenceState: 1,
    tshpscSensitivity: 2,
    tshpsMinimumRange: 3,
    tshpsMaximumRange: 4,
    tshpsTargetDistance: 9,
    tshpsDetectionDelay: 101,
    tshpsFadingTime: 102,
    tshpsIlluminanceLux: 104,
    tshpsCLI: 103, // not recognize
    tshpsSelfTest: 6, // not recognize

smittong, 2022-07-26:
Hello

I connected the presence sensor ("_TZE200_ikvncluo") back via the SmartLife app to get the dp_id values from Tuya IOT.  This is the resulting debugging decode:  If someone could take these to make a configurable quirk for the "_TZE200_ikvncluo" that would be awesome. I'm not sure this is 100% correct from Tuya IOT - also no info on Lux levels for example.  Note the example dp_4 is the maximum distance value and as reported by @sparkydave1981 above the value of 600 is for the default value of 6 meters.  And dp_3 is the minimum distance and if you wanted that to be 1 meter you would have to enter 100.  According to the product specs the max detect range is 6 meters, but the Tuya IOT and SmartLife app allows for a value up to 9.5 meters (950).  Could the units be set to "cm" (centimeters) to match the device values?

{
  "result": {
    "category": "hps",
    "functions": [
      {
        "code": "sensitivity",
        "dp_id": 2,
        "type": "Integer",
        "values": "{\"min\":0,\"max\":9,\"scale\":0,\"step\":1}"
      },
      {
        "code": "near_detection",
        "dp_id": 3,
        "type": "Integer",
        "values": "{\"unit\":\"m\",\"min\":0,\"max\":950,\"scale\":2,\"step\":15}"
      },
      {
        "code": "far_detection",
        "dp_id": 4,
        "type": "Integer",
        "values": "{\"unit\":\"m\",\"min\":0,\"max\":950,\"scale\":2,\"step\":15}"
      }
    ],
    "status": [
      {
        "code": "presence_state",
        "dp_id": 1,
        "type": "Enum",
        "values": "{\"range\":[\"none\",\"presence\"]}"
      },
      {
        "code": "sensitivity",
        "dp_id": 2,
        "type": "Integer",
        "values": "{\"min\":0,\"max\":9,\"scale\":0,\"step\":1}"
      },
      {
        "code": "near_detection",
        "dp_id": 3,
        "type": "Integer",
        "values": "{\"unit\":\"m\",\"min\":0,\"max\":950,\"scale\":2,\"step\":15}"
      },
      {
        "code": "far_detection",
        "dp_id": 4,
        "type": "Integer",
        "values": "{\"unit\":\"m\",\"min\":0,\"max\":950,\"scale\":2,\"step\":15}"
      },
      {
        "code": "checking_result",
        "dp_id": 6,
        "type": "Enum",
        "values": "{\"range\":[\"checking\",\"check_success\",\"check_failure\",\"others\",\"comm_fault\",\"radar_fault\"]}"
      },
      {
        "code": "target_dis_closest",
        "dp_id": 9,
        "type": "Integer",
        "values": "{\"unit\":\"m\",\"min\":0,\"max\":1000,\"scale\":2,\"step\":1}"
      }
    ]
  },
  "success": true,
  "t": 1658791225182,
  "tid": "5cce9b550c7011edaf8fbae27635b1d4"
}

screenshot from the SmartLife App showing the configurable options for the device:

[image]

PlusPlus-ua, 2022-08-25:
I've almost completed a quirk for this device. Here is a source code:

<details>

```
"""Tuya mmw radar occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaDPType,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""
[... 243 lines cut ...]
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```
</details>

Here is screenshot of exposed entities:
[image]

Have few questions. I have exposed configuration entities using AnalogOutput cluster. Is it possible to prevent adding "Number" string before entity name? Is it possible to choose which type of user interface will be used for entity (slider or edit box)?

Waiting for testing results.

MattWestb, 2022-08-25:
The second question i think its depending of the unit you have configurated is ZHA selecting the type.

For the first one is HA adding its deom the class and its not possible adding any extra information.
2 PR have being made for adding it "information" for displaying in ZHA but both is being discarded also if tagging them with "tuya" or one "ZHA" manufacture number for doing it 100% Zigbee compatible is not helping.

PlusPlus-ua, 2022-08-25:
> The second question i think its depending of the unit you have configurated is ZHA selecting the type.

Have investigated this. It doesn't depend on unit, only on number of "steps". If the number of "steps" is more than 256 it will be edit box.

MattWestb, 2022-08-25:
So its with people that cant read code and only speculating ;-((

vietpv89, 2022-08-29:
> I've almost completed a quirk for this device. Here is a source code:
> 
> Here is screenshot of exposed entities: [image]
> 
> Have few questions. I have exposed configuration entities using AnalogOutput cluster. Is it possible to prevent adding "Number" string before entity name? Is it possible to choose which type of user interface will be used for entity (slider or edit box)?
> 
> Waiting for testing results.

Hello [PlusPlus-ua](https://github.com/PlusPlus-ua), how to use your quirk ?
i'm already modify configuration
[image]
and copy your code, put it here
[image]
But my device still no change any thing, not same your screen shot.
[image]

pls help !

PlusPlus-ua, 2022-08-30:
Hi @vietpv89, according to your screenshot, my code is not used. It seems like your ZHA still use file modified as mentioned in https://github.com/zigpy/zha-device-handlers/issues/1645#issuecomment-1192850979 Name of the class in my code TuyaMmwRadarOccupancy. Recheck please content of ts0601_motion.py file.

[image]

javicalle, 2022-08-30:
Also delete the `__pycache__` folder every time you modified the file and restart HA.

PlusPlus-ua, 2022-09-04:
@vietpv89 Do you have any new response?

sparkydave1981, 2022-09-16:
> 

Any update on this quirk? I'm wanting to try and get the sensor to switch back to "clear" faster after motion stops. Is that ID: 0x0021 ?

javicalle, 2022-09-19:
> I'm wanting to try and get the sensor to switch back to "clear" faster after motion stops. Is that ID: 0x0021 ?

I would say that is the DP_102 (fading time or hold time)

sparkydave1981, 2022-09-27:
> I would say that is the DP_102 (fading time or hold time)

I'm not sure which setting that is...  I was thinking it's the `pir_o_to_u_delay`
[image]

javicalle, 2022-09-27:
> I was thinking it's the `pir_o_to_u_delay`

That would be the 'standar' value, but nobody has verified or tested which DP is using Tuya, and nobody has mapped the DP to that attribute.
My guess is that the property you want to modify is the `dp_102` of the cluster `MmwRadarManufCluster`. If it works, maybe we can map the DP to that attribute.

sparkydave1981, 2022-09-28:
For that cluster attribute, the default was 600.  I tried setting it to 30 to see what happens.  This is my first time playing with clusters so I'm not sure I've done it correctly.  If I enter 30 into either box and hit 'set attribute', the 'value' box still shows 600.  Is that correct?  Do I just leave 30 in the 'override' box?
[image]

javicalle, 2022-09-28:
You must edit the "Value" value and leave empty the "Manufacturer Code Override".
Then push the "Set zigbee attribute".

To check, just push the "get zigbee attribute". If value don't change, the debug logs will be needed.

sparkydave1981, 2022-09-29:
> You must edit the "Value" value and leave empty the "Manufacturer Code Override".
> Then push the "Set zigbee attribute".

Yep, so that's what I initially tried but it still says '600' which is the default value.

How do I get the debug logs?  what would I be looking for in the logs?

javicalle, 2022-09-29:
Take a look here:
* https://www.home-assistant.io/integrations/zha/#debug-logging
* https://www.home-assistant.io/integrations/logger/#viewing-logs

Once enabled, look at the logs when you try to update the value. Call the command several times to be able to identify the generated logs and post it here.

sparkydave1981, 2022-10-28:
With those debug settings my log gets spammed so hard it's impossible to do much at all!  I had a 2GB log in a matter of 16 hours or so...

sparkydave1981, 2022-10-28:
> Waiting for testing results.

Any update on this?  Sending values via the device "manage zigbee device" UI doesn't seem to actually update the device for me. (I haven't tried your quirk as yet)

PlusPlus-ua, 2022-10-28:
> Any update on this? 

Yes, I've updated version. It looks like TuyaAttributesCluster from zhaquirks.tuya.mcu doesn't support float values, so I've changed all settings to int. So, delays measured now in milisecons and ranges in centimeters. I'ts a bit ugly but works. I also has problem with Illuminance, I not sure in conversation formula, but it may be a defect of mine sensor.

Here is my current code:

```
"""Tuya mmw radar occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaDPType,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

[... 241 lines cut ...]
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

sparkydave1981, 2022-10-28:
> Here is my current code:

Thank you.  How would I add this to my HA setup?  When I first got these sensors I added a custom quirk called ts0601_motion.py in order to get a working sensor with just the basic two entities.  Would I add your code as a new file?

sparkydave1981, 2022-10-28:
> my log

Not sure how much of this is related but I see a bunch of mmWave stuff in there...

```
2022-10-28 13:30:10.503 DEBUG (MainThread) [zigpy.zcl] [0x5F29:1:0xef00] Decoded ZCL frame: MmwRadarManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=139, datapoints=[TuyaDatapointData(dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xd6\x01\x00\x00', *payload=470))]))
2022-10-28 13:30:10.503 DEBUG (MainThread) [zigpy.zcl] [0x5F29:1:0xef00] Received command 0x02 (TSN 29): set_data_response(data=TuyaCommand(status=0, tsn=139, datapoints=[TuyaDatapointData(dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xd6\x01\x00\x00', *payload=470))]))
2022-10-28 13:30:10.504 DEBUG (MainThread) [zigpy.zcl] [0x5F29:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=29, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-28 13:30:10.504 DEBUG (MainThread) [zigpy.zcl] [0x5F29:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-10-28 13:30:10.505 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x5F29), dst_ep=1, source_route=None, extended_timeout=False, tsn=29, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\x1d\x0b\x02\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-28 13:30:10.505 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0x5f29>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'091d02008b09020004000001d6'
2022-10-28 13:30:10.505 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 183, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x5f29 endpoint=1>, 260, 61184, 1, b'\x18\x1d\x0b\x02\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-10-28 13:30:10.507 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_request[2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 183]
2022-10-28 13:30:10.507 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 183]
2022-10-28 13:30:10.516 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-10-28 13:30:10.516 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-10-28 13:30:10.516 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-10-28 13:30:10.517 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 183, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x5f29 endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2022-10-28 13:30:10.517 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 183: 00
2022-10-28 13:30:10.517 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xb7 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x5f29 endpoint=1>, status: 0x00
2022-10-28 13:30:11.258 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-10-28 13:30:11.258 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-10-28 13:30:11.258 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2022-10-28 13:30:11.260 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xc2a8>, 1, 260, 1794, b'\x08U\n\x00\x04*\x1d\x00\x00', 0, 175, 191, 196, 142, 246, 2, -70]
2022-10-28 13:30:11.260 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xc2a8>, 1, 260, 1794, b'\x08U\n\x00\x04*\x1d\x00\x00', 0, 175, 191, 196, 142, 246, 2, -70]
2022-10-28 13:30:11.260 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC2A8), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=1794, data=Serialized[b'\x08U\n\x00\x04*\x1d\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=191, rssi=-70)
[... cut ...]
2022-10-28 13:30:16.620 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 193, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x5f29 endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2022-10-28 13:30:16.620 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 193: 00
2022-10-28 13:30:16.620 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xc1 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x5f29 endpoint=1>, status: 0x00
```

PlusPlus-ua, 2022-10-28:
> Thank you. How would I add this to my HA setup? When I first got these sensors I added a custom quirk called ts0601_motion.py in order to get a working sensor with just the basic two entities. Would I add your code as a new file?

Just remove ts0601_motion.py, since it overwrites file from zhaquirks package, and create new file ts0601_radar.py with content of my code.

javicalle, 2022-10-28:
> I also has problem with Illuminance, I not sure in conversation formula, but it may be a defect of mine sensor.

The usual conversion for tuya it seems to be this:
https://github.com/zigpy/zha-device-handlers/blob/249ec6d01bfdb0938486bbf27b9b2e6615ba5288/zhaquirks/tuya/ts0601_motion.py#L191-L195

smittong, 2022-10-31:
Hello @PlusPlus-ua great bit of code.  I have it working well.  Only issue I've noticed is no Illuminance sensor value is available?  (This isn't an issue for me as I'm using this currently as a security sensor and will be getting rid of my ugly PIR sensors). Is it possible to remove the word "Number" from each of the fields just to make it look more tidy?

My sensor output:

[image]

Thank you.

aquemy, 2022-10-31:
@smittong I have also test successfully the quirk and I have the Illuminance value (see below). 
However, to have my model working I had to add the "models_info" to the signature, otherwise the quirk was not loaded. 
Not sure is this is the problem but in case it is, my `TuyaMmwRadarOccupancy` class:

Don't forget to add your model (from your screen it looks like _TZE200_ikvncluo:
```python
class TuyaMmwRadarOccupancy(CustomDevice):
    """Millimeter wave occupancy sensor."""

    signature = {
        #  endpoint=1, profile=260, device_type=81, device_version=1,
        #  input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10]
        "models_info": [
            ("_TZE200_ar0slwnd", "TS0601"),
            ("_TZE200_sfiy5tfs", "TS0601"),
            ("_TZE200_mrf6vtua", "TS0601"),
            ("_TZE200_ztc6ggyl", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaMmwRadarCluster,
                    TuyaIlluminanceMeasurement,
                    TuyaOccupancySensing,
                    TuyaMmwRadarTargetDistance,
                    TuyaMmwRadarSensitivity,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMinRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

[image]

aquemy, 2022-10-31:
> Usually it's not needed, if model name and signature of endpoints matches, quirk should be loaded.

Somehow it was the only way I managed to get the quirk loaded on my side.

javicalle, 2022-10-31:
> > Don't forget to add your model (from your screen it looks like _TZE200_ikvncluo:
> 
> Usually it's not needed, if model name and signature of endpoints matches, quirk should be loaded.

Well, you can try to match the model and signature like this:
```python
class TuyaMmwRadarOccupancy(CustomDevice):
    """Millimeter wave occupancy sensor."""

    signature = {
        MODEL: "TS0601",
        .../...
```

Or match the model, manufacturer and signature like this:
```python
class TuyaMmwRadarOccupancy(CustomDevice):
    """Millimeter wave occupancy sensor."""

    signature = {
        "models_info": [
            ("_TZE200_ar0slwnd", "TS0601"),
            ("_TZE200_sfiy5tfs", "TS0601"),
            ("_TZE200_mrf6vtua", "TS0601"),
            ("_TZE200_ztc6ggyl", "TS0601"),
        ],
        .../...
```

But IMHO not defining the manufacturer seems quite risky to me since many `TS0601` devices can be adjusted to that signature (I have found more of 10 of them):
```python
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaNewManufCluster.cluster_id,  # <-- Tuya manufacturer cluster
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
```

So I would suggest (at least for the `TS0601` devices) to adjust the quirk for a model and manufacturer.

sparkydave1981, 2022-11-01:
> Is it possible to remove the word "Number" from each of the fields just to make it look more tidy?

You can do this in the device config by manually overriding the names for each entity.

[image]
[image]

smittong, 2022-11-01:
Hello @sparkydave1981 thanks for that tip on the renaming, brilliant.

wormiedk, 2022-11-02:
> Yes, I've updated version. It looks like TuyaAttributesCluster from zhaquirks.tuya.mcu doesn't support float values, so I've changed all settings to int. So, delays measured now in milisecons and [...]
@PlusPlus-ua Your quirk works excellently however for me the illuminance value is updated VERY frequently at least every 10 seconds but sometimes much more often. Is there a way to limit this?

blair287, 2022-11-23:
> @smittong I have also tested successfully the quirk and I have the Illuminance value (see below). However, to have my model working I had to add the "models_info" to the signature, otherwise the quirk was not loaded. Not sure is this is the problem but in case it is, my `TuyaMmwRadarOccupancy` class:
> 
> Don't forget to add your model (from your screen it looks like _TZE200_ikvncluo:
> 
> ```python
> class TuyaMmwRadarOccupancy(CustomDevice):
>     """Millimeter wave occupancy sensor."""
> 
>     signature = {
>         #  endpoint=1, profile=260, device_type=81, device_version=1,
>         #  input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10]
>         "models_info": [
>             ("_TZE200_ar0slwnd", "TS0601"),
>             ("_TZE200_sfiy5tfs", "TS0601"),
>             ("_TZE200_mrf6vtua", "TS0601"),
>             ("_TZE200_ztc6ggyl", "TS0601"),
>         ],
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaNewManufCluster.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>         },
>     }
> 
> 
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaMmwRadarCluster,
>                     TuyaIlluminanceMeasurement,
>                     TuyaOccupancySensing,
>                     TuyaMmwRadarTargetDistance,
>                     TuyaMmwRadarSensitivity,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             2: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarMinRange,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             3: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarMaxRange,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             4: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarDetectionDelay,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             5: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarFadingTime,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>         }
>     }
> ```
> 
> [image]

This just generates an error saying custom device not defined and prevents zha from loading.

smittong, 2022-11-23:
Hello @blair287.  I did add the models info about a month ago but didn't seem to make any difference, but I noted a couple of days ago the illuminance sensor appeared without me doing anything.

blair287, 2022-11-23:
> Yes, I've updated version. It looks like TuyaAttributesCluster from zhaquirks.tuya.mcu doesn't support float values, so I've changed all settings to int. So, delays measured now in milisecons and [...]

Used your code it loads the quirk illumination is missing but it also doesn't work for occupancy never changes from clear.

blair287, 2022-11-24:
> Hello @blair287. I did add the models info about a month ago but didn't seem to make any difference, but I noted a couple of days ago the illuminance sensor appeared without me doing anything.

Ah yes it did appear after a while.

Hopefully this can be supported inside ZHA soon by default 

Would be great if the times were in seconds and not ms

KevinVanthuyne, 2022-12-13:
> Yes, I've updated version. It looks like TuyaAttributesCluster from zhaquirks.tuya.mcu doesn't support float values, so I've changed all settings to int. So, delays measured now in milisecons and [...]

This quirk works for me in combination with adding the manufacturer info as mentioned by @aquemy.

Is there a reason why this quirk has not been merged to the official quirks yet?

aquemy, 2022-12-13:
@KevinVanthuyne I am actually now facing an issue with this quirk: https://github.com/zigpy/zha-device-handlers/issues/1863#issuecomment-1347885668

Can you confirm that the configuration via the sliders works and is not reset automatically? Can you see the service called in the logbook? Not only for the sensitivity?

I am having a hard time understanding if the issue is my own environment or the quirk itself.

aquemy, 2022-12-14:
Thanks for the confirmation @KevinVanthuyne. 

After updating to the the latest version of HA, removing all quirks and devices, restarting my hardware and repairing every sensor, everything started to work again. This is weird though.

blair287, 2022-12-31:
My sensor is working but all the configuration sliders so unavailable now so cannot be adjusted which is weird.

jimmyeao, 2023-01-03:
I haven't been able to get  this code working, seems I cant have a custom quirk for a device that is already "integrated" into ZHA.
using a  TS0601 identified as  _TZE204_ztc6ggyl
All I get is illuminance and occupancy, no other settings.
Its picking up `Quirk: zhaquirks.tuya.ts0601_motion.MmwRadarMotion` instead of `ts0601_radar.py`

blair287, 2023-01-03:
> I haven't been able to get this code working, seems I cant have a custom quirk for a device that is already "integrated" into ZHA. using a TS0601 identified as _TZE204_ztc6ggyl All I get is [...]

Because you need to add that device signature to it I have same model use this, also remember to delete the pycash in tje custom quirks folder then restart HA.

`"""Tuya mmw radar occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaDPType,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""

[... 241 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
`

jimmyeao, 2023-01-03:
That did the trick, thank you!

acseven, 2023-02-02:
> Because you need to add that device signature to it. I have same model use this, also remember to delete the pycache in the custom quirks folder then restart HA.

I've added the [quirk code mentioned here ](https://github.com/zigpy/zha-device-handlers/issues/1645#issuecomment-1369787576) to a specific file:
 
[image]

As far as I can tell the file is loaded fine (I made sure to delete that folder's __pycache__ folder as well when present before restarting HA), and I get this device info and entities:

[image]

Although it is mostly working, there are some missing entities; when comparing to  this [Zigbee2MQTT listing](https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor.html), these aren't available, being the first one kind of important to create some automations:

 - target_distance

[image]

 - self_test

Beside this, and as visible in the screenshot above, the illuminance sensor doesn't return a correct measurement. Regardless of the light intensity in the room the measured value is 2 lx. 

Can these be fixed?

Additional screenshots:

[image]

CobraDunn, 2023-02-19:
Quirk works great. with _TZE200_holel4dk for the Moes unit I have. Thanks to all for the work on this.

rabin-io, 2023-03-02:
I get this error message after reloading the config, 
```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_radar.py:42
First occurred: 10:20:58 (3 occurrences)
Last logged: 16:14:56

Error setting up entry socket://10.10.10.29:8888 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_radar.py", line 42, in <module>
    from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/mcu/__init__.py)

```

rabin-io, 2023-03-02:
Here is my fix, hope this will help

```python
"""Tuya mmw radar occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""

    def __init__(self, *args, **kwargs):
        """Init."""
[... 230 lines cut ...]
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

smittong, 2023-03-03:
> Here is my fix, hope this will help (replaced the inline code with a link to a gist to keep the comment small) → https://gist.github.com/rabin-io/a279f93903a3574aedc0c4d7f860febc

Thank you @rabin-io for sharing your updated code.  I had the same error after updating to 2023.3.0 and took out all my Zigbee devices. Working perfect again now, thank you.

rabin-io, 2023-03-04:
@PlusPlus-ua I'm working with your code for more than a month, and I found it to be stable and usable, is there any PR to marge your code to the main branch?

sparkydave1981, 2023-03-14:
After upgrading my Home Assistant from 2023.2.5 to 2023.3.3 I had an error in HA due to this ZHA quirk and the only way to get HA to load the ZHA integration was to remove this quirk, but now these sensors do not work.  Anyone else have this problem?

TheJulianJES, 2023-03-14:
Future HA versions (likely 2022.3.4 or 2022.3.5) won't crash your ZHA instance anymore if you have a bad custom quirk.
Still, custom quirks are only meant as a temporary solution until a proper quirk gets merged into this repo (so ZHA itself).

Take a look at the [link sent by rabin-io](https://github.com/zigpy/zha-device-handlers/issues/1645#issuecomment-1452021468). It looks like that custom quirk should work on the latest releases.

sparkydave1981, 2023-03-14:
> Take a look at the [link sent by rabin-io](https://github.com/zigpy/zha-device-handlers/issues/1645#issuecomment-1452021468). It looks like that custom quirk should work on the latest releases.

Thank you!  I really should have looked at the recent comments to see this before I posted.

smittong, 2023-03-14:
> The original code stopped working for me with 2023.3.0, Here is my fix, hope this will help (replaced the inline code with a link to a gist to keep the comment small) → https://gist.github.com/rabin-io/a279f93903a3574aedc0c4d7f860febc

Hello @rabin-io is it possible to include this device signature:  ("_TZE200_ikvncluo", "TS0601"),  That's a Moes Prescence Sensor and I can confirm this code works. Just not sure how to merge it into your code for others to use.

rabin-io, 2023-03-14:
@smittong gist updated,

https://gist.github.com/rabin-io/a279f93903a3574aedc0c4d7f860febc#file-ts0601_radar-py-L265

sparkydave1981, 2023-03-14:
> @smittong gist updated,
> 
> https://gist.github.com/rabin-io/a279f93903a3574aedc0c4d7f860febc#file-ts0601_radar-py-L265

Thank you rabin-io, this now work for me (the devices in the OP).  Much appreciated mate.

Are you going to create a PR to get this merged into core ZHA?

rabin-io, 2023-03-15:
I'm not the original author of the code, I'm just using it same as you, I can open a PR for it, but I think it should come from @PlusPlus-ua which I think is the original author.

PlusPlus-ua, 2023-03-15:
Hi, 

Sorry guys, I'm a bit out of context now.

@rabin-io works like a charm on two of my sensors. My old code has problem with _TZE204_ztc6ggyl one, you - not.

Please open PR, you have polished my code, add me as co-author if you like.

[4 more replies]

closed 2023-10-22 by github-actions[bot]