zigpy/zha-device-handlers #1650: [Device Support Request] Zigbee Bulb Model TS0505B Mfg: _TZ3210_remypqqm Start Up On Off

DStein809, 2022-07-10:
**Is your feature request related to a problem? Please describe.**
Hello, new to HA and coding in general so sorry in advance if I did something wrong or need additional details when trouble shooting. My Tuya smart bulb has ability for start_up_on_off (id: 0X4003) but isn't enabled and in need of quirk.

**Describe the solution you'd like**
When the bulb starts up (aka powers on) I'd like the ability to select on, off, or last state. Similar to Philips Hue bulbs under configuration.

[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_remypqqm",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.7.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
[... 534 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-07-22:
Are you sure that device support that feature? I'm not sure that would be supported just because it have the `0X4003` attribute 🤔 
It seems that device have a Tuya cluster `0xef00` but it would be better if there is any evidence that device supports it before asking for logs and tests.

MattWestb, 2022-07-22:
Its more then likely not supporting ZCL attributes as all tuya devices we have seen but i using the tuya version that is implanted in the INIT https://github.com/zigpy/zha-device-handlers/blob/4b65304438decc625ddc21bd968108258537d25f/zhaquirks/tuya/__init__.py#L864.

I think its one "RGB+CCT" light from the tuya type and we have on quirk for the LIDL "A" version (A is very likely Zigbee certificated and B not) so it easy testing if its working by adding the cluster in the LIDL quirk and testing if its working.
LIDL quirk https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py.
I was testing with one LIDL E14 RGB+CCT and if i remember is working OK but it in my boy of not used or bad Zigbee devices for the moment then i using IKEA CWS3 for color lights.

TheJulianJES, 2023-01-04:
Similar to https://github.com/zigpy/zha-device-handlers/issues/1651

closed 2023-07-10 by github-actions[bot]