zigpy/zha-device-handlers #1654: [Device Support Request] ZNCZ12LM

multicm, 2022-07-15:
I have seen similar devices listed as issues here but not quite this exact model number ZNCZ12LM.
It is an [Aqara Smart Plug](https://www.amazon.com/gp/aw/d/B07CJ2MM6Z?psc=1&ref=ppx_pop_mob_b_asin_title). It will join to the network and can be controlled but much of the energy reporting metrics are showing as unavailable

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    },
    "100": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x000f"
      ],
      "out_clusters": [
        "0x0004",
        "0x000f"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maus01",
  "class": "zhaquirks.xiaomi.aqara.plug_maus01.Plug"
}
```

[... 510 lines cut ...]
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

multicm, 2022-07-15:
I updated to 2022.7.5 (zigpy 0.47.3) removed the device and re-paired it and the issue persists

TheJulianJES, 2022-07-17:
I must have opened my issue just after yours. I'm working on a fix but it won't be implemented officially for quite a while probably (as it would break other MAEU01 plugs).
It should be fairly easy to provide you with some custom quirk download links for now (they should work but could break with future HA updates).
See https://github.com/zigpy/zha-device-handlers/issues/1655 and https://github.com/zigpy/zha-device-handlers/pull/1656
I'll update you when the custom quirks are ready to be tested.

TheJulianJES, 2022-07-17:
@multicm You can see the ``Testing`` section of this PR: https://github.com/zigpy/zha-device-handlers/pull/1656
Let me know if this also works for US plugs.

multicm, 2022-07-17:
> I must have opened my issue just after yours. I'm working on a fix but it won't be implemented officially for quite a while probably (as it would break other MAEU01 plugs). It should be fairly easy to provide you with some custom quirk download links for now (they should work but could break with future HA updates). See #1655 and #1656 I'll update you when the custom quirks are ready to be tested.
> 
> Edit: Actually, your signature mentions a `lumi.plug.maus01` US plug and not an EU plug like mentioned in the title. So I'm not sure this will work for you
> 
> Edit 2: Maybe the quirk could fix stuff for you too

I apologize, yes I am looking for the US plug fix, that was a typo in the title. I just fixed it.

TheJulianJES, 2022-07-17:
> I am going to test something on Monday. I think my issue is with the basic zigbee integration with home assistant only allowing 1 sensor per device (I read this somewhere...)

I think this should no longer be an issue. If I'm correct, the issue is that ZHA (the integration in HA) creates no sensor because the (locally/faked) ``active_power`` didn't have a default value without the changes in the custom quirk.

​

> so I am getting a conbee on Monday and will try it with zigbee2mqtt and see if that works.

You can try that. It's also possible that the plug behaves differently depending on coordinator firmware (seen that with the MAEU01). Also note that if you want to switch Z2M in the long-term, use a CC2652 based stick. The Conbee isn't really recommended to use with ZHA nor Z2M.

multicm, 2022-07-17:
> @multicm You can see the `Testing` section of this PR: #1656 Let me know if this also works for US plugs.

I tested this but didn't see any change. I followed the included instructions.

There was one minor change in that volts appear now, but they stay 0

TheJulianJES, 2022-07-23:
Don't think this would fix it either, but there's likely some new firmware available at the link [here](https://github.com/Koenkk/zigbee-OTA/blob/74d6c15b14a78319eb086f2c215f38c71d572571/index.json#L1827).

dereitz, 2022-10-20:
Has anyone had success with MAUS1 plugs?  I'm particularly interested in the ability to set the power memory option.

@TheJulianJES if you're willing to work with me, I'd be happy to test.  Do you still recommend trying the update you provided back in Jul?

TheJulianJES, 2022-10-21:
Due to some issues with newer Aqara firmware versions on the EU plugs with IKEA remotes, I don't recommend updating the firmware for now.
@dereitz Just to confirm, your model also reports as a ``lumi.plug.maus01`` (and not simply a ``lumi.plug``)?

If so, I've made some test changes which could make the power memory option work.
To test these changes:
- update to Home Assistant Core 2022.10.5 (or later)
- configure a ZHA custom quirk folder
- download this file: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/tjj/xiaomi_us_plug_fixes/zhaquirks/xiaomi/aqara/plug_maus01.py
- put the downloaded ``plug_maus01.py`` file in the ``zha_custom_quirks`` folder (or in a sub folder)
- restart HA
- see on the device page if the custom quirk was applied (official quirk always starts with ``zhaquirks.xiaomi.aqara`` and the custom quirk should be much shorter depending on how the file/(sub-)folder is named)
- test what sensors work
- (possibly remove and) re-pair the plug and see if further sensors work now
- test the ``power_outage_memory`` attribute:

Open the ZHA device page, click on the three dots, click "Manage Zigbee Device" and select the ``OppleCluster`` and ``power_outage_memory``:
[image]
Then, click "Read Attribute" and see if it returns ``Bool.false``. If it does, remove that from the field and put a ``1`` in there and click ``Write Attribute``.
Read the attribute again and see if it now returns ``Bool.true``. (To reset the changes: use a ``0`` for writing to disable the function again).

(Always leave the second "Manufacturer Code Override" field completely empty (like in my screenshot))

dereitz, 2022-10-21:
Wow, thank you so much for the quick, detailed response!!!

> Due to some issues with newer Aqara firmware versions on the EU plugs with IKEA remotes, I don't recommend updating the firmware for now. @dereitz Just to confirm, your model also reports as a `lumi.plug.maus01` (and not simply a `lumi.plug`)?

Correct, mine shows as `lumi.plug.maus01`

]> If so, I've made some test changes which could make the power memory option work. To test these changes:
> * update to Home Assistant Core 2022.10.5 (or later)
>     * configure a ZHA custom quirk folder
>     * download this file: [...]

All confirmed, I'm running 2022.10.5, I enabled the ZHA custom folder, placed the file in a subfolder, restarted, and I now see `Quirk: plug_maus01.Plug` once I expand "Zigbee info" under the Device Info section.

>     * test what sensors work

I get the following sensors:
[image]

>     * (possibly remove and) re-pair the plug and see if further sensors work now

After removing and re-pairing, I see the following:
[image]

> * test the `power_outage_memory` attribute:
>
>
> Open the ZHA device page, click on the three dots, click "Manage Zigbee Device" and select the `OppleCluster` and `power_outage_memory`: [image] Then, [...]

When I try to read, it simply says "None".  When I enter "1" and Write Attribute, it gives me a checkmark, but when I go to read the value again, it says "None".  I never get any Bool.true or Bool.false results at all.

Thanks again for the help!!!

TheJulianJES, 2022-10-21:
Ah, that's a bummer. Thanks for all the screenshots though! So far, it seems like the firmware running on those plugs seems to differ quite a bit from the EU plug firmware. It's possible that a firmware update could help but I doubt it so far.
Zigbee2MQTT also doesn't seem to support "power outage memory" for the ``lumi.plug.maus01`` (but does support it for the ``lumi.plug``. It's a completely different attribute compared to the EU plug one there)

I don't think there's much I can do at the moment. It's possible that the total power consumption/smart metering could be fixed when the second AnalogInput cluster is properly replaced with another one that relays the info on the consumption bus (to a fake (Smart) MeteringCluster) but I don't think power outage memory can be added, as it doesn't seem to be supported in firmware at all.

I'm interested in how old the firmware is though:
Could you read ``app_version``, ``date_code`` (and ``sw_build_id``) from the ``Basic`` cluster and post the responses here?

dereitz, 2022-10-21:
> I'm interested in how old the firmware is though: Could you read `app_version`, `date_code` (and `sw_build_id`) from the `Basic` cluster and post the responses here?

Absolutely!  Here you go:

app_version: 17
date_code: 12-24-2021
sw_build_id: None

Rogue136198, 2022-11-06:
I also have a few of these devices. With your custom quirk I have them working for energy monitoring but I am trying to get the default power state set. What can I do to help get this working?

dereitz, 2022-11-06:
I ended up returning my Aqara.  I found that the Sengled plugs work out of the box and are cheaper.  Good luck!

Rogue136198, 2022-11-06:
> I ended up returning my Aqara. I found that the Sengled plugs work out of the box and are cheaper. Good luck!

Okay, Good to know. I'll give the Sengled ones a try.

TheJulianJES, 2023-01-04:
This should partly be fixed in Home Assistant [2023.1.0b5](https://github.com/home-assistant/core/releases/tag/2023.1.0b5) and later by:
- https://github.com/zigpy/zha-device-handlers/pull/2057

You might need to fully remove the plug and re-pair it to get the functionality working.

Note that the issue also states: "There are several other functions supported by the device that I'll add in a subsequent PR." 

Also, this is likely a duplicate of https://github.com/zigpy/zha-device-handlers/issues/1636

closed 2023-07-10 by github-actions[bot]