zigpy/zha-device-handlers #1657: [Device Support Request] OWON Technology Inc. PC321,need more information to be display

adi90x, 2022-07-17:
Hello,

Reoppening #977
At the moment zha will only report one consumption while 3 are available.

Is it possible to do something ?
Signature of the device is available in #977
And manfucaturer doc is available there : https://github.com/froggyfly/pc321-zigbee/blob/main/PC321zigbee.docx

@froggyfly : Did you find another solution ?

Regards

fergalish, 2022-08-31:
Hi, I'm trying to install one of these OWON 3-phase/triple single phase power meters. It's adds to ZHA with no problem, but the only sensors I'm seeing are instantaneous power and total energy consumed, each a sum of the three physical input sensors.

I'm still learning HA & ZHA, so I have no idea how to edit a device signature etc.

Curious to know if there's a way to proceed.

Thanks.

udivankin, 2023-01-26:
Seems to be implemented in z2m

BUT

there's two different PC321 zigbee clamp meters - PC321-Z and PC321-Z-TY (Tuya), which are according to @Koenkk two completely different devices.

Here's the long thread PC321-Z
https://github.com/Koenkk/zigbee2mqtt/issues/9525

And here is for PC321-Z-TY (Tuya)
https://github.com/Koenkk/zigbee2mqtt/issues/14790

exelsis423, 2023-05-16:
Hello,
I reopen this request.
I have this equipment too the OWON PC321 Z.

After having searched a lot, I tried to write a quirk. The first way I tried was following the documentation with using the metering cluster. This the quirk:
`import zigpy.types as t

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.foundation import ZCLAttributeDef, ZCLCommandDef
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    )
from zigpy.zcl.clusters.smartenergy import (
    Metering, 
    )
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
    )
    
from zhaquirks import LocalDataCluster

class OwonSimpleMetering(CustomCluster):
    cluster_id = 0x0702
    name = "Owon Simple Metering dbl"
    ep_attribute = "electrical_measurement" 

    attributes: dict[int, ZCLAttributeDef] = {
        0x0000: ZCLAttributeDef("current_summ_delivered", type=t.uint48_t, access="r", mandatory=True),
        0x0017: ZCLAttributeDef("Inlet_temperature", type=t.uint24_t, access="r", mandatory=True),
        0x0200: ZCLAttributeDef("Status", type=t.bitmap8, access="r", mandatory=True),
        0x0300: ZCLAttributeDef("unit_of_measure", type=t.enum8, access="r", mandatory=True),
        0x0301: ZCLAttributeDef("multiplier", type=t.uint24_t, access="r", mandatory=True),
        0x0302: ZCLAttributeDef("divisor", type=t.uint24_t, access="r", mandatory=True),
        0x0303: ZCLAttributeDef("summation_formatting", type=t.bitmap8, access="r", mandatory=True),
        0x0304: ZCLAttributeDef("demand_formatting", type=t.bitmap8, access="r", mandatory=True),
        0x0306: ZCLAttributeDef("metering_device_type", type=t.bitmap8, access="r", mandatory=True),
        0x0400: ZCLAttributeDef("Instantaneous_Demand", type=t.uint24_t, access="rw", mandatory=True),
        0x1000: ZCLAttributeDef("report_map", type=t.bitmap8, access="rw", mandatory=True, is_manufacturer_specific=True),
        0x2000: ZCLAttributeDef("L1_phase_power", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x2001: ZCLAttributeDef("L2_phase_power", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x2002: ZCLAttributeDef("L3_phase_power", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x2100: ZCLAttributeDef("L1_phase_reactive_power", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x2101: ZCLAttributeDef("L2_phase_reactive_power", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x2102: ZCLAttributeDef("L3_phase_reactive_power", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x2103: ZCLAttributeDef("reactive_power_summation_of_the_3_phases", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3000: ZCLAttributeDef("L1_phase_voltage", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3001: ZCLAttributeDef("L2_phase_voltage", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3002: ZCLAttributeDef("L3_phase_voltage", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3100: ZCLAttributeDef("L1_phase_current", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3101: ZCLAttributeDef("L2_phase_current", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3102: ZCLAttributeDef("L3_phase_current", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3103: ZCLAttributeDef("current_summation_of_the_3_phases", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x3104: ZCLAttributeDef("leakage_current", type=t.uint24_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4000: ZCLAttributeDef("L1_phase_energy_consumption", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4001: ZCLAttributeDef("L2_phase_energy_consumption", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4002: ZCLAttributeDef("L3_phase_energy_consumption", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4100: ZCLAttributeDef("L1_phase_reactive_energy_consumption", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4101: ZCLAttributeDef("L2_phase_reactive_energy_consumption", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4102: ZCLAttributeDef("L3_phase_reactive_energy_consumption", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x4103: ZCLAttributeDef("reactive_energy_summation_of_the_3_phases", type=t.uint48_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x5000: ZCLAttributeDef("the_latest_historical_record_time", type=t.uint32_t, access="r", mandatory=True, is_manufacturer_specific=True),
        0x5001: ZCLAttributeDef("the_oldest_historical_recorded_time", type=t.uint32_t, access="r", mandatory=True, is_manufacturer_specific=True),
[... 194 lines cut ...]
    replacement = {
        #SKIP_CONFIGURATION: True,
        ENDPOINTS: {
[... cut ...]
                    Identify.cluster_id,
                    OwonPowerMeasurement,
                    OwonClearMetering,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id],
            },
        },
    }`

With this new quirk, I can read attributes as the first one, but I still don't have any entity. I block on this point, I cannot pass the good parameter to have for example PHASE_A_MEASUREMENT.
If someone could help on this point to try to finalize the quirk, I could share it.

Thanks

exelsis423, 2023-05-16:
Sorry for the presentation, I used quote but all isn't taken completely for each example...

udivankin, 2023-07-29:
any idea when it makes its way to hass update? and what are the steps to make it work in case it was already configured with only 3 entities?

fergalish, 2023-09-04:
> Hi, I found a way to write a quirk, but it's doesn't descover entitier. So it's needed to create sensors, and an automation to have these sensors updated (mine is every minute).
> 
> Here is the code, code for sensors and automation is in comments at the end

Thanks for this. I implemented these, removed the PC321, restarted HA, and it is working as described.

closed 2024-03-09 by github-actions[bot]

KPI33, 2024-05-31:
Never integrated?

zoic21, 2025-01-30:
Hello,

For information I create an quirks v2 : 

```
"""Quirk for Owon PC321"""

from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import EntityPlatform, EntityType, QuirkBuilder
import zigpy.types as t
from zigpy.zcl.clusters.smartenergy import Metering
from zigpy.quirks.v2.homeassistant import (
    UnitOfApparentPower,
    UnitOfElectricCurrent,
    UnitOfPower,
    UnitOfEnergy,
    UnitOfElectricPotential,
)

class OwonPC321MeteringCluster(CustomCluster, Metering):
    """OwonPC321MeteringCluster cluster."""

    cluster_id = 0x0702
    name = "OwonPC321Metering"
    ep_attribute = "smartenergy_metering"

    attributes = Metering.attributes.copy()
    attributes.update(
        {
        0x2000: ("phase_1_power", t.uint24_t, True),
        0x2001: ("phase_2_power", t.uint24_t, True),
        0x2002: ("phase_3_power", t.uint24_t, True),
        0x2100: ("phase_1_reactive_power", t.uint24_t, True),
        0x2101: ("phase_2_reactive_power", t.uint24_t, True),
        0x2102: ("phase_3_reactive_power", t.uint24_t, True),
        0x2103: ("reactive_power_summation_of_the_3_phases", t.uint24_t, True),
        0x3000: ("phase_1_voltage", t.uint24_t, True),
        0x3001: ("phase_2_voltage", t.uint24_t, True),
        0x3002: ("phase_3_voltage", t.uint24_t, True),
        0x3100: ("phase_1_current", t.uint24_t, True),
        0x3101: ("phase_2_current", t.uint24_t, True),
        0x3102: ("phase_3_current", t.uint24_t, True),
        0x3103: ("current_summation_of_the_3_phases", t.uint24_t, True),
        0x3104: ("leakage_current", t.uint24_t, True),
        0x4000: ("phase_1_energy_consumption", t.uint48_t, True),
        0x4001: ("phase_2_energy_consumption", t.uint48_t, True),
        0x4002: ("phase_3_energy_consumption", t.uint48_t, True),
        0x4100: ("phase_1_reactive_energy_consumption", t.uint48_t, True),
        0x4101: ("phase_2_reactive_energy_consumption", t.uint48_t, True),
        0x4102: ("phase_3_reactive_energy_consumption", t.uint48_t, True),
        0x4103: ("reactive_energy_summation_of_the_3_phases", t.uint48_t, True),
        }
    )

(
    QuirkBuilder("OWON Technology Inc.", "PC321")
    .replaces(OwonPC321MeteringCluster)
    .sensor(
        "phase_1_power",
        OwonPC321MeteringCluster.cluster_id,
        unit=UnitOfPower.WATT,
        translation_key="active_power_phase_1",
        fallback_name="Active power phase 1",
    )
    .sensor(
        "phase_2_power",
        OwonPC321MeteringCluster.cluster_id,
        unit=UnitOfPower.WATT,
        translation_key="active_power_phase_2",
        fallback_name="Active power phase 2",
[... 120 lines cut ...]
        "phase_2_reactive_energy_consumption",
        OwonPC321MeteringCluster.cluster_id,
        translation_key="reactive_energy_consumption_2",
        unit=UnitOfPower.WATT,
        fallback_name="Reactive energy consumption 2",
    )
    .sensor(
        "phase_3_reactive_energy_consumption",
        OwonPC321MeteringCluster.cluster_id,
        translation_key="reactive_energy_consumption_3",
        unit=UnitOfPower.WATT,
        fallback_name="Reactive energy consumption 3",
    )
    .sensor(
        "reactive_energy_summation_of_the_3_phases",
        OwonPC321MeteringCluster.cluster_id,
        translation_key="reactive_energy_consumption_summation",
        unit=UnitOfPower.WATT,
        fallback_name="Reactive energy consumption summation",
    )
    .add_to_registry()
)
```

With this v2 it's not needed anymore to have automation or template all entity are automatically create and update.

BartGysens, 2025-06-10:
Working - confirmed
Please add to integration.
Thank you.