zigpy/zha-device-handlers #1661: [BUG] zhaquirks.tuya.ts0601_valve.ParksidePSBZS attribute timer_duration missing

mglaabde, 2022-07-27:
**Describe the bug**
I have a strange issue with two identical Parkside Irrigation Zigbee Smart Valves, which come up in HA as zhaquirks.tuya.ts0601_valve.ParksidePSBZS
One comes with the attribute timer_duration (and the related Entity) and it can be used as intended, whereas the other one misses that attribute/Entity.

**To Reproduce**
N/A

**Expected behavior**
Having attribute/entity timer_duration in each zhaquirks.tuya.ts0601_valve.ParksidePSBZS device

**Screenshots**
[image]

<details>
Device signature of both devices is the same:
### Device 9-Parkside-Irrigation-Back (this one comes with the timer_duration attribute/entity as expected):
```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_htnnfasr",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_valve.ParksidePSBZS"
}
```

### Device 9-Parkside-Irrigation-Front (this one misses the attribute/entity):
```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_htnnfasr",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_valve.ParksidePSBZS"
<details>
<!-- Device signature can be acquired by clicking on the "Zigbee Device Signature" button in the device settings view -->
<summary>Device signature</summary>
```

</details>

<details>

<summary>Diagnostic information of both devices</summary>

[zha-c75dae519ffd301f071295c2f25c56db-_TZE200_htnnfasr TS0601-1d3d630694311be16d326839d7768180.json.txt](https://github.com/zigpy/zha-device-handlers/files/9200193/zha-c75dae519ffd301f071295c2f25c56db-_TZE200_htnnfasr.TS0601-1d3d630694311be16d326839d7768180.json.txt)
[zha-c75dae519ffd301f071295c2f25c56db-_TZE200_htnnfasr TS0601-c0d8e18371d419989d842a5aae1162ad.json.txt](https://github.com/zigpy/zha-device-handlers/files/9200194/zha-c75dae519ffd301f071295c2f25c56db-_TZE200_htnnfasr.TS0601-c0d8e18371d419989d842a5aae1162ad.json.txt)

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-07-28:
Have you tried to repair device or put it in pairing mode and pair again?
Maybe, because is battery powered you need to keep it online during the pairing process pushing some device button...

mglaabde, 2022-07-28:
Yes. Multiple times

mglaabde, 2022-07-28:
What I need to mention is that I used the one that is showing this entry first via DeCONZ add-on. But I think I did not have the attribute their. I naturally also needed to pair it again with ZHA. I cannot remember that I have seen timer_duration right from the beginning. Is it possible that it appeared later (after an update or so), without re-pairing? Because the working one is paired way longer than the one additional one, which I've got just recently.

mglaabde, 2022-07-28:
related to: 
https://github.com/zigpy/zha-device-handlers/issues/933
https://github.com/zigpy/zha-device-handlers/pull/1574

javicalle, 2022-07-28:
I can see that both devices have the same `app_version` and `mcu_version`, so no seems to be an updated version problem.

It seems that one device have settled a `timer_duration`. Maybe if the other one have a `null` value is not being initiallized, and because of this no `timer_duration` entity is created.

I believe that we can force to have a value in the quirk just like the `frost_lock_reset`.

Can you try to add the `ts0601_valve.py` file as a local quirk and edit the `bind` method like?:
```python
    async def bind(self):
        """
        Bind cluster.
        When adding this device tuya gateway issues factory reset,
        we just need to reset the frost lock, because its state is unknown to us.
        """
        result = await super().bind()
        await self.write_attributes({self.attributes_by_name["frost_lock_reset"].id: 0})
        await self.write_attributes({self.attributes_by_name["timer_duration"].id: 0})
        return result
```

mglaabde, 2022-07-28:
Did that.
Even deleted and newly connected the valve, but attribute does not show up.

What is strange is that when using the Cluster attribute setting possibility, I cannot modify it:
[image]
(Yes, currently its NULL)
But I cannot edit it by use of the cluster attribute way for the working valve either (but with the existing entity it works as expected)

mglaabde, 2022-07-28:
`Device 0x219e (84:71:27:ff:fe:d6:20:cb) joined the network
Device 84:71:27:ff:fe:d6:20:cb changed id (0x89c7 => 0x219e)
[0x219e] Skipping initialization, device is fully initialized
Device is initialized <ParksidePSBZS model='TS0601' manuf='_TZE200_htnnfasr' nwk=0x219E ieee=84:71:27:ff:fe:d6:20:cb is_initialized=True>
[0x219e:zdo] ZDO request ZDOCmd.Device_annce: [0x219E, 84:71:27:ff:fe:d6:20:cb, 128]
'binary_sensor' component -> 'FrostLock' using ['tuya_manufacturer']
'button' component -> 'ZHAIdentifyButton' using ['identify']
'sensor' component -> 'TimeLeft' using ['tuya_manufacturer']
'sensor' component -> 'Battery' using ['power']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
'button' component -> 'FrostLockResetButton' using ['tuya_manufacturer']
'number' component -> 'TimerDurationMinutes' using ['tuya_manufacturer']
'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
device - 8606:84:71:27:ff:fe:d6:20:cb entering async_device_initialized - is_new_join: True
device - 8606:84:71:27:ff:fe:d6:20:cb has joined the ZHA zigbee network
[8606](TS0601): started configuration
[8606:ZDO](TS0601): 'async_configure' stage succeeded
[8606:1:0xef00]: Performing cluster binding
[0x219E:1:0xef00] write_attributes --> record: Attribute(attrid=0xEF14, value=TypeValue(type=Bool, value=<Bool.false: 0>))
[0x219E:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='frost_lock_reset', attr_value=<Bool.false: 0>, expect_reply=False)
[0x219E:1:0xef00] get_dp_mapping --> found DP: 109
[0x219E:1:0xef00] from_cluster_data: 109, DPToAttributeMapping(ep_attribute='tuya_manufacturer', attribute_name='frost_lock_reset', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
[0x219E:1:0xef00] ztype: Bool.false
[0x219E:1:0xef00] from_value: [1, 0]
[0x219E:1:0xef00] raw: b'\x00'
[0x219E:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=214, dp=109, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))
[0x219E:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=215, command_id=0, *is_reply=False)
[0x219E:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=214, dp=109, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
[0x219E:1:0xef00] write_attributes --> record: Attribute(attrid=0xEF11, value=TypeValue(type=uint32_t, value=0))
[0x219E:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='timer_duration', attr_value=0, expect_reply=False)
[0x219E:1:0xef00] get_dp_mapping --> found DP: 5
[0x219E:1:0xef00] from_cluster_data: 5, DPToAttributeMapping(ep_attribute='tuya_manufacturer', attribute_name='timer_duration', dp_type=<TuyaDPType.VALUE: 2>, converter=None, dp_converter=None, endpoint_id=None)
[0x219E:1:0xef00] ztype: 0
[0x219E:1:0xef00] from_value: [4, 0, 0, 0, 0]
[0x219E:1:0xef00] raw: b'\x00\x00\x00\x00'
[0x219E:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=216, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
[0x219E:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=217, command_id=0, *is_reply=False)
[0x219E:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=216, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[8606:1:0xef00]: bound 'tuya_manufacturer' cluster: Status.SUCCESS
[8606:1:0xef00]: Configuring cluster attribute reporting
[8606:1:0xef00]: finished channel configuration
[8606:1:0x0001]: Performing cluster binding
[8606:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[8606:1:0x0001]: Configuring cluster attribute reporting
[8606:1:0x0001]: Successfully configured reporting for '{'battery_voltage': (3600, 10800, 1), 'battery_percentage_remaining': (3600, 10800, 1)}' on 'power' cluster: [ConfigureReportingResponseRecord(status=0)]
[8606:1:0x0001]: finished channel configuration
[8606:1:0x0003]: Configuring cluster attribute reporting
[8606:1:0x0003]: finished channel configuration
[8606:1:0x0000]: Configuring cluster attribute reporting
[8606:1:0x0000]: finished channel configuration
[8606:1:0x0006]: Performing cluster binding
[8606:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[8606:1:0x0006]: Configuring cluster attribute reporting
[8606:1:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=0)]
[8606:1:0x0006]: finished channel configuration
[8606:1:0x0019]: finished channel configuration
[8606:1:0xef00]: 'async_configure' stage succeeded
[8606:1:0x0001]: 'async_configure' stage succeeded
[8606:1:0x0003]: 'async_configure' stage succeeded
[8606:1:0x0000]: 'async_configure' stage succeeded
[8606:1:0x0006]: 'async_configure' stage succeeded
[8606:1:0x0019]: 'async_configure' stage succeeded
[8606](TS0601): completed configuration
[... cut ...]  *is_reply=True)
[0x425F:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
[8606](TS0601): Device seen - marking the device available and resetting counter
[8606](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x4153](Plug 01): Device seen - marking the device available and resetting counter
[0x4153](Plug 01): Update device availability -  device available: True - new availability: True - changed: False
`

javicalle, 2022-07-29:
Please, put your previous comment inside ` ``` ` marks to make it readable.

The `timer_duration` entity is created by ZHA (and not the quirk):
* https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/number.py#L515-L530

I believe that will not be created if attribute is null. Maybe attribute must be initiallized elsewhere.

Can you look at the logs when you try to modify the device attributtes?

javicalle, 2022-07-31:
There are some interesting logs here:
```
timer_duration is not supported - skipping TimerDurationMinutes entity creation
start_up_on_off is not supported - skipping ZHAStartupOnOffSelectEntity entity creation
```
That must be due to:
```python
channel.cluster.get(cls._zcl_attribute) is None
```

But the device is trying to write the attribute here:
```python
[0x219E:1:0xef00] write_attributes --> record: Attribute(attrid=0xEF11, value=TypeValue(type=uint32_t, value=0))
[0x219E:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='timer_duration', attr_value=0, expect_reply=False)
[0x219E:1:0xef00] get_dp_mapping --> found DP: 5
[0x219E:1:0xef00] from_cluster_data: 5, DPToAttributeMapping(ep_attribute='tuya_manufacturer', attribute_name='timer_duration', dp_type=<TuyaDPType.VALUE: 2>, converter=None, dp_converter=None, endpoint_id=None)
[0x219E:1:0xef00] ztype: 0
[0x219E:1:0xef00] from_value: [4, 0, 0, 0, 0]
[0x219E:1:0xef00] raw: b'\x00\x00\x00\x00'
[0x219E:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=216, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
[0x219E:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=217, command_id=0, *is_reply=False)
[0x219E:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=216, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
```

Once again it seems that this part is not working. And I have my usual supect: `is_manufacturer_specific=True`

Could you try to modify in your local quirk the `ParksidePSBZS` class like this:
```python
class ParksidePSBZS(EnchantedDevice, NoManufacturerCluster):
    """LIDL Parkside water without implemented scheduler."""
```
you will need to import the new class (in the same file):
```python
from zhaquirks.tuya import NoManufacturerCluster, TuyaLocalCluster
```

Save changes, delete cache and restart.
If no new device appears, try to modify (and read) the value from the cluster manager view.
Logs from the operation would be welcomed.

mglaabde, 2022-08-01:
Thank you @javicalle for your support and advice.
Unfortunately, your proposal did not work:
```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_valve.py:211
First occurred: 21:33:32 (1 occurrences)
Last logged: 21:33:32

Error setting up entry ConBee II, s/n: DE2469242 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 353, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_valve.py", line 211, in <module>
    class ParksidePSBZS(EnchantedDevice, NoManufacturerCluster):
TypeError: Cannot create a consistent method resolution
order (MRO) for bases LocalLogMixin, ListenableMixin
```

mglaabde, 2022-08-01:
Before, I tried once again to set the attribute manually by use of cluster attribute functionality.
Seems like there's a type issue, since the current value is None and thus a string:
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py:518
First occurred: 21:17:01 (6 occurrences)
Last logged: 21:19:40

[0xED0F:1:0xef00] Failed to convert attribute 0xEF11 from None (<class 'str'>) to type <class 'zigpy.types.basic.uint32_t'>: invalid literal for int() with base 10: 'None'
[0xED0F:1:0xef00] Failed to convert attribute 0xEF12 from (<class 'str'>) to type <class 'zigpy.types.basic.uint32_t'>: invalid literal for int() with base 10: ''
```
May that helps. Any way to set it while ignoring the current type?

javicalle, 2022-08-01:
> Unfortunately, your proposal did not work:

Ouuuuch, my fault.

Keep the import but use it in the `ParksideTuyaValveManufCluster` class:
```python
class ParksideTuyaValveManufCluster(NoManufacturerCluster, TuyaMCUCluster):
    """Manufacturer Specific Cluster for the _TZE200_htnnfasr water valve sold as PARKSIDE."""
```
Restore the `ParksidePSBZS` class definition:
```python
class ParksidePSBZS(EnchantedDevice):
    """LIDL Parkside water without implemented scheduler."""
```

Save, delete cache and restart.
If not working (but there is no errors) try also to re-pair the device.

mglaabde, 2022-08-02:
Did it. Unfortunately is still not working (see below): 

```
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'binary_sensor' component -> 'FrostLock' using ['tuya_manufacturer']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'TimeLeft' using ['tuya_manufacturer']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'Battery' using ['power']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'RSSISensor' using ['basic']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'LQISensor' using ['basic']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'FrostLockResetButton' using ['tuya_manufacturer']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'TimerDurationMinutes' using ['tuya_manufacturer']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x7C23:84:71:27:ff:fe:d6:20:cb entering async_device_initialized - is_new_join: True
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x7C23:84:71:27:ff:fe:d6:20:cb has joined the ZHA zigbee network
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7C23](TS0601): started configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:ZDO](TS0601): 'async_configure' stage succeeded
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0000]: Configuring cluster attribute reporting
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0000]: finished channel configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0003]: Configuring cluster attribute reporting
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0003]: finished channel configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0xef00]: Performing cluster binding
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0xef00]: bound 'tuya_manufacturer' cluster: Status.SUCCESS
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0xef00]: Configuring cluster attribute reporting
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0xef00]: finished channel configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0001]: Performing cluster binding
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0001]: bound 'power' cluster: Status.SUCCESS
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0001]: Configuring cluster attribute reporting
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0001]: Successfully configured reporting for '{'battery_voltage': (3600, 10800, 1), 'battery_percentage_remaining': (3600, 10800, 1)}' on 'power' cluster: [ConfigureReportingResponseRecord(status=0)]
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0001]: finished channel configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0006]: Performing cluster binding
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0006]: Configuring cluster attribute reporting
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=0)]
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0006]: finished channel configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0019]: finished channel configuration
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0000]: 'async_configure' stage succeeded
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0x0003]: 'async_configure' stage succeeded
2022-08-02 12:18:01 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C23:1:0xef00]: 'async_configure' stage succeeded
[... cut ...]
Normally, the valve comes with pre-configured 1 min (which is the main reason, why `timer_duration` attribute is appreciated, in order to being able to adapt this, because otherwise I need to re-trigger it continuously (< 1 min) in order to keeping the valve open).
However, this Parkside Valve, I've purchased not newly, but already being used before. Might be that it got a different configuration stored by use of the Parkside Zigbee Hub and related App. There's no information on factory reset available. (However, I still would hope that we can change all internal settings in order to have it fully identical and functional to the other valve)

DieterTHeck, 2022-08-25:
Hi,

I have basically the same issue - but only a single device.
It was never "initialized" with the vendor's tuya interface / App.

time_left sensor is available - it shows 0 when off, and always showed 1 when enabled.
So the valve turned back to off after 1 minute. (my tactical solution: send "ON" message every 55 seconds).

timer_duration is *NOT* shown.

I tried to GET/SET it via Cluster commands as indicated above, but it always shows NONE
This is independent of wether the valve is currently ON or OFF.

[image]

Now funnily, after SET Value to 15, the Cluster panel still shows None when querying.
BUT: the time_left sensor shows now 15 minutes when I swithc the valve on.
I can replicate this by setting whatever value I want. It shows next time as time_left.

[image]

Still, I'd be interested in getting timer_duration into the UI as proper entity.

javicalle, 2022-08-27:
@mglaabde I don't see the `write_attributes` in your logs. Have you tried to re pair your device?

I believe that this device also have some function to turn on without any time out, but I don't remember where I see it.

mglaabde, 2022-08-29:
Hi all, I'd like to provide an update:
I've got one Silvercrest Zigbee Hub and paired the valve with it. It enabled me to set the duration to an integer value. Afterwards, I was able to re-pair it successfully with ZHA, and with the custom quirks the device came up with the `timer_duration` attribute, which could also be successfully set.
I think this supports the thesis that it is an initial issue with data types that hinders the entity creation.

However, I don't know, what happened with the other valve that I've bought before. I only had an automation in place that re-triggered the `on` every 20 sec to ensure not to close it after the default time of 1 min. I am quite sure that the possibility to set the `timer_duration` was not there from the beginning, but appeared somewhen, even without the quirk.

mglaabde, 2022-08-29:
@DieterTHeck: Have you tried to delete and re-pair the device, after setting the `timer_duration > 1`?

DieterTHeck, 2022-08-31:
> @DieterTHeck: Have you tried to delete and re-pair the device, after setting the `timer_duration > 1`?

Yes, I tried first "reconfigure Device" and then Delete + Re-Pair.
Both did not change anything.

DieterTHeck, 2022-08-31:
> I am quite sure that the possibility to set the `timer_duration` was not there from the beginning, but appeared somewhen, even without the quirk.

Over at Zigbee2MQTT I saw this. https://github.com/Koenkk/zigbee2mqtt/issues/7695#issuecomment-1180640196
So try sending a null to it. Havent tested myself.

Is there actually a way to send these cluster commands 
[image]
in a automation / NodeRed or whatever?

closed 2023-07-10 by github-actions[bot]