zigpy/zha-device-handlers #1667: [Device Support Request] Moes Temperature And Humidity Sensor With LED Screen TS0201 (_TZ3000_itnrsufe)

PlusPlus-ua, 2022-07-29:
**Is your feature request related to a problem? Please describe.**
Device from https://www.aliexpress.com/item/1005004443497564.html reports relative humidity in a wrong way. Instead of reporting value\*100 (as it described in ZCL) it reports value\*10. As a result ZHA converts it to actual_relative_humidity/10. Right now LED screen on the device showing 42% but HA showing 4.2%

**Describe the solution you'd like**
I think zhaquirk is needed for the device. The zhaquirk should apply multiplication by 10 for relative humidity value.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405",
        "0xe002"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_itnrsufe",
  "model": "TS0201",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.7.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Kiev",
    "os_name": "Linux",
    "os_version": "5.15.55",
    "supervisor": "2022.07.1",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hisense_tv": {
      "version": "22.05.09",
      "requirements": [
        "wakeonlan==2.0.1"
      ]
[... 314 lines cut ...]
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1029, SrcAddr=0x26B0, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=163, SecurityUse=<Bool.false: 0>, TimeStamp=9513272, TSN=0, Data=b'\x08\xC6\x0A\x00\x00\x21\xC9\x01', MacSrcAddr=0x26B0, MsgResultRadius=29)
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Received ZCL frame: b'\x08\xC6\x0A\x00\x00\x21\xC9\x01'
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=198, command_id=10, *is_reply=True)
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Decoded ZCL frame: RelativeHumidity:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=457))])
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Received command 0x0A (TSN 198): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=457))])
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Attribute report received: measured_value=457
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=198, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy.zcl] [0x26B0:1:0x0405] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-07-29 16:23:58 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.tz3000_itnrsufe_ts0201_fe09270d_humidity, old_state=<state sensor.tz3000_itnrsufe_ts0201_fe09270d_humidity=4.7; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=_TZ3000_itnrsufe TS0201 fe09270d humidity @ 2022-07-29T16:23:57.050111+03:00>, new_state=<state sensor.tz3000_itnrsufe_ts0201_fe09270d_humidity=4.6; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=_TZ3000_itnrsufe TS0201 fe09270d humidity @ 2022-07-29T16:23:58.177734+03:00>>
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x26B0), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=1029, TSN=198, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\xC6\x0B\x0A\x00')
2022-07-29 16:23:58 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
```

</details>

javicalle, 2022-07-29:
There is a possible quick and dirty quirk for your device:
```python
"""Neo Tuya Temperature, Humidity and Illumination Sensor."""

from zigpy.profiles import zha
from zigpy.profiles.zha import DeviceType
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.measurement import (
    RelativeHumidity,
    TemperatureMeasurement,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class ValueAlarm(t.enum8):
    """Temperature and Humidity alarm values."""

    ALARM_OFF = 0x02
    MAX_ALARM_ON = 0x01
    MIN_ALARM_ON = 0x00

class NeoTemperatureHumidityAlarmCluster(CustomCluster):
    """Neo Temperature and Humidity Alarm Cluster (0xE002)."""

    name = "Neo Temperature and Humidity Alarm Cluster"
    cluster_id = 0xE002

    attributes = {
        # Alarm settings
        0xD00A: ("alarm_temperature_max", t.uint16_t, True),
        0xD00B: ("alarm_temperature_min", t.uint16_t, True),
        0xD00C: ("alarm_humidity_max", t.uint16_t, True),
        0xD00E: ("alarm_humidity_min", t.uint16_t, True),
        # Alarm information
        0xD00F: ("alarm_humidity", ValueAlarm, True),
        0xD006: ("temperature_humidity", ValueAlarm, True),
        # Unknown
        0xD010: ("unknown", t.uint8_t, True),
    }

class RelativeHumidityX10(CustomCluster, RelativeHumidity):
    """Handles invalid values for Humidity."""

    def _update_attribute(self, attrid, value):
        # x10 factor in `measured_value` (attrid=0)
        if attrid == 0:
            value = value * 10
        super()._update_attribute(attrid, value)

class MoesTemperatureHumidtyIlluminanceSensorWithScreen(CustomDevice):
    """Moes Tuya ZigBee Smart Home Temperature And Humidity Sensor With LED screen."""

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type="0x0302"
        #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0402", "0x0405", "0xe002"]
        #  output_clusters=["0x0003", "0x000a", "0x0019"]>
        MODELS_INFO: [("_TZ3000_itnrsufe", "TS0201")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: DeviceType.LIGHT_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    NeoTemperatureHumidityAlarmCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidityX10,
                    NeoTemperatureHumidityAlarmCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

If you are not familiar, there are some guides about it. Maybe the most popular can be:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

You will need to create the `ts0201.py` file in your local `custom_zha_quirks` folder.

Save & restart.

Then check if device signature changes, that would mean that the device loads the quirk.

PlusPlus-ua, 2022-07-30:
@javicalle thanks, it did the trick. I think it has a sense to merge ts0201_neo.py and ts0201_zemismart.py in one file and add there this new class. What do you think?

javicalle, 2022-07-30:
> I think it has a sense to merge ts0201_neo.py and ts0201_zemismart.py in one file and add there this new class.

That's exactly the same thing I was thinking. I second to put all classes in the same `ts0201.py` file.
I don't really know if all the attributes from `NeoTemperatureHumidityAlarmCluster` are usefull, but since cluster ID is the same I would prefer to have just one cluster definition.

PlusPlus-ua, 2022-07-30:
> I don't really know if all the attributes from `NeoTemperatureHumidityAlarmCluster` are usefull, but since cluster ID is the same I would prefer to have just one cluster definition.

In Tuya Smart app alarm settings for this device is grayed, but I think it's better to keep it as is.

javicalle, 2022-07-31:
Sorry @PlusPlus-ua but I may have misunderstood your intention. Would you mind creating the PR with the new version?
I can give you support if you have any questions.

Regards.

PlusPlus-ua, 2022-07-31:
Yes, I'll prepare PR. Just have spent rest of weekend with my family, so had no time for it.

javicalle, 2022-07-31:
No problem. The important things first!
It was just to confirm it.

Thanks.

PR #1672 (2022-08-10): Add support of Tuya Moes temperature and humidity sensor with screen.
Support of Tuya Moes temperature and humidity sensor with screen is added. Files ts0201_neo.py and ts0201_zemismart.py are merged into one ts0201.py and new classes added there.

Fix: #1667
closed 2022-08-10 by dmulcahey