zigpy/zha-device-handlers #1668: [Device Support Request] SASWELL SAS980SWT-7-Z01 (_TZE200_akjefhj5, TS0601) Smart Irrigation Valve

ScratMan, 2022-07-29:
### Discussed in https://github.com/zigpy/zha-device-handlers/discussions/1660

<div type='discussions-op-text'>

<sup>Originally posted by **ScratMan** July 25, 2022</sup>
Hello,

I need some help to develop the quirk for the SASWELL SAS980SWT-7-Z01 irrigation timer valve, with the vendor ID _TZE200_akjefhj5 and model TS0601.
[image]

https://www.saswell.com/smart-irrigation-wifi-water-timer-sas980swt-7-z01_p147.html

I own two units, and I can control it through Tuya SmartLife app, but I'd like to get rid of Tuya app and use Home Assistant instead through ZHA integration.
I've seen that this valve is already available with different vendors (RTX ZVG1 for example), and I could find in ZHA that there is a quirk that looks very similar to the result the scan of my valves but with different vendor ID :
`MODELS_INFO: [("_TZE200_81isopgh", "TS0601")],` https://github.com/zigpy/zha-device-handlers/issues/1556
This valve exists in the [ts0601_valve.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_valve.py) file in ZHA device handlers, but unfortunately it seems it doesn't fully matches my valve's description, changing only the vendor ID didn't work, and I had to make some modifications based on some extractions from Tuya logs.

I could grab the following data from my valves : 

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>,
	complex_descriptor_available=0, user_descriptor_available=0, reserved=0,
	aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
	mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>,
	manufacturer_code=4098, maximum_buffer_size=82,
	maximum_incoming_transfer_size=82, server_mask=11264,
	maximum_outgoing_transfer_size=82,
	descriptor_capability_field=<DescriptorCapability.NONE: 0>,
	*allocate_address=True, *is_alternate_pan_coordinator=False,
	*is_coordinator=False, *is_end_device=True, *is_full_function_device=False,
	*is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False,
	*is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",  # Basic
        "0x0001",  # Power configuration
        "0x0004",  # Groups
        "0x0005",  # Scenes
        "0x0006",  # OnOff
        "0x0702",  # Smart Energy: Metering
        "0xef00"  # TuyaNewManufCluster
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_akjefhj5",
  "model": "TS0601",
  "class": "ts0601_TZE200_akjefhj5_valve.TuyaValve"
}
```

In the tuya logs I could extract the DP codes using Chrome developers tool, but unfortunately, it doesn't indicates in which cluster the data are available (in bold those showing data received from valves in Tuya developers platform):

-   **1: "switch",**
-   3: "flow_state",   # No data in tuya logs
-   4: "failure_to_report",   # No data in tuya logs
-   **5: "water_once",   # last irrigation volume**
-   **7: "battery",  # indicates the level of the battery in % in Tuya logs**
-   8: "battery_state",   # No data in tuya logs
-   9: "accumulated_usage_time",   # No data in tuya logs
-   10: "weather_delay",   # No data in tuya logs
-   **11: "irrigation_time",   # Corresponds to remaining time**
-   **12: "work_state",   # idle or manual**
[... 159 lines cut ...]

```
</details>

With the file as shown above I can get the valve in HA as below : 

[image]

And regarding what works and what doesn't work : 

- I can switch on and off the valve using the on_off switch entity.
- The battery level doesn't work and reports unknown.
- The `smartenergy_metering` sensor is supposed to report the volume of water during the last opening of the valve, but it reports garbage. I could fill 5 liters in a bucket, and it indicated 114, then I added 5 more liters and it said 55 ; while it said 615 and 543 when I added some water in my pool (around 150/160 liters both time, in 30mn).
- I can read the time left while the valve is open using the cluster management tool in HA with `TuyaValveManufCluster` : 
[image]
[image]
- I can also write the duration of next irrigation while valve is off, using the `time_left` attribute by writing the value in the field and using 1 in the replacement code before performing the write. Once the new time value written, it remains set up when reading the attribute, and this value is used when switching the valve on. Once the valve is switched off (either manually or after timer end), the value is then reset to 600 seconds.
- The last valve open duration reported in clusters manager works correctly.
- The state of the valve is reported correctly in clusters manager (1 = on, 2 = off)

The doubling power management cluster doesn't seem to have the correct battery level configured, but I can't figure where this data is located in the clusters available in the data from the valve to change it in the quirk.

Could someone please help me to understand how to find the correct data point for battery level and water volume consumed and how to make the attributes be reported as sensor entities in HA ? When looking at the python code from the quirks, it looks really abstract to me, despite my knowledge in python. The behavior of the quirk looks strange, when I try to enable the clusters 1 (power configuration) and 6 (on_off) it breaks the quirk, and the modified clusters like `TuyaValveManufCluster` disappear from the interface. But it seems the `PowerConfiguration` cluster should be closer to the one used in the valve, or am I wrong ?

Thanks for your help.</div>

ScratMan, 2022-07-29:
OK, I made a lot of trial and error with my main HA instance (too hard to implement the conbee on my development instance) and I managed to get the battery reporting to work correctly. Had to hack a bit and overload the handle_message() method into my MCU cluster to be able to use the battery bus and update the correct attribute from power configuration cluster.

```python
    def handle_message(
        self,
        hdr,
        args,
        *,
        dst_addressing,
    ):
        if getattr(getattr(args, "data", None), "dp", None) == 7:
            _LOGGER.debug(
              "%s - TuyaValveManufCluster handling message triggers" 
              "battery_reported event with value=%s",
              self.endpoint.device.nwk,
              args.data.data.payload
            )
            self.endpoint.device.battery_bus.listener_event("battery_reported",
              args.data.data.payload
            )
        return super().handle_message(hdr, args, dst_addressing=dst_addressing)

```

Now working:
- ON / OFF
- "time left" timer read and write through cluster manager
- last open valve duration readout through cluster manager
- battery percentage

Not working yet:
- Water volume reporting is reporting an integer, need to find the way to convert to liters.
- No entity available in UI for timer and last open valve duration.
- Can't set timer value through UI.

Now, I need to understand how to make my `TuyaValveManufCluster` cluster create some entities in HA for its attributes `time_left` and `last_valve_open_duration`.

Here is the file : 

<details>
  <summary>ts0601_TZE200_akjefhj5_valve.py</summary>

```python
"""Tuya Valve."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, OnOff
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import  Bus, LocalDataCluster, DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaLocalCluster,
    LocalDataCluster
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
[... 132 lines cut ...]
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    # PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaValveWaterConsumed,  # attr 0 = last volume delivered
                    NewDoublingPowerConfigurationCluster,
                    #DoublingPowerConfigurationCluster,
                    # OnOff.cluster_id,
                    TuyaValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```
</details>

javicalle, 2022-07-30:
Hi, I will try to give you support in all the issues, but lately I don't have much free time and I think that the diferent timezone is not going to help (UTC+2 here). 

My first recommendation (if you're not doing it already) work with a local quirk in your installation, avoiding modifying the original class. That way it will be easier to create a PR and detect the changes. If you need to modify existing classes, do it in the local file.
And create backup files every time you have stable functionality. When working with so many functionalities, it is not uncommon to reaching a point where you will not know what you have modified but everything stops working...

Create a new class name with their own signature. `SaswellValve` can be fine and we can change it later, but that way we will avoid confusion with already existing classes.

Enable debug logs for ZHA. It will be very verbose, but ZHA logs will have valuable information about device reports (DP, data types, values...). The logs will also report if there are unhandled messages (but I think that these como in WARN level).
More info here:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Now I will try to review all the remaining issues.

## Battery report

You have managed to make it work, but with an old implementation (bus) that we are trying to replace.
I think it should work for you with the `TuyaPowerConfigurationCluster`:
* add the import from `zhaquirks.tuya.mcu`
* map it in the `dp_to_attribute`:
```python
        7: DPToAttributeMapping(
           TuyaPowerConfigurationCluster.ep_attribute,
           "battery_percentage_remaining",
           TuyaDPType.VALUE,
        ),
```
* add in the replacement part

The value can not be getted from device. You have to wait for the device to report it. 
Normally it reports the value each time the device is activated. 
Here the ZHA logs will help us.

## Time left

I don't know if this is the same case, but in other valves they have 2 attributes:
* `timer`: with the value assigned to the timer (how many time will run the valve)
* `remaining_time`: with the remaining time until valve stop

I am going to try to help you in this matter, but I am not sure what atribute we are dealing with here (or if there are 2 or just 1).

If the `time_left` is working through 'cluster manager', all your implementation is fine. 
To get a device in the UI usually changes in HA core is needed. 
The good new is that someone already has done the job:
* https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/number.py#L521-L529
The bad new is that only works for one specific device:
* https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/number.py#L515-L521

Soooo, what can be done here?
My suggestion would be:
* change in the quirk the attribute name to `timer_duration` (and probably to `t.uint32_t`)
```python
    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # 7: ("battery_percentage_remaining", t.uint16_t, True),
            0xEF0B: ("timer_duration", t.uint32_t, True),
            0xEF0C: ("state", t.enum8, True),
            0xEF0F: ("last_valve_open_duration", t.uint16_t, True),
        }
    )

.../...

        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "timer_duration",
            TuyaDPType.VALUE,
        ),
```
* (once verified) create a PR in HA core to add the manufacturer (or maybe to remove the manufacturer constraint):
https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/number.py#L515-L530

And, just in case that there was already a `remaining_time` attribute for this device, alse exists a `time_left` sensor in HA:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/sensor.py#L808-L821

The `last_valve_open_duration`, just need to implement a sensor in HA. I would use a more common name, this way the HA entity could match more devices. I believe that the `ikea_airpurifier` would be a good starting point:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/sensor.py#L823-L830

javicalle, 2022-07-30:
## Water metering

For the water metering, we are using the `Metering` cluster. And (in your implementation) we are reporting to the `current_summ_delivered` attribute.

It seems that device is reporting the `water_once` value, but depending on how it is doing the report, it may not be calculating correctly.
If the device reports the value several times, it could be that we are adding the same value multiple times.
To better understand the behavior of the device and the meaning of the reports, once again, it is best to go to the source, which is the ZHA logs. There we would see how the device is reporting attributes and its raw values.

dieneuser, 2022-08-08:
@ScratMan
Thank you for your solution. I own the model with device signature _TZE200_81isopgh but could not see a value for the battery level with the device handler in [https://github.com/zigpy/zha-device-handlers/issues/1556].
With your code and adding my device signature it works fine.

ScratMan, 2022-08-08:
@dieneuser , you're welcome.
Did you manage to get the relationship between the "water_once" data point bytes (dp 5) and the real water volume on your valve ?

dieneuser, 2022-08-08:
@ScratMan 

> Did you manage to get the relationship between the "water_once" data point bytes (dp 5) and the real water volume on your valve ?

No, I just added my device signature to your code and restarted HA:
[image]

The valve on/off (Wasserventil) and battery/power level is enough for me for now.

nilsreiter, 2023-06-19:
The same device is apparently also sold by [HiHome](https://en.hihome.eu/products/hihome-zigbee-kraan-voor-slimme-irrigatie). I got the same manufacturer and model ids, and it registers in zha but doesn't show any entities. 

Will try more on the weekend.

Did you @ScratMan get your second valve to work?

ScratMan, 2023-06-19:
> The same device is apparently also sold by [HiHome](https://en.hihome.eu/products/hihome-zigbee-kraan-voor-slimme-irrigatie). I got the same manufacturer and model ids, and it registers in zha but [...]

I could get a lot of things working, yes, but with ZigBee2MQTT. Unfortunately ZHA doesn't work well to implement the entities as it relies on HA core itself to recognize entities and add them to device's panel.

See here if you want to try with Z2M : https://github.com/Koenkk/zigbee-herdsman-converters/issues/5727
All following entities are working :
- On/off
- timer setting slider
- timer set sensor
- time left sensor
- last valve open duration sensor
- timer state
- water consumption

I'm still working on the cycling and normal timers programming. They are working, but there is a small bug on the weekdays selection (understood, need to fix it and test); and the valve local time sync doesn't seem to work, as the timers are triggered 6 hours before the requested time (but it acts the same way with Tuya Smart Life app, so it might not be related to my implementation).

closed 2023-12-23 by github-actions[bot]

nicktrigger, 2024-10-13:
I own this device and the same one with a  slightly different model number.

Did you manage to get thisnworking properly?

zha seems to only connect but gives no controls

nicktrigger, 2024-10-13:
The other dive i have is TS0601 by _TZE204_qtnjuoae

Tuya has the same controls, interface etc for them but id like to remove them from smart life to take the cloud out of the picture and use native zigbee