zigpy/zha-device-handlers #1670: [Device Support Request] Aubees Tuya Temperature and Humidity LCD sensor

hemy81, 2022-07-31:
This is a device request for the Aubees Tuya Temperature and Humidy LCD sensor from Ali: https://de.aliexpress.com/item/1005004059457595.html?spm=a2g0o.productlist.0.0.5b6f72c4OnhJgH&algo_pvid=46b8c263-c6f4-4865-a96e-652f9ca89e9d&algo_exp_id=46b8c263-c6f4-4865-a96e-652f9ca89e9d-6&pdp_ext_f=%7B%22sku_id%22%3A%2212000027896131186%22%7D&pdp_npi=2%40dis%21CHF%2114.1%215.43%21%21%21%21%21%402101e9d516592967255884625e5a92%2112000027896131186%21sea

**Describe the solution you'd like**
So far the device is detected within ZHA. I can see thee entities: Temperature, Humidity and Battery. But all values are 0.0. It would be nice to see the real numbers. I have experimented with clusters, but could not get anything out. Does anybody have a solution?

<details>

<summary>Diagnostic information</summary>

```yaml
Although the device is detected I get the following message in the log of ZHA during pairing.
![8B952D21-DED6-4357-B225-AFFEFD8D654C](https://user-images.githubusercontent.com/13441102/182043218-90e3ff87-bc99-498d-b840-aa56fc18bf3d.jpeg)

Ignoring message (b'09070200190102000400000111') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:bd:69:c9:08:6f, 1)'
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-07-31:
Please, attach the device's signature. You can get it from the device's details view.

hemy81, 2022-07-31:
Sorry:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qoy0ekbd",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

javicalle, 2022-08-02:
It seems a quite normal Zigbee device.
Have you tried reconfiguring the device while keeping the device awake? You can do it short pressing the reset pinhole while pairing.

javicalle, 2022-08-06:
In issue #1682 other user was sugested to use the `TuyaTempHumiditySensor` quirk, but it seems that is not working.
@ernstmosch check if after applying the quirk the device signature has changed from the `"class": "zigpy.device.Device"` value.

In this issue, @hemy81 has reported to get some logs like:
```
Ignoring message (b'09070200190102000400000111') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:bd:69:c9:08:6f, 1)'
```

These logs would give to us information about the device reporting.

Despite that device have the temp & hum clusters it seems to be reporting in the (hidden) Tuya manufacturer cluster.

If can you confirm the reporting cluster a quirk can be made, but probably would need to convert the values, so will need the info from logs and the 'correct' ones to made the conversion.

javicalle, 2022-08-06:
My (not tested, not validated) proposal:

```python
class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=0x0302
        # input_clusters=[0x0000, 0x0001, 0x0402, 0x0405]
        # output_clusters=[0x000a, 0x0019]>
        MODELS_INFO: [("_TZE200_qoy0ekbd", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

Enable the custom quirks configuration, copy the current `ts0601_sensor.py` inside the folder and edit it adding the proposed class at the end of the file.
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py

Save changes, delete cache and restart.
Check if the device signature has changed to the new `TuyaTempHumiditySensor_Square` and look to any relevant info in the logs.

hemy81, 2022-08-06:
@javicalle Thank you very much for your suggestion! Let me just summarise what I did:
- enabled ZHA custom quirks in the config
- I took your suggested code and added it to the end of the ts0601_sensor.py file in the custom_zha_quirks folder. 

This resulted in a ZHA error, which you find attached here:

Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_sensor.py:85 
First occurred: 14:10:25 (1 occurrences) 
Last logged: 14:10:25

Error setting up entry ConBee II - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2481023-if00, s/n: DE2481023 - dresden elektronik ingenieurtechnik GmbH - 1CF1:0030 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 353, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_sensor.py", line 71, in <module>
    class TuyaTempHumiditySensor_Square(CustomDevice):
  File "/config/custom_zha_quirks/ts0601_sensor.py", line 85, in TuyaTempHumiditySensor_Square
    PowerConfiguration.cluster_id,
NameError: name 'PowerConfiguration' is not defined

I also tried to replace the "class TuyaTemp..." in the original ts0601_sensor.py file, with the same error. ZHA refuses to work at all here.

javicalle, 2022-08-06:
Caught it!
Replace in the new class the `PowerConfiguration` with `TuyaPowerConfigurationCluster2AAA`.

Already fixed in the previous code.

hemy81, 2022-08-06:
Thanks for the prompt reply. I applied the changes and there was no error message. I removed the device and added it again in the ZHA integration. It was found, but all the measured values are still 0.0.

[image]

javicalle, 2022-08-06:
Has the quirk been applied to the device?
Check if the device's signature has changed.

Maybe you would need to enable the debug logs.

hemy81, 2022-08-06:
Thanks for you answer. the Zigbee-Signature has not changed. So I guess it is still using the ZHA device handler and not the quirk, if I understand this correct. How can I apply the quirk to the device, this is not clear to me.

javicalle, 2022-08-06:
The quirk can't be forced, the device signature must match the quirk signature.
Try to remove the `TemperatureHumidityManufCluster` from the signature, like this:
```python
    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=0x0302
        # input_clusters=[0x0000, 0x0001, 0x0402, 0x0405]
        # output_clusters=[0x000a, 0x0019]>
        MODELS_INFO: [("_TZE200_qoy0ekbd", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

Save changes, delete cache and restart.
Check if the device's signature changes.

If not, enable debug logs and pair again the device. Attache the logs from the pairing process.

ernstmosch, 2022-08-07:
Hey  @javicalle You really MADE MY DAY.
Thank You So much for bringing me forward in understanding the ZHA and quirks thing.

After Applying you quirk from 10 hours ago my ZHA Device shows me Temp and Humidity,

Battery charge is not transferred to ZHA and Humidity is not shown as 37% it pops up at 3,7% 

But im absolutely fine with that or put it in other words will see at that later, Now that you put a "personal quirk" between me an the zigbee/ZHA/HA infrastructure.

I cannot thank you enough! This Thing is not only about getting a single device to work it is about to show how it can be done.

javicalle, 2022-08-07:
Once again, all the required info seems to be in the `zigbee-herdsman-converters` library:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L1952-L1980

Interesting that humidity fix:
```python
            case tuya.dataPoints.tthHumidity:
                return {humidity: calibrateAndPrecisionRoundOptions(
                    (value / (meta.device.manufacturerName === '_TZE200_bjawzodf' ? 10 : 1)),
                    options, 'humidity')};
```

Probably only battery levels (and not % values) will be available.

hemy81, 2022-08-07:
@javicalle thanks a lot for the fix. Very much appreciated! I also learned a lot during this fix. I can confirm what @ernstmosch found. The temperature, humidity and even the battery % are transferred. The humidity is an order of magnitude off, which is fine as we can fix this. However I changed the temperature and the humidity and no new values are reported to HA. It seems that it only shows the numbers from the initial configuration. Can you confirm this @ernstmosch ?

ernstmosch, 2022-08-08:
@hemy81 im not sure if i got your question right but this is what my sensor does:

[image]

i know the reason for that, but i didnt get my mind up how to smoothen that wave...

And once again thanx to @javicalle @hemy81 and everybody around HA!

MattWestb, 2022-08-08:
@ernstmosch Is the device reporting 0 (zero or null) between the normal reports ?
If its doing that then i think its good putting one filter in the quirk that taking them way and you is getting one nice graph in HA.

I think the reporting is not good if i remember what the other devs was finding then getting  some tuya LCD sensors working in there system.

ernstmosch, 2022-08-09:
> @ernstmosch Is the device reporting 0 (zero or null) between the normal reports ? If its doing that then i think its good putting one filter in the quirk that taking them way and you is getting one [...]

Hi @MattWestb 
I found the line in the quirk to get Humidity in to percent

```
   2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 100,  # decipercent to percent
        ),
```

And i found that the Device is setting its Values to 0.0 in its "idle" times:

`2022-08-09 12:11:48.660 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_temperature, old_state=<state sensor.thermi_temperature=24.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-09T12:04:56.233719+02:00>, new_state=<state sensor.thermi_temperature=0.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-09T12:11:48.659719+02:00>>`

But my dev-capabilities combined with inrense search-engine usage are far crappy to filter the Value in the two sensors out.

But Again: THANK EVERYBODY around here!

javicalle, 2022-08-09:
As I have asked for several times, please enable the debug logs and attach the relevant info.

If the device reports incorrect values, it can be fixed, but we need to know what is the device reporting. We can't just drop the 0 value (because 0 is a valid temperature).

ernstmosch, 2022-08-09:
Sorry @javicalle 

i have set:

```
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```
in Config.

and i thaught that the line from "\config\homeasisstant.log which i mentione in my last post;

`2022-08-09 12:11:48.660 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_temperature, old_state=<state sensor.thermi_temperature=24.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-09T12:04:56.233719+02:00>, new_state=<state sensor.thermi_temperature=0.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-09T12:11:48.659719+02:00>>`
 
Was the Info you have been asking for - if not please tell me what to or tell me that i should not post anymore in this thread.

javicalle, 2022-08-09:
We need the logs from the device report. It must be something like:
```
2022-08-09 20:14:33 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=2820, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=180), 255, -53, 0x68e4, 255, 255, b'\x18\xd7\x01\x0b\x05\x00)\x04\x00\x08\x05\x00!\x1a\x00\x05\x05\x00!\xdd\x00']
2022-08-09 20:14:33 DEBUG (MainThread) [zigpy.zcl] [0x68E4:1:0x0b04] Received ZCL frame: b'\x18\xd7\x01\x0b\x05\x00)\x04\x00\x08\x05\x00!\x1a\x00\x05\x05\x00!\xdd\x00'
2022-08-09 20:14:33 DEBUG (MainThread) [zigpy.zcl] [0x68E4:1:0x0b04] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=215, command_id=1, *is_reply=True)
2022-08-09 20:14:33 DEBUG (MainThread) [zigpy.zcl] [0x68E4:1:0x0b04] Decoded ZCL frame: TuyaZBElectricalMeasurement:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x050B, status=<Status.SUCCESS: 0>, value=TypeValue(type=int16s, value=4)), ReadAttributeRecord(attrid=0x0508, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=26)), ReadAttributeRecord(attrid=0x0505, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=221))])
```

The interesting ones must contain something like `[0xABCD:1:0x0402]` where `0xABCD` would be your device `Nwk` (you can get it from 'Zigbee info' in the device view). The `0x0402` is the TemperatureMeasurement cluster.

There must have several reports along the time. It seems that some can be with the correct value and some must be something else:
* correct: ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=int16s, value=26))
* wrong: ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=int16s, value=**¿¿??**)) 

The `attrid=0x0000` is the measured_value (for the Temperature cluster). If there is not `attrid=0x0000` reports with wrong values, look for any other reports for the Temperature cluster.
Also, all the values reported in the logs, must be in the HA graph. Maybe there is something wrong with the data (non integer values?).

hemy81, 2022-08-09:
Dear all, sorry for the late reply, I just turned on logging, will send more data when available.
@ernstmosch's fix worked for the humidity! Thanks!

I have two of these sensors running and both work, expect that HA only receives values from time to time (see graph).
[image]
Even if I change the temperature/humidity by breathing on the device the changes are not registered immediately. I thought that once temp. or hum. changes a zigbee command is send, which is likely not the case.

ernstmosch, 2022-08-10:
Hello Everybody

my logs a bit "loded but i hope the needed info is in it

here is how my sensor jumps to the correct value:

`2022-08-10 12:19:16.648 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 61184, b'\tU\x02\x00\xe5\x01\x02\x00\x04\x00\x00\x00\xe8', 0, 175, 255, 197, 216, 99, 0, -70]
2022-08-10 12:19:16.650 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received ZCL frame: b'\tU\x02\x00\xe5\x01\x02\x00\x04\x00\x00\x00\xe8'
2022-08-10 12:19:16.651 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=85, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-10 12:19:16.654 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame: TemperatureHumidityManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=229, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xe8\x00\x00\x00', *payload=232)))
2022-08-10 12:19:16.655 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received command 0x02 (TSN 85): set_data_response(data=TuyaCommand(status=0, tsn=229, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xe8\x00\x00\x00', *payload=232)))
2022-08-10 12:19:16.661 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=85, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-10 12:19:16.662 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-08-10 12:19:16.666 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09550200e501020004000000e8'
2022-08-10 12:19:16.723 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 94, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x3E71 endpoint=1>, 260, 61184, 1, b'\x18U\x0b\x02\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-10 12:19:16.865 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x5e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x3e71 endpoint=1>, status: 0x00
2022-08-10 12:19:17.034 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 61184, b"\tV\x02\x00\xe6\x02\x02\x00\x04\x00\x00\x00'", 0, 175, 255, 197, 216, 99, 0, -70]
2022-08-10 12:19:17.035 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received ZCL frame: b"\tV\x02\x00\xe6\x02\x02\x00\x04\x00\x00\x00'"
2022-08-10 12:19:17.037 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=86, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-10 12:19:17.039 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame: TemperatureHumidityManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=230, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b"'\x00\x00\x00", *payload=39)))
2022-08-10 12:19:17.039 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received command 0x02 (TSN 86): set_data_response(data=TuyaCommand(status=0, tsn=230, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b"'\x00\x00\x00", *payload=39)))
2022-08-10 12:19:17.043 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=86, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-10 12:19:17.044 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-08-10 12:19:17.048 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09560200e60202000400000027'
2022-08-10 12:19:17.049 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 95, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x3E71 endpoint=1>, 260, 61184 [... cut ...]
2022-08-10 12:31:01.955 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_temperature, old_state=<state sensor.thermi_temperature=23.2; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-10T12:19:16.656819+02:00>, new_state=<state sensor.thermi_temperature=0.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-10T12:31:01.954807+02:00>>
```

ernstmosch, 2022-08-10:
or after reading your last post one more time 

filter on: [0x3e71:1:0x0402]

```
2022-08-10 12:46:04.544 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received ZCL frame: b'\x18\\\n\x00\x00)\x00\x00'
2022-08-10 12:46:04.545 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=92, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-10 12:46:04.546 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Decoded ZCL frame: TuyaTemperatureMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=0))])
2022-08-10 12:46:04.546 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received command 0x0A (TSN 92): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=0))])
2022-08-10 12:46:04.547 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Attribute report received: measured_value=0
```

filter on thermi temperature

```
2022-08-10 12:45:52.465 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_temperature, old_state=<state sensor.thermi_temperature=0.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-10T12:31:01.954807+02:00>, new_state=<state sensor.thermi_temperature=23.6; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-10T12:45:52.464884+02:00>>
2022-08-10 12:46:04.548 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_temperature, old_state=<state sensor.thermi_temperature=23.6; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-10T12:45:52.464884+02:00>, new_state=<state sensor.thermi_temperature=0.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-10T12:46:04.547856+02:00>>

```

[image]

Thank you

javicalle, 2022-08-11:
I can't say that something is wrong but some report values seems at least strange to me:
* `b'\x18X\n\x00\x00)\x00\x00'`
* `b'\x18\\\n\x00\x00)\x00\x00'`
* `b'\x18W\n\x00\x00!\x00\x00'`

@dmulcahey sorry to ping to you directly, but before drop all the `0` values (which would be legitimate values for a temperature) could you check if there is something wrong in the decode frame part? Or can be some kind of zero/default value when parsing an incomplete frame?

Thanks in advanced.

MattWestb, 2022-08-11:
If filtering the device looks working OK the greater then 0.01 and lesser - 0.01 is being reported OK and only some reports with 0 is being missed and shall not being any large problem in the graph in HA.

javicalle, 2022-08-12:
> If filtering the device looks working OK the greater then 0.01 and lesser - 0.01 is being reported OK and only some reports with 0 is being missed and shall not being any large problem in the graph in HA.

Yes, I can agree with that but I was worry that we were masking some other problem. Since I don't control the frame parsing part, I wanted a second opinion about it 🤷🏻‍♂️

javicalle, 2022-08-12:
Wait, there are a pattern here... 💣 
When the device reports through the Tuya cluster (0xEF00), the values appear to be correct.
It seems that the problem comes from the values reported by the cluster of temperature (0x0402) and humidity (0x0405) that have the values `0`. And it seems that they are replies (`*is_reply=True`) to something, maybe some polling mechanism?

ernstmosch, 2022-08-13:
I dont actually understand what it means but the times in log suggest that [0x3E71:1:0x0402] is responsible for the zeros while [0x3E71:1:0xef00] has to deal with values.

[0x3E71:1:0x0402] (when sensor jumps to 0)
`
2022-08-13 06:27:21.056 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 1026, b'\x18\xac\n\x00\x00)\x00\x00', 0, 175, 248, 115, 78, 100, 0, -68]
2022-08-13 06:27:21.057 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received ZCL frame: b'\x18\xac\n\x00\x00)\x00\x00'

2022-08-13 07:12:28.734 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 1026, b'\x18\xb5\n\x00\x00)\x00\x00', 0, 175, 184, 73, 80, 100, 0, -74]
2022-08-13 07:12:28.735 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received ZCL frame: b'\x18\xb5\n\x00\x00)\x00\x00'
`
[0x3E71:1:0xef00] at a timestamp when sensors jump to values:

`2022-08-13 06:58:38.675 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received ZCL frame: b'\t\xb1\x02\x00\xe9\x01\x02\x00\x04\x00\x00\x00\xe8'
2022-08-13 06:58:38.676 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=177, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-13 06:58:38.678 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame: TemperatureHumidityManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=233, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xe8\x00\x00\x00', *payload=232)))
2022-08-13 06:58:38.681 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received command 0x02 (TSN 177): set_data_response(data=TuyaCommand(status=0, tsn=233, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xe8\x00\x00\x00', *payload=232)))
2022-08-13 06:58:38.684 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=177, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-13 06:58:38.685 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-08-13 06:58:38.944 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received ZCL frame: b'\t\xb2\x02\x00\xea\x02\x02\x00\x04\x00\x00\x00%'
2022-08-13 06:58:38.945 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=178, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-13 06:58:38.947 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Decoded ZCL frame: TemperatureHumidityManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=234, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'%\x00\x00\x00', *payload=37)))
2022-08-13 06:58:38.948 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Received command 0x02 (TSN 178): set_data_response(data=TuyaCommand(status=0, tsn=234, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'%\x00\x00\x00', *payload=37)))
2022-08-13 06:58:38.952 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=178, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-13 06:58:38.953 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)`

ernstmosch, 2022-08-13:
and here is the unfiltered log of a zero event:

```
2022-08-13 06:27:21.030 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-13 06:27:21.032 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-13 06:27:21.035 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-13 06:27:21.040 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 1029, b'\x18\xab\n\x00\x00!\x00\x00', 0, 175, 252, 35, 79, 100, 0, -68]
2022-08-13 06:27:21.041 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Received ZCL frame: b'\x18\xab\n\x00\x00!\x00\x00'
2022-08-13 06:27:21.042 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=171, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-13 06:27:21.044 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Decoded ZCL frame: TuyaRelativeHumidity:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=0))])
2022-08-13 06:27:21.045 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Received command 0x0A (TSN 171): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=0))])
2022-08-13 06:27:21.045 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Attribute report received: measured_value=0
2022-08-13 06:27:21.050 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_humidity, old_state=<state sensor.thermi_humidity=37.0; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=Thermi Humidity @ 2022-08-13T06:13:08.765099+02:00>, new_state=<state sensor.thermi_humidity=0.0; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=Thermi Humidity @ 2022-08-13T06:27:21.049819+02:00>>
2022-08-13 06:27:21.052 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, ep: 1, profile: 0x0104, cluster_id: 0x0405, data: b'18ab0a0000210000'
2022-08-13 06:27:21.052 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-13 06:27:21.056 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 1026, b'\x18\xac\n\x00\x00)\x00\x00', 0, 175, 248, 115, 78, 100, 0, -68]
2022-08-13 06:27:21.057 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received ZCL frame: b'\x18\xac\n\x00\x00)\x00\x00'
2022-08-13 06:27:21.058 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=172, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-13 06:27:21.059 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Decoded ZCL frame: TuyaTemperatureMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=0))])
2022-08-13 06:27:21.059 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received command 0x0A (TSN 172): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=0))])
2022-08-13 06:27:21.060 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Attribute report received: measured_value=0
2022-08-13 06:27:21.061 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.thermi_temperature, old_state=<state sensor.thermi_temperature=23.2; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-13T06:13:08.537182+02:00>, new_state=<state sensor.thermi_temperature=0.0; state_class=measurement, unit_of_measurement=°C, device_class=temperature, friendly_name=Thermi Temperature @ 2022-08-13T06:27:21.060799+02:00>>
2022-08-13 06:27:21.064 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, ep: 1, profile: 0x0104, cluster_id: 0x0402, data: b'18ac0a0000290000'
```

MattWestb, 2022-08-13:
Both temp and humidity is sending 0.0.
```
state sensor.thermi_temperature=0.0
state sensor.thermi_humidity=0.0
```
so i think both need filter for 0.0 and all other shall being OK and its only bad if the humidity (not possible in the real world) or the temperature is long measured 0.0 for long time and the graph shall looking OK all the other time.

javicalle, 2022-08-13:
I believe that we can drop all values from clusters 0x0402 and 0x0405.
It could be done removing the clusters in the replacement part, but probably then we will get errors from not cluster defined for recieved data. Maybe a custom drop values cluster would be helpfull here.

But it seems to me that only @ernstmosch is having the 0 values issue. Have you tried to repair the device (trying to keep it awake during the proccess if possible).

ernstmosch, 2022-08-14:
@hemy81 could you please be so nice and post your quirk here?
I think that i have messed something up in mine for getting the zeros.

````
"""Tuya temp and humidity sensor with e-ink screen."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaDPType, TuyaMCUCluster

# NOTES:

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 100,  # decipercent to centipercent
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }

class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_qoy0ekbd", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```
`

javicalle, 2022-08-14:
Hi @ernstmosch, could you test with this other version:

```python
"""Tuya temp and humidity sensor with e-ink screen."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaDPType, TuyaMCUCluster

# NOTES:

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            endpoint_id=2,
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            endpoint_id=2,
            converter=lambda x: x * 100,  # decipercent to centipercent
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }

class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_qoy0ekbd", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                ],
                OUTPUT_CLUSTERS: [],
            }
        },
    }
```

That would duplicate the sensors.
I expect to have two new temp&hum sensors without zeros, but I want to know if the others have the zero values or would have no values at all.

Thanks in advanced.

hemy81, 2022-08-14:
@javicalle I attached the quirk I am using atm:

```
"""Tuya temp and humidity sensor with e-ink screen."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaDPType, TuyaMCUCluster

# NOTES:
# The data comes in as a string on cluster, if there is nothing set up you may see these lines in the logs:
# Unknown message (b'19830100a40102000400000118') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          28.0 degrees
# Unknown message (b'19840100a5020200040000022c') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          55.6% humid
# Unknown message (b'19850100a60402000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          100% battery

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 100,  # decipercent to percent
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }

class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [("_TZE200_bjawzodf", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
    
class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=0x0302
        # input_clusters=[0x0000, 0x0001, 0x0402, 0x0405]
        # output_clusters=[0x000a, 0x0019]>
        MODELS_INFO: [("_TZE200_qoy0ekbd", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

ernstmosch, 2022-08-14:
Hi @javicalle let me start with another Thank You!

regarding your Test the firts sensor is constantly showing zeros and the "_2" Sensors show a nice Graph fitting to felt real-waörlt values.

Do you want me fetch some detailed log the pleas telle me.

Hi @hemy81 thank very much.

I think theres quite i lot i DONT KNOW about zigbee-architecture.

Let me correct my last quote I KNOW.

Thanks, Ernst

javicalle, 2022-08-14:
@hemy81 can you tell to us if your quirk is having 0 values in the temp or hum readings?
@ernstmosch can you check if the fisrt sensors are getting values periodically or it is sticket at some old 0 value. I want to check if with the new quirk the first sensor is reporting values or not. I want to understand where this readings come from, they are from the device or some polling mechanism? The info you can share will give us some clues.

Thanks in advanced.

ernstmosch, 2022-08-14:
@javicalle i thaught i made it clear by "constantly": since activating yor quirk ist never changed away from rero.

ernstmosch, 2022-08-14:
i guess thats a tero on senser one:
```
2022-08-14 20:07:23.379 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 1029, b'\x18\xd3\n\x00\x00!\x00\x00', 0, 175, 255, 223, 153, 100, 0, -70]
2022-08-14 20:07:23.381 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Received ZCL frame: b'\x18\xd3\n\x00\x00!\x00\x00'
2022-08-14 20:07:23.382 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=211, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-14 20:07:23.384 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Decoded ZCL frame: RelativeHumidity:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=0))])
2022-08-14 20:07:23.384 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Received command 0x0A (TSN 211): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=0))])
2022-08-14 20:07:23.385 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0405] Attribute report received: measured_value=0
2022-08-14 20:07:23.389 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, ep: 1, profile: 0x0104, cluster_id: 0x0405, data: b'18d30a0000210000'
2022-08-14 20:07:23.393 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, 1, 260, 1026, b'\x18\xd4\n\x00\x00)\x00\x00', 0, 175, 255, 141, 153, 100, 0, -70]
2022-08-14 20:07:23.394 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received ZCL frame: b'\x18\xd4\n\x00\x00)\x00\x00'
2022-08-14 20:07:23.395 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=212, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-14 20:07:23.396 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Decoded ZCL frame: TemperatureMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=0))])
2022-08-14 20:07:23.396 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Received command 0x0A (TSN 212): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=0))])
2022-08-14 20:07:23.397 DEBUG (MainThread) [zigpy.zcl] [0x3E71:1:0x0402] Attribute report received: measured_value=0
2022-08-14 20:07:23.398 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3e71>, ep: 1, profile: 0x0104, cluster_id: 0x0402, data: b'18d40a0000290000'
```

hemy81, 2022-08-14:
@javicalle I do not have any zeros in HA.

javicalle, 2022-08-15:
Can all of you confirm that your device is the `_TZE200_qoy0ekbd`?

Both quirk are identically for this device (but they have been named diferent: `TuyaTempHumiditySensor` vs `TuyaTempHumiditySensor_Square`).
I can't see any reason why one device is reporting these 0 values _constantly_ 😉 
@ernstmosch can you try to remove the device from HA, reseting the device (pull out batteries), restart HA  and pair the device again. If you can attach the logs from the pairing proccess.

ernstmosch, 2022-08-15:
@javicalle: I removed re-paired an got my (doubled) Sensors back

doubled because of the replacement you offerd me:

```
replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                ],
                OUTPUT_CLUSTERS: [],
            }
        },
    }
```
One of the Sensors is showing the Temp properly and correct without zeros
And one shows unknown at the moment. Ill have a next look in 2 hous

ernstmosch, 2022-08-15:
one on unknown:
[image]
and one with a pictureperfect graph:
[image]

javicalle, 2022-08-15:
I bet that now you can go back to the old quirk (the one with just one endpoint) and you will have the 2 sensors without 0 values.

You would need to delete the extra sensor from HA.

ernstmosch, 2022-08-15:
Thank you so much, Sir! (i really learned a lot about zigbee, zha and HA during this request!)

moonsense715, 2022-09-02:
Hey guys, I would also like to thank you for this quirk, I just applied it today and it worked flawlessly.
Excellent work, fix confirmed.

I just have a cosmetical question, would it be possible to turn a e.g. 53.0% humidity report to 53%? If yes, how?
The battery is already displayed without the extra (nonsense) 0 floating point value. I tried int(x * 100) and math.trunc(x * 100) at the converter, but they made no difference.

(I know that in configuration.yaml I could fix this, but I would like to fix it globally for the device type rather than having to do it for every sensor manually, like how the quirk works for all my sensors already without having to input entity ids, etc.)

dmulcahey, 2022-09-02:
Use a template sensor. Everyone has different preferences and using a template sensor is the best way to handle this

javicalle, 2022-11-01:
Can someone with the functional quirk (without the 0 values) make a PR with the code?
This way other users with the problem will be able to use the device thanks to that person.

ernstmosch, 2022-11-02:
Hi Javicalle, 
im not sure how and where to put my PullRequest as my github experience is to small:

[image]

does it help if i post my working py code here once again?

[ts0601_sensor.py.txt](https://github.com/zigpy/zha-device-handlers/files/9918701/ts0601_sensor.py.txt)

willing to help, Ernst

javicalle, 2022-11-02:
Not the best way if there are too many changes, but you can edit the file directly from the web:
* https://github.com/zigpy/zha-device-handlers/edit/dev/zhaquirks/tuya/ts0601_sensor.py

Then you can just follow the indications.

But if the change is massive, probably will be better run all the code validators locally before publish the code in github.

I can take care of it if you don't feel comfortable with that part.

ernstmosch, 2022-11-02:
Hi Javicalle,

i dont feel very comfortable in submitting this code on github, because i somehown have to merge the two devices "_TZE200_bjawzodf" (from original source) and "_TZE200_qoy0ekbd" (mine) in the quirk. 

Even after locally testing i wouldnt know if i ruined the other device.

If you woul finish the pull request i woul feel much better.

kind regarde, Ernsti

javicalle, 2022-11-04:
I'm finishing the quirk version.

Could you test if that version is fine with the humidity value?

<details open>
<summary>ts0601_sensor.py</summary>

```python
"""Tuya temp and humidity sensor with e-ink screen."""

from typing import Any, Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaDPType, TuyaMCUCluster

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

    def update_attribute(self, attr_name: str, value: Any) -> None:
        """Apply a correction factor to value."""

        if attr_name == "measured_value":
            value = value * (
                self.endpoint.device.RH_MULTIPLIER
                if hasattr(self.endpoint.device, "RH_MULTIPLIER")
                else 100
            )

        return super().update_attribute(attr_name, value)

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            # converter=lambda x: x * 10,  --> move conversion to TuyaRelativeHumidity cluster
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }

class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    RH_MULTIPLIER = 10

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [("_TZE200_bjawzodf", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    # RH_MULTIPLIER = 100

    signature = {
        MODELS_INFO: [
            ("_TZE200_qoy0ekbd", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                # "profile_id": 260, "device_type": "0x0302",
                # "in_clusters": ["0x0000","0x0001","0x0402","0x0405"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TemperatureHumidityManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

</details>

ernstmosch, 2022-11-04:
Hi Javicalle,

your quirk works picture perfect on my "_TZE200_qoy0ekbd".

Thank you so much for letting me learn so much about this issue.

greetings from hesse, Ernst

PR #1887 (2022-11-04): New ts0601_sensor square
Add the new ts0601 sensor definition `_TZE200_qoy0ekbd`.
To avoid duplicate code, I have moved the Tuya -> Zigbee value conversion from the `dp_to_attribute` to the `TuyaRelativeHumidity` cluster. Not happy with the solution but I don't find any better one to avoid duplicate the `TemperatureHumidityManufCluster`.

Now the device can define a `RH_MULTIPLIER` to apply to value. If not defined, defaults to `100`.

Fixes: #1670 
Fixes: #1724
closed 2022-11-04 by dmulcahey