zigpy/zha-device-handlers #1671: [BUG] FYRTUR block-out roller blind - Position reporting

V4n1X, 2022-08-01:
**Describe the bug**
The position of the does not get reported, stop of the cover is not possible because of a missing position.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to stop the cover while moving or try to get the actual position of the cover.

**Expected behavior**
Report of the current position, that a stop is also possible

**Screenshots**
-

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0102",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "FYRTUR block-out roller blind",
  "class": "zhaquirks.ikea.blinds.IkeaTradfriRollerBlinds2"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.7.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.55",
    "supervisor": "2022.07.0",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
[... 314 lines cut ...]
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
A fix for the deconz official software that i tested with my conbee2 stick, so I can confirm that this is working: https://riesinger.dev/posts/ikea-fyrtur-homeassistant-deconz/
After that fix, position, battery, and stop position is working.

Adminiuga, 2022-08-02:
Are these paired through Ikea provided ZigBee repeaters?
Have you tried reconfiguring the devices while keeping the devices awake?

I can't reproduce it with Ikea Trendsen blinds, those report attributes almost in real time.

MattWestb, 2022-08-02:
All IKEA blind and covers is using the same OTA firmware but it can being that different models have different setting for getting the function OK with different hardware (motor controller and low long it shall running) but the ground function shall being the same in the Zigbee side with configuring and reporting.

Runepx4, 2022-09-17:
I have the same issue as @V4n1X.

tbruinsma, 2022-09-23:
Same issue here. Position, battery are not reporting/available.

alexanderphoenix, 2022-11-23:
Just moved from Z2M to ZHA and have encountered this issue as well.

MattWestb, 2022-11-23:
If battery and position is not reported OK do one `reconfigurate` from the device card and wake the blind up if its sleeping then sending the commands.

If the blind is having one real Zigbee router as parent you shall getting in the long on the device card somthing like this:
```
IKEA TREDANSEN block-out cellul blind Checkin event was fired
14:47:35 - 9 minutes ago
```
every hour and if not reparing the blind and reconfigure after its being OK joined.

Battery reporting is not made oft then is one sleeping end device so it can taking little time for it showing up.

oscar-b, 2023-07-01:
This seems to work now in Home Assistant 2023.6.3, the position didn’t report correctly in 2023.3.

brantje, 2024-01-03:
This is still an issue

closed 2024-07-08 by github-actions[bot]