zigpy/zha-device-handlers #1673: [Device Support Request] Zignito Zigbee Plug 13Amps with Energy Monitoring.

DaveTiff, 2022-08-01:
**Is your feature request related to a problem? Please describe.**

They are discovered by ZHA and report back Volts, Amps & Power. The Power factor value remains "Unknown".

You can turn the socket On & Off with HA. But if you turn the socket On or Off via the devices local switch, the On / Off status is not reflected in HA. i.e. HA does not show the actual On - Off state.

**Describe the solution you'd like**

Can they be added to ZHA

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_1h2x4akh",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.7.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
    "supervisor": "2022.07.0",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
[... 508 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
[image]
[image]

TheJulianJES, 2023-07-03:
@DaveTiff If you're on HA Core 2023.6.0 or later, try to "reconfigure" the plug using the button in the UI. Then try to see if the state changes when the button on the plug is pressed.

DaveTiff, 2023-07-04:
Yes, that corrected the issue, now working fine

closed 2023-07-04 by DaveTiff