zigpy/zha-device-handlers #1674: [Device Support Request] Make attribute time_left writable for tuya.ts0601_valve.ParksidePSBZS

BenPru, 2022-08-01:
**Is your feature request related to a problem? Please describe.**
In the normal gui the attribute time_left (Internal device timer to turn off) is readonly.
In the "Manage cluster" "timer_time_left" is writable.

**Describe the solution you'd like**
Make the attribute time_left writable in the normal gui.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.

  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_htnnfasr",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_valve.ParksidePSBZS"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.

{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.7.7",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "run_as_root": true
  },
  "custom_components": {
    "tinkerforge_switch": {
      "version": "2021.03.18",
      "requirements": [
[... 435 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-08-01:
Please, can you enable the debug log level and attach the logs from operating the GUI and the cluster manager.

BenPru, 2022-08-02:
I´m confused... I connected a new, never used device and I can´t set the "time_left" attribute in the cluster manager.
It has 1 minute by default.
[image]

But in the past I have changed it for other devices:
[image]

javicalle, 2022-08-27:
I believe that the default value for the `time_left` attribute is 1 min.
You can change the value via the manage cluster attributes view.

BenPru, 2022-08-29:
> I believe that the default value for the `time_left` attribute is 1 min. You can change the value via the manage cluster attributes view.

In an older zha or ha version I was able to change it via cluster manager. But with the current zha and ha version it doesn't work. But I have other properties that I also can not change.
But it were good to have a normal attribute "time_left" to change it via automation etc.

BenPru, 2022-08-30:
Here is the debug output: @javicalle 
```
2022-08-30 21:47:01.835 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-30 21:47:01.836 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-30 21:47:01.839 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xcc12>, 1, 260, 61184, b'\t\x02\x02\x00\xf7\x06\x02\x00\x04\x00\x00\x01\xf3', 0, 175, 199, 193, 1, 210, 0, -69]
2022-08-30 21:47:01.841 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Received ZCL frame: b'\t\x02\x02\x00\xf7\x06\x02\x00\x04\x00\x00\x01\xf3'
2022-08-30 21:47:01.841 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=2, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-30 21:47:01.842 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Decoded ZCL frame: ParksideTuyaValveManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=247, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xf3\x01\x00\x00', *payload=499)))
2022-08-30 21:47:01.843 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Received command 0x02 (TSN 2): set_data_response(data=TuyaCommand(status=0, tsn=247, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xf3\x01\x00\x00', *payload=499)))
2022-08-30 21:47:01.844 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=2, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-30 21:47:01.845 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-08-30 21:47:01.849 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 2 under 253 request id, data: b'18020b0200'
2022-08-30 21:47:01.850 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xcc12>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09020200f706020004000001f3'
2022-08-30 21:47:01.850 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 253, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0xCC12 endpoint=1>, 260, 61184, 1, b'\x18\x02\x0b\x02\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-30 21:47:01.853 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 253]
2022-08-30 21:47:01.863 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-08-30 21:47:01.864 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-08-30 21:47:01.867 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 253: 00
2022-08-30 21:47:01.867 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xfd 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xcc12 endpoint=1>, status: 0x00
2022-08-30 21:47:03.206 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] write_attributes --> record: Attribute(attrid=0xEF12, value=TypeValue(type=uint32_t, value=500))
2022-08-30 21:47:03.206 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='timer_time_left', attr_value=500, expect_reply=False, manufacturer=10)
2022-08-30 21:47:03.207 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] get_dp_mapping --> found DP: 6
2022-08-30 21:47:03.207 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] from_cluster_data: 6, DPToAttributeMapping(ep_attribute='tuya_manufacturer', attribute_name='timer_time_left', dp_type=<TuyaDPType.VALUE: 2>, converter=None, dp_converter=None, endpoint_id=None)
2022-08-30 21:47:03.207 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] ztype: 500
2022-08-30 21:47:03.207 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] from_value: [4, 0, 0, 1, 244]
2022-08-30 21:47:03.207 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] raw: b'\x00\x00\x01\xf4'
2022-08-30 21:47:03.208 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=254, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x01\xf4', *payload=4093706240))
2022-08-30 21:47:03.209 DEBUG (MainThread) [zigpy.zcl] [0xCC12:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=10,  [... cut ...]
2022-08-30 21:47:04.847 DEBUG (MainThread) [zigpy.zcl] [0x1D69:3:0x0008] Attribute report received: current_level=20
2022-08-30 21:47:04.847 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1D69:3:0x0008]: received attribute: 0 update with value: 20
2022-08-30 21:47:04.848 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1d69>, ep: 3, profile: 0x0104, cluster_id: 0x0008, data: b'18470a00002014'

Any ideas?
```

BenPru, 2022-09-06:
> I believe that the default value for the `time_left` attribute is 1 min. You can change the value via the manage cluster attributes view.

Yes, now I could change it via cluster manager. But it is a good idea to have it in the normal gui.
At the moment I use a script for this:
alias: Bewässerungs Timer setzen
fields:
  ieee:
    example: 84:71:27:ff:fe:cf:d2:32
  minutes:
    example: 10
sequence:
  - service: zha.set_zigbee_cluster_attribute
    data:
      ieee: "{{ ieee }}"
      endpoint_id: 1
      cluster_id: 61184
      cluster_type: in
      attribute: 61201
      value: "{{ minutes }}"
mode: parallel
icon: mdi:valve
max: 10

closed 2023-07-10 by github-actions[bot]