zigpy/zha-device-handlers #1675: [Device Support Request]

RhavoX, 2022-08-03:
**Is your feature request related to a problem? Please describe.**
The Aqara P1 allegedly supports temperature sensing, it is not present in ZHA though.

**Describe the solution you'd like**
I would like to receive the output of the temperature sensor

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0406",
        "0x0500",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.ac02",
  "class": "zhaquirks.xiaomi.aqara.motion_ac02.LumiMotionAC02"
}
```

</details>

Guenni75, 2022-12-02:
Hi. How did you get this thing t work?
I have one, but with ZHA, it only shows 1lx and motion detection doesn't works.
With deCONZ it works, but the integration sucks. Only problems!!

dmulcahey, 2022-12-02:
> Hi. How did you get this thing t work?
> I have one, but with ZHA, it only shows 1lx and motion detection doesn't works.
> With deCONZ it works, but the integration sucks. Only problems!!

Make sure your stick has the newest available firmware on it then reset and re-join the device

TheJulianJES, 2022-12-02:
Regarding the original issue (temperature not showing):

The temperature sensor provided by the device is an internal inaccurate sensor and shouldn't be used really.
It's likely this functionality could be implemented by just adding ``DeviceTemperature.cluster_id,`` in ``replacement`` here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/motion_ac02.py

closed 2023-07-11 by github-actions[bot]