zigpy/zha-device-handlers #1677: [BUG] TS011F _TZ3000_cfnprab5 (XENON 4AC+2USB power strip) Sockets not individually controllable

c11umw, 2022-08-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Pair the power strip via HomeAssistant ZHA
2. Once detected, locate the device in http://homeassistant.local:8123/config/devices/dashboard
3. Enter the device page and each of the switches are present
4. If you click one off, all will trigger off in 1-5 sequence a fraction of a second apart.
5. If you click one on, they all turn back on in the same 1-5 sequence.

**Expected behavior**
Each socket on the strip should be independently controllable separate to the others

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
[... 657 lines cut ...]
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

</details>

It is this device:  https://zigbee.blakadder.com/Xenon_SM-SO306.html

c11umw, 2022-08-03:
My specific model is SM-SO306-K
(UK Specific?)

MattWestb, 2022-08-03:
Your device is very likely needing "tuya magic spell" being casted on it then it joining  the network.
LIDL have one similar power strip that is working OK and they have made on new batch with updated firmware that is doing exactly the same as yous is doing.
One dev was doing one quirk for it and is shall working OK https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/ts011f_plug.py.

If you like testing you can configure local quirks in HA and patching the device signature in the LIDL quirk so it matching your device and ZHA is loading the quirk and you can testing if its working or not.

c11umw, 2022-08-07:
Thank @MattWestb
I had a go. Not sure what I was doing, but I had a go.

Couldn't find much about Tuya Magic Spell, but hints led me to installing ZHA Toolkit via HACS (was that right?) and I tried doing... stuff.

I added to the _/config/custom_components/zha_toolkit_ folder a _ts011f_plug.py_ file (similar to the Lidl one) which also had a 4 socket+USB power strip entry and put my device signature in it ... but it made no difference.
Delete device, reboot, re-add device, all still go on/off with an individual switch.

If I knew what I was doing there I'd have a chance of knowing if it should work or not.

c11umw, 2022-08-08:
Thank @MattWestb
I had a go. Not sure what I was doing, but I had a go.

I found how to add ZHA Quirks from this link. https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226

I added to the /config/custom_zha_quirks/ folder and added the Lidl file, but it was giving me Zigbee errors on reboot. Possibly as it's a 3 socket+4USB device, not 4 socket+2 USB?
However I did find another file which was accepted by HA:
https://github.com/zigpy/zha-device-handlers/blob/57a68a2b57afabc0b5dfc49ef4231263594e310e/zhaquirks/tuya/ts011f_plug.py#L403

Delete device, reboot, re-add device, all still go on/off with an individual switch.

But it still didn't work after I changed the signature for **class Plug_4AC_2USB(CustomDevice):**

Not sure if I need to make other changes to the file, not just the "MODELS_INFO" part....?

MattWestb, 2022-08-08:
Great then you have the local quirk working !!

The tuya TS011F is not having the "magic spell" (reading  and writing attribute then the device is joining the network).

You must edging the signature of the LIDL quirk so the system is loading it (endpoints and cluster must matching). 
Then doing that you must deleting the fixed TS011F so the system is not loading that instead.

apgha, 2022-08-10:
Hi, I have just received the Tuya TS011F and I am facing the sae issue. @c11umw, did you manage to make it work? Thanks.

javicalle, 2022-08-11:
Duplicate of #1632

closed 2022-08-11 by dmulcahey