zigpy/zha-device-handlers #1679: [BUG] Moes "TS0601" radiator valve - Boost and safety lock triggers wrong entity

freequenzart, 2022-08-04:
**Describe the bug**
I have added a second Valve Moes "TS0601" to my Home Assitant by ZHA and by a Sonoff ZHABridge with Tasmota. A Single Valve works perfect. But if I add the second, the boost and the lock Switch of the first and the second valve triggers only the second valve functions.

On/Off, Temperatur and so on working fine.

Both functions have other ID's
eg Boost: 
switch.tze200_e9ba97vf_ts0601_f016c6fe_on_off  (first Valve)
switch.tze200_e9ba97vf_ts0601_72b271fe_on_off  (second Valve)

**To Reproduce**
Steps to reproduce the behavior:
1. Add a second valve
2. Click on Boost or Lock in HA in the first valve
3. only the sceond valve opens fully, not the first, where I clicked

**Expected behavior**
Click on Boost or Lock in HA in the first valve should trigger the first valve, not ne new one

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

<details>

<summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.7.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
    "supervisor": "2022.07.0",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.31.1",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.77",
      "zigpy-deconz==0.18.0",
      "zigpy==0.47.3",
      "zigpy-xbee==0.15.0",
      "zigpy-zigate==0.9.0",
      "zigpy-znp==0.8.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
[... 629 lines cut ...]
            "endpoint_attribute": "on_off",
            "attributes": {
              "0x0000": {
                "attribute_name": "on_off",
                "value": 1
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "on_off"
              },
              "0x4003": {
                "attribute_name": "start_up_on_off"
              }
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

jacekk015, 2022-09-01:
Probably quirk or included in ZHA code doesn't give any ID's to TRV's. That's why you trigger always last added device. Are you using quirk?

freequenzart, 2022-09-08:
I just use the current Home Assisstant on rp4. So i use the code from zha. in the beginning i was using the valve.py but after that was integratet, i switched. How can I have a look on the IDs?

closed 2023-07-10 by github-actions[bot]

alaerock, 2024-11-08:
I have the same issue with 3 TRVs `TS0601`:
- only one will be triggered on `boost hit`
- when `boost hit` time is finished, the switch is only turned `off` for the first one (that was not triggered), the clicked one switch still `on` without hit. 
- I already removed all TRVs and re-added them with different names, only the same one is triggered if other TRVs are triggered.

jacekk015, 2024-11-08:
@alaerock 
https://github.com/jacekk015/zha_quirks

freequenzart, 2024-11-08:
Thank you. I had nearly forgotten my issue. The update to the *[s0601_trv_zonnsmart.py](https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_zonnsmart.py)* was the solution. This issue can be closed.

Regards