zigpy/zha-device-handlers #1680: [BUG]

JoryHogeveen, 2022-08-04:
**Describe the bug**

Since I'm not sure where to report I also reported here: https://github.com/home-assistant/core/issues/76205

I have 3 groups of GU10 spots.

1 group is IKEA > This group works well!
2 groups are eWeLink > These have issues.

When I turn on the ZHA group entity the children respond normally. However, when I turn off the ZHA group then the children still lists as being ON (even though the light is off!).

After a few minutes one-by-one the devices register as OFF again, most likely since they report this themselved.

**Expected behavior**
The group children should turn off as well :)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0210",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300"
      ],
      "out_clusters": []
    },
    "13": {
      "profile_id": 49246,
      "device_type": "0xe15e",
      "in_clusters": [
        "0x1000"
      ],
      "out_clusters": [
        "0x1000"
      ]
    }
  },
  "manufacturer": "eWeLight",
  "model": "ZB-CL01",
  "class": "zigpy.device.Device"
}
```

</details>

**Additional context**
No idea what to find further.

MattWestb, 2022-08-04:
Its looks like the eWeLink is not reporting off state as Zigbee standard and ZHA is pulling its status some minutes later and updating the device status and then all is off ZHA is making the group off.

I have read some similar problem but i cant remember with brands it was. 
Philips HUE is not reporting at all and ZHA is pulling the lights for getting  the status from then and i thin its one or 2 more brands that have the same fix.

JoryHogeveen, 2022-08-04:
Hi @MattWestb 
Thank you for your quick reply again!
How was it fixed for these devices? I honestly thought that ZHA would simply turn off the children within HA.

MattWestb, 2022-08-04:
Philips HUE is "hard coded" for pulling the status from the device in ZHA.
I think @TheJulianJES (sorry for pinging you directly) is knowing more of the "bad" no standard Zigbee lights and how its working.

ZHA is waiting that the device in light groups is updating there state and if the device is not reporting its changed state ZHA is not putting the light group in off state.
Zigbee light devices shall reporting there changes state to the coordinator (or configured device / address) if its being changed so the coordinator dont need spamming the network with pulls all the time to all lights in the system for getting its status.

You can try reconfigure the device from the device card and see if they is start reporting OK but i think its not helping then on state is reported OK.

TheJulianJES, 2022-08-04:
Thanks for the detailed report!
Do you know if these eWeLink lights in ZHA groups were also broken in HA 2022.7.x?

When a ZHA light group is turned on/off, it waits about a second (default turn on/off transition time of most lights) and then polls all individual group members for their status.
(For single Hue lights, this is done in a 3-5 minute interval. For all other "single" lights, it's done every 45 to 70 minutes IIRC.) If an attribute reports is received after that 1.25 second delay, it also gets parsed.

For the eWeLink lights, can you check the following attribute (ZHA -> Devices -> eWeLink light -> three dots -> Manage Clusters -> select OnOff cluster -> select on_off attribute -> Read attribute:
[image]
The attribute should read ``Bool.false`` when the light is off and ``Bool.true`` when it's on. (Please check if this works)

Otherwise, turn on debug logging (restart HA), then wait a bit and turn on/off a ZHA group. Then send the relevant log part here.
https://www.home-assistant.io/integrations/zha/#debug-logging

JoryHogeveen, 2022-08-04:
Hi @TheJulianJES 

Checked the cluster and this works as you mentioned!
So the spots turn off but ZHA doesn't yet check it's status?
I also tested turning the group on and off. Once the group was off (and the spot still reports as on) I checked the cluster again an ZHA instantly updated and turned the light off in HA.
For this reason I think (guess) the bulb isn't the issue. Even if I get the cluster value instantly after turning off the group it reports as off (`Bool.false`) and HA is updated.

Oh and yes, the same issue occurs on 2022.7.x

dmulcahey, 2022-08-04:
Please reconfigure each bulb and post screenshots of the dialogs. Also provide a debug log as described by @TheJulianJES

JoryHogeveen, 2022-08-04:
Hi @dmulcahey 

Reconfiguration can be done without issues. See screenshot below.

I'd very much like provide log data but I'll need a bit of help on how to filter out the relevant records. I have 60 devices in ZHA so if I simply log everything it will be a mess I suppose.

[image]

TheJulianJES, 2022-08-04:
Check if the behavior is now different after reconfiguration.

Regarding the logs: each Zigbee bulb has a unique Nwk id (like ``0xe77c``). That can be found on the device page and in some log lines.

If possible, just let HA/ZHA settle down a bit after restarting with debug logging. Then just turn on/off the ZHA group light (wait some time in between) and send the entire "log chunks" of where a group is turned on/off (even if there are other ZHA devices mixed in between).

dmulcahey, 2022-08-04:
I can handle the mess :) Just give us the nwk addresses as @TheJulianJES mentions.

JoryHogeveen, 2022-08-04:
Ok no problem! I'll run HA with logger asap!
I currently cannot access my PI remotely to change yaml files since I'm not at home :/

dmulcahey, 2022-08-04:
No rush. Thanks!

TheJulianJES, 2022-08-04:
No hurries. FYI, the following service calls should also do the same:

<details>
<summary>Set to debug</summary>

```yaml
service: logger.set_level
data:
  homeassistant.core: debug
  homeassistant.components.zha: debug
  bellows.zigbee.application: debug
  bellows.ezsp: debug
  zigpy: debug
  zigpy_deconz.zigbee.application: debug
  zigpy_deconz.api: debug
  zigpy_xbee.zigbee.application: debug
  zigpy_xbee.api: debug
  zigpy_zigate: debug
  zigpy_znp: debug
  zhaquirks: debug
```

</details>

<details>
<summary>Reset to info</summary>

```yaml
service: logger.set_level
data:
  homeassistant.core: info
  homeassistant.components.zha: info
  bellows.zigbee.application: info
  bellows.ezsp: info
  zigpy: info
  zigpy_deconz.zigbee.application: info
  zigpy_deconz.api: info
  zigpy_xbee.zigbee.application: info
  zigpy_xbee.api: info
  zigpy_zigate: info
  zigpy_znp: info
  zhaquirks: info
```

</details>

JoryHogeveen, 2022-08-05:
Hi @dmulcahey and @TheJulianJES 

I'm a bit flabbergasted at the moment. Just updated to 2022.8.1 and it seems the issue is gone :/
However, I don't see and ZHA related changes in the logs..

While I hate to admit it, I'm now thinking the restart might have simply fixed it..
Could it be that the Zigbee network had too much spam?

MattWestb, 2022-08-05:
I have also having problem for some versions ago that controller (IKEA and tuya remotes) was "silent" (was not getting zha_event) but was starting reporting after HA restart.

ZHA change log is in HA release notes https://rc.home-assistant.io/changelogs/core-2022.8 only need searching for "ZHA" and is normally linking to zigpy release notes if its changes made in the Zigbee libs.

Glad that your spots start reporting OK and hope its staying so !!

JoryHogeveen, 2022-08-05:
Hope so you! Sorry to waste your time.
If I encounter the same issue again I will start with the logs directly :)

closed 2022-08-05 by JoryHogeveen