zigpy/zha-device-handlers #1681: [Device Support Request] Freint Electricity Meter Interface (Develco ZHEMI101)

RefineryX, 2022-08-04:
**Is your feature request related to a problem? Please describe.**
I believe that the Freint Electricity Meter Interface (Develco ZHEMI101) is already somewhat supported, meaning I can add it to my Home Assistant via ZHA. However, not all sensors and functionality is present....

**Describe the solution you'd like**
1. It appears that the battery sensor is supported but does get get exposed via ZHA. According to the manual, there is a status attribute on the metering cluster for reporting “Low battery”.

2. The device supports changing  the ```PulseConfiguration``` from the default ```1000``` to something else such as ```3200```. However, changing this in Home Assistant does not save and the default ```1000` remains. 

3. Upon adding the device, it exposes a ```Devicetemperature``` sensor. There is no features to allow the monitoring of the device.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.0: 0>, mac_capability_flags=<MACCapabilityFlags.NONE: 0>, manufacturer_code=0, maximum_buffer_size=0, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0002",
        "0x0003",
        "0x0005"
      ],
      "out_clusters": [
        "0x0001"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0702"
      ],
      "out_clusters": [
        "0x000a"
      ]
    }
  },
  "manufacturer": "Develco",
  "model": "ZHEMI101",
  "class": "zigpy.device.Device"
} ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.8.0",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.10.0-16-amd64",
    "run_as_root": true
[... 352 lines cut ...]
**Technical manual:**
https://www.develcoproducts.com/media/1782/zhemi101-technical-manual-external-meter-interface.pdf

### **Extracts from Manual**

**_Battery Information_**

[image]

_**Pulse Confirmation Information:**_

[image]

_**Pulse Confirmation in Home Assistant**_

_This does not save and the default remains present_

[image]

### _**HA Reconfig**_
[image]

### Other info
I saw zigbee2mqtt added support recently
https://github.com/Koenkk/zigbee2mqtt/issues/8638

javicalle, 2022-08-06:
Could you try to put the 4117 value as `Manufacturer Code Override` when you tries to set the value from the cluster management view?

RefineryX, 2022-08-06:
@javicalle tried this a few times with the manufacture code of ```4117```. It allows me to input but it does not save.

**Saving the new value with manufacturer code:**

[image]

**Getting the new ZigBee Attribute:**

[image]

javicalle, 2022-08-06:
I believe that here is the relevant info about Z2M support:
* https://github.com/Koenkk/zigbee-herdsman-converters/pull/3337/files

There is nothing that ring my bells.

Can you enable the debug log level and attach the result from writing and reading the value?

RefineryX, 2022-08-06:
Sure, I will do that today. Please could you advise what I add to my config file to enable zha logging? I use a conbee II stick. I did see this link though not sure which one to add:

https://www.home-assistant.io/integrations/zha/#debug-logging

RefineryX, 2022-08-06:
I hope these debug logs help.

<details>
<summary>Extract from Debug Logs</summary>

```2022-08-06 20:00:48.646 DEBUG (MainThread) [zigpy.zcl] [0x9377:2:0x0702] Decoded ZCL frame: Metering:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=25831)), Attribute(attrid=0x0400, value=TypeValue(type=int24s, value=14062)), Attribute(attrid=0x0200, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>)), Attribute(attrid=0x0100, value=TypeValue(type=uint48_t, value=281474976710655)), Attribute(attrid=0x0102, value=TypeValue(type=uint48_t, value=281474976710655))])
2022-08-06 20:00:48.647 DEBUG (MainThread) [zigpy.zcl] [0x9377:2:0x0702] Received command 0x0A (TSN 67): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=25831)), Attribute(attrid=0x0400, value=TypeValue(type=int24s, value=14062)), Attribute(attrid=0x0200, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>)), Attribute(attrid=0x0100, value=TypeValue(type=uint48_t, value=281474976710655)), Attribute(attrid=0x0102, value=TypeValue(type=uint48_t, value=281474976710655))])
2022-08-06 20:00:48.648 DEBUG (MainThread) [zigpy.zcl] [0x9377:2:0x0702] Attribute report received: current_summ_delivered=25831, instantaneous_demand=14062, status=<bitmap8.0: 0>, current_tier1_summ_delivered=281474976710655, current_tier2_summ_delivered=281474976710655
2022-08-06 20:00:48.650 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.develco_zhemi101_smartenergysummation, old_state=<state sensor.develco_zhemi101_smartenergysummation=25.81; state_class=total_increasing, device_type=Electric Metering, status=NO_ALARMS, unit_of_measurement=kWh, device_class=energy, friendly_name=Develco ZHEMI101 Smartenergysummation @ 2022-08-06T20:00:43.670189+01:00>, new_state=<state sensor.develco_zhemi101_smartenergysummation=25.831; state_class=total_increasing, device_type=Electric Metering, status=NO_ALARMS, unit_of_measurement=kWh, device_class=energy, friendly_name=Develco ZHEMI101 Smartenergysummation @ 2022-08-06T20:00:48.650170+01:00>>
2022-08-06 20:00:48.651 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.develco_zhemi101_smartenergymetering, old_state=<state sensor.develco_zhemi101_smartenergymetering=12568; state_class=measurement, device_type=Electric Metering, status=NO_ALARMS, unit_of_measurement=W, device_class=power, friendly_name=Develco ZHEMI101 Smartenergymetering @ 2022-08-06T20:00:43.670844+01:00>, new_state=<state sensor.develco_zhemi101_smartenergymetering=14062; state_class=measurement, device_type=Electric Metering, status=NO_ALARMS, unit_of_measurement=W, device_class=power, friendly_name=Develco ZHEMI101 Smartenergymetering @ 2022-08-06T20:00:48.651295+01:00>>
2022-08-06 20:00:48.660 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.live_electricity_usage, old_state=<state sensor.live_electricity_usage=3928; state_class=measurement, unit_of_measurement=W, device_class=power, friendly_name=Live Electricity Usage @ 2022-08-06T20:00:43.673357+01:00>, new_state=<state sensor.live_electricity_usage=4394; state_class=measurement, unit_of_measurement=W, device_class=power, friendly_name=Live Electricity Usage @ 2022-08-06T20:00:48.660498+01:00>>
2022-08-06 20:00:48.661 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x9377>, ep: 2, profile: 0x0104, cluster_id: 0x0702, data: b'18430a000025e7640000000000042aee360000021800000125ffffffffffff020125ffffffffffff'
2022-08-06 20:00:53.621 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-06 20:00:53.622 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-06 20:00:53.625 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [63, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x9377>, 2, 260, 1794, b'\x18D\n\x00\x00%\xf2d\x00\x00\x00\x00\x00\x04*P"\x00\x00\x02\x18\x00\x00\x01%\xff\xff\xff\xff\xff\xff\x02\x01%\xff\xff\xff\xff\xff\xff', 0, 175, 255, 45, 129, 143, 1, -28]
2022-08-06 20:00:53.626 DEBUG (MainThread) [zigpy.zcl] [0x9377:2:0x0702] Received ZCL frame: b'\x18D\n\x00\x00%\xf2d\x00\x00\x00\x00\x00\x04*P"\x00\x00\x02\x18\x00\x00\x01%\xff\xff\xff\xff\xff\xff\x02\x01%\xff\xff\xff\xff\xff\xff'
2022-08-06 20:00:53.627 DEBUG (MainThread) [zigpy.zcl] [0x9377:2:0x0702] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=68, command_id=10, *is_reply=True)
2022-08-06 20:00:53.628 DEBUG (MainThread) [zigpy.zcl] [0x9377:2:0x0702] Decoded ZCL frame: Metering:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=25842)), Attribute(attrid=0x0400, value=TypeValue(type=int24s, value=8784)), Attribute(attrid=0x0200, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>)), Attribute(attrid=0x0100, value=TypeValue(type=uint48_t, value=281474976710655)), Attribute(attrid=0x0102, value=TypeValue(type=uint48_t, value=281474976710655))])
[... cut ...]
2022-08-06 20:01:20.494 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB01B](LCT010): Update device availability -  device available: True - new availability: True - changed: False
2022-08-06 20:01:20.500 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x301B](TRADFRI SHORTCUT Button): Device seen - marking the device available and resetting counter
2022-08-06 20:01:20.500 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x301B](TRADFRI SHORTCUT Button): Update device availability -  device available: True - new availability: True - changed: False
```
</details>

RefineryX, 2022-08-10:
@javicalle were these logs okay? Happy to pull others if needed.

closed 2023-02-13 by github-actions[bot]