zigpy/zha-device-handlers #1682: [Device Support TS0601 by _TZE200_qoy0ekbd Request] tuya lcd temperature and humidity sensor

ernstmosch, 2022-08-05:
**Is your feature request related to a problem? Please describe.**

The Sensor is found by zha-discovery and 3 Sensors are been made (Temprature/Humidity/Battery)
But The Sensors dont get any values over zigbee.

**Describe the solution you'd like**
i would like the device to provide the values it displays on lcd over zigbee.

<details>

<summary>Device signature</summary>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qoy0ekbd",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
in my webretrieval i found solutions from the zigbee2mqtt side where they made there version of a quirk where they made the device from manucaturer '_TZE200_qoy0ekbd' behave the same way like manufacturer '_TZE200_bjawzodf'

    

>   {
>           fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bq5c8xfe'},
> !             {modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'},
> !             {modelID: 'TS0601', manufacturerName: '_TZE200_qoy0ekbd'}],
>           model: 'TS0601_temperature_humidity_sensor',
>           vendor: 'TuYa',
>           description: 'Temperature & humidity sensor',
> 

but i have to damit that my knowledge is to limiteted to do the same

MattWestb, 2022-08-05:
Try local quirk and adding your device in this line: https://github.com/zigpy/zha-device-handlers/blob/57a68a2b57afabc0b5dfc49ef4231263594e310e/zhaquirks/tuya/ts0601_sensor.py#L79
Restarting HA and look if the system is loading the quirk.

javicalle, 2022-08-05:
Maybe duplicate of #1670
The device's signature may not match with the quirk.
Any relevant info in the logs?

ernstmosch, 2022-08-06:
Javicalle your right it is an duplicate of #1670

Please excuse me that i didnt see it.

I can´t find relevant info in the logs. Maybe i dont look in the proper log - im still a little confused by HA logging.

ernstmosch, 2022-08-06:
> Try local quirk and adding your device in this line:
> 
> https://github.com/zigpy/zha-device-handlers/blob/57a68a2b57afabc0b5dfc49ef4231263594e310e/zhaquirks/tuya/ts0601_sensor.py#L79
> 
> 
> Restarting HA and look if the system is loading the quirk.

Hi MattWestB

I added a line in "ts0601_sensor.py" and after reboot HA created a folder "__pycache__", so i think HA used my custom quirk.

`class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_bjawzodf", "TS0601"),
            ("_TZE200_qoy0ekbd", "TS0601"),
        ],
        ENDPOINTS: {
            1: {`

But nothing changed - all 3 sensors remain in silence.

(i resetted the device and went through new discovery)

javicalle, 2022-08-06:
Can we continue the discussion in the other issue and keep all the info together?

closed 2022-08-14 by ernstmosch