zigpy/zha-device-handlers #1683: [Device Support Request] New TS0043 variant _TZ3000_kkahwiyu 3 button switch/remote

julianjwong, 2022-08-07:
**Is your feature request related to a problem? Please describe.**
The _TZ3000_kkahwiyu variant of the TS0043 isn't supported by Home Assistant. Device can pair but no ZHA events fire and buttons do not respond to automation.

**Describe the solution you'd like**
Would be great if someone could update the quirk to handle this model. I am happy to test custom quirks.

<details>

<summary>Device signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xe000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_kkahwiyu",
  "model": "TS0043",
  "class": "zigpy.device.Device"
}

```

</details>

<details>

<summary>Diagnostic information</summary>

```
[... 343 lines cut ...]
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}

```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

riddlenl, 2022-09-19:
+1!

Got this 3 gang switch, but all I get is 4 "switches" which don't do anything in HA. The 3 gang switches do work as intended.

Rush56, 2023-07-03:
To fix this a quirk is needed. My 3 gang touch zigbee switch apparently had firmware for 4 gang switch. Therefore it was not recognized in HA. I had to modify quirk for it, that fixed it.

julianjwong, 2023-07-03:
Been working for a while for me. Not sure which release fixed it.

closed 2023-07-03 by julianjwong