zigpy/zha-device-handlers #1686: Garden Valve

infinitejest, 2022-08-08:
Dear ZHA developers: 

I bought this device on Ali Express: 

https://www.aliexpress.com/item/3256804035783288.html?spm=a2g0o.order_list.0.0.7c961802trDImq

This is the technical information found on zigbee2mqtt website
https://www.zigbee2mqtt.io/devices/QT06.html

ZHA is able to find the device in the Home assistant UI. However, the only entities found are:  
Lqi
Rssi

Entities expected such as battery; on/off are not found. I know some python but not sure I can figure out how to develop a bridge.

 Just wondering if you have any suggestions to at least find the on/off so I can turn on and off remotely through Home assistant Hub. 

Here is the device Signature: 
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_sh1btabb",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

javicalle, 2022-08-09:
Duplicate of #1622

infinitejest, 2022-08-14:
The issue was not resolved. I put the solution in #1622

infinitejest, 2022-08-19:
If i was able to get more entities like start time.  Any tips on how to. Make home assistant add, extra attributes.  Its showing in device diagnostic but not on UI

mbo18, 2022-11-27:
This issue can be closed. Support has been added in ZHA and should be available in HA in next release

javicalle, 2022-11-27:
Duplicate

closed 2022-11-27 by javicalle