zigpy/zha-device-handlers #1687: [Device Support Request] RTX Zigbee Tuya ZIR aka. TS1201

sobkas, 2022-08-09:
**Is your feature request related to a problem? Please describe.**
Adding support for Tuya TS1201 link: https://allegro.pl/oferta/uniwersalny-pilot-ir-tuya-zigbee-smart-remote-360-11845030905

**Describe the solution you'd like**
Everything is described at https://github.com/Koenkk/zigbee2mqtt/issues/11633

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xf000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe004",
        "0xed00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3290_7v1k4vufotpowp9z",
  "model": "TS1201",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.55",
    "supervisor": "2022.07.0",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
    "chassis": "vm",
[... 275 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

nimrodolev, 2022-08-30:
+1 Here!

This is also sold as `Moes UFO-R11`.

Would very much like to have support for this one.

akarabach, 2023-01-22:
Any updates on it ?

ferehcarb, 2023-02-07:
Hi, I'm working on it, learning mode is functional, I need to write the code to send IR frames.
I'll do a PR when it'll be finished.

ferehcarb, 2023-02-09:
Quirk is functional, here: https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py
Learning mode must be set with "Manage Zigbee Device":
[image]
When you click on "Issue Zigbee Command" the blue led on the device should light up.

Then, you press the button on your IR remote and the IR code is available in homeassistant logs, the blue led should turn down.

You can replay the IR code with:
[image]

Warning, very early code, I must clean up and better integrate it with HA.

ranton1990, 2023-02-09:
Hey THX a lot for that. I am trying to follow your guide but i don't seem to get any codes in my logs. The blue LED lights up for about 20 seconds if I dont point a remote at it and even earlier if I do, but nothing in the logs. Do you mean the standard home-assistant.log, which is also shown in the settings?

ferehcarb, 2023-02-09:
> Hey THX a lot for that. I am trying to follow your guide but i don't seem to get any codes in my logs. The blue LED lights up for about 20 seconds if I dont point a remote at it and even earlier if [...]

Yes, in the standard logs, file `/config/home-assistant.log`, you should see "Command 0x05: Ir message really totally received:"  followed by the IR code (base64 encoded).
Your log config must be at least to `info` I think, I use `debug` for zha on my side:
```
logger:
  default: warning
  logs:
    homeassistant.components.zha: warning
    zigpy: debug
    custom_components.zha_toolkit: debug
    custom_zha_quirks: debug
    zhaquirks: debug
```

ranton1990, 2023-02-09:
Thx, Zigpy logging must be `info`.

ranton1990, 2023-02-09:
Here is a sample script to send a code:
<pre>
alias: Vacuum Start
sequence:
  - service: zha.issue_zigbee_cluster_command
    data:
      cluster_type: in
      endpoint_id: 1
      command: 2
      ieee: 38:5b:44:ff:fe:33:11:8a
      command_type: server
      params:
        code: >-
          C7ILsgsfAskBHwLGBYADwAtAB+APA8ArwAfgDyfgIxfAS0AHQDdAB8ADB/pPsguyCx8C4AEXArMCBWCzB8kByQGzAskBQAGAH0AHCR8CyQHJAR8CxgXgBQNAE+A3A+ADU0BL4AMP4AmnAR8CwCNAB+APA0An4AMD4A8n4CMX4AdP4AMP4JWnBLMCxgUfYAPglacJHwLGBR8CxgUfAg==
      cluster_id: 57348
mode: single
icon: mdi:robot-vacuum
</pre>

You will need to switch out the ieee as well as the code. The cluster_id was translated from 0xe004

ferehcarb, 2023-02-09:
@ranton1990 Is it working for your device ?
I developed the last pieces of code outside my home and didn't test it on a real device...

ranton1990, 2023-02-09:
@ferehcarb Yes it is, I can successfully start my Robo Vacuum and send it back to its station.

loclamor, 2023-03-13:
Hi @ferehcarb i'm trying to use your quirk (first time for me, maybe i missed something), but i have the following errors on stratup : 
[image]
Any ideas what i'm wrong ?

MattWestb, 2023-03-13:
You have copy the URL for the quirk and not the real file so the system cant loading it.
In the GIT clicking on the file and the raw button and copy the raw data and saving it in your local quirk file then you doing getting the http things with.

loclamor, 2023-03-13:
Yes, no more error with the correct file :+1: 

But when adding my device it is not using the quirk : 
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xf000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe004",
        "0xed00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3290_7v1k4vufotpowp9z",
  "model": "TS1201",
  "class": "zigpy.device.Device"
}
```

MattWestb, 2023-03-13:
Then you must updating the signature in the quirk so it matching your device then the system can loading it.

ferehcarb, 2023-03-13:
Hi, the signature for "_TZ3290_7v1k4vufotpowp9z" is already in the quirk.
Did you specify the custom directory for quirks in your config ?

loclamor, 2023-03-13:
Yes, and i can see a "__pycache__" folder in my custom quirk folder.

I have delete and add again the device but still the same class.

[image]
[image]

loclamor, 2023-03-13:
@ferehcarb 
Could this help ?
[image]

loclamor, 2023-03-13:
@ferehcarb I think the problem is this : 
```
Checking quirks for _TZ3290_7v1k4vufotpowp9z TS1201 (dc:8e:95:ff:fe:f4:28:56)
Considering <class 'ts1201.ZosungIRBlaster'>
Fail because endpoint list mismatch: {1} {1, 242}
```
The signature of my device did not match the ENDPOINTS list in the quirk

ferehcarb, 2023-03-13:
Well seen.
Try to remove this lines:
- https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py#L436
- https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py#L457
It seems there isn't PowerConfiguration cluster on your device.

ferehcarb, 2023-03-13:
Arf... you have another endpoint (242) too.
You should add to "ENDPOINTS" dictionary signature I think.

loclamor, 2023-03-13:
I wasn't know quirks 2 days ago 😅
What should i add as endpoint for 242 ?for profil, device type and clusters ?

ferehcarb, 2023-03-14:
Neither me, I wasn't know what it was 2 months ago ;-)
I pushed a [new version](https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py) with your device signature, give it a try.

loclamor, 2023-03-14:
@ferehcarb Thx ! Seems to have worked : 
[image]
I'll try with an IR remote tonigth at home or tomorow :)

ferehcarb, 2023-03-14:
Bye the way, now you can get the learned IR code with an attribute:
[image]
You don't need any more to catch it in the hommeassistant logs.

loclamor, 2023-03-15:
@ferehcarb I have this error when triggering the IRLearn : 
[image]
And i cannot set the value to `true` : 
[image]

ferehcarb, 2023-03-15:
What is this sorcery ?
Can you put zha in debug mode and post logs please ?
Something like this:
```
logger:
  default: warning
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    custom_components.zha_toolkit: debug
    custom_zha_quirks: debug
    zhaquirks: debug
````

loclamor, 2023-03-15:
I think related logs is there : 
```
2023-03-15 09:33:40.442 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Received ZCL frame: b'\x08\x1a\n\x01\x00 F\xe2\xff \x1f\xe4\xff \x00'
2023-03-15 09:33:40.443 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=26, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-15 09:33:40.445 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=70)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=31)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])
2023-03-15 09:33:40.446 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Received command 0x0A (TSN 26): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=70)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=31)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])
2023-03-15 09:33:40.447 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Attribute report received: app_version=70, 0xFFE2=31, 0xFFE4=0
2023-03-15 09:33:40.450 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=26, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-15 09:33:40.451 DEBUG (MainThread) [zigpy.zcl] [0x56C3:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2023-03-15 09:33:42.645 DEBUG (MainThread) [homeassistant.components.zha.api] Requested attributes for: cluster_id: 57348, cluster_type: 'in', endpoint_id: 1, response: [{'id': 0, 'name': 'last_learned_ir_code'}]
2023-03-15 09:33:42.646 DEBUG (MainThread) [homeassistant.components.zha.api] Requested commands for: cluster_id: 57348, cluster_type: 'in', endpoint_id: 1, response: [{'type': 'server', 'id': 0, 'name': 'data'}, {'type': 'server', 'id': 1, 'name': 'IRLearn'}, {'type': 'server', 'id': 2, 'name': 'IRSend'}]
2023-03-15 09:33:44.091 DEBUG (MainThread) [zigpy.zcl] [0xC78F:1:0x0b04] Received ZCL frame: b'\x18\xf8\n\x0f\x05!\xac\r'
2023-03-15 09:33:44.092 DEBUG (MainThread) [zigpy.zcl] [0xC78F:1:0x0b04] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=248, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-15 09:33:44.093 DEBUG (MainThread) [zigpy.zcl] [0xC78F:1:0x0b04] Decoded ZCL frame: ElectricalMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x050F, value=TypeValue(type=uint16_t, value=3500))])
2023-03-15 09:33:44.093 DEBUG (MainThread) [zigpy.zcl] [0xC78F:1:0x0b04] Received command 0x0A (TSN 248): Report_Attributes(attribute_reports=[Attribute(attrid=0x050F, value=TypeValue(type=uint16_t, value=3500))])
2023-03-15 09:33:44.094 DEBUG (MainThread) [zigpy.zcl] [0xC78F:1:0x0b04] Attribute report received: apparent_power=3500
2023-03-15 09:33:45.028 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Received ZCL frame: b'\x08g\n\x00\x00\x10\x00'
2023-03-15 09:33:45.029 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=103, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-15 09:33:45.031 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2023-03-15 09:33:45.031 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Received command 0x0A (TSN 103): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2023-03-15 09:33:45.032 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Attribute report received: on_off=<Bool.false: 0>
2023-03-15 09:33:45.033 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=103, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-15 09:33:45.034 DEBUG (MainThread) [zigpy.zcl] [0x9A87:1:0x0006] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2023-03-15 09:33:48.677 DEBUG (MainThread) [zigpy.zcl] [0x7C0E:1:0xe004] Receive command with args: () and kwargs {}
[... cut ...]
await result
File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 1273, in issue_zigbee_cluster_command
response = await zha_device.issue_cluster_command(
File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 677, in issue_cluster_command
response = await cluster.command(
File "/config/zha_quirks/ts1201.py", line 119, in command
if kwargs["on_off"]:
KeyError: 'on_off'
```

full log is here : https://gist.github.com/loclamor/07218efd03849c66696785f107f451aa

ferehcarb, 2023-03-15:
You can try to call the service with the "Developper Tools":
[image]

loclamor, 2023-03-15:
@ferehcarb I was not up-to-date... i'm now on Home Assistant 2023.3.4 : 
Zigbee cluster : 
[image]
cluster command : 
[image]

Seems promizing... I will try soon !

loclamor, 2023-03-16:
Thank you for your work @ferehcarb !
I have manage to learn a code from my climatiser remote and succesfully emit it from ha ! 😍
now i have to manage an automation with all possible temperatures and fan speed, for heat and freeze 😅

ferehcarb, 2023-03-16:
Good news, thank you for your perseverance, we've got now another device supported by the quirk.
I think yours is not battery powered but by charger, right ?

loclamor, 2023-03-16:
@ferehcarb sadly yes !

davidbb, 2023-04-01:
Just a quick comment to say the quirk worked for me. I was able to learn new IR codes and replay them as described in https://github.com/zigpy/zha-device-handlers/issues/1687#issuecomment-1424357574 

My device is the battery powered version, reported as:
```
TS1201
by _TZ3290_ot6ewjvmejq5ekhl
```
Thanks for putting this together @ferehcarb!

ferehcarb, 2023-04-03:
Good news, I think it's time to create a PR.

blair287, 2023-04-04:
> Good news, I think it's time to create a PR.

I've tested this I'm having an issue with commands that require a long press it's only sending a short press anyway to fix this please?

davidbb, 2023-04-04:
@blair287 have you tried sending the same command in rapid succession? you could write a home assistant script that issues a set of zigbee cluster commands N times to simulate your key hold or long press. That's usually all that the remote is doing.

blair287, 2023-04-05:
> @blair287 have you tried sending the same command in rapid succession? you could write a home assistant script that issues a set of zigbee cluster commands N times to simulate your key hold or long press. That's usually all that the remote is doing.

Broadlink shows different commands for long and short but they aren't in tje same format this returns a short press even though your still holding the button

ranton1990, 2023-04-06:
Not sure when it first happened, but today I noticed this while starting HA:
`2023-04-06 13:35:17.633 ERROR (MainThread) [zhaquirks] Unexpected exception importing custom quirk 'ts1201'

Traceback (most recent call last):

  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 418, in setup

    importer.find_module(modname).load_module(modname)

  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper

  File "<frozen importlib._bootstrap_external>", line 1063, in load_module

  File "<frozen importlib._bootstrap_external>", line 888, in load_module

  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim

  File "<frozen importlib._bootstrap>", line 719, in _load

  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked

  File "<frozen importlib._bootstrap_external>", line 883, in exec_module

  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed

  File "/config/custom_zha_quirks/ts1201.py", line 53, in <module>

    class ZosungIRControl(CustomCluster):

  File "/config/custom_zha_quirks/ts1201.py", line 65, in ZosungIRControl

    0x00: foundation.ZCLCommandDef(

TypeError: ZCLCommandDef.__init__() got an unexpected keyword argument 'is_reply'`

ranton1990, 2023-04-06:
After removing all `is_reply` lines in the py file, it works again perfectly.

ferehcarb, 2023-04-06:
> After removing all `is_reply` lines in the py file, it works again perfectly.

We must replace `is_reply=True` by `direction = Direction.Client_to_Server` or something like this , and the invert for `is_reply=False`.
The zigbbe lib code has changed and dropped this attribute I think.
is_reply is necessary, otherwise, without it, frames from the devices are repeated and there is much delay to learn or send the IR code.

ferehcarb, 2023-04-07:
@ranton1990 , fixed with [this commit](https://github.com/ferehcarb/zha-device-handlers/commit/0087a3cc57c817a1ab41b3ea3d95fa4deac8eaa3)

blair287, 2023-04-07:
> @ranton1990 , fixed with [this commit](https://github.com/ferehcarb/zha-device-handlers/commit/0087a3cc57c817a1ab41b3ea3d95fa4deac8eaa3)

Any idea how to get long presses working at all?

Manbearpig92, 2023-04-09:
> Hi!
> 
> I have a similar zigbee IR Remote Controller. It looks different, but identifies as `TS1201 by _TZ3290_ot6ewjvmejq5ekhl`: https://www.ebay.com/itm/275310058694
> 
> I can confirm that it works with this custom quirk. I tried it with a Yamaha amplifier (power on/off and volume up/down). Both learning and sending IR codes.
> 
> Thank you for your work!

what did you do for the config after 

> Hi!
> 
> I have a similar zigbee IR Remote Controller. It looks different, but identifies as `TS1201 by _TZ3290_ot6ewjvmejq5ekhl`: https://www.ebay.com/itm/275310058694
> 
> I can confirm that it works with this custom quirk. I tried it with a Yamaha amplifier (power on/off and volume up/down). Both learning and sending IR codes.
> 
> Thank you for your work!

how did you configure the IR Blaster? I'm not sure what settings i need to select on all the drop down menus. I have the same blaster as you except its the Wired version.

Manbearpig92, 2023-04-09:
> Here is a sample script to send a code:
> 
> alias: Vacuum Start
> sequence:
>   - service: zha.issue_zigbee_cluster_command
>     data:
>       cluster_type: in
>       endpoint_id: 1
>       command: 2
>       ieee: 38:5b:44:ff:fe:33:11:8a
>       command_type: server
>       params:
>         code: >-
>           C7ILsgsfAskBHwLGBYADwAtAB+APA8ArwAfgDyfgIxfAS0AHQDdAB8ADB/pPsguyCx8C4AEXArMCBWCzB8kByQGzAskBQAGAH0AHCR8CyQHJAR8CxgXgBQNAE+A3A+ADU0BL4AMP4AmnAR8CwCNAB+APA0An4AMD4A8n4CMX4AdP4AMP4JWnBLMCxgUfYAPglacJHwLGBR8CxgUfAg==
>       cluster_id: 57348
> mode: single
> icon: mdi:robot-vacuum
> You will need to switch out the ieee as well as the code. The cluster_id was translated from 0xe004

did you need to manage device before doing this or did you just add the device to ZHA and make that script?

Manbearpig92, 2023-04-09:
I have this IR blaster:
https://www.aliexpress.com/item/1005003813684787.html?spm=a2g0o.order_list.order_list_main.15.575d1802a2M0rC

I am unable to get IR learn and IR send Clusters in any of these drop down menu's. How do I get this? I'm running the newest Home assistant update with a sky connect
[image]

Manbearpig92, 2023-04-11:
@ranton1990 do you have a link to the one you have?

Manbearpig92, 2023-04-12:
@ferehcarb I checked all in those drop down menus and I couldn't find it

Manbearpig92, 2023-04-12:
@ferehcarb 

I am unfortunately unable to get to the physical device and it's found on my network but unable to connect. I can access the logs but do you just need the IEEE and NWK?

ferehcarb, 2023-04-12:
> @ferehcarb
> 
> I am unfortunately unable to get to the physical device and it's found on my network but unable to connect. I can access the logs but do you just need the IEEE and NWK?

No, I need the manufacturer code, like [here](https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py#L420) and [here](https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py#L481) .
If your manufacturer code is not in the list, it won't work.
The code is in the signature tab:
[image]

ranton1990, 2023-04-12:
> @ranton1990 do you have a link to the one you have?

Should be this one: 
https://de.aliexpress.com/item/1005005230661089.html

antfurn, 2023-04-16:
Hi, does anyone know why if there a reason why the zha_toolkit : attr_read command can read the 'learned' IR code:

```This error originated from a custom integration.

Logger: custom_components.zha_toolkit.zcl_attr
Source: custom_components/zha_toolkit/zcl_attr.py:303
Integration: 🧰 ZHA Toolkit (documentation, issues)
First occurred: 22:23:48 (2 occurrences)
Last logged: 22:28:14

Result: [ReadAttributeRecord(attrid=0x0000)] - Attribute 0 not in read ({0: 'DUAjlBEoAigCXQJ9Bl0CQAdAA8ALQAcDfQYoAoABgAuAD8ALAV0CgAsBXQKAAwUoAigCXQLAAwR9Bl0CKGABQAeAAwJ9BijgAAMBKAJADwldAt6cQCO5CCgC'}, [ReadAttributeRecord(attrid=0x0000)])```

More details in the issue I raised here: https://github.com/mdeweerd/zha-toolkit/issues/157

ferehcarb, 2023-04-17:
> Hi, does anyone know if there a reason why the zha_toolkit : attr_read command can't read the 'learned' IR code:
> ....
> 
> More details in the issue I raised here: [mdeweerd/zha-toolkit#157](https://github.com/mdeweerd/zha-toolkit/issues/157)

Hi, I think the answer is in this [comment](https://github.com/mdeweerd/zha-toolkit/issues/157#issuecomment-1510518025), if you test the change and it's ok, tell it me, I'll update the code.

antfurn, 2023-04-17:
Yes, [mdeweerd](https://github.com/mdeweerd) suggested change works for me.

[image]

ferehcarb, 2023-04-18:
> Yes, [mdeweerd](https://github.com/mdeweerd) suggested change works for me.

Quirk updated, thank you for the report.

Manbearpig92, 2023-04-21:
> No, I need the manufacturer code, like [here](https://github.com/ferehcarb/zha-device-handlers/blob/dev/zhaquirks/tuya/ts1201.py#L420) and [...]

  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xf000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe004",
        "0xed00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3290_acv1iuslxi3shaaj",
  "model": "TS1201",
  "class": "zigpy.device.Device

ferehcarb, 2023-04-22:
@Manbearpig92 quirk updated, give it a try.

LeVraiRoiDHyrule, 2023-05-03:
Hi, I just got a Zigbee IR blaster but I'm lost with all these messages. I have a TS1201 too. In "Manage Zigbee Device", all I have is this:
[image]
[image]

I can't find how to get IR controls to appear. I was using Zigbee2MQTT before and it was working. Could someone help me please ? How can I add more "clusters" ?

ferehcarb, 2023-05-04:
> I can't find how to get IR controls to appear. I was using Zigbee2MQTT before and it was working. Could someone help me please ? How can I add more "clusters" ?

Hi, in the signature you've got: "class": "zigpy.device.Device", it should be "class": "ts1201.ZosungIRBlaster"
I think the custom quirk is not used. Have you configured the path for custom quirks and put the file in the right place (ts1201.py) ?
Something like this in configuration.yaml:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

LeVraiRoiDHyrule, 2023-05-06:
Thanks a lot for the explaination, I've just added the file, and rebooted.

I now have the following in my list:
[image]

But I don't have any ZosungIRBlaster, is that normal ? I've choosed ZosungIRControl for now.
Is there further configuration to do ?

ferehcarb, 2023-05-07:
> Ok apparently ts1201.ZosungIRBlaster quirk is correctly applied. Thanks a lot. But is there more things to do to have controls from the HA UI ? I would like to be able to send codes from a button on HA, or being able to send codes within automations.

Read the thread, you've got an example in [this comment](https://github.com/zigpy/zha-device-handlers/issues/1687#issuecomment-1424510023).

[77 more replies]

PR #2336 (2024-08-27): Add support for Tuya TS1201 IR blaster
TS1201 is an IR blaster, this quirk is able to send and to receive raw IR codes using Zigbee commands.
Received IR code is available in a Zigbee attribute.
Device is also known as:
- Moes UFO-R11
- Aubess ZXZIR-02

Should fix #1687 fix #1782 and fix #1955
closed 2024-08-27 by TheJulianJES