zigpy/zha-device-handlers #1688: [Device Support Request] SM-SO306E-1Z 4AC + 2USB Power Strip

yruss972, 2022-08-09:
**Is your feature request related to a problem? Please describe.**
Device controls do not work after pairing.

**Describe the solution you'd like**
Please add support for the device.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
[... 736 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

```

</details>

**Additional context**
The attempted quirk matched but didn't solve the problems.

javicalle, 2022-08-09:
The proposed quirk looks good to me.
It would create 5 switches in HA.
What is the behavior? I guess you can't control them from HA, but does the status in HA get updated if you turn them on from the device?
Any error in the logs?

yruss972, 2022-08-09:
With or without the quirk, HA creates 5 switches for the device but they don't have any effect.
HA doesn't reflect the state of the outlets if I toggle them directly on the device.
I've since tried using EnchantedDevice instead of CustomDevice in the quirk but there doesn't seem to be any improvement.

There are errors in the log, ie:
```Error handling '_save_attribute' event with (a4:c1:38:7b:49:01:46:95, 1, 0, 4, '_TZ3000_cfnprab5') params: FOREIGN KEY constraint failed```
```Error handling '_save_attribute' event with (a4:c1:38:7b:49:01:46:95, 1, 0, 5, 'TS011F') params: FOREIGN KEY constraint failed```
```
[0xf79d] Delivery error for seq # 0xbe, on endpoint id 0 cluster 0x0021: message send failure
[0xF79D:3:0x0006]: Failed to bind 'on_off' cluster: [0xf79d:0:0x0021]: Message send failure
```
I've attached a sample of each log (tried to clean them up so it's only the relevant device)
[EnchantedDeviceLog.txt](https://github.com/zigpy/zha-device-handlers/files/9294980/EnchantedDeviceLog.txt)
[CustomDeviceLog.txt](https://github.com/zigpy/zha-device-handlers/files/9295010/CustomDeviceLog.txt)

Note: 
1. The device keeps rejoining the network. I don't know if this is expected or not.
2. In my trials, I did manage to temporarily gain control of the outlets via HA while the pairing process was in progress :?

Please lmk what I can try to help get this working :)

yruss972, 2022-08-11:
Would it be helpful to temporarily bring up a clean HA instance with debug logging enabled and provide the log of the pairing?
Alternatively, if I could find a Tuya GW and connect the device to it, is there any information I could provide from their developer portal which would help?

javicalle, 2022-08-11:
> here are errors in the log, ie: `Error handling '_save_attribute' event with (a4:c1:38:7b:49:01:46:95, 1, 0, 4, '_TZ3000_cfnprab5') params: FOREIGN KEY constraint failed` `Error handling [...]
These messages are normal when you try to 'add' a device that has been already registered.

> ```
> [0xf79d] Delivery error for seq # 0xbe, on endpoint id 0 cluster 0x0021: message send failure
> [0xF79D:3:0x0006]: Failed to bind 'on_off' cluster: [0xf79d:0:0x0021]: Message send failure
> ```
> 
Not sure about these others.

I don't think that HA logs could get more info. Maybe sniffing the pairing proccess with a Tuya gateway could give more info, but I'm not the ona capable to analize it 🤷🏻‍♂️

javicalle, 2022-08-11:
Duplicate of #1632

Could continue the conversation there?

javicalle, 2022-11-27:
Duplicated

closed 2022-11-27 by javicalle