zigpy/zha-device-handlers #1689: [Device Support Request] Smartwings WM25/L-Z

warllo54, 2022-08-10:
**Is your feature request related to a problem? Please describe.**

Pressing the open and close buttons in the UI will not open nor close the blinds however using the position slider you can adjust the blinds. The blinds are also inverted, open is closed and closed is open

**Describe the solution you'd like**

I would like the open and close buttons to work, for the state to not be inverted.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Smartwings",
  "model": "WM25/L-Z",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.8.0",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "4.2.8",
    "run_as_root": true
  },
  "custom_components": {
    "alexa_media": {
      "version": "4.1.0",
      "requirements": [
        "alexapy==1.26.1",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
[... 361 lines cut ...]
2022-08-10 12:03:32.133 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x86 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x41ea endpoint=1>, status: 0x00
2022-08-10 12:03:32.156 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.masterbedroomsmartplug_value_electric_consumed_2, old_state=<state sensor.masterbedroomsmartplug_value_electric_consumed_2=0.0; state_class=measurement, meter_type=1, meter_type_name=ELECTRIC, unit_of_measurement=W, device_class=power, friendly_name=MasterBedroomSmartPlug: Value (Electric, Consumed) @ 2022-08-10T12:03:31.992250-05:00>, new_state=<state sensor.masterbedroomsmartplug_value_electric_consumed_2=0.0; state_class=measurement, meter_type=1, meter_type_name=ELECTRIC, unit_of_measurement=W, device_class=power, friendly_name=MasterBedroomSmartPlug: Value (Electric, Consumed) @ 2022-08-10T12:03:32.155892-05:00>>
2022-08-10 12:03:32.570 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-10 12:03:32.570 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-10 12:03:32.572 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x41ea>, 1, 260, 1, b'\x08Z\n!\x00 U', 0, 175, 239, 190, 106, 252, 0, -64]
2022-08-10 12:03:32.572 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Received ZCL frame: b'\x08Z\n!\x00 U'
2022-08-10 12:03:32.573 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=90, command_id=10, *is_reply=True)
2022-08-10 12:03:32.574 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Decoded ZCL frame: PowerConfiguration:Report_Attributes(attribute_reports=[Attribute(attrid=0x0021, value=TypeValue(type=uint8_t, value=85))])
2022-08-10 12:03:32.574 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Received command 0x0A (TSN 90): Report_Attributes(attribute_reports=[Attribute(attrid=0x0021, value=TypeValue(type=uint8_t, value=85))])
2022-08-10 12:03:32.574 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Attribute report received: battery_percentage_remaining=85
2022-08-10 12:03:32.575 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.smartwings_wm25_l_z_battery, old_state=<state sensor.smartwings_wm25_l_z_battery=43; state_class=measurement, battery_voltage=0.0, unit_of_measurement=%, device_class=battery, friendly_name=Smartwings WM25/L-Z Battery @ 2022-08-10T11:02:43.488219-05:00>, new_state=<state sensor.smartwings_wm25_l_z_battery=42; state_class=measurement, battery_voltage=0.0, unit_of_measurement=%, device_class=battery, friendly_name=Smartwings WM25/L-Z Battery @ 2022-08-10T12:03:32.575158-05:00>>
2022-08-10 12:03:32.576 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=90, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-08-10 12:03:32.576 DEBUG (MainThread) [zigpy.zcl] [0x41EA:1:0x0001] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-08-10 12:03:32.578 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 90 under 135 request id, data: b'185a0b0a00'
2022-08-10 12:03:32.579 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x41ea>, ep: 1, profile: 0x0104, cluster_id: 0x0001, data: b'085a0a21002055'
2022-08-10 12:03:32.579 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 135, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x41EA endpoint=1>, 260, 1, 1, b'\x18Z\x0b\n\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-10 12:03:32.582 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 135]
2022-08-10 12:03:32.590 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-08-10 12:03:32.590 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-08-10 12:03:32.592 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 135: 00
```

</details>

**Additional context**

warllo54, 2022-08-11:
Proposed Quirk

```
"""Device handler for Smartwings WM25L-Z roller blinds."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    Ota,
    PowerConfiguration,
    Scenes,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class InvertedWindowCoveringCluster(CustomCluster, WindowCovering):
    """WindowCovering cluster implementation.
    This implementation inverts the reported covering percent for non standard
    devices that don't follow the reporting spec.
    """

    cluster_id = WindowCovering.cluster_id
    CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008

    def _update_attribute(self, attrid, value):
        if attrid == self.CURRENT_POSITION_LIFT_PERCENTAGE:
            value = 100 - value
        super()._update_attribute(attrid, value)

class WM25LBlinds(CustomDevice):
    """Custom device representing Smartwings WM25/L-Z roller blinds."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=514
        # device_version=1
        # input_clusters=[0, 1, 3, 4, 5, 32, 258]
        # output_clusters=[3, 25]>
        MODELS_INFO: [
            ("Smartwings", "WM25/L-Z"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    InvertedWindowCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        }
    }
```

javicalle, 2022-08-11:
Can you please try to get this attribute values from the `InvertedWindowCoveringCluster`:
* 0x0007 'config_status'
* 0x0017 'window_covering_mode'

Thanks in advance.

warllo54, 2022-08-11:
Thanks for your response! I really appreciate the help. Here's the requested info.
- 0x0007 | ConfigStatus.Online|Operational

- 0x0017 | WindowCoveringMode.16|Motor_in_maintenance_mode

javicalle, 2022-08-11:
I think that (in general) all the quirks to invert the cover behavior is nonsense, since the operation of each cover will depend on how the motor has been installed.
For this reason all the covers have a mechanism to reverse the direction of rotation of the motor (normally with the remote).

If that's okay with you, I'd like to suggest trying to reverse the direction of rotation of the blind without using a quirk.
To do this, it would be necessary to remove the quirk from your installation (clear the cache and reboot) and try to assign a new value to the `window_covering_mode` attribute.
The value to assign I think would be `1`

I have no device to test and no experience with zigbee cover devices, but IMO there must be a better way to manage all that inverted cover issues.

warllo54, 2022-08-11:
I will certainly give this a try however without the quirk for some reason the cover-open and close commands don't work. You can control the blinds via percentage though.  I also noticed that Home Assistant shows the battery percentage as 42% while reading the percentage remaining attribute shows 84%, which I'm guessing would need to be fixed with a quirk. I just don't understand why manufactures won't stick the the Zigbee spec...

javicalle, 2022-08-11:
> I also noticed that Home Assistant shows the battery percentage as 42% while reading the percentage remaining attribute shows 84%, which I'm guessing would need to be fixed with a quirk.

It is common in Tuya devices to have to fix the battery value. You can use the `DoublingPowerConfigurationCluster` for it.

Probably the quirk will be need anyway, but maybe the inverted part can be addressed in a better way. 👍🏻

warllo54, 2022-08-11:
After disabling the quirk and setting window_covering_mode attribute to 1 the state is no longer inverted. However the open and close buttons no longer provide control of the shade. Setting the position using the slider does control the shades. Any suggestion on making the cover.cover_close and cover.cover_open services work? Once again thank you very much for the suggestions.

I did re-write the quirk to only correct the battery information and that is working perfectly.

javicalle, 2022-08-12:
Could you share with us the values of the attributes again? 

> Any suggestion on making the cover.cover_close and cover.cover_open services work?

Ummm, could you try to add an attribute to your cover?
* `assumed_state: true`

That can be done from the 'developers view' selecting the cover and adding to the device's attributes.

Meanwhile I'll check how can be fixed.

javicalle, 2022-08-12:
I'm not sure why are the controls disabled.
Could you share the values from your device?:
* from the WindowCovering cluster attribute: `current_position_lift_percentage`
* from the HA device status (from the developer tools view): `state` and `current_position`

Are some of this values changing when you operate the cover with the slider?

warllo54, 2022-08-12:
from the WindowCovering cluster attribute: ```current_position_lift_percentage:0```

From Dev Tools View: 
```
state: open
current_position: 100
friendly_name: MasterBedroomBlinds
supported_features: 15
```

The values only change when operating the cover with the slider. When using the slider the blinds function as expected. When using the open and close button the state will switch to opening or closing but the blinds don't perform any action.

javicalle, 2022-08-12:
🤦🏻‍♂️ I had misunderstood the problem. So the buttons are _enabled_ in HA, but when you press them they only work when the quirk is configured, is that correct?

Could you attach the HA logs when you press the button in HA with the quirk and without the quirk (i.e. when it works and when it doesn't work).

warllo54, 2022-08-12:
You're correct.  The buttons are _enabled_ in HA, but only work when pressed if the quirk is configured with the inverted blind code as shown.
```
    def _update_attribute(self, attrid, value):
        if attrid == self.CURRENT_POSITION_LIFT_PERCENTAGE:
            value = 100 - value
        super()._update_attribute(attrid, value)
```
I find this strange because after setting the ```window_covering_mode``` to 1 as suggested, the state in HA developer tab is no longer inverted, even without the quirk.

I believe the quirk will still be needed to double the battery percentage  but one step at a time :smiley:

I won't be home until tomorrow so I won't be able to get you the log info until then.

Thanks again!

warllo54, 2022-08-13:
Logs without quirk

```
2022-08-13 12:06:39.934 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5389:1:0x0020]: executed 'set_long_poll_interval' command with args: '(24,)' kwargs: '{}' result: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-08-13 12:06:39.935 DEBUG (MainThread) [zigpy.zcl] [0x5389:1:0x0020] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=25, command_id=1, *is_reply=False)
2022-08-13 12:06:39.935 DEBUG (MainThread) [zigpy.zcl] [0x5389:1:0x0020] Sending request: fast_poll_stop()
2022-08-13 12:06:39.935 DEBUG (MainThread) [zigpy.device] [0x5389] Extending timeout for 0x19 request
2022-08-13 12:06:39.936 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 25 under 26 request id, data: b'011901'
2022-08-13 12:06:39.936 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 26, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x5389 endpoint=1>, 260, 32, 1, b'\x01\x19\x01', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-13 12:06:39.941 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 26]
2022-08-13 12:06:39.948 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-08-13 12:06:39.949 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-08-13 12:06:39.951 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 26: 00
2022-08-13 12:06:39.952 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x1a 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x5389 endpoint=1>, status: 0x00
2022-08-13 12:06:40.271 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=weather.thermostat, old_state=<state weather.thermostat=partlycloudy; temperature=69, temperature_unit=°F, humidity=81, pressure=30.0, pressure_unit=inHg, wind_bearing=38, wind_speed=15.66, wind_speed_unit=mph, visibility=9.94, visibility_unit=mi, precipitation_unit=in, forecast=[{'condition': 'partlycloudy', 'wind_bearing': 38, 'datetime': '2022-08-13T17:06:10.270642+00:00', 'temperature': 78, 'templow': 58, 'wind_speed': 15.66}, {'condition': 'partlycloudy', 'wind_bearing': 152, 'datetime': '2022-08-14T17:06:10.270642+00:00', 'temperature': 82, 'templow': 55, 'wind_speed': 6.71}, {'condition': 'partlycloudy', 'wind_bearing': 157, 'datetime': '2022-08-15T17:06:10.270642+00:00', 'temperature': 82, 'templow': 57, 'wind_speed': 13.42}, {'condition': 'rainy', 'wind_bearing': 158, 'datetime': '2022-08-16T17:06:10.270642+00:00', 'temperature': 78, 'templow': 60, 'wind_speed': 13.42}, {'condition': 'partlycloudy', 'wind_bearing': 186, 'datetime': '2022-08-17T17:06:10.270642+00:00', 'temperature': 85, 'templow': 62, 'wind_speed': 15.66}], attribution=Ecobee weather provided by FI:727500 at 2022-08-13 16:32:14 UTC, friendly_name=Thermostat @ 2022-08-13T11:49:39.831653-05:00>, new_state=<state weather.thermostat=partlycloudy; temperature=69, temperature_unit=°F, humidity=81, pressure=30.0, pressure_unit=inHg, wind_bearing=38, wind_speed=15.66, wind_speed_unit=mph, visibility=9.94, visibility_unit=mi, precipitation_unit=in, forecast=[{'condition': 'partlycloudy', 'wind_bearing': 38, 'datetime': '2022-08-13T17:06:40.271361+00:00', 'temperature': 78, 'templow': 58, 'wind_speed': 15.66}, {'condition': 'partlycloudy', 'wind_bearing': 152, 'datetime': '2022-08-14T17:06:40.271361+00:00', 'temperature': 82, 'templow': 55, 'wind_speed': 6.71}, {'condition': 'partlycloudy', 'wind_bearing': 157, 'datetime': '2022-08-15T17:06:40.271361+00:00', 'temperature': 82, 'templow': 57, 'wind_speed': 13.42}, {'condition': 'rainy', 'wind_bearing': 158, 'datetime': '2022-08-16T17:06:40.271361+00:00', 'temperature': 78, 'templow': 60, 'wind_speed': 13.42}, {'condition': 'partlycloudy', 'wind_bearing': 186, 'datetime': '2022-08-17T17:06:40.271361+00:00', 'temperature': 85, 'templow': 62, 'wind_speed': 15.66}], attribution=Ecobee weather provided by FI:727500 at 2022-08-13 16:32:14 UTC, friendly_name=Thermostat @ 2022-08-13T11:49:39.831653-05:00>>
2022-08-13 12:06:40.420 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-13 12:06:40.421 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-13 12:06:40.423 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5389>, 1, 260, 32, b'\x08\x19\x0b\x01\x00', 0, 175, 255, 227, 21, 253, 0, -62]
2022-08-13 12:06:40.424 DEBUG (MainThread) [zigpy.zcl] [0x5389:1:0x0020] Received ZCL frame: b'\x08\x19\x0b\x01\x00'
2022-08-13 12:06:40.425 DEBUG (MainThread) [zigpy.zcl] [0x5389:1:0x0020] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_gen [... cut ...] 13T12:08:22.011505-05:00>>
2022-08-13 12:08:22.012 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.foyer_detection_fps, old_state=<state sensor.foyer_detection_fps=5; unit_of_measurement=fps, icon=mdi:speedometer, friendly_name=Foyer Detection Fps @ 2022-08-13T12:08:12.015775-05:00>, new_state=<state sensor.foyer_detection_fps=3; unit_of_measurement=fps, icon=mdi:speedometer, friendly_name=Foyer Detection Fps @ 2022-08-13T12:08:22.012121-05:00>>
```

warllo54, 2022-08-13:
Logs with quirk enabled

```
2022-08-13 12:20:05.545 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1dff>, ep: 1, profile: 0x0104, cluster_id: 0x0300, data: b'083d0a0300218d37040021b6130700217201'
2022-08-13 12:20:05.546 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 143, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x1DFF endpoint=1>, 260, 768, 1, b'\x18=\x0b\n\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-13 12:20:05.548 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 143]
2022-08-13 12:20:05.557 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-08-13 12:20:05.558 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-08-13 12:20:05.559 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 143: 00
2022-08-13 12:20:05.560 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x8f 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x1dff endpoint=1>, status: 0x00
2022-08-13 12:20:05.583 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.masterbedroomsmartplug_value_electric_consumed_2, old_state=<state sensor.masterbedroomsmartplug_value_electric_consumed_2=0.0; state_class=measurement, meter_type=1, meter_type_name=ELECTRIC, unit_of_measurement=W, device_class=power, friendly_name=MasterBedroomSmartPlug: Value (Electric, Consumed) @ 2022-08-13T12:20:05.419363-05:00>, new_state=<state sensor.masterbedroomsmartplug_value_electric_consumed_2=0.0; state_class=measurement, meter_type=1, meter_type_name=ELECTRIC, unit_of_measurement=W, device_class=power, friendly_name=MasterBedroomSmartPlug: Value (Electric, Consumed) @ 2022-08-13T12:20:05.583318-05:00>>
2022-08-13 12:20:06.294 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA14F](RT RGBW): Device seen - marking the device available and resetting counter
2022-08-13 12:20:06.295 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA14F](RT RGBW): Update device availability -  device available: True - new availability: True - changed: False
2022-08-13 12:20:16.944 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-13 12:20:16.945 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-13 12:20:16.947 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xa14f>, 1, 260, 8, b'\x08N\n\x00\x00 \x07', 0, 175, 223, 211, 22, 253, 0, -66]
2022-08-13 12:20:16.948 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Received ZCL frame: b'\x08N\n\x00\x00 \x07'
2022-08-13 12:20:16.949 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=78, command_id=10, *is_reply=True)
2022-08-13 12:20:16.949 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Decoded ZCL frame: LevelControl:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=7))])
2022-08-13 12:20:16.950 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Received command 0x0A (TSN 78): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=7))])
2022-08-13 12:20:16.950 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Attribute report received: current_level=7
2022-08-13 12:20:16.950 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xA14F:1:0x0008]: received attribute: 0 update with value: 7
2022-08-13 12:20:16.951 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=78, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-08-13 12:20:16.951 DEBUG (MainThread) [zigpy.zcl] [0xA14F:1:0x0008] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-08-13 12:20:16.952 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 78 under 144 request id, data: b'184e0b0a00'
2022-08-13 12:20:16.952 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xa14f>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'084e0a00002007'
2022-08-13 12:20:16.952 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 144, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0xA14F endpoint=1>, 260, 8, 1, b'\x18N\x0b\n\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
[... cut ...]
2022-08-13 12:21:14.344 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x869B](E11-G13): Update device availability -  device available: True - new availability: True - changed: False
2022-08-13 12:21:18.359 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF169](TRADFRI bulb GU10 WW 400lm): Device seen - marking the device available and resetting counter
2022-08-13 12:21:18.359 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF169](TRADFRI bulb GU10 WW 400lm): Update device availability -  device available: True - new availability: True - changed: False
```

javicalle, 2022-08-13:
Not analyzed the logs yet but to be all the info clear:
* the quirk that you are testing, have the inverted code implemented?
* my guess is that the device must keep the `window_covering_mode` between HA restarts. Maybe get lost if you repair the device. Can you confirm? But you say something about the 'developer tab', could you explain this point?
* without the quirk the open/close buttons aren't working at all, correct? Is not a matter that cover is open and controls inverted then the close button (that will trigger an open command) is not responding. You have tried both buttons and aren't responding

javicalle, 2022-08-13:
To future review (`UNSUP_CLUSTER_COMMAND` ¿11?):
```
2022-08-13 12:20:41.341 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Received ZCL frame: b'\x08\x92\x0b\x00\x00'
2022-08-13 12:20:41.341 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=146, command_id=11, *is_reply=True)
2022-08-13 12:20:41.342 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Decoded ZCL frame: InvertedWindowCoveringCluster:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-08-13 12:20:41.344 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x8279>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08920b0000'
2022-08-13 12:20:41.345 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8279:1:0x0102]: executed 'up_open' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-08-13 12:20:41.345 DEBUG (MainThread) [homeassistant.components.zha.cover] state=opening
2022-08-13 12:20:41.345 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.smartwings_wm25_l_z_cover, old_state=<state cover.smartwings_wm25_l_z_cover=closed; current_position=0, friendly_name=Smartwings WM25/L-Z Cover, supported_features=15 @ 2022-08-13T12:15:35.370407-05:00>, new_state=<state cover.smartwings_wm25_l_z_cover=opening; current_position=0, friendly_name=Smartwings WM25/L-Z Cover, supported_features=15 @ 2022-08-13T12:20:41.345868-05:00>>
2022-08-13 12:20:41.352 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-13 12:20:41.353 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-13 12:20:41.355 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x8279>, 1, 260, 258, b'\x08\x92\x0b\x00\x81', 0, 175, 127, 214, 22, 253, 0, -78]
2022-08-13 12:20:41.356 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Received ZCL frame: b'\x08\x92\x0b\x00\x81'
2022-08-13 12:20:41.356 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=146, command_id=11, *is_reply=True)
2022-08-13 12:20:41.357 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Decoded ZCL frame: InvertedWindowCoveringCluster:Default_Response(command_id=0, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
2022-08-13 12:20:41.357 DEBUG (MainThread) [zigpy.zcl] [0x8279:1:0x0102] Received command 0x0B (TSN 146): Default_Response(command_id=0, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
2022-08-13 12:20:41.360 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x8279>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08920b0081'
```

warllo54, 2022-08-13:
- Yes, the quirk has the inverted code implemented. 

<details>
<summary>Quirk as implemented</summary>

```
"""Device handler for Smartwings WM25L-Z roller blinds."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    Ota,
    PowerConfiguration,
    Scenes,
)

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class InvertedWindowCoveringCluster(CustomCluster, WindowCovering):
    """WindowCovering cluster implementation.
    This implementation inverts the reported covering percent for non standard
    devices that don't follow the reporting spec.
    """

    cluster_id = WindowCovering.cluster_id
    CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008

    def _update_attribute(self, attrid, value):
        if attrid == self.CURRENT_POSITION_LIFT_PERCENTAGE:
            value = 100 - value
        super()._update_attribute(attrid, value)

class WM25LBlinds(CustomDevice):
    """Custom device representing Smartwings WM25/L-Z roller blinds."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=514
        # device_version=1
        # input_clusters=[0, 1, 3, 4, 5, 32, 258]
        # output_clusters=[3, 25]>
        MODELS_INFO: [
            ("Smartwings", "WM25/L-Z"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
#                    WindowCovering.cluster_id,
                    InvertedWindowCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        }
    }
```
</details>

- Yes, the window covering mode _persists_ through ha reboots. If I reconfigure use the reconfigure button on the device the ```window_covering_mode retains the WindowCoveringMode.Motor_direction_reversed```
 
- Without the quirk, if the blinds are _physically_ open, the state in Home Assistant is Open and pressing either the UP or Down button has no effect. 
 
- Without the quirk enabled If the state of the blinds in Home Assistant is open and the physical state of the blind is open, calling the cover.open_cover service from the developer tools services page will physically close the blinds.
 
- With the quirk enabled, if the blinds are _physically_ **OPEN** the state in Home Assistant is _**closed**_ and pressing the **UP** button physically closes the blinds.

D3v01dZA, 2022-08-23:
Thought I'd chime in here, I just got a set of these today.

Physically open reads as 100% in home assistant.
Open/close is reversed, opening physically closes them.

Without assumed_state I can't actually control them from the default more-info page since the close actually physically opens them and the close button is disabled.

warllo54, 2022-08-24:
I reset mine so the included remotes controls are backwards but the state and open and close buttons now work.

D3v01dZA, 2022-08-25:
What do you mean by reset? Did you have to reset it in a special way to get it to reverse?

warllo54, 2022-08-25:
When I received my blinds, the included remote controls was working correctly, however the state in HomeAssistant was inverted. I factory reset the blinds, holding the button on the blinds for 14 seconds and the controls were inverted on the included remote however the state and controls are correct in HomeAssistant. In the included remote programming guide it talks about reversing the motor direction.

D3v01dZA, 2022-08-25:
Thanks, this fixes it!

tube0013, 2022-10-02:
I just got these blinds. I used the quirk above and the remote to invert the motor and all is good. Though I have the battery on one plugged in and it's not showing any increase in charge after like 10 hours but thats not a quirk issue.

It it probably worth making a pr with this code.

haywiremk, 2023-02-08:
ARRGGHHHH.  I want to say ZHA is going to always be sub-par to zigbee2mqtt for HA users with the HA attitude towards documentation around quirks.  Make it easy for community to fill in the gap until an official quirk is merged in. Best yet make this more UI friendly!  Thanks all here for starting me in the right direction.  Documenting what it took to get SmartWings in ZHA so other can just cut and paste.

I do have one help request.  I notice sometime the blinds stop responding and it takes a ZHA reload to make that one blind work again.  Any suggestions would be helpful.

First to use custom quirks add to configuration.yaml
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks  #any folder you want

I have blinds with motor on the right and unspooling against the window (us solar users and wise people to get the charge port pointed indoors).  To keep the remote in the correct up/down orientation and make this integration work here is the quirk to use.  Working for me at least. Seems the up/down commands are flipped when this way

`"""Device handler for Smartwings WM25L-Z roller blinds."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    Ota,
    PowerConfiguration,
    Scenes,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class InvertedWindowCoveringCluster(CustomCluster, WindowCovering):
    """WindowCovering cluster implementation.
    This implementation inverts the reported covering percent for non standard
    devices that don't follow the reporting spec.
    """

    cluster_id = WindowCovering.cluster_id
    CMD_GO_UP = 0x0001 #reported as down_close in ZHA commands
    CMD_GO_DOWN = 0x0000 #reported as up_open in ZHA commands

    async def command(
        self, command_id, *args, manufacturer=None, expect_reply=True, tsn=None
    ):
        """Override default commands for up and down. they are backwards."""
        if command_id == self.CMD_GO_UP:
            return await super().command(self.CMD_GO_DOWN, *args)
        if command_id == self.CMD_GO_DOWN:
            return await super().command(self.CMD_GO_UP, *args)
        return await super().command(
            command_id,
            *args,
            manufacturer=manufacturer,
            expect_reply=expect_reply,
            tsn=tsn
        )

class WM25LBlinds(CustomDevice):
    """Custom device representing Smartwings WM25/L-Z roller blinds."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=514
        # device_version=1
        # input_clusters=[0, 1, 3, 4, 5, 32, 258]
        # output_clusters=[3, 25]>
        MODELS_INFO: [
            ("Smartwings", "WM25/L-Z"),
        ],
[... 114 lines cut ...]
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    InvertedWindowCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        }
    }`

So use the one for the correct remote operation/blind orientation.  Its wine-o-clock so will look at if the battery is wrong in the next week and update this so.  What makes HA awesome the community support.  Just wish the team around quirks understood this after reading some rude maintainer comments.  Hunting down how to change commands was a PITA as the repo has ZERO examples.

tube0013, 2023-02-23:
@haywiremk this is working well for me, and got the slider to make more sense! thanks.

the battery should work with just one minor change to what you have above.  The original endpoints shouldn't have the DoublingPowerConfigurationCluster.

Do you want to convert this into a PR? can likely make it into the next HA release. If not I can do i it, just let me know.

```
"""Device handler for Smartwings WM25L-Z roller blinds."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    Ota,
    PowerConfiguration,
    Scenes,
)
from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class InvertedWindowCoveringCluster(CustomCluster, WindowCovering):
    """WindowCovering cluster implementation.
    This implementation inverts the reported covering percent for non standard
    devices that don't follow the reporting spec.
    """

    cluster_id = WindowCovering.cluster_id
    CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008
    CMD_GO_TO_LIFT_PERCENTAGE = 0x0005

    def _update_attribute(self, attrid, value):
        if attrid == self.CURRENT_POSITION_LIFT_PERCENTAGE:
            value = 100 - value
        super()._update_attribute(attrid, value)
    
    async def command(
        self, command_id, *args, manufacturer=None, expect_reply=True, tsn=None
    ):
        """Override default command to invert percent lift value."""
        if command_id == self.CMD_GO_TO_LIFT_PERCENTAGE:
            percent = args[0]
            # Invert the percentage value 
            percent = 100 - percent
            v = (percent,)
            return await super().command(command_id, *v)
        return await super().command(
            command_id,
            *args,
            manufacturer=manufacturer,
            expect_reply=expect_reply,
            tsn=tsn
        )

class WM25LBlinds(CustomDevice):
    """Custom device representing Smartwings WM25/L-Z roller blinds."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=514
        # device_version=1
        # input_clusters=[0, 1, 3, 4, 5, 32, 258]
        # output_clusters=[3, 25]>
        MODELS_INFO: [
            ("Smartwings", "WM25/L-Z"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    InvertedWindowCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        }
    }
```

haywiremk, 2023-02-23:
Go ahead and submit the PR.  Good catch on the initial endpoints.  I think we way it scraps just works but wrong.  Not much docs around quirks and dug this out of another quirk.  One final thing is there likely needs to be TWO versions.  I have a right mount motor(best for solar) with roll backside and just use the command flip version with my remotes being proper up and down.  I think if you have a left mount motor with rollbackside the quirk could be backwards.  The other two frontsides will just be one of these.  Not sure if the remote flip after a reset fixes that so one quirk will always work.  Likely need someone with leftside-rollback to test or just leave a link to this ticket as a comment in the quirk for the lefties.  Always those lefties!

haywiremk, 2023-02-24:
Please see my first script that inverts the up/down commands vs flipping all the statuses and setting the remote backwards. I am using that one to keep the remote correct.

tube0013, 2023-02-24:
I'll reset my motor to default and test. I had it inverted for the previous quirk , switched to the last one you posted and all worked the same with added bonusthe slider was fixed to not be backwards.

tube0013, 2023-03-04:
I played around with my shades today I set the motor direction back to default with the remote and ended up removing the quirk completely for some testing, and clearing py_cache in custom_quirks folder. the buttons seem to be functioning albeit it backwards and the slider works. battery is halved as expected.

Setting the window_covering_mode (id: 0x0017) attribute to 1 doesn't seem to do anything, buttons still reversed. I'm interested in trying the config_status (id: 0x0007) attribute as it has a inverted control option. but I'm stuck trying to figure out what to write as the attribute to enable that.

[image]

TheJulianJES, 2023-03-04:
Not sure what your ``config_status`` currently is, but the 7 bits are ordered like this: ``6543210``.
So just to enable bit 2, it's ``0000100`` and since the clusters UI (only?) does decimal, convert it: ``100`` in binary -> ``4`` in decimal. That would disable everything but bit 2. (You might want to enable more though)
Note: When reading, it might return it in a "nicer way", as zigpy does some type conversion.

(But read the attribute and save the output (here) before writing anything)

tube0013, 2023-03-04:
@TheJulianJES thank you,

so this is the log for what I get back by default:

```
2023-03-04 15:03:12.455 DEBUG (MainThread) [zigpy.zcl] [0x9D6C:1:0x0102] Decoded ZCL frame: WindowCovering:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0007, status=<Status.SUCCESS: 0>, value=TypeValue(type=bitmap8, value=<bitmap8.2|1: 3>))])
```

which in the UI shows as 

```

So what would I send to keep those which are bit0 and bit1 from the spec and also add the reversed commands for bit2
ConfigStatus.Online|Operational
```

tube0013, 2023-03-04:
FWIW It does seem that z2m just inverts it.  I did combine the 2 proposed quirks above that invert the commands and lift lift percentage and those did make the stock biind/shade function as expected. but if this could be done just by modifying some attributes I figured that would be more desired.

TheJulianJES, 2023-03-04:
Operational and online are bits 0 and 1. So that would be `11` in binary which is `3` in decimal.
To add bit 2 (reverse bit), it would be `111` in binary, so `7`` in decimal. So try to write `7` to that attribute.

tube0013, 2023-03-04:
thanks.

Doesn't seem to change anything. so I guess inverting in the quirk is best. let me put back in place and confirm all is working and I'll up date the PR.

tube0013, 2023-03-04:
PR updated. I think this should work with most smartwings so they work out of the box as expected.

closed 2023-03-12 by TheJulianJES

erodewald, 2024-08-15:
Hi all, I've got 5 of the Zigbee WM25/L-Z rollershades (and 1 of the Matter, but that's not relevant here). 3 of the 5 have been very flaky so in the process I've discovered ZHA quirks (awesome!) and have been learning how to install them and ensure they're being utilized in HA. Nonetheless, it hasn't seemed to help. These 3 all show the same issues: inverted controls vs status (so the only thing that works is setting a percentage). Despite using the quirk, I can't seem to force it to invert. 

I decided to reset them and re-join them to my SmartThings hub using [this SmartWings-endorsed Edge Driver](https://github.com/Mariano-Github/Edge-Drivers-Beta/blob/main/zigbee-window-treatment/fingerprints.yml). It works flawlessly here (and didn't, until I installed the custom Driver, by the way). 

I've reset it, re-joined it back to HA (with the quirk installed). Like before, the close/open commands do not work. The "Inverted" configuration toggle won't stay toggled either.  I'd really appreciate some help navigating this issue — the WAF is plummeting and I'm burning precious good will.

owen2, 2024-12-03:
It looks like new ones are shipping with different firmware, so the first set of these I bought were backwards and the quirk fixes them (firmware 0x0000002).

The second set I bought are not backwards, but and the quirk makes them backwards. The newer ones do not report their firmware version to Home Assistant. 

My workaround is to always use set position instead of open and close, because that works on both sets.

jobi, 2025-01-31:
> The second set I bought are not backwards, but and the quirk makes them backwards. The newer ones do not report their firmware version to Home Assistant.

I just bought one and they have the same issue, position is reported correctly, setting the position works correctly, but I need to use the open service to close them, the close service to open them

> My workaround is to always use set position instead of open and close, because that works on both sets.

Same, but it prevents using the standard tile

stovedoctor, 2025-01-31:
I switched to zigbee2mqtt for smart wings.

A few little tricks to get the up and down state correct. 
Not sure if this works with zha
But read the Alexa skill instructions in the manual to switch up and down directions in HA {not the remote up and down}
Using the remote programing

stovedoctor, 2025-01-31:
[image]

Alexa invert settings

PR #2220 (2023-03-12): Add Smartwings WM25L-Z quirk
Fixes: #1689 

Initial support for Smartwings blinds/shades - they all seem to use the same zigbee motor/variant. In some or all cases the the blinds need to be inverted with the included physical RF remote as per the instructions included with the blinds/shades.

Thanks to @warllo54 and @haywiremk for the actual work here, this was pulled for #1689 and it works for me 😀