zigpy/zha-device-handlers #1690: [Device Support Request]

kekonn, 2022-08-10:
**Is your feature request related to a problem? Please describe.**
The device is not supported yet, but it is available in [Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/LED2005R5.html).

**Describe the solution you'd like**
Support for dimming, on/off and colour temperature. 

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRIbulbGU10WS345lm",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.8.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Brussels",
    "os_name": "Linux",
    "os_version": "5.15.0-1012-raspi",
    "supervisor": "2022.08.3",
[... 537 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
```

</details>

**Additional context**

The entities attached to the lights are named really weird: 
[image]

dmulcahey, 2022-08-30:
What isn’t supported?

kekonn, 2022-08-30:
It's not so much unsupported as not displaying properly. See the labels in my screenshots. This is probably solved with assigning a quirk from a similar Tradfri LED to this one.

dmulcahey, 2022-08-30:
Quirks won’t affect those labels at all. That’s all in ZHA. Better names will be used in the future after we reconcile all of the HA entity naming changes.

closed 2022-08-30 by dmulcahey