zigpy/zha-device-handlers #1691: [Device Support Request] EcoDim ED-10013

JannickNijholt, 2022-08-11:
Hi! I bought the EcoDim ED-10013. This ZigBee device for turning lights on and off is already supported in [Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/ED-10013.html), but not yet in ZHA.
<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4644, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "EcoDim",
  "model": "ED-10013",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.7.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
[... 409 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
```

</details>

Thank you so much!

Dutchstyler, 2022-09-02:
Found this dim controller to and find this request. Just want to say that this support will be awesome 👌

MattWestb, 2022-09-02:
Its looks like one very nice Light Dimmer !!
What is not working in ZHA then all locks very Zigbee standard and not as Xiaomi or tuya ?

Z2M is using On, Off, brightness_move_up, brightness_move_down and brightness_stop for EP 1 and 2 that is normal Zigbee Light commands.

Use `Development tools` > `Events` and putting `zha_events` in Listening to events and you is knowing what command you can see the commands its sending  and using then in automatons.

JannickNijholt, 2022-09-02:
I managed to alter this Blueprint and make it work using ZHA: https://community.home-assistant.io/t/zha-ecodim-ed-10014-4-channel-remote/351853

The only thing is that dimming does not seem to work. A native integration inside of ZHA would be great.

MattWestb, 2022-09-02:
@JannickNijholt Is you getting `brightness_move_up`, `brightness_move_down` and `brightness_stop events` then listening to `zha_events` ?

Also try "reconfigure device" from device card (dont forgetting waking the device up then its one sleeper) then some controllers is not getting proper binding then joining  the system.

closed 2023-03-08 by github-actions[bot]

jerry2boz, 2023-10-04:
@JannickNijholt i'm new to Home Assistant, added ZHA and MQTT to HA with 2 ZigBee controllers. I managed to add the blueprint for the 8 button ECODim. But i don't succeed to connect the ecodim ED-10014 to MQTT. What have i do to, to make it appearing in MQTT?

jerry2boz, 2023-10-04:
@JannickNijholt  or make it possible with ZHA?