zigpy/zha-device-handlers #1692: [Device Support Request] Xiaomi Aqara E1 double key wireless switch

suvl, 2022-08-11:
As with zigbee2mqtt I was expecting this Aqara double switch to have at least two actions:

1. left click
2. right click
3. both click

double clicks of the three combinations, I believe, are also supported by the device, but I am not completely sure.

right now, the events that are emitted when the left key is pressed three times are these:
<details>
<summary>Current press events</summary>

```yaml
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:02:6e:bb:44",
        "unique_id": "00:15:8d:00:02:6e:bb:44:1:0x0006",
        "device_id": "0e6c85509d0f12d8c35b2fd2c76a8cd7",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 0,
            "attribute_name": "on_off",
            "value": 0
        },
        "params": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-11T11:41:30.029367+00:00",
    "context": {
        "id": "01GA6BB8HDMSRAQNS2GM71A8DQ",
        "parent_id": null,
        "user_id": null
    }
},

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:02:6e:bb:44",
        "unique_id": "00:15:8d:00:02:6e:bb:44:1:0x0006",
        "device_id": "0e6c85509d0f12d8c35b2fd2c76a8cd7",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 0,
            "attribute_name": "on_off",
            "value": 1
        },
        "params": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-11T11:41:30.029005+00:00",
    "context": {
        "id": "01GA6BB8HD4ZKD44ZD6G1MZSKB",
        "parent_id": null,
        "user_id": null
    }
}, 

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:02:6e:bb:44",
        "unique_id": "00:15:8d:00:02:6e:bb:44:1:0x0006",
        "device_id": "0e6c85509d0f12d8c35b2fd2c76a8cd7",
        "endpoint_id": 1,
[... 590 lines cut ...]
          },
          "0x0012": {
            "endpoint_attribute": "multistate_input",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0006": {
            "endpoint_attribute": "on_off",
            "attributes": {
              "0x0000": {
                "attribute_name": "on_off",
                "value": 0
              }
            },
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

**Additional context**
Link to the zigbee2mqtt implementation: [here](https://www.zigbee2mqtt.io/devices/WXKG17LM.html).

javicalle, 2022-08-18:
According to the code, this device supports short, long and double press for left, right and both:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/remote_b286acn01.py#L214..L224

What are the events that you miss?
Have you looked at the automations available for that device? It seems that the device already implement it and you don't have to handle the events.

suvl, 2022-08-18:
@javicalle as you can see from the issue, all I get event wise when I click any of the two buttons are these `attribute_updated` events 
[image]
it won't even tell me what button was pressed.

tell you what, I'm going to reconfigure the device to see if it was a bad interview.

suvl, 2022-08-18:
@javicalle OOTB with zigbee2mqtt I get an MQTT event that tells me what button was pressed: left, right or both. Trying to replicate that with ZHA.

I was able to replicate the same behaviour with this trigger:

```yaml
  - platform: event
    event_type: zha_event
    event_data:
      device_id: 0e6c85509d0f12d8c35b2fd2c76a8cd7
      endpoint_id: 1
      args:
        value: 1
```

For what I understand from the code, this event should be mapped to the following one:

```yaml
  - device_id: 0e6c85509d0f12d8c35b2fd2c76a8cd7
    domain: zha
    platform: device
    type: remote_button_short_press
    subtype: button_1
```
but this event is never triggered.

javicalle, 2022-08-18:
Every rocker button will send events from a diferent `"endpoint_id": 1/2/3,`. Not sure about how diferent actions must come, I believe that would be as diferent `"command"`.

But ill will be much better handle it from the device event and not as a zha_event:
[image]

From the automation view:
* 'Trigger type': device
* select your device
* in the 'trigger' combo you must see all the available events for the device

suvl, 2022-08-18:
@javicalle I agree with you however what I was trying to tell you is that despite there being a `zha_event` being triggered there is no `device` event from the `zha` platform that is being triggered. I've tried to listen to all events and do not see anything other than the `zha_event` being triggered. So using the device trigger type is not working.

javicalle, 2022-08-18:
I see. You are having only events from the `OnOff` cluster and nothing from the `MultistateInputCluster`.
I don't know if there's a way to activate this (I don't see anything in the code).

Can you enable the debug logs and look for anything related to the device?

javicalle, 2022-08-18:
According to blackadder and Z2M this device only reports `single_click` events:
* https://zigbee.blakadder.com/Aqara_WXKG02LM.html
* https://www.zigbee2mqtt.io/devices/WXKG02LM_rev1.html#action-enum

You can identify the different buttons from the `endpoint_id`.

DeadlySin2, 2022-09-28:
Same isue for me. Please add the support of this devixce.

javicalle, 2022-09-28:
@DeadlySin2 your device don't have the same signature. Please open another issue with all the information.

andyjenkinson, 2022-10-09:
I'm also having problems with this device (lumi.sensor_86sw2). There are two versions of the device, this is the original 2016 model, I understand the 2018 model identifies as lumi.remote.b286acn02 and I believe there are hardware differences in the way that they function, however in ZHA they are both using the same quirk.

The info scattered around the web suggests that the later version supports double-press and hold events, so there's clearly a difference somehow that the quirk is not directly addressing, and that it is necessary to configure the device via Zigbee command using a manufacturer-specific endpoint to change the 'operation mode' (I'm unsure as lots of the discussion online also mixes with the H1 devices, e.g. https://github.com/zigpy/zha-device-handlers/issues/940). Again the quirk does not do anything to expose this change of mode. Apparently zigbee2mqtt exposes a YAML configuration option to disable 'legacy mode'.

Like you, I am just getting attribute updated events from endpoints 1 and 2 on the OnOff cluster (cluster ID 6). The 'both press' event doesn't fire from endpoint 3 either. This suggests to me that the device is running in ZLL (sometimes referred to as LO) mode instead of ZHA mode which perhaps the quirk expects. Except the problem is, I'm not completely sure that both versions of the hardware work the same way in this regard, and even though the device exposes a manufacturer specific cluster on endpoint 1 (0xffff), it does not list any attributes so I do not know how or what to send as a command to change it. The cluster details mentioned for the [H1 version](https://github.com/zigpy/zha-device-handlers/issues/940) don't match what I see exposed for this device.

closed 2023-04-14 by github-actions[bot]