zigpy/zha-device-handlers #1693: [Device Support Request] Decoupled mode for lumi.switch.l1aeu1

OliverHi, 2022-08-13:
I have a lumi.switch.l1aeu1 single rocker Aqara H1 EU switch without neutral wire paired with ZHA and it is missing a couple of things:
- temperature is off (showing 0.3°C all the time)
- there seems to be no cluster config to set it to decoupled mode where a click just sends an event but does not turn off the relay

The on/off button is working well otherwise. It seems like Zigbee2MQTT is supporting this so it should be possible. Please let me know if any additional information is needed

<details>
<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l1aeu1",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.7.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
[... 362 lines cut ...]
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

Ulrar, 2022-09-01:
Same with the two rocker version (lumi.switch.l2aeu1).
It's been mentioned in https://github.com/zigpy/zha-device-handlers/pull/1391 but strangely re-pairing the device changes the signature sometimes, it's not always finding all the clusters. That's likely true for the 1 rocker version as well.

As for the temperature it seems Z2M exposes a calibration field, it'll probably be way off until we can get that calibration into HA.

moroz-slavomir, 2022-09-11:
Hey guys. Seems these EU buttons without neutral share the same clusters and attributes as their counterparts with neutral from https://github.com/zigpy/zha-device-handlers/pull/1391. Only the endpoints are a little bit different. By tweaking the endpoints descriptors slightly I was able to match with both 1 and 2 face devices and the [OppleSwitchClusterer](https://github.com/zigpy/zha-device-handlers/blob/b98948e7b6538ab31d87330bba1793e0eb65de11/zhaquirks/xiaomi/aqara/opple_switch.py#L84) seems to work fine. 

What worked for me:
- operation_mode: 0 - decoupled, 1 - coupled
- reverse_indicator_light - 0 - normal, 1 - reversed (works only on endpoint 1, shared setting)
- automation triggers - single, double press

If you want to try you can create a folder `custom_zha_quirks` in your config folder. Paste [this python file](https://github.com/moroz-slavomir/zha-device-handlers/blob/aqara-h1/zhaquirks/xiaomi/aqara/switch_h1.py) inside. Add 
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
``` 
to you configuration.yaml and restart HA. 

I haven't written any python code ever before this attempt and also I'm not really familiar with zha and contribution guidelines, so I don't expect I will be making it official PR just yet. This way you can at least test it or use it until official support is done.

Ulrar, 2022-09-13:
Hey @moroz-slavomir,

You succeeded where I failed, and I can see why, I forgot these are hexadecimal .. I was so close.
Your code works great, thanks !

I do wish these had a way to override the decoupled mode to force the relay without going through home assistant and supported binding, but that's nothing to do with the quirk.

Ulrar, 2022-10-21:
This has been working well, any chance to get that merged in ?

moroz-slavomir, 2023-05-19:
@ardstuebck  Maybe if you have also an aqara hub, try updating the firmware. It changed the endpoints on my switch. Before the update the quirk would not work.

strass, 2023-11-19:
> @ardstuebck Maybe if you have also an aqara hub, try updating the firmware. It changed the endpoints on my switch. Before the update the quirk would not work.

I tried using [your quirk](https://github.com/moroz-slavomir/zha-device-handlers/blob/aqara-h1/zhaquirks/xiaomi/aqara/switch_us.py) for my b2laus01 with no luck. Any ideas on what I can do to troubleshoot?

moroz-slavomir, 2023-11-21:
> > @ardstuebck Maybe if you have also an aqara hub, try updating the firmware. It changed the endpoints on my switch. Before the update the quirk would not work.
> 
> I tried using [your quirk](https://github.com/moroz-slavomir/zha-device-handlers/blob/aqara-h1/zhaquirks/xiaomi/aqara/switch_us.py) for my b2laus01 with no luck. Any ideas on what I can do to troubleshoot? I don't have a hub - it's connected directly to my HA setup. Is there a way to update firmware?

I can confirm this was also the case for me. Needed to update the firmware first. After that the endpoints on the device have changed and quick started to work.

ardstuebck, 2023-11-23:
> I can confirm this was also the case for me. Needed to update the firmware first. After that the endpoints on the device have changed and quick started to work.

Hi, @moroz-slavomir witch device do you have and which firmware? I have a lumi.switch.l2aeu1 (Aqara H1 EU Double Rocker without Neutral) with Firmware 0.0.0_1411. I bought a Aqara E1 Hub to update the Firmware, but 0.0.0_1411 is the latest. 

 <details>
      <summary>The device Signatur in HA</summary>
      <br>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l2aeu1",
  "class": "zigpy.device.Device"
}
```
</details>

Anyone else with this issues?

moroz-slavomir, 2023-11-24:
> > > > @ardstuebck Maybe if you have also an aqara hub, try updating the firmware. It changed the endpoints on my switch. Before the update the quirk would not work.
> > > 
> > > 
> > > I tried using [your quirk](https://github.com/moroz-slavomir/zha-device-handlers/blob/aqara-h1/zhaquirks/xiaomi/aqara/switch_us.py) for my b2laus01 with no luck. Any ideas on what I can do to troubleshoot? I don't have a hub - it's connected directly to my HA setup. Is there a way to update firmware?
> > 
> > 
> > I can confirm this was also the case for me. Needed to update the firmware first. After that the endpoints on the device have changed and quick started to work.
> 
> Hi, @moroz-slavomir witch device do you have and which firmware? I have a lumi.switch.l2aeu1 (Aqara H1 EU Double Rocker without Neutral) with Firmware 0.0.0_1411. I bought a Aqara E1 Hub to update the Firmware, but 0.0.0_1411 is the latest.
> 
> The device Signatur in HA
> 
> ```json
> {
>   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
>   "endpoints": {
>     "1": {
>       "profile_id": "0x0104",
>       "device_type": "0x0100",
>       "input_clusters": [
>         "0x0000",
>         "0x0002",
>         "0x0003",
>         "0x0004",
>         "0x0005",
>         "0x0006",
>         "0x0009"
>       ],
>       "output_clusters": [
>         "0x000a",
>         "0x0019"
>       ]
>     },
>     "2": {
>       "profile_id": "0x0104",
>       "device_type": "0x0100",
>       "input_clusters": [
>         "0x0000",
>         "0x0003",
>         "0x0004",
>         "0x0005",
>         "0x0006"
>       ],
>       "output_clusters": []
>     },
>     "242": {
>       "profile_id": "0xa1e0",
>       "device_type": "0x0061",
>       "input_clusters": [],
>       "output_clusters": [
>         "0x0021"
>       ]
>     }
>   },
>   "manufacturer": "LUMI",
>   "model": "lumi.switch.l2aeu1",
>   "class": "zigpy.device.Device"
> }
> ```
> 
> Anyone else with this issue?

I'm sorry, but I can't verify aqara firmware version anymore as I don't have the hub. It's possible they might have removed the update. Your signature resembles what my device looked like upon purchase. With firmware upgrade it now looks like this: 

<details>
<summary>Signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "41": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "42": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "51": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l2aeu1",
  "class": "switch_h1.AqaraSwitchH1NoNeutral_2G"
}
```
</details>

The extra added cluster `"0xfcc0"` is important for controlling the device via this quirk.

ardstuebck, 2023-11-25:
@moroz-slavomir Thank you for the tip. Unfortunately, there is no newer update for other regions either. I have also acquired an Aqara Smart Wall Switch E1 Dual Button. The same problem occurs here. Could it be related to my Zigbee stick? I am using a TI CC2531, Z-Stack Home 1.2.

Thanks again for your support!

strass, 2023-11-25:
I'm in the same boat - I didn't try other regions but the app reports I'm on the latest firmware. Decoupled mode exists in the official app but when I move the switch back to Home Assistant the signature is the same as before...

I was hoping to keep everything in HA but I'd really like that virtual switch mode. Any ideas?

ardstuebck, 2023-12-20:
The Quirk now works for me. I don't know if it's because I switched to the Sky Connect Stick. After repairing it several times it worked. Thanks for your help

closed 2024-06-28 by github-actions[bot]

strass, 2024-07-16:
Still hoping this might happen

Ulrar, 2026-01-15:
Coming back to ZHA recently, this still seems to be an issue.
I've tried the one from above but could not get it working this time, unfortunately.

So I've had copilot make a new one for me, here it is in case anyone is interested : 
```python
"""Custom quirk for Aqara H1 Double Rocker Switch (no neutral) with decoupled mode entity.

Adds operation_mode select entities and button event support.
"""

from zigpy.profiles import zha
from zigpy.quirks.v2 import QuirkBuilder, EntityType

from zhaquirks.const import (
    ARGS,
    ATTR_ID,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    CLUSTER_ID,
    COMMAND,
    COMMAND_BUTTON_DOUBLE,
    COMMAND_BUTTON_HOLD,
    COMMAND_BUTTON_SINGLE,
    COMMAND_DOUBLE,
    COMMAND_SINGLE,
    ENDPOINT_ID,
    PRESS_TYPE,
    VALUE,
)
from zhaquirks.xiaomi import LUMI
from zhaquirks.xiaomi.aqara.opple_switch import (
    BOTH_BUTTONS,
    OppleOperationMode,
    OppleIndicatorLight,
    OppleSwitchCluster,
)
from zhaquirks.xiaomi.aqara.opple_remote import MultistateInputCluster

# Device automation trigger definitions matching XiaomiOpple2ButtonSwitchBase exactly
DEVICE_AUTOMATION_TRIGGERS = {
    # Button 1 (left) - endpoint 41
    (COMMAND_BUTTON_SINGLE, BUTTON_1): {
        ENDPOINT_ID: 41,
        CLUSTER_ID: 18,
        COMMAND: "41_single",
        ARGS: {ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
    },
    (COMMAND_BUTTON_DOUBLE, BUTTON_1): {
        ENDPOINT_ID: 41,
        CLUSTER_ID: 18,
        COMMAND: "41_double",
        ARGS: {ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_DOUBLE, VALUE: 2},
    },
    # Button 2 (right) - endpoint 42
    (COMMAND_BUTTON_SINGLE, BUTTON_2): {
        ENDPOINT_ID: 42,
        CLUSTER_ID: 18,
        COMMAND: "42_single",
        ARGS: {ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
    },
    (COMMAND_BUTTON_DOUBLE, BUTTON_2): {
        ENDPOINT_ID: 42,
        CLUSTER_ID: 18,
        COMMAND: "42_double",
        ARGS: {ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_DOUBLE, VALUE: 2},
    },
    # Both buttons - endpoint 51
    (COMMAND_BUTTON_SINGLE, BOTH_BUTTONS): {
        ENDPOINT_ID: 51,
        CLUSTER_ID: 18,
        COMMAND: "51_single",
        ARGS: {ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
    },
    (COMMAND_BUTTON_DOUBLE, BOTH_BUTTONS): {
        ENDPOINT_ID: 51,
        CLUSTER_ID: 18,
        COMMAND: "51_double",
        ARGS: {ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_DOUBLE, VALUE: 2},
    },
    # Hold events come from cluster 0xFCC0 on endpoint 1
    (COMMAND_BUTTON_HOLD, BUTTON): {
        ENDPOINT_ID: 1,
        CLUSTER_ID: 0xFCC0,
        ARGS: {ATTR_ID: 0x00FC, VALUE: 0},
    },
}

(
    QuirkBuilder(LUMI, "lumi.switch.l2aeu1")
    # Replace with OppleSwitchCluster on main endpoints
    .replaces(OppleSwitchCluster, endpoint_id=1)
    .replaces(OppleSwitchCluster, endpoint_id=2)
    # Add button event endpoints (these don't exist in the device signature)
    .adds_endpoint(41, device_type=zha.DeviceType.ON_OFF_LIGHT_SWITCH)
    .adds_endpoint(42, device_type=zha.DeviceType.ON_OFF_LIGHT_SWITCH)
    .adds_endpoint(51, device_type=zha.DeviceType.ON_OFF_LIGHT_SWITCH)
    # Add MultistateInputCluster to button endpoints
    .adds(MultistateInputCluster, endpoint_id=41)
    .adds(MultistateInputCluster, endpoint_id=42)
    .adds(MultistateInputCluster, endpoint_id=51)
    # Endpoint 1 - Button 1 operation mode
    .enum(
        attribute_name="operation_mode",
        enum_class=OppleOperationMode,
        cluster_id=0xFCC0,
        endpoint_id=1,
        entity_type=EntityType.CONFIG,
        translation_key="operation_mode",
        fallback_name="Operation Mode (Button 1)",
    )
    # Endpoint 2 - Button 2 operation mode
    .enum(
        attribute_name="operation_mode",
        enum_class=OppleOperationMode,
        cluster_id=0xFCC0,
        endpoint_id=2,
        entity_type=EntityType.CONFIG,
        translation_key="operation_mode_2",
        fallback_name="Operation Mode (Button 2)",
    )
    # Indicator light direction
    .enum(
        attribute_name="reverse_indicator_light",
        enum_class=OppleIndicatorLight,
        cluster_id=0xFCC0,
        endpoint_id=1,
        entity_type=EntityType.CONFIG,
        translation_key="indicator_light",
        fallback_name="Indicator Light Mode",
    )
    # Add device automation triggers for button events
    .device_automation_triggers(DEVICE_AUTOMATION_TRIGGERS)
    .add_to_registry()
)
```

This supports the two relays like the official quirk, but also adds the missing light indicator mode and operation mode for the buttons : 

[image]

And when you pick decoupled, that button starts generating zha_event that do show up properly in the automation's device picker : 

[image]

Works like a charm, for me anyway.